atar-axis/xpadneo #1: Wrong mapping

atar-axis, 2018-01-27:
Sometimes the controller is sending different mappings, we already know how they look - but not how we can distinguish at device probing.

We therefore added a lot of debugging infos - whenever you recognize a wrong mapping, please send us your `dmesg` output!

Spiffyk, 2018-02-12:
The wrongly mapped controller seems not to be recognized by the `xpadneo` driver. Instead, in `dmesg`, it is reported by the `hid-generic` driver if I understood that correctly?

```
[   85.920369] xpadneo: hello there!
/// ... some probably unrelated nvidia-modeset outputs ...
[  486.383215] hid-generic 0005:045E:02FD.0005: unknown main item tag 0x0
[  486.383435] input: Xbox Wireless Controller as /devices/pci0000:00/0000:00:14.0/usb1/1-11/1-11:1.0/bluetooth/hci0/hci0:256/0005:045E:02FD.0005/input/input25
[  486.388442] hid-generic 0005:045E:02FD.0005: input,hidraw3: BLUETOOTH HID v9.03 Gamepad [Xbox Wireless Controller] on a4:34:d9:50:cb:ec
```

atar-axis, 2018-02-12:
at least it is loaded now :) but the incorrect mapping in your case is not related to the mapping inside xpadneo, it is because xpadneo is not binded to the gamepad and hid-generic spits out a lot of... well... bullshit xD

have you already updated the `hid.ko` module (please reboot afterwards)? this is necessary because  we have to tell the hid-core that the correct driver for the xbox controller is xpadneo, not hid-generic.

---

you can, if you want, bind the driver "by hand" (as a temporary solution):

```
echo 0005:045E:02FD > /sys/bus/hid/drivers/hid-generic/unbind
echo 0005:045E:02FD > /sys/bus/hid/drivers/xpadneo/bind
```

this is the same as the Udev-rule does, but that is just a workaround.

Spiffyk, 2018-02-12:
Ok, right after manual binding I get the following in `dmesg`: [xpadneo-dmesg.txt](https://github.com/atar-axis/xpadneo/files/1716747/xpadneo-dmesg.txt)
Strangely enough, jstest-gtk reports 0 buttons. I have not tried it in-game yet to see if it is just a jstest problem or something else.
I'm pretty sure I did update my `hid.ko` and `bluetooth.ko` files as well but I may have slipped up somewhere on the way.

I will try again later and report on what I've found out.

atar-axis, 2018-02-12:
okay the output looks good to me! everything is mapped correctly i think.
instead of `jstest-gtk`, you should maybe try `evtest`.

thank you for your patience ;)

atar-axis, 2018-02-13:
@Spiffyk any news on that? does it work?

atar-axis, 2018-02-13:
ok, you can just copy the `src/99-xpadneo.rule` to `/etc/udev/rules.d/` and run `sudo udevadm control --reload` afterwards - then it should load automatically.

concerning the wrong mapping: can you please run `evtest /dev/input/event<highest number available>`? then press any of those buttons that do not work - thank you!

Spiffyk, 2018-02-14:
I did try the udev rule file but no luck, it still does not work for some reason. Maybe Ubuntu is doing something differently with Bluetooth.

I'll try the mapping in a few minutes.

Spiffyk, 2018-02-14:
Alright, here are your logs, hope they're helpful:
[xpadneo-dmesg.txt](https://github.com/atar-axis/xpadneo/files/1724683/xpadneo-dmesg.txt)
[xpadneo-evtest.txt](https://github.com/atar-axis/xpadneo/files/1724684/xpadneo-evtest.txt)

atar-axis, 2018-02-14:
Thanks!

You can also try the `99-...-raspberry.rule`-File.

`xpadneo-dmesg.txt` is absolutely helpful since I am sure now, that at least internally xpadneo works as expected.  The buttons are recognized and events are generated in theory.

But `xpadneo-evtest.txt` is a bit confusing, it looks like the event-file is not the one which is generated by xpadneo.

Did you do the `unbind`/`bind` thing before?
* if no: please do so and repost the event-file
* if yes: is there a symbolic link in `/sys/bus/hid/drivers/hid-xpadneo` to the device?

rootyElf, 2018-02-22:
It seems that the button mapping is somehow connected to an internal state the controller has which is set when the controller is paired. I have previously mentioned it [here](https://github.com/paroj/xpad/issues/53#issuecomment-341937279). Did you already have a look into this?

kakra, 2018-02-25:
I've added your patches to my set of kernel patches (based on v4.14), in the way I reformatted the commit descriptions and fixed some typos. Then I worked on a fix to correct mapping of X and Y buttons on my controller. I also added a few other patches. Here is the result:

https://gist.github.com/kakra/1e41897b8026fda3467f121f13680aab

In case you're interested, here's also the dmesg debug log before my fixes:

https://gist.github.com/kakra/d5325c0c6ae28acc1ed4e8b9acce17d5

Could you fork the Linux kernel sources and put your patches into a branch? It would me more easy to follow your patches and rebase them when I change kernel versions. We could also try to get them into shape for generel kernel inclusion.

atar-axis, 2018-02-25:
@kakra 

> Then I worked on a fix to correct mapping of X and Y buttons on my controller.

There are (at least) two ways to name the Butons, the most generic one is
BTN_NORTH, BTN_EAST, BTN_SOUTH and BTN_WEST.

The other one is to use BTN_X, BTN_Y, BTN_A and BTN_B.

The problem is, that both are connected in this way:

```
#define BTN_X			BTN_NORTH
#define BTN_Y			BTN_WEST
```
https://elixir.bootlin.com/linux/v4.5/source/include/uapi/linux/input-event-codes.h#L384

As you can see, that's exaclty the other way around as you would expect it (BTN_X is BTN_WEST on Xbox gamepads - i don't know who defined it in this way and why...)

By simply switching BTN_NORTH and BTN_WEST it is none of both types.
We should therefore simply use BTN_X, BTN_Y, BTN_A and BTN_B if that is, what Steam expects.

> In case you're interested, here's also the dmesg debug

Unfortunately it isn't useful to me since your gamepad uses the right mapping at the moment :D 

> With these patches, the controller works correctly in Steam games via wine, even with force feedback 

Thank you very much - it's good to know.

> But the guide button doesn't open Linux Steam any longer

Hum, `BTN_MODE` was just my personal favourite :D I can switch to whatever Steam expects.

A general question:
Could you please fork my repo for your changes - otherwise it is really a pain in the ... to find out what *exactly* you changed ;) I will do the same for the kernel patches - as you suggested - that's a good idea.

kakra, 2018-02-25:
I'm using Gentoo. My workflow is to branch changes off the release versions of the kernel, apply patches and then export patch series as one-file-patch (using `git format-patch --stdout`) to the package manager to get them into the distribution kernel. That's why your workflow of putting patch files into git doesn't really fit my workflow.

I'll first try your changes now, using `BTN_{A,B,X,Y}` instead of `BTN_{SOUTH,EAST,WEST,NORTH}` looks like a good thing to do. If this works, I'll try exporting patches to your forked repo.

I don't think that Steam expects a special button, `BTN_MODE` looks just right. It's just I don't see it in `xev` and that seems to be the problem. Only `evtest` will show the events. `xev` no longer sees button events from the gamepad with xpadneo.

With the `hid-microsoft` (patched) or `hid-generic` driver, this gamepad actually sends two events for the guide button: `BTN_MODE` and `KEY_HOME`. But mappings are totally messed up and force feedback isn't working. And I think Steam doesn't expect `KEY_HOME`. With a logitech gamepad (F710), it also sends `BTN_MENU` only and it brings up Steam on the first press, on the second press it launches big picture mode. This is very convenient when having your gamepad nearby the couch. ;-)

But I no longer use my F710 due to issues with worn-out thumb stick clicking. But I can dig it out and try to reconstruct which events were sent.

If you need a starting point, I'll push my kernel patches:
https://github.com/kakra/linux/tree/rebase-4.14/xpadneo

atar-axis, 2018-02-25:
> I'm using Gentoo. My workflow is to branch changes off the release versions of the kernel, apply patches and then export patch series as one-file-patch (using git format-patch --stdout) to the [...]

I think this workflow would fit the DKMS approach i mentioned in #9, will work on that in some weeks - I am a bit busy at the moment.

> I'll first try your changes now, using BTN_{A,B,X,Y} instead of BTN_{SOUTH,EAST,WEST,NORTH} looks like a good thing to do. If this works, I'll try exporting patches to your forked repo.

Thanks, I'm curious if it works as expected.

> I don't think that Steam expects a special button, BTN_MODE looks just right. It's just I don't see it in xev and that seems to be the problem. Only evtest will show the events. xev no longer sees button events from the gamepad with xpadneo.

I don't really understand why... evtest works on a very low level, so if it works there, it should anywhere - except something on a higher layer is broken. I will give it a try as soon as i get access to some gamepad! Evtest does reconize the X button, right?

>  actually I don't swap north and west but the keycodes that the gamepad sends.

Yep, but it's maybe a bit confusing when you send a `BTN_NORTH` event when the `WEST` button is pressed - I am sure you understand my concerns.

kakra, 2018-02-25:
Until now, I only tested Steam in wine. It works (with the latests master patches to use SDL for game controllers) with xpadneo. Previously (without xpadneo), I saw axes swapped with each other.

But now, with xpadneo, native Linux games are totally messed up and see the same axes swapped. Also, X and Y is swapped (exactly what I changed by my patch to your driver). So probably my patch needs to be reverted.

I don't understand where all the mess comes from: My F710 behaves correctly in both Wine and native Linux games. The Xbox controller doesn't: It's either wrong in Wine (with hid-microsoft) or wrong in Linux (with hid-xpadneo).

Strange thing:

evtest is correct with xpadneo, jstest is correct with xpadneo, both show correct names and if I do parallel tests on both my F710 and Xbox controller, the same axes and buttons trigger the same screen output, codes match, names match.

But in Linux games, it's totally messed up now.

Axes mappings:

X = X (correct)
Y = Y (correct)
Z <> Rx (Z now moves the right X stick)
Rx <> Ry (Rx now moves the right Y stick)
Rz ??? (couldn't figure out what it does, it's probably Z)

This results in games expecting the following order of enumeration:

X Y Rx Ry Z Rz

while xpadneo reports:

X Y Z Rx Ry Rz

jstest and evtest both report the same enumeration order, for both controllers. And that was always the case, no matter if xpadneo or hid-microsoft.

So I don't know where this confusion comes from. I can only guess there's some black magic happening with higher layers trying to correct wrong behavior of some hardware controllers.

Before Wine natively supported SDL controllers, I used DLL injection to replace xinput*.dll with a library from koku. And I used hid-microsoft by that time. Windows games had the above axes messiness, so I used the following export to fix this in SDL:

`export SDL_GAMECONTROLLERCONFIG="050000005e040000fd02000003090000,Xbox Wireless Controller,platform:Linux,x:b3,a:b0,b:b1,y:b4,back:b10,guide:b12,start:b11,dpleft:h0.8,dpdown:h0.4,dpright:h0.2,dpup:h0.1,leftshoulder:b6,lefttrigger:a5,rightshoulder:b7,righttrigger:a4,leftstick:b13,rightstick:b14,leftx:a0,lefty:a1,rightx:a2,righty:a3,"`

Additionally, jstest saw buttons A,B,C,X,Y,Z and a lot more axes than normal. I fixed them is jscal, by remapping blind axes and buttons to the end. Everything was good.

````
jscal -u 8,0,1,2,5,9,10,16,17,15,304,305,307,308,310,311,314,315,316,317,318,306,309,312,313
jscal -s 8,1,255,28672,36862,21844,21844,1,255,28672,36862,21844,21844,1,255,28672,36862,21844,21844,1,255,28672,36862,21844,21844,1,3,448,574,1394469,1394469,1,3,448,574,1394469,1394469,1,0,0,0,536870912,536870912,1,0,0,0,536870912,536870912
````

So you see, it had axes order 0,1,2,5,9,10,16,17 and I moved buttons 306,309,312,313 to the end. I'm not sure what those ids are, maybe hid usage ids? Maybe keycodes? At least 304... seem to be the codes for BTN_* from the linux source code. And 0... seems to match with "relative axes" codes.

So actually, with hid-microsoft, I saw the left and right trigger as gas and brake. In jstest it was reported as:

X,Y,Z,Rz,Gas,Brake

and magically corresponded to

X,Y,Rx,Ry,Z,Rz

So I even had no Rx and Ry but it worked.

But Wine doesn't know about gas and brake and couldn't handle those axes, and swapped the others.

Now, the controller reports no gas and no brake, and the Linux games broke.

Actually, it matches what you do with mapping the input hid usage reports 0x30 to 0x39: It matches 0, 1, 2, 5, 9 to X, Y, Rx, Ry, Hat - exactly as I did with jscal.

It looks like software is still looking at the ev codes and tries to map this again, now messing up with what the xpadneo mapping layer does.

I'm totally confused and don't know where to fix this. Or even why that happens.

atar-axis, 2018-02-25:
> So I don't know where this confusion comes from. I can only guess there's some black magic happening with higher layers trying to correct wrong behavior of some hardware controllers.

That's exactly what happens I guess.

The problem is, as already discussed above, that the gamepad mapping is different in Windows and Linux:
On Linux `0x32` of `GENERIC_DESKTOP` is associated with `Z`
On Windows `0x32` of `GENERIC_DESKTOP` is associated with `Rx`

I think, that Steam thinks it is in Windows and therefore it uses the Windows-Button mapping to associate the input events with physical buttons, *but* your controller is paired with linux and therefore the gamepad behaves the other way around...

Therefore, my solution would look like that:
1) Let's finally find out how the descriptor looks like if the controller was paired to an windows OS before (see https://github.com/atar-axis/xpadneo/issues/1#issuecomment-367636301)
2) I will add a parameter to let you switch between both mappings "by hand" - so you can choose the right one when you play within wine

kakra, 2018-02-25:
PS: The X button works for me...

I have some updates. After some digging the Linux games issue is resolved. Steam Linux games now work.

It looks like Steam keeps its secret little database, and it actually recognizes the xpadneo provided controller, and it applies the SDL mapping fix that I used for Wine.

I don't yet know where it pulls the information from but I went to `~/.steam/steam/config/config.vdf` and found the section `SDL_GamepadBind`. I deleted the Xbox controller from there, just to find the same entry again after starting Steam. I now replaced the mapping inside that entry to match the mapping from my F710 controller, and now it works. Even Steam comes up when pressing `BTN_MENU`.

This makes me think if it is really correct to remap the axes and buttons within xpadneo? Instead, SteamOS has a kernel fix to bind the modern Xbox Controllers to `hid-microsoft`, which also includes a quirks fix to correctly send button events for start, select, and menu (because otherwise the controller only sends keyboard events). And Steam provides the correct controller mappings for SDL. The same which I would provide to Wine then. All should be well.

So, why not drop xpadneo and instead add force feedback to hid-microsoft?

It looks like the controller behaves "correctly" in both Windows and Linux modes, provided you're using the `hid-microsoft` driver. But it's probably SDL which provides the wrong mapping because it cannot distinguish Linux and Windows mode of the controller.

The overall problem is that many parties try to fix the same problem at different layers.

If xpadneo did anything at all, it should probably only compensate for the different modes and add force feedback, and let the rest be done by SDL and Steam, as it was already solved there.

That is, SDL expects the controller to have BTN_GAS and BTN_BRAKE.

What I don't want is switching the driver on the fly.

My current approach would be to simply provide a system wide correct SDL controller mapping and be done with it. Steam using a SDL gamepad database, and Wine not using it, it the biggest impact of confusion.

What do you think? Should xpadneo try to comply with standard gamepads and thus disturb all the deployed SDL configurations already out there? Or should it try to fix the mode switching bahavior and map to what SDL has already deployed? This also means, jscal is needed to fix the `/dev/input/js*` interface.

kakra, 2018-02-25:
BTW: I think the cleanest solution would be to let xpadneo provide a standard controller mapping and then everyone just has to drop the SDL mapping corrections. That way, Android deployments also would not be affected as Android just doesn't have to use xpadneo and everything stays the same there.

The other way would be to port force feedback over to `hid-microsoft` and add a flag to switch on mode correction.

kakra, 2018-02-25:
Source of controller DB may be here: https://github.com/gabomdq/SDL_GameControllerDB

atar-axis, 2018-02-25:
> So, why not drop xpadneo and instead add force feedback to hid-microsoft?

IMHO opinion this approach - which is widely used in Linux - I know, is absolutely crazy. If a single part of `hid-microsoft` is prone to an attack like _driver fuzzing_, then the whole module is useless until the exploitable part is fixed...

The other reason why I don't like the idea is, that I would like to develop support for the headset extension, the USB-adapter and so on.

If everything works, we can think about that again - fine?

> The overall problem is that many parties try to fix the same problem at different layers. ... If xpadneo did anything at all, it should probably only compensate for the different modes and add force feedback

That's absolutely true, I fully agree on that. So... okay, let us just make sure, that all modes are mapped in the same manner - in a way, that is already known to SDL, ...

> That is, SDL expects the controller to have BTN_GAS and BTN_BRAKE.

It's not a big deal to change that, are you sure that this is what SDL and all the others expect?

> This also means, jscal is needed to fix the /dev/input/js* interface.

You mean the ordering is wrong, right? Sorry, too much information at the moment ;D

atar-axis, 2018-02-25:
I created a Milestone for that and splitted up this issue, therefore I will close this one - please use the new, more specific ones (or create one that fits).

closed 2018-02-25 by atar-axis

atar-axis, 2018-02-27:
@rootyElf Could you please do me favor: post the output of /sys/kernel/debug/hid/<controller>/rdesc here: https://github.com/atar-axis/xpadneo/issues/10

rootyElf, 2018-03-02:
Sorry, I haven't been keeping up with my GitHub notifications. I see that you had some progress with #10 in the meantime. Do you still need my help?

atar-axis, 2018-03-02:
@rootyElf No, but thank you!! I solved it yesterday - will upload it tomorrow I guess.

gytisrepecka, 2020-04-19:
I was reading through details posted here, wondering @atar-axis is switching "by hand" was implemented?

In my case any joystick (XBOX One Elite 2, GameSir, some other no-names) I use in Linux Mint / Ubuntu appears to be working fine, but games I run through PlayOnLinux or Lutris gets axis messed up in all games similarly: RT does what analog right Y axis is supposed to do while analog right Y axis does what analog right X supposed to do. Any clues how to fix that?

Analog left works as expected, but I cannot get analog right to work in the same way as console would :(

kakra, 2020-04-19:
What you describe (differences between multiple applications) are usually the result of NOT using xpadneo as the driver: The old js interface used by some older games and some wine versions (only for some older games, too) would be messed up, some games running through Steam are correct (because Steam fixes it through SDL, but this only works for games not using the input/event interface, legacy js stays broken)... xpadneo fixes this by faking the device identifier, so legacy js and input/event axis and button codes match, and SDL does no longer tries to fix it itself.

The problem is with some applications still detecting the controller as having seemingly "broken" layout, this is currently the case with Chrome when using its controller support: Chrome has a similar mapping table as SDL but it only examines VID/PID, not the device version ID as SDL does. So Chrome tries to correct the already correct mapping of xpadneo and messes that up again.

So yes: Steam has fixups for non-xpadneo drivers but most other software doesn't. So probably you're not using the xpadneo driver otherwise you would see correct mapping from the device and no mapping fixups only in Steam. Steam corrects the mapping through the SDL layer only.

gytisrepecka, 2020-04-19:
Thanks for detailed explanation!

So I understand xpadneo is driver running on host OS (in my case it is Linux Mint) - after installing xpadneo I can see XBOX Elite 2 controlled in `jstest-gtk` and is able to calibrate it there.

But when it comes to running Windows games through Wine (PlayOnLinux or Lutris in between), is there some layer which would force Wine to communicate to controller using xpadneo driver?

E.g. I have GOG version of Mafia 2 on PlayOnLinux. Most that I can set up regarding controller is Wine's **Control Panel > Game controllers** where I see two entries:
- Xbox Elite Wireless Controller (event)
- Xbox Elite Wireless Controller (js)

Although using **Test joystick** shows axis are messed up: **X,Y** is supposed to be **L analog** while pressing RT makes Y axis go up. Same as **Z, Rz** is **R analog** but pressing LT moves Z to the left.

kakra, 2020-04-19:
You should blacklist whatever driver currently is used for the controller, or unbind it. The output of `lsmod` could help. There's docs about how to generate some useful debug output to get qualified help: https://atar-axis.github.io/xpadneo/#troubleshooting

The problem you're seeing is also outlined there: https://atar-axis.github.io/xpadneo/#sdl-mapping but current `xpadneo` fixes that, you don't need to follow the instruction there. It's only for reference describing the observed problems and which components it affects.