atar-axis/xpadneo #10: Distinguish between Windows and Linux mode

atar-axis, 2018-02-25:
I need some information about the Windows mode to distinguish between them,
unfortunately i don't own a gamepad myself, so I would really appreciate it, if anyone
could do the following for me:

- Pair your gamepad with Windows
- Find out your LinkKey as thezeroth described here:
  https://superuser.com/questions/229930/finding-bluetooth-link-key-in-windows-7-to-double-pair-a-device-on-dualboot-com
- Shut down gamepad
- Reboot to Linux, copy your LinkKey from Windows to linux: https://askubuntu.com/questions/253949/how-can-i-avoid-having-to-pair-my-bluetooth-mouse-all-the-time/741821#741821
- Start your controller (should automatically connect to linux/xpadneo)

Send me your dmesg output afterwards (everything that starts with `xpadneo`)

atar-axis, 2018-02-25:
I once extracted the Windows mapping using USBlyzer, but I am not sure if this one is the same as in linux:

`05 01 09 05 A1 01 A1 00 09 30 09 31 15 00 27 FF FF 00 00 95 02 75 10 81 02 C0 A1 00 09 33 09 34 15 00 27 FF FF 00 00 95 02 75 10 81 02 C0 05 01 09 32 15 00 26 FF 03 95 01 75 0A 81 02 15 00 25 00 75 06 95 01 81 03 05 01 09 35 15 00 26 FF 03 95 01 75 0A 81 02 15 00 25 00 75 06 95 01 81 03 05 09 19 01 29 0A 95 0A 75 01 81 02 15 00 25 00 75 06 95 01 81 03 05 01 09 39 15 01 25 08 35 00 46 3B 01 66 14 00 75 04 95 01 81 42 75 04 95 01 15 00 25 00 35 00 45 00 65 00 81 03 A1 02 05 09 97 15 00 25 01 75 04 95 01 91 02 15 00 25 00 91 03 09 70 15 00 25 64 75 08 95 04 91 02 09 50 66 01 10 55 0E 26 FF 00 95 01 91 02 09 A7 91 02 65 00 55 00 09 7C 91 02 C0 05 01 09 80 A1 00 09 85 15 00 25 01 95 01 75 01 81 02 15 00 25 00 75 07 95 01 81 03 C0 05 06 09 20 15 00 26 FF 00 75 08 95 01 81 02 C0`

atar-axis, 2018-02-27:
I tried the method above, but I still cannot reproduce it...

**If anyone has an idea how I can change the behaviour to the one from window, please let me know.**

atar-axis, 2018-02-27:
I found an old log:
```
05 01 09 05 a1 01 85 01 09 01 a1 00 09 30 09 31 15 00 27 ff ff 00 00 95 02 75 10 81 02 c0 09 01 a1 00 09 33 09 34 15 00 27 ff ff 00 00 95 02 75 10 81 02 c0 05 01 09 32 15 00 26 ff 03 95 01 75 0a 81 02 15 00 25 00 75 06 95 01 81 03 05 01 09 35 15 00 26 ff 03 95 01 75 0a 81 02 15 00 25 00 75 06 95 01 81 03 05 01 09 39 15 01 25 08 35 00 46 3b 01 66 14 00 75 04 95 01 81 42 75 04 95 01 15 00 25 00 35 00 45 00 65 00 81 03 05 09 19 01 29 0a 15 00 25 01 75 01 95 0a 81 02 15 00 25 00 75 06 95 01 81 03 05 01 09 80 85 02 a1 00 09 85 15 00 25 01 95 01 75 01 81 02 15 00 25 00 75 07 95 01 81 03 c0 05 0f 09 21 85 03 a1 02 09 97 15 00 25 01 75 04 95 01 91 02 15 00 25 00 75 04 95 01 91 03 09 70 15 00 25 64 75 08 95 04 91 02 09 50 66 01 10 55 0e 15 00 26 ff 00 75 08 95 01 91 02 09 a7 15 00 26 ff 00 75 08 95 01 91 02 65 00 55 00 09 7c 15 00 26 ff 00 75 08 95 01 91 02 c0 85 04 05 06 09 20 15 00 26 ff 00 75 08 95 01 81 02 c0 00
```
```
05 01 09 05 a1 01 85 01 09 01 a1 00 09 30 09 31 15 00 27 ff ff 00 00 95 02 75 10 81 02 c0 09 01 a1 00 09 32 09 35 15 00 27 ff ff 00 00 95 02 75 10 81 02 c0 05 02 09 c5 15 00 26 ff 03 95 01 75 0a 81 02 15 00 25 00 75 06 95 01 81 03 05 02 09 c4 15 00 26 ff 03 95 01 75 0a 81 02 15 00 25 00 75 06 95 01 81 03 05 01 09 39 15 01 25 08 35 00 46 3b 01 66 14 00 75 04 95 01 81 42 75 04 95 01 15 00 25 00 35 00 45 00 65 00 81 03 05 09 19 01 29 0f 15 00 25 01 75 01 95 0f 81 02 15 00 25 00 75 01 95 01 81 03 05 0c 0a 24 02 15 00 25 01 95 01 75 01 81 02 15 00 25 00 75 07 95 01 81 03 05 0c 09 01 85 02 a1 01 05 0c 0a 23 02 15 00 25 01 95 01 75 01 81 02 15 00 25 00 75 07 95 01 81 03 c0 05 0f 09 21 85 03 a1 02 09 97 15 00 25 01 75 04 95 01 91 02 15 00 25 00 75 04 95 01 91 03 09 70 15 00 25 64 75 08 95 04 91 02 09 50 66 01 10 55 0e 15 00 26 ff 00 75 08 95 01 91 02 09 a7 15 00 26 ff 00 75 08 95 01 91 02 65 00 55 00 09 7c 15 00 26 ff 00 75 08 95 01 91 02 c0 85 04 05 06 09 20 15 00 26 ff 00 75 08 95 01 81 02 c0 00
```

atar-axis, 2018-02-28:
I found, that the third mapping (which is currently called `map_hid_to_input_unknown` in source), is the one i get in Ubuntu 17.01. Funny... really - how does the Controller distinguish between the other sides OS? And why!?

Anyway, the current problem is: This "mode" uses the *same* hid-descriptor as the other one (the one i called `map_hid_to_input_rs335` - the funny thing is, both are correct and fit the descriptor, at least in the sense of "which buttons are reported".

Next I will check which mode sends the correct report, by "correct" I mean "it fits the descriptor bitwise".

kakra, 2018-03-01:
As per my findings, I can maybe clean this up a little bit:

The mode switch happens during pair time, that is: If you pair it in Windows, then copy the linkkey over, the controller stays in Windows mode. If you pair it in Linux natively, the controller switches to Android mode.

Older firmware versions expose a different behavior what is why you may have seen a third mode.

I don't think the product id ever changes, except you've done a controller firmware update. While in Windows, that may happen behind your back without you even noticing. It may look like the product id has changed next time you look in Linux.

I suggest putting one variable out of the equation: Xpadneo at current stage should be developed only against the latest firmware version of the controller, and it should be developed with Android mode first in mind because I guess majority of users won't use the linkkey hack. Both facts should be documented. If xpadneo somehow detects that the device is not in Android mode, it can/should fall back to Windows mode. All other mappings are probably older firmware or different controller model. We should encourage users to update the controller firmware first by pairing the device temporarily to a Windows machine and running the Xbox Windows app to update the controller.

Is there a way to read the controller firmware version from the device descriptor?

atar-axis, 2018-03-01:
> The mode switch happens during pair time, that is: If you pair it in Windows, then copy the linkkey over, the controller stays in Windows mode. If you pair it in Linux natively, the controller switches to Android mode.

That fits @rootyElf's assertion, but when I tried it ony my machine, I couldn't reproduce that. Where you able to observe that behaviour on your machine?

> I don't think the product id ever changes, except you've done a controller firmware update.
Right, it changes from 0x02E0 to 0x02FD when you update to the newest firmware (that's at least what I observed on @Dinopron91's controller).

> I suggest putting one variable out of the equation: Xpadneo at current stage should be developed only against the latest firmware version of the controller, and it should be developed with Android mode first in mind because I guess majority of users won't use the linkkey hack.
The problem is: That this sometimes also happens without copying the linkkey. 

> If xpadneo somehow detects that the device is not in Android mode, it can/should fall back to Windows mode.
The case is, that it is really not that easy to detect if it is in Windows mode or not since - sometimes - the wrong descriptor is sent by the device, and it is the descriptor which is normally used to distinguish between different behaviors.

>Is there a way to read the controller firmware version from the device descriptor?
See above, it's not the descriptor which changes, but other BT attributes (the product id) - but I only saw that once, therefore....

---

What I did tonight is that I captured the Bluetooth communication between the gamepad and an windows machine (in fact I captured the USB communication - since my bluetooth module is luckily attached to the USB-bus).
As a result, i now can confirm, that the 307 byte long report is exactly the *windows report* - it becomes a bit more clear now.

atar-axis, 2018-03-01:
Ok, finally the LinkKey thing worked here too, so the situation now looks the following:
* The gamepad sends us the linux descriptor (337 bytes long)
* but it behaves like it where in windows (the actual reports send are windows events, which do not fit the linux descriptor)

The good thing is, that it is possible to differentiate the actual reports by their length:
* the windows report has 16 bytes
* the linux report has 17 bytes

In order to make the decriptor and the actual report fit, we could...
1) change the report descriptor to the one which is used by windows
2) manipulate the incoming reports as they arrive (which i would prefer)
  "translate" the windows reports to the linux ones

---

Next step is to inspect the behaviour i got in Ubuntu some days before.

atar-axis, 2018-03-01:
Okay it looks like the third behaviour (is saw in Ubuntu) does not really exist but was produced by the "linux mapping" on the "windows events".

I therefore can complete the mapping now and will do so tonight/tomorrow. Will reference this issue then.

kakra, 2018-03-03:
I updated my kernel to 4.15 with your latest version (as of commit 40266c3) of `hid-xpadneo.c` and my gamepad is properly detected in Android mode and it looks like it's properly mapped. I could even remove my `swap X/Y` patch (in fact, I even had to).

atar-axis, 2018-03-05:
I uploaded a first fix for the windows/linux mapping problem. This is a dirty one, since the mapping is already done when we finally can tell if the gamepad sends the correct reports or not...

I will take a look deeper into the hid-core of linux to find a better solution (something like call the mapping hook a second time, by hand), but at least for now it should be fixed - unfortunately i hadn't time to test it thoroughly yet, but I will do so withing the next days.

atar-axis, 2018-03-05:
The distinction between Windows is done by the Service Discovery Protocol (SDP).
[bt_conn_xbox_win_filtered.zip](https://github.com/atar-axis/xpadneo/files/1781910/bt_conn_xbox_win_filtered.zip)

atar-axis, 2018-03-05:
I tested the fix, it still has two little problems:
1) The automatic fix does not work when the gamepad is restarted
2) Ry is reccognized as Rz and Rx as Z

I know where the problem is and will overwork it ASAP.

atar-axis, 2018-03-06:
I fixed it and did some tests, should work now as expected - no matter to which OS it was paired before. 
It's still not a very clean solution - but it looks like the only one I can think of (changing the report descriptor)  is not possible anymore at that time (since hid_parse and report_fixup where already invoked when we recognize the wrong behaviour).

closed 2018-03-06 by atar-axis

atar-axis, 2018-03-15:
Im was poking around in the latest firmware the last days and I found some interesting thing, one of that is this: `Android HID Desc.`, `MS HID Desc` - hence the second descriptor is indeed for `Android`, not `Linux` in general.

kakra, 2018-03-16:
It is indeed for Android, that was already documented some longer time ago. It is a feature of the gamepad. It's just that Android runs on Linux, and thus the gamepad detects all Linux kernels out there as Android upon pairing.

commit b963a63e (2018-03-06): fixed #10, still a bit dirty but ok for now