atar-axis/xpadneo #11: Define a consistent mapping

atar-axis, 2018-02-25:
We have to determine a consistent mapping.

* Should we use `BTN_NORTH` or `BTN_Y`?
* Should we use `ABS_Z` or `ABS_GAS`?
* Should we use `BTN_DPAD_UP` or `ABS_HAT0Y`?

* Is the order in the `jsX` file important? How can we influence that?
* What does Steam expect? What does SDL expect?

kakra, 2018-02-26:
Steam provides SDL mappings by an internal vendor-bundled database to games automatically, it does this on all platforms. And it uses SDL itself as I could figure out by coincidence.

I would pretend that the most-common use-case is using xpadneo to run Steam games in Linux. So we should stick to the mapping that Steam expects.

The other use-case is running Windows games through wine. It would be important to document how to pass SDL controller mappings to wine, and distribute mappings matching what xpadneo uses with the driver. From wine 3.3 onwards, it will support using SDL for gamepads, so passing SDL gamepad mappings via env is easy.

What other use-cases can we think of?

BTW: Native Linux games will always be a problem if they don't use SDL. Maybe the driver could implement a flip switch (defaulting to off) which switches to standard event codes.

There are very few games which seem to depend on order in `jsX`, e.g. Metro Last Light and Cradle seem to be such games: They expect gamepads in `DINPUT` mode (that mode does not have analog triggers but only digital triggers, have fewer axes, and the reported order of buttons is different). So having a mode parameter in the driver would be nice to support modes `xbox`, `android`, `dinput`.

atar-axis, 2018-03-13:
Thank you for the answer, I will comment and think about the rest a bit later when I finished the other issues, but I remembered something I read somewhere and I will post it here to remember when I come back here:

> Apart from that, we should stick with the names that the physical controller has on the buttons: The event code for BTN_SOUTH and BTN_A is identical according to Linux sources. These buttons are not directional (multiple can be pressed at once, even across), and this A button has a green A printed on it. So it should be called BTN_A in the driver source.

**VS.**

[](https://www.kernel.org/doc/html/v4.13/input/gamepad.html)

>Every gamepad device has at least 2 action buttons. This means, that every device reports BTN_SOUTH (which BTN_GAMEPAD is an alias for). Regardless of the labels on the buttons, the codes are sent according to the physical position of the buttons.
> If all 4 action-buttons are present, they can be aligned in two different formations. If diamond-shaped, they are reported as BTN_NORTH, BTN_WEST, BTN_SOUTH, BTN_EAST according to their physical location. If rectangular-shaped, the upper-left button is BTN_NORTH, lower-left is BTN_WEST, lower-right is BTN_SOUTH and upper-right is BTN_EAST.

kakra, 2018-03-13:
If all games have swapped X and Y buttons just because the Linux codes insists that the Y button should be west, and the X button north, then resistance is futile (tm). It's just against all expectations. I think the design is many years old. I can't think of a controller which has the buttons in rectangular shape, all are diamond shaped. The kernel is wrong here, hardware was implemented differently. Nobody cared.

BTN_X and BTN_NORTH are just aliases for the same hex code, let's not try to be smart. Let's call them BTN_X etc and be done with it.

Have a look here:
https://www.google.de/search?biw=2560&bih=1311&tbm=isch&sa=1&ei=Ax6nWqebFobO6ASmvJeIDw&q=controller+button+layouts+x+y+a+b&oq=controller+button+layouts+x+y+a+b&gs_l=psy-ab.3...0.0.0.38471.0.0.0.0.0.0.0.0..0.0....0...1c..64.psy-ab..0.0.0....0.ncgHhCzDC7M

Except for very unusual clunky controllers nobody would use today, the buttons are X left, Y top, B right, A bottom. The few exceptions that have X and Y swapped (which would fit the linux bill) also have A and B swapped (which is where it collides).

Maybe the north/south/east/west naming was made for controllers having shapes like Play Station controllers. Maybe those buttons have swapped roles there. Then everything would fit. The shapes there apply to roles: O and X mean yes and no, [] symbolizes a paper sheet meaning menu, /\ symbolizes point of view, or angle. Yes and no matches mostly the usage of A and B (A confirms, B cancels in menus). The Y button is used by some games as a weapon menu which would fit the west button on PS controllers. So that would explain it.

atar-axis, 2018-03-13:
> If all games have swapped X and Y buttons just because the Linux codes insists that the Y button should be west, and the X button north, then resistance is futile (tm). It's just against all [...]

It is not that hardware was implemented differently, but that someone added e.g. a `BTN_A` as a synonym for `BTN_SOUTH`. That is completely legal in my eyes for e.g. the Wii Controller, and the Cobra Gamepad and others which simply do not follow the diamond-shaped layout, but it is not for the rest.
[image]
[image]

If you take a look [https://elixir.bootlin.com/linux/v3.1/ident/BTN_A](here), you will see that the `xpad` driver is one of the rare drivers for a diamond shaped layout which uses `BTN_A` and so on. All the others are joysticks or somehow special.

The one who is wrong is xpad - not the others :smile: 

Therefore, I would suggest to offer a `xpad` layout (which is in general the one we use at the moment in fact). As a second layout I would suggest one which follows kernel standard, I would call this `standard`.

atar-axis, 2018-03-22:
Let's implement the following mappings

### `xpad` mapping
`A` → 
`B` → 
`X` → 
`Y` → 
`Start` → 
`Select` → 
`Xbox` → 
`L1` → 
`L2` → 
`R1` → 
`R2` → 
`L Thumb` → 
`R Thumb` → 
`Left Stick X` → 
`Left Stick Y` → 
`Right Stick X` → 
`Right Stick Y` → 
`Dpad Up` → 
`Dpad Right` → 
`Dpad Down` → 
`Dpad Left` → 

### `kernel` mapping
`A` → 
`B` → 
`X` → 
`Y` → 
`Start` → 
`Select` → 
`Xbox` → 
`L1` → 
`L2` → 
`R1` → 
`R2` → 
`L Thumb` → 
`R Thumb` → 
`Left Stick X` → 
`Left Stick Y` → 
`Right Stick X` → 
`Right Stick Y` → 
`Dpad Up` → 
`Dpad Right` → 
`Dpad Down` → 
`Dpad Left` → 

### `android preset` mapping
`A` → 
`B` → 
`X` → 
`Y` → 
`Start` → 
`Select` → 
`Xbox` → 
`L1` → 
`L2` → 
`R1` → 
`R2` → 
`L Thumb` → 
`R Thumb` → 
`Left Stick X` → 
`Left Stick Y` → 
`Right Stick X` → 
`Right Stick Y` → 
`Dpad Up` → 
`Dpad Right` → 
`Dpad Down` → 
`Dpad Left` → 

### `microsoft preset` mapping
`A` → 
`B` → 
`X` → 
`Y` → 
`Start` → 
`Select` → 
`Xbox` → 
`L1` → 
`L2` → 
`R1` → 
`R2` → 
`L Thumb` → 
`R Thumb` → 
`Left Stick X` → 
`Left Stick Y` → 
`Right Stick X` → 
`Right Stick Y` → 
`Dpad Up` → 
`Dpad Right` → 
`Dpad Down` → 
`Dpad Left` →

atar-axis, 2018-05-21:
The `xpad` and `kernel` mapping are straightforward: we know how xpad maps an physical button to an event and we know how the kernel documentation prefers it.

Unfortunately, the `android` and `microsoft` presets are not *really* on the same layer... they describe how a physical button is mapped to an `HID usage`.

It goes like: `physical button` <--> `HID usage` <--> `Input Event`

The mapping between the `HID_usage` and the `Input Event` is not described by it. Anyway - I would simply use the default mapping there.

Thereal problem is, that the `android` preset is simply wrong! It e.g. assigns `usage_page: SIMULATION, usage: GAS` to the right shoulder trigger, but also `usage_page: GENERIC_DESKTOP, usage: Rz` to right stick horizontal axis...
As a result, we have a mapping which does not map `ABS_RX` or `ABS_RY`, but `ABS_Z`/`ABS_RZ` *and* `ABS_BRAKE`/`ABS_GAS`

I am still not sure how if it makes senses to offer a mapping which is in fact corrupted - the only reason I can think of, is that maybe some higher layer is trying to fix the misbehaviour too.

kakra, 2018-05-21:
As far as I can tell, "native" Android controllers (which you pair to your phone) have additional buttons (C and Z right of A,B and X,Y). That explains why the button order is mangled up. They also expose the triggers as gas and brake. There are additional shoulder buttons (L2 and R2) like on an PS4 controller. Now the crazy part is: Someone who made up his mind about it totally intuitively guessed Rx and Ry are right X and right Y - but of course it's rudder/roll X and Y for a third pair of axes of movement. OTOH, one could say "why the heck did controller driver designers put the rudder axes on the triggers?" - those aren't even axes. Not to mention that it makes no sense to have a pair of axes on two different rudders. But that was probably done for compatibility with all those controllers out there already.

The Android controller designer guy is probably the one who broke compatibility. I cannot see any reasoning behind this. The problem is that standard configuration expects the three X axes first, then the Y axes. That is, X1, X2, Rx, Y1, Y2, Ry, which is then exposed by drivers as (X1,Y1),(X2,Y2),(Rx,Ry). But the controller sends X1,X2,Y1,Y2,Rx,Ry or something similar weird. Usually, with the wrong mapping, the right Y axis is dead while I can use one trigger to change the right X axis.

It would probably work (and indeed does with some games), if games would not rely on offsets into an array of values but the actual meaning of those values exposed by the event layer.

And yes, then there's the Z axes, too.

It's hard to wrap your head around. The driver should simply correct this. In Windows mode, everything is probably okay and the gamepad exposes a standard mapping. I never tried that.

The Wine code has means of mapping Rx,Ry onto Z1,Z2 or vice-versa, depending on how many axes are reported - probably to compensate for those two common mappings. But it only works for the joystick interface, not for games using the xinput API. Those get passed through the Linux HID data. So Windows has exactly the same split architecture as Linux: an old API, and a new one = jsX vs eventX = winjoy vs xinput. The latter one has the additional layer (HID usage), the first one maps directly from button to reported event.

atar-axis, 2018-05-22:
Wait, what?! Okay now everything becomes... even more confusing...
I'm not sure if I understood that correctly, what you are talking about is how the "original" Android controller does report it's buttons/axis - right?

> They also expose the triggers as gas and brake.

If you say "triggers" you are talking about the L1/R1 buttons - right? What the heck? I always thought that the GAS/BRAKE is meant to be on the L2/R2 axis. That would explain why the HID report is reporting both, ABS_GAS/ABS_BRAKE and ABS_Z/ABS_RZ.

> Now the crazy part is: Someone who made up his mind about it totally intuitively guessed Rx and Ry are right X and right Y - but of course it's rudder/roll X and Y for a third pair of axes of movement. 

I'm out... Rx and Ry are on the shoulders? That's totally in contrary to what the Linux Gamepad Specification says (_The left analog-stick is reported as ABS_X, ABS_Y. The right analog stick is reported as ABS_RX, ABS_RY._)

Can you maybe tell me where I can find some information about the "andoid controller" and its mapping? 
I still do not really understand why they do it as they do.

atar-axis, 2018-05-22:
Does the Xbox One Controller really work on an Android phone correctly? Because i never saw a hid-event  which fits the android mapping, the controller always sends an event which is fitting the windows one.

atar-axis, 2018-05-22:
okay I think I will offer the following mappings:- 
`kernel` - as defined in the Linux Gamepad Specification. This Mapping will be the default one (*"All new gamepads are supposed to comply with this mapping. Please report any bugs, if they don’t."*).
`xpad` - simulates the behaviour which is known from the legacy xpad driver
and `none` - which leaves is up to the HID/input system and behaves just as wrong the hid-generic driver, this is maybe necessary for higher layers that fix the misbehaviour themself.

Is this fine for everybody?

closed 2019-03-20 by atar-axis