atar-axis/xpadneo #9: Integrate DKMS for simple compilation

atar-axis, 2018-02-25:
Maybe we should use DKMS to make it a bit more easier for users to
install this driver on their system:

https://en.wikipedia.org/wiki/Dynamic_Kernel_Module_Support

kakra, 2018-02-25:
I'm not sure if it's worth yet to think about it... Before other kernel patches (bluetooth pairing fix, binding to xpadneo), it doesn't make so much sense to have this outside of the kernel. Yes, there're workarounds like turning off ERTM and rebind by udev rule. But a DKMS module should come without kernel workaround hacks. And in the future, you never know which hacks to enable or better not to enable.

But if we had a kernel fork with the patches in a branch, I'd be happy to provide instructions for Gentoo on how to feed the kernel patches to the package manager.

And, first we should clarify what xpadneo is going to fix at the core, see issue #1. Should it emulate a standard gamepad mapping, with all the headache this means to users running Steam with its SDL gamepad database. Or should it detect and try to map between Android and Windows mode, so all SDL-related deployments could stay in place? Should we also document how Wine interacts with SDL from 3.3 onward, providing SDL controller mappings to users for easy deployment?

atar-axis, 2018-05-20:
made a working DKMS version, see branch:dkms

closed 2018-05-20 by atar-axis