atar-axis/xpadneo #14: Enable battery reports

kakra, 2018-03-05:
The module currently doesn't report that is has a battery sensor, and doesn't report battery events.

The original xpad driver can do this, tho it reported fresh and full batteries as only 50% full. So the new controller probably uses a different scale.

atar-axis, 2018-03-05:
The battery report is `ID 04`, which is is only one byte long. The values i saw until now where always around 80-90. I will observe that and make some tests to find out the correct scale, should be easy then to include that.

atar-axis, 2018-03-05:
The report descriptor says the range is between 0 and 255, I cannot confirm that - I think that it already turned off at around 60-70 once.

```
Usage Page (Generic Device Controls) 05 06  

Usage (Battery Strength) 09 20  
Logical Minimum (0) 15 00  
Logical Maximum (255) 26 FF 00  
Report Size (8) 75 08  
Report Count (1) 95 01  
Input (Data,Var,Abs,NWrp,Lin,Pref,NNul,Bit) 81 02  
```

atar-axis, 2018-03-05:
At least in Linux it behaves like the following:
It stops all rumbling effects at `0x84`, it shuts down when reaching `0x83` (for a moment it becomes `0x04`).

I will have to:
1) Check if the behaviour is the same for windows
2) Check how the value is with new, full, quality batteries

atar-axis, 2018-03-06:
All Information above is only true for AA batteries! I will therefore impement this kind first, I am very sure that the "play and charge"-batteries differ since it uses different connectors.

atar-axis, 2018-03-06:
Started development: d8087aaf939077f0e885390a34aee34b84b1a0d8

atar-axis, 2018-03-07:
It is nearly ready, but I still don't get an battery indication in Steam.
At #16 you (@kakra) wrote:

> It also works for the Xbox controller if binding the controller to hid-microsoft. But it only reports 50% with a warning that batteries are almost depleted and, then it may (I cannot remember) to 25%, shortly after it turns off.

Can you explain me how exactly you do that? I can inspect the differences then and may find what is missing.

atar-axis, 2018-03-07:
Basic reports are implemented now, for everything else: let's open a new issue.

closed 2018-03-07 by atar-axis

atar-axis, 2018-03-09:
https://github.com/atar-axis/xpadneo/issues/22#issuecomment-371688371