atar-axis/xpadneo #16: Steam: No Rumble available

atar-axis, 2018-03-06:
I installed Steam on Windows and Linux (native and runtime),
when I use "identify" my gamepad rumbles in Windows, but in Linux it doesn't.
This is not a general force feedback problem since using `fftest` the force feedback works.

@kakra Any thoughts on that?

kakra, 2018-03-06:
The battery indicator works for some gamepads. And it works for Logitech mice and keyboards (using K360 and M600 here). So generally, the infrastructure is here.

It also works for the Xbox controller if binding the controller to hid-microsoft. But it only reports 50% with a warning that batteries are almost depleted and, then it may (I cannot remember) to 25%, shortly after it turns off.

Rumble in Steam itself doesn't work for identify. This only works for the original Steam controller (which I also own). Also, my F710 doesn't rumble for identify. This is probably a Steam Linux implementation detail. Technically, it could work but isn't properly implemented.

In games, tho, rumble works.

There may be two pitfalls:

The controller may be mapped differently depending if you're running games through the desktop client or through big picture mode. Steam is able to handle keyboard/mouse emulation when you add a non-steam game to the library. So everything seems to work here, I think Steam injects a patched libsdl2 or environment settings for SDL into the games run through Steam.

And there's the next pitfall: Steam injects SDL-specific environment variables. The controller may have added a "wrong" mapping to your Steam config file (one based on the original hid-microsoft mapping, as the controller binds to that driver in SteamOS). You can correct this:

1. Completely stop the Steam client
1. Locate your Steam global config file: `~/.steam/steam/config/config.vdf`
2. Open the file and find a string `"SDL_GamepadBind"`
3. These are SDL gamepad bindings, find the entry with your controller name: `Xbox One Wireless Controller`
4. Replace it with the following entry:

```
050000005e040000fd02000003090000,Xbox One Wireless Controller,a:b0,b:b1,back:b6,dpdown:h0.4,dpleft:h0.8,dpright:h0.2,dpup:h0.1,guide:b8,leftshoulder:b4,leftstick:b9,lefttrigger:a2,leftx:a0,lefty:a1,rightshoulder:b5,rightstick:b10,righttrigger:a5,rightx:a3,righty:a4,start:b7,x:b2,y:b3,
```

Take care not to strip the closing `"` quote sign.

You can now start Steam again, your controller should have a standard mapping now in Steam. If you already made controller mappings in Big Picture Mode, you may need to adjust them.

The id of the entry should be the same for your controller but it may vary. If it does, keep the id of your original entry, only replace the key and axes mapping. Rumble should work in games with this entry.

kakra, 2018-03-06:
There's also a small PR I submitted to wine: https://github.com/wine-mirror/wine/pull/10

It enables the xpadneo driver to be correctly recognized in latest wine versions. That way, wine sees the correct mapping throughout all games. Steam in wine may need the same SDL mapping correction as shown above. Rumble is fully supported there, too, tested primarily with Witcher 3 currently.

The controller shows some strange behavior in some Steam games running through wine: The right stick emulates a mouse but still sends controller reports. This confuses some games, Dishonored and Kingdoms of Amalur are such examples. I'm pretty sure this is something that Steam does (it seems to use a controller template that enables mouse camera through the right stick). Since there is no Big Picture Mode supported in wine, it cannot be corrected currently, at least I'm yet to find out which config file to edit, and how.

But this means that the same quirk may apply to native Linux Steam. It's easy to fix this: Just disable controller configuration support for the Xbox controller in Big Picture Mode.

atar-axis, 2018-03-06:
> The battery indicator works for some gamepads. And it works for Logitech mice and keyboards (using K360 and M600 here). So generally, the infrastructure is there.

Perfect! Thanks!

> Rumble in Steam itself doesn't work for identify. This only works for the original Steam controller (which I also own). Also, my F710 doesn't rumble for identify. This is probably a Steam Linux implementation detail. Technically, it could work but isn't properly implemented.

Very useful too, thank you!

atar-axis, 2018-03-14:
I will close this as it is the same as the battery report thing - missing support by steam, not the driver.

closed 2018-03-14 by atar-axis