atar-axis/xpadneo #18: LED brightness control

kakra, 2018-03-06:
According to https://github.com/paroj/xpad, the controller seems to support the LED brightness. I mean, have you ever played with this at night: It's really shiny. The driver should support controlling the LED.

atar-axis, 2018-03-06:
Are you sure about that? I mean, there is absolutely no indication in the HID descriptor that one can control the brightness!

kakra, 2018-03-06:
A friend told me that the controller should indicate in the four corners of the X logo which controller number you're using. This works at least with the Xbox, not sure about Windows. So it should be possible but maybe not through the normal bluetooth hid connection? Maybe this is possible only with the dongle. Also, I'm not sure if he owns the new Xbox One S controller but the older one.

atar-axis, 2018-03-06:
> A friend told me that the controller should indicate in the four corners of the X logo which controller number you're using.

Hum... I know that the old Xbox 360 Controller does that, but I never saw that anywhere else. There is also just one LED behind the Xbox-button. 

"Diming the light" seems to an option which is only available for the elite controller, at least according to some comments here: https://xbox.uservoice.com/forums/251650-console-hardware-accessories/suggestions/6029021-dim-the-bright-power-light-on-the-xbox-one-control?page=3&per_page=20

atar-axis, 2018-03-07:
Looks like it is possible indeed: https://www.neogaf.com/threads/xbox-one-lights-go-dimmer.723997/

But as you already said, I believe not over BT.

closed 2018-03-10 by atar-axis

atar-axis, 2018-03-14:
Another hint that it is possible: https://github.com/JRHeaton/XboxOneControllerDev

Slavko-P, 2022-01-28:
> Another hint that it is possible: https://github.com/JRHeaton/XboxOneControllerDev

Any updates regarding this hint?

kakra, 2022-01-28:
Current Xbox controllers have a vendor-defined descriptor extension which may expose the LED brightness and RGB control. This hasn't been explored yet but it's on the todo list. USB hints won't help here.

Slavko-P, 2022-01-28:
> Current Xbox controllers have a vendor-defined descriptor extension which may expose the LED brightness and RGB control. This hasn't been explored yet but it's on the todo list. USB hints won't help here.

Unrelated to xpadneo:  So in theory if I did it over USB, the setting would be permanent?

kakra, 2022-01-28:
I don't think so, this state would be cleared when the controller disconnects. You could send a custom HID packet via udev when the controller connects, xpadneo create a raw hid device node. I've seen the vendor extensions only for the Elite 2 controller currently which can store custom profiles inside the controller itself, this is probably encoded in reports 0x06, 0x07, 0x08, 0x09, and/or 0x0a. One of those can probably program LED color and brightness. The HID protocol is very similar to the GIP protocol but uses different report numbers, and GIP has a sequence number embedded into the protocol itself for synchronization (afaik, HID does that in an outer layer):
https://github.com/atar-axis/xpadneo/blob/master/docs/descriptors/xbe2_linux.md

So in theory, you could peek into the GIP protocol e.g. in the xone source code to get an idea how it works for HID. You may also need to record wireshark traces of the Windows app to see how it uses these reports.

Also, it looks like I didn't decode report descriptor 11 into the table:
```
0x06, 0x00, 0xFF,  //   Usage Page (Vendor Defined 0xFF00)
0x09, 0x06,        //   Usage (0x06)
0xA1, 0x02,        //   Collection (Logical)
0x85, 0x0B,        //     Report ID (11)
0x09, 0x14,        //     Usage (0x14)
0x15, 0x00,        //     Logical Minimum (0)
0x25, 0x64,        //     Logical Maximum (100)
0x75, 0x08,        //     Report Size (8)
0x95, 0x01,        //     Report Count (1)
0xB1, 0x02,        //     Feature (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position,Non-volatile)
0xC0,              //   End Collection
```

It basically describes a HID report with a single (count = 1) byte value (report size = 8 bits) and a value range of 0 to 100. Not sure what happens if you'd write `0x0B 0x20` (ID + data) to the raw HID device, maybe it sets the LED brightness, maybe the controller explodes.

There's a code example how to send rumble as raw data in `misc/examples/c_hidraw`, you could add more features to it if you like.

If your particular controller model doesn't have the required report IDs in the HID descriptor, you won't be able to send them: The controller will ignore them, or maybe even crash. You'll get a binary descriptor in /sys/module/hid_xpadneo/drivers/hid:xpadneo/0005:045E:*/report_descriptor (preprocessed by xpadneo). I think if you have a debug kernel, you can also get a decoded descriptor from debugfs. Otherwise use the online decoder as linked in the descriptor docs.

moviuro, 2023-11-24:
@kakra I don't understand at all what's happening in `hidraw.c`.

Did someone actually achieve turning the LED off when the controller is using bluetooth? My report descriptor below:

<details>
<summary>
Thanks https://eleccelerator.com/usbdescreqparser/
</summary>

```
0x05, 0x01,        // Usage Page (Generic Desktop Ctrls)
0x09, 0x05,        // Usage (Game Pad)
0xA1, 0x01,        // Collection (Application)
0x85, 0x01,        //   Report ID (1)
0x09, 0x01,        //   Usage (Pointer)
0xA1, 0x00,        //   Collection (Physical)
0x09, 0x30,        //     Usage (X)
0x09, 0x31,        //     Usage (Y)
0x15, 0x00,        //     Logical Minimum (0)
0x27, 0xFF, 0xFF, 0x00, 0x00,  //     Logical Maximum (65534)
0x95, 0x02,        //     Report Count (2)
0x75, 0x10,        //     Report Size (16)
0x81, 0x02,        //     Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
0xC0,              //   End Collection
0x09, 0x01,        //   Usage (Pointer)
0xA1, 0x00,        //   Collection (Physical)
0x09, 0x33,        //     Usage (Rx)
0x09, 0x34,        //     Usage (Ry)
0x15, 0x00,        //     Logical Minimum (0)
0x27, 0xFF, 0xFF, 0x00, 0x00,  //     Logical Maximum (65534)
0x95, 0x02,        //     Report Count (2)
0x75, 0x10,        //     Report Size (16)
0x81, 0x02,        //     Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
0xC0,              //   End Collection
0x05, 0x01,        //   Usage Page (Generic Desktop Ctrls)
0x09, 0x32,        //   Usage (Z)
0x15, 0x00,        //   Logical Minimum (0)
0x26, 0xFF, 0x03,  //   Logical Maximum (1023)
0x95, 0x01,        //   Report Count (1)
0x75, 0x0A,        //   Report Size (10)
0x81, 0x02,        //   Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
0x15, 0x00,        //   Logical Minimum (0)
0x25, 0x00,        //   Logical Maximum (0)
0x75, 0x06,        //   Report Size (6)
0x95, 0x01,        //   Report Count (1)
0x81, 0x03,        //   Input (Const,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
0x05, 0x01,        //   Usage Page (Generic Desktop Ctrls)
0x09, 0x35,        //   Usage (Rz)
0x15, 0x00,        //   Logical Minimum (0)
0x26, 0xFF, 0x03,  //   Logical Maximum (1023)
0x95, 0x01,        //   Report Count (1)
0x75, 0x0A,        //   Report Size (10)
0x81, 0x02,        //   Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
0x15, 0x00,        //   Logical Minimum (0)
0x25, 0x00,        //   Logical Maximum (0)
0x75, 0x06,        //   Report Size (6)
0x95, 0x01,        //   Report Count (1)
0x81, 0x03,        //   Input (Const,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
0x05, 0x01,        //   Usage Page (Generic Desktop Ctrls)
0x09, 0x39,        //   Usage (Hat switch)
0x15, 0x01,        //   Logical Minimum (1)
0x25, 0x08,        //   Logical Maximum (8)
0x35, 0x00,        //   Physical Minimum (0)
0x46, 0x3B, 0x01,  //   Physical Maximum (315)
0x66, 0x14, 0x00,  //   Unit (System: English Rotation, Length: Centimeter)
0x75, 0x04,        //   Report Size (4)
0x95, 0x01,        //   Report Count (1)
0x81, 0x42,        //   Input (Data,Var,Abs,No Wrap,Linear,Preferred State,Null State)
0x75, 0x04,        //   Report Size (4)
0x95, 0x01,        //   Report Count (1)
[... 57 lines cut ...]
0x75, 0x08,        //     Report Size (8)
0x95, 0x01,        //     Report Count (1)
0x91, 0x02,        //     Output (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position,Non-volatile)
0x09, 0xA7,        //     Usage (0xA7)
0x15, 0x00,        //     Logical Minimum (0)
0x26, 0xFF, 0x00,  //     Logical Maximum (255)
0x75, 0x08,        //     Report Size (8)
0x95, 0x01,        //     Report Count (1)
0x91, 0x02,        //     Output (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position,Non-volatile)
0x65, 0x00,        //     Unit (None)
0x55, 0x00,        //     Unit Exponent (0)
0x09, 0x7C,        //     Usage (0x7C)
0x15, 0x00,        //     Logical Minimum (0)
0x26, 0xFF, 0x00,  //     Logical Maximum (255)
0x75, 0x08,        //     Report Size (8)
0x95, 0x01,        //     Report Count (1)
0x91, 0x02,        //     Output (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position,Non-volatile)
0xC0,              //   End Collection
0xC0,              // End Collection

// 283 bytes

// best guess: USB HID Report Descriptor
```
</details>

kakra, 2023-11-24:
It looks like the vendor extensions are gone from HID with current firmwares. I'll explore this after v0.10 is released.

abbluiz, 2025-05-18:
Not possible to control the LED over _Bluetooth_? Guess I'll have to put a sticker on top 🙂

kakra, 2025-05-18:
Yeah, Microsoft removed all vendor extensions from the HID descriptor. So even if there has been a way, it's no longer possible.