atar-axis/xpadneo #19: Support setting timer for power off timeout

kakra, 2018-03-06:
It sometimes may be better to control the power off timeout of the controller, especially since Linux games usually cannot handle game controller devices files go missing and reappear. Some games may even crash on such events.

Additionally, Steam seems to support turning controllers off when quitting big picture mode. If possible, the driver should support that, too.

atar-axis, 2018-03-06:
Yeah I think we can send some idle events to the controller from time to time, I know that windows does since i captured the communication yesterday.

I already tried that "shut down when leaving big picture mode" in Windows, but even if Steam offers that option, it doesn't work.

atar-axis, 2018-03-12:
## Update:
* The controller shuts down even while busy doing FF effects,
* the package in Windows I mentioned above was nothing but an "stop all FF effects" package to all possible controller addresses (no stay alive as I thought previously)
* I tried some packages which are reported to crash the controller - but none of them worked so far.

atar-axis, 2018-06-06:
I don't think that it this is possible via bluetooth. Closed until I find a way (or any of you).

closed 2018-06-06 by atar-axis