atar-axis/xpadneo #20: Values at startup

atar-axis, 2018-03-07:
Tools which rely on the js-subsystem report wrong values for most axis until the first event is processed.
This is most probably because the joystick subsystem uses some default values until the first event is processed. 

### before
[image]

### after
[image]

To fix that, we may send some input events at startup to reflect the idle/startup position.

atar-axis, 2018-03-07:
@tcptomato: commit 7051a8f07cfae96ea5db265c45dcfb39afbd5819 fixes the problem about the steering angle at startup (the rover always steered to an maximum angle at startup).

closed 2018-03-07 by atar-axis

tcptomato, 2018-03-07:
Will flash it tomorrow :)

commit 7051a8f0 (2018-03-07): added startup input events, fixes #20