atar-axis/xpadneo #21: Steam: No Battery Indicator

atar-axis, 2018-03-07:
The driver is now registering a power_supply for the gamepad (#14) at startup of the device.
You can access it's current state in `/sys/bus/hid/devices/<gamepad>/power_supply/`.

Unfortunately the battery is now visible in Steam, we don't know why yet.

@kakra, at #16 you wrote:

> It also works for the Xbox controller if binding the controller to hid-microsoft. But it only reports 50% with a warning that batteries are almost depleted and, then it may (I cannot remember) to 25%, shortly after it turns off.

Are you talking about bluetooth or usb? Can you explain me how exactly you did that and how I can reproduce that? I will inspect the differences then and may find what steam expects and is missing.

atar-axis, 2018-03-08:
> With the hid-microsoft driver, the battery indicator shows up in the KDE systray in the battery and energy saving category.

Strange, since there is no special handling (not even a single line) for any power supply in hid-microsoft. Has to be something generic, maybe it is in another stacked driver.

> As with identify rumble, battery indicator in Steam only works with the Steam controller for me, it didn't work with hid-microsoft. But KDE desktop notifications worked for that driver (but Steam controller doesn't show up there).

Ah! Okay I misunderstood that - do you know by any chance which driver is loaded for the Steam gamepad?

Thanks!

kakra, 2018-03-09:
I now see the Xbox battery in KDE but it is shown as "Unknown" while with the hid-microsoft driver it shows up as "Xbox Controller":
[image]

kakra, 2018-03-09:
It looks like simply "manufacturer" and "model_name" may be missing:

```
hidpp_battery_0/:
capacity_level  device  manufacturer  model_name  online  power  powers  scope  serial_number  status  subsystem  type  uevent

'xpadneo_batt_3a:61:31:3a:30:30/':
capacity_level  device  power  powers  present  scope  status  subsystem  type  uevent
```

atar-axis, 2018-03-09:
Perfect but somehow strange because I still do not see it here (despite thw fact that I am using KDE too) - anyway, will add those missing attributes in the evening.

atar-axis, 2018-03-09:
> I also think this comes from a stacked driver.

The underlying part on the driver-stack was **hid-input** itself, which has a builtin generic HID_BATTERY_STRENGTH support (see [here](https://elixir.bootlin.com/linux/v3.15/source/drivers/hid/hid-input.c#L332) and [here](https://cateee.net/lkddb/web-lkddb/HID_BATTERY_STRENGTH.html)).

The problem you observed (empty on half battery) - is because the generic part need some information about what is minimum and what is maximum and it tries to use the `logical minimum` / `logical maximum` from the HID report. Unfortunately that is completely wrong in our case since the whole report descriptor is somehow buggy.

The good news is: We can learn from the hid-input implementation for our version ;) I will not use the builtin generic solution since the whole thing is a bit more complicated (battery pack and disposable batteries result in different values, both are influences by the charging cable, and so on)

atar-axis, 2018-03-10:
> It looks like simply "manufacturer" and "model_name" may be missing

added, hope it works - I still don't see any peripheral batteries, but that's not a problem related to xpadneo.

kakra, 2018-03-10:
I think you need powerdevil in KDE for it to work.

kakra, 2018-03-10:
Looks good now in KDE:
[image]

atar-axis, 2018-03-10:
Thanks! I will keep it open until I fixed my problem too (i have powerdevil installed, but the problem is that it's internal `hasPeripheralBattery` is set to `false` somehow), or at least as long as I am not totally sure that this is not a problem which is caused by the driver.

I also would like to rethink some Properties (Online, Present) - missing code documentation is sometimes really a problem in linux kernel code :-1: 

Anyhow - I am glad it works for you :smile:

atar-axis, 2018-03-10:
The underlying Problem is, that `Solid` does not recognize the battery as an `Solid::Battery` - I don't know yet why

atar-axis, 2018-03-10:
@kakra Can you do me a favor and give me the output of the following commands:
`upower --dump`

kakra, 2018-03-10:
Here you go:

```
$ upower --dump
Device: /org/freedesktop/UPower/devices/keyboard_hidpp_battery_0
  native-path:          hidpp_battery_0
  model:                Wireless Keyboard K360
  serial:               4004-ad-e8-6b-cb
  power supply:         no
  updated:              Sa 10 Mär 2018 15:23:05 CET (93 seconds ago)
  has history:          yes
  has statistics:       yes
  keyboard
    present:             yes
    rechargeable:        yes
    state:               discharging
    warning-level:       none
    battery-level:       normal
    percentage:          55%
    icon-name:          'battery-good-symbolic'

Device: /org/freedesktop/UPower/devices/keyboard_xpadneo_batt_3ao61o31o3ao30o30
  native-path:          xpadneo_batt_3a:61:31:3a:30:30
  model:                Xbox Wireless Controller
  power supply:         no
  updated:              Sa 10 Mär 2018 15:24:31 CET (7 seconds ago)
  has history:          yes
  has statistics:       yes
  keyboard
    present:             yes
    rechargeable:        yes
    state:               discharging
    warning-level:       none
    battery-level:       high
    percentage:          70%
    icon-name:          'battery-full-symbolic'

Device: /org/freedesktop/UPower/devices/DisplayDevice
  power supply:         no
  updated:              Do 08 Mär 2018 07:19:00 CET (201938 seconds ago)
  has history:          no
  has statistics:       no
  unknown
    warning-level:       none
    icon-name:          'battery-missing-symbolic'

Daemon:
  daemon-version:  0.99.5
  on-battery:      no
  lid-is-closed:   no
  lid-is-present:  no
  critical-action: HybridSleep
```

atar-axis, 2018-03-10:
Hum strange... can you please give me the following too:

`solid-hardware list`
`solid-hardware nonportableinfo /org/freedesktop/UPower/devices/gaming_input_xpadneo_batt_<see at list>`

kakra, 2018-03-10:
Here it is, tho I only have `solid-hardware5`, and query is actually empty:
https://gist.github.com/d8c427b8f35f358375e891a6f6f0cabc

atar-axis, 2018-03-11:
Thank you very much, the last one was indeed wrong:
3 `solid-hardware query 'IS Battery'`

kakra, 2018-03-11:
Here you go:

```
$ solid-hardware5 query 'IS Battery'
udi = '/org/freedesktop/UPower/devices/keyboard_hidpp_battery_0'
udi = '/org/freedesktop/UPower/devices/mouse_hidpp_battery_1'
udi = '/org/freedesktop/UPower/devices/keyboard_xpadneo_batt_3ao61o31o3ao30o30'
```

kakra, 2018-03-11:
BTW: The Steam controller does not show up here. So it has its own means of reporting battery. The Steam client doesn't seem to use standard interfaces here.

atar-axis, 2018-03-12:
[image]

I finally found the underlying problem...

The KDE battery notification is based on **Powerdevil**.
Powerdevil is base on **Solid** and Solid has Backends to e.g. **Upower**.

atar-axis, 2018-03-12:
I consider closing this issue, the driver is working as expected - the problems are caused by `Solid` (the bug above) and `Steam` (missing implementation of the power_supply system).

atar-axis, 2018-03-12:
Closed, thanks @kakra for your help in finding the problem!

closed 2018-03-12 by atar-axis

kakra, 2018-03-12:
I'd prefer if Valve would bring this up to Linux standard interfaces...

atar-axis, 2018-03-12:
Yeah me too :)

kakra, 2018-03-14:
I wonder why Gentoo Linux correctly shows it as battery despite missing such a patch. Maybe because UPower identifies the device as keyboard for me?

atar-axis, 2018-03-14:
Yep that's why I guess - but the question is why... is you UPower older than 6 months? They changed it at Version `.99.6` from keyboard to gaming_input

kakra, 2018-03-14:
I'm still at:

```
$ equery l upower
 * Searching for upower ...
[IP-] [  ] sys-power/upower-0.99.5:0/3
```