atar-axis/xpadneo #25: Can't install on raspbian stretch kernel 4.9 - various issues

JonBlue, 2018-03-25:
After running `task` it gets stuck on **patch_src**, I can fix this by adding  `--reject --whitespace=fix` to git apply but then it gets stuck on **copy_config_from_local** saying it can't find the **/boot/config-{{.K_VER}}/.config** file. I think in Raspbian there is only a build/configure.txt file which I tried feeding it by manually but that brings out all sorts of other issues by asking me to manually configure the build process asking **Cross-compiler tool prefix (CROSS_COMPILE) [] (NEW)** and about a hundred other questions and ultimately fails again.

atar-axis, 2018-03-26:
Hi there, looks like you are trying to compile the driver natively on a Raspberry - right?
Never tried that and it isn't supported yet by my Taskfile to compile it there directly.

But cross-compilation is! So, why don't you use `task raspi3` to cross-compile it on your local machine? :)
Please make sure that it is the right Raspbian version you have installed on the Raspberry Pi:
v4.9 - https://github.com/raspberrypi/linux/archive/raspberrypi-kernel_1.20171029-1.tar.gz

Give it a try, on the other hand you are welcome to add support for Raspbian directly.

JonBlue, 2018-03-28:
Hey, thanks for the answer and my bad, I did read the instruction but I somehow missed the reference to cross compiling under the Build instructions. Cross compiling is great as most of us access the raspberry via ssh anyways.

So it is my understanding that all I have to do is cross compile it then install the modules in Raspbian using `insmod` and run the Udev-rule.  

Is the above correct?

atar-axis, 2018-03-28:
> So it is my understanding that all I have to do is cross compile it then install the modules in Raspbian using insmod and run the Udev-rule.

Yea, just run `task raspi3` at first - compilation then takes a while since you have to compile *all* modules (in order to create the missing module.symvers). After the building process finished, you can find three files in `out/arm/`:

1. 99-xpadneo-raspi.rules
2. bluetooth.ko
3. hid-xpadneo.ko

Place the Udev-rule e.g. in `/etc/udev/rules.d` (it will be loaded automatically on the next restart).

Run "modinfo -n bluetooth" to find out where your current bluetooth module is installed placed, replace it (backup!) by the patched one in the `out` folder.

Place `hid-xpadneo` somewhere in extramodules (don't know how the correct path at the moment, sorry).

Run `depmod` to register the new modules afterwards and restart.

---

If you are unsure, take a look at `src/install.sh` which does the same (except the udev rule) on your local machine - maybe that script does also work on Raspbian, never tried. Again, you are welcome to modify it to work on Raspbian too.

atar-axis, 2018-03-28:
Btw: please report back if it works, I will close that issue then :)

atar-axis, 2018-03-28:
Ah, perfect :) Nice to hear that it works now for you!

The reason is that this was the latest version at thetime we installed raspbian on our raspberries at work ^^ Feel free to parametrize that if you want :smile:

closed 2018-03-28 by atar-axis