atar-axis/xpadneo #28: Build on Raspberry Pi 3

Jbbrack03, 2018-05-20:
My goal is to install this driver on a Raspberry Pi 3 running Raspbian Stretch. It doesn't appear that you are supposed to build this drive on the Pi itself. Is that accurate? There doesn't appear to be a version of go-task that is compatible with the Pi. Without being able to use the Taskfile, how would you install this driver on a Raspberry Pi? Sorry if this seems like a silly question. I appreciate your work, and I'm excited to see this in action.

atar-axis, 2018-05-20:
hey josh! we mostly pre-compile an ARM version on an ordinary PC using a cross-compiler! you can simply upload it then via scp or sdcard. Check out the README for more information :)

closed 2018-05-20 by atar-axis

atar-axis, 2018-05-20:
i made a DKMS version which should compile easily on any raspberry pi - take a look at the dkms branch. unfortunately it is still missing a way to automatically patch the `hid` and `bluetooth` module.

Jbbrack03, 2018-05-20:
That’s great! This driver is really needed, and I look forward to trying it.

atar-axis, 2018-05-21:
Looking forward to hear about any issues - will give it a try too as soon as I get access to an RaspberryPi.

Jbbrack03, 2018-05-21:
Quick question, so compiling the source code should generate the rule, HID, and Bluetooth files. Are you saying that your script applies the rule, but that you still need to manually copy the Bluetooth and HID files to the module directories? Is that what you mean by patching?

atar-axis, 2018-05-21:
I already fixed that, it is now fully automatic :P

Jbbrack03, 2018-05-21:
Thats’s awesome news! I’m going to test it again shortly.

atar-axis, 2018-05-21:
Fortunately the Raspi Version of the UDEV rule works on x86 too as it seems, therefore i replaced the rule in favour of the raspi version. Now everything should work, on both systems, without patching or replacing something by hand - hopefully.

atar-axis, 2018-05-21:
```
cd ~
git clone https://github.com/atar-axis/xpadneo.git -b xpadneo-dkms
cd xpadneo-dkms
sudo cp -r hid-xpadneo-0.2.0 /usr/src
```

Jbbrack03, 2018-05-21:
> fatal: Remote branch xpadneo-dkms not found in upstream origin

atar-axis, 2018-05-21:
sorry, fixed it above

Jbbrack03, 2018-05-21:
Works like a charm! Controller works flawlessly on the Raspberry Pi 3! Thank you for your hard work and your help!

atar-axis, 2018-05-21:
Thank you for testing and your response! I'm glad that it works :)

atar-axis, 2018-05-26:
hey josh! Is it your article on flickstiq? :) Because the installation process slightly chanced.

Jbbrack03, 2018-05-26:
Hi, yes it is

atar-axis, 2018-05-26:
perfect!

1. I merged and remived the dkms branch, so it is now: 
`git clone https://github.com/atar-axis/xpadneo.git`

2. Instead of copying the folder and installing it yourself, you can now simply run the script
`chmod +x install.sh`
`./install.sh`

Jbbrack03, 2018-05-26:
That's perfect! I just updated the article. It's a nice change that prevents the user from needing to manually append the version number. And it just simplifies the install. Thank you for your hard work!

atar-axis, 2018-05-26:
You are welcome!

chris-han, 2019-01-23:
I'm able to get the controller connected (stable logo light). However, when I check the input devices by ls /dev/input, I don't see the js0 there. How do you check/test the the device?

atar-axis, 2019-01-23:
hum, what does `dmesg` say?

chris-han, 2019-01-23:
Restart the controller and checked again. It's there. thanks!