atar-axis/xpadneo #29: Add a automatic way to fix hid and bluetooth while using the DKMS version

atar-axis, 2018-05-20:
We currently do not offer an automatic way to patch the kernel modules *hid* and *bluetooth*
while using the DKMS version of the driver.

At the master branch - using the Taskfile - this is done by the script which downloads the correct kernel, applies the patches, compiles them and replaces the related modules in the system. Event though this is working well in most cases, I would prefer to separate *kernel patching* and *driver installation* a bit more than it is using this way. Hence, I outsourced the driver installation to DKMS, which will automatically recompile the driver on every kernel update. Unfortunately, this is not (yet) true for the kernel modules which need to get patched...

I think we have in general three ways to solve that issue:
1. Use the workarounds like "disable ertm" and "udev unbind/bind" as an automatic solution instead of patching the kernel modules
2. Supply or automatically download the (latest) hid-core.c and l2cap_core.c from the official kernel - maybe by DKMS's `POST_ADD=` hook. Apply patches using DKMS's `PATCH=` command and replace the original modules somehow...

atar-axis, 2018-05-20:
the dkms branch should work, using the workarounds automatically.

I would appreciate if someone helps me bringing the patch for l2cap upstream, never did that before @kakra ? :smile:

closed 2018-05-20 by atar-axis

kakra, 2018-05-21:
Well, you can still document what settings need to be adjusted when not patching the kernel. We already have to unbind the device from the generic hid driver in that case. There's no pressure in pushing the l2cap patch mainline for the same reason: You can just disable ERTM.

The SteamOS kernel already disables this by default. But OTOH, Valve also has their very own fixups to the controller which is limited to games run through the Steam client only, and they need to use SDL. Otherwise, games still se wrong mapping. SDL also introduced a pure user-space fixup for the button mappings which won't fix mapping for legacy joystick interface. Wine is a user of this interface, too, unless games use the xinput layer AND the xinput SDL driver has been enabled, in which case you can use the SDL fixup.

I think all these fixes are at the wrong layer, because it leaves the legacy joystick interface with wrong mappings. And it makes things even more confusing, because depending on whether one of the several fixups at different layers applies or not, your gamepad behaves fine or doesn't, see Wine gaming as an example. :-(

But the l2cap fix should go mainline when xpadneo enters mainline.

atar-axis, 2018-05-21:
>  And it makes things even more confusing
Oh damn yeah, absolutely.

> There's no pressure in pushing the l2cap patch mainline for the same reason: You can just disable ERTM.
That's how I did it at the dkms branch now. I really think DKMS is the right way to go, muss less error-prone and absolutely easy to use. Anyway, I will keep the kernel_patches folder for documentation puposes.

Cheers