atar-axis/xpadneo #34: xpadneo driver crashes when connecting multiple controllers

cybercyst, 2018-06-06:
**Describe the bug**
Connecting multiple xbox one s controllers by bluetooth crashes the driver

**To Reproduce**
Steps to reproduce the behavior:
1. Connect one xbox one s controller to bluetooth
2. Connect another xbox one s controller to bluetooth
3. Bluetooth daemon crashes... check dmesg output

**Expected behavior**
I should be able to connect multiple bluetooth controllers

**System information:**
Linux strix 4.16.13-2-ARCH #1 SMP PREEMPT Fri Jun 1 18:46:11 UTC 2018 x86_64 GNU/Linux

**Additional context**
dmesg output:
https://pastebin.com/Vizda1Uh

atar-axis, 2018-06-06:
It still does not work, I borrowed an controller from fortiss - will give it another try tonight

atar-axis, 2018-06-06:
okay, should work now! please give it a try ;)

cybercyst, 2018-06-07:
It does! Awesome work. This is another feature windows doesn't support that this driver does. 😀

atar-axis, 2018-06-07:
thanks for confirming! will close it then

closed 2018-06-07 by atar-axis