atar-axis/xpadneo #35: Controller connects successfully, but generates no input

nassek, 2018-06-11:
I installed the driver by following the readme, controller is able to connect and rumbles but it doesn't generate any keypresses. I tried running Steam Big Picture to check it. I also ran _evtest_; output is empty after the line `Testing ... (interrupt to exit)`.

**System info:**
`Xubuntu 16.04`, `Linux 4.13.0-43-generic #48~16.04.1-Ubuntu SMP Thu May 17 12:56:46 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux`

dmesg output: [here](https://ghostbin.com/paste/v3e2m)

atar-axis, 2018-06-11:
hi nessek! I'm sorry to hear that.
The problem seems to be the battery initialization:
```
[ 2166.217028] xpadneo 0005:045E:02FD.000B: battery initialization failed
[ 2166.217115] xpadneo: probe of 0005:045E:02FD.000B failed with error -26908
```
I think I may already know where the problem is, will get back as soon as I fixed it (or if I need further debugging information).

atar-axis, 2018-06-11:
Okay, give it a try - please ;D

Please remove the last version by
* running `./uninstall.sh` (optional)
* `modprobe -r hid_xpadneo` to unload the old version from the kernel
* `git pull` to get the new version
*  `./install` to install the new version

If the problem persists, it would be helpful if you enable debug output - it it isn't already (see https://github.com/atar-axis/xpadneo#troubleshooting).

Thank you for getting back in advance!

nassek, 2018-06-11:
It's working now! Thanks for fixing the issue so fast :D
I still have some button mapping problems, but I think I'll be able to sort them out myself.

closed 2018-06-11 by nassek

atar-axis, 2018-06-11:
Ah, wonderful! :)

Regarding the button mapping:
Could  you please run `evtest /dev/input/event<#>` and tell me which buttons are wrong?
Furthermore the behaviour of `jstest /dev/input/js<#>` would be helpful too.

I am always **very interested** in wrong mappings because this is something we aim to fix with this driver :)

atar-axis, 2018-06-11:
@nassek!

Please **DO NOT** update your controller firmware via the xbox or windows app, you have an older firmware which is very helpful for me - since i have seen this firmware just once (and never again).

Your dmesg is already very very helpful. I will definitely add your firmware version - but I may need some debug outputs. Hope that's fine.

atar-axis, 2018-06-11:
-- deleted --

kakra, 2018-06-11:
@nassek Please, under all circumstances consider the pitfalls outlined in the README regarding button mapping. There's all sorts of corrections in different layers applied or not applied depending on versions of libraries used. The one and only fix is to first get rid of all "corrections" done at other layers.

atar-axis, 2018-06-11:
you can also give this branch a try: https://github.com/atar-axis/xpadneo/tree/input_version
it includes the fix/workaround I mentioned above, it works on my machine and I would really appreciate if you could confirm that it is working on yours too.

nassek, 2018-06-11:
First of all, in my case only Steam's mappings were messed up - _evtest_ and _jstest_ reported correct keypresses.
I tried the input_version branch, and had to change one line to get the module to compile, as in pull request #36

After installing the new module, Steam handles keypresses correctly.

atar-axis, 2018-06-11:
perfect ;) thanks for that!

kakra, 2018-06-11:
So my proposed version hack idea works around this whole mess successfully... Nice. :-)

atar-axis, 2018-06-11:
@kakra Looks like - wonderful :D I will therefore add the parameter, merge the version back to master and close your issue. Thanks to you both!

atar-axis, 2018-06-12:
@nassek May I ask you to test the new `master` one last time? I merged your pull request into `input_version`, added some other things and merged everything back to `master` then.
Thank you very much :)

nassek, 2018-06-12:
Updated to master ` c2885f4`, it works as supposed to. Good job guys ;)