atar-axis/xpadneo #38: Xbox button light shows connection, but controller does not vibrate and the connection is not successful.

josuetleoro, 2018-06-19:
I am using this driver for an application where a reconnection might happen often, i.e., the user will turn off the controller if not using it. Most of the time the controller connects correctly and it vibrates to indicate a successful connection. However, sometimes when the controller is turned on the xbox button light turns on indicating a successful connection but it does not vibrate. When this happens, it takes some time for the xbox button light to change from blinking to fully on. Once the controller is connected jstests works but with the wrong button mapping. fftest says that no function is implemented. 

The path for power supply is /sys/class/power_supply/hid-XX:XX:XX:XX:XX:XX-battery instead of /sys/class/power_supply/xpadneo_batt_XX:XX:XX:XX:XX:XX as it would normally be.

I have also noticed that this issue occurs when I have not used the controller for some time and it turns off automatically or during the first connection after a reboot.

Is this a bluetooth related issue or it is related to the driver?

atar-axis, 2018-06-19:
Hey there! Thank you for your feedback.
If the controller does not vibrate (3 times) but the light is fully on, then the gamepad got bounded to some other driver in your system - most probably **hid_generic** - which is the, well, generic hid driver :smile:.
The question now is: What happened to hid_xpadneo?

To answer that question I really need the `dmesg` output.
Furthermore, please activate debugging output in beforehand and reboot: https://atar-axis.github.io/xpadneo/index.html#debugging

Thank you in advance for getting back.
Flo

atar-axis, 2018-06-19:
A fully on light does mean nothing more than that the bluetooth connection is established. It does not necessarily mean that any driver is bound to the device.

On the other hand, you have an `js<X>` file in your `/dev/input` - so at least one driver _is_ loaded and bound to a gamepad/joystick. I assume that there is no other gamepad connected to your system, so this is the Xbox One S controller. The driver which is loaded is most probably not **hid_xpadneo** since the gamepad was not vibrating, but **hid_generic** (which is the, well, generic hid driver).

What I need now is
* your kernel version: `uname -r`
* the `dmesg` output (please activate the debugging output before, see [here](https://atar-axis.github.io/xpadneo/index.html#debugging)).

josuetleoro, 2018-06-20:
Thanks for your fast response to this issue!. The kernel version I am using is 4.13.0-45. I was not able to activate the debugging output because the file /sys/module/hid_xpadneo/parameters/debug_level does not exist on my system.

atar-axis, 2018-06-20:
Hey there, no problem!
The file is created once the driver is loaded ;) But you can activate it by
* `echo "options hid_xpadneo debug_level=3" | sudo tee /etc/modprobe.d/xpadneo.conf`

Can you please tell me the result of 
* `modinfo hid_xpadneo`

Cheers, Flo

josuetleoro, 2018-06-20:
Here are the outputs from dmesg and modinfo:
[dmesg_output.txt](https://github.com/atar-axis/xpadneo/files/2118345/dmesg_output.txt)
[hid_xpadneo_output.txt](https://github.com/atar-axis/xpadneo/files/2118346/hid_xpadneo_output.txt)
Thanks in advance,
Josue

josuetleoro, 2018-06-20:
I have noticed that when the problem of the connection takes place I have to turn the controller off and turn it on again for the connection to be successful. You can see below the dmesg output just after the connection was successful.
[dmesg_output2.txt](https://github.com/atar-axis/xpadneo/files/2118382/dmesg_output2.txt)
Many thanks.
Josue

atar-axis, 2018-06-20:
It looks like this not a problem by the driver itself, but by the **udev** rule, which is responsible to bind the device to the right driver. xpadneo gets loaded, but it does not get noticed about the new device.

1. please run `udevadm monitor` and start your gamepad, upload the result
2.  * try to replace the content of `/etc/udev/rules.d/99-xpadneo.rules` with:
      ```
      ACTION=="add", KERNEL=="0005:045E:02FD.*|0005:045E:02E0.*", SUBSYSTEM=="hid", 
      DRIVER=="hid-generic", \
      RUN+="/bin/bash -c 'echo UDEV RULE START > /dev/kmsg'", \
      RUN+="/bin/bash -c 'echo $kernel > /sys/bus/hid/drivers/hid-generic/unbind'", \
      RUN+="/bin/bash -c 'echo UDEV RULE UNBINDIND DONE > /dev/kmsg'", \
      RUN+="/bin/bash -c 'echo $kernel > /sys/bus/hid/drivers/xpadneo/bind'", \
      RUN+="/bin/bash -c 'echo UDEV RULE END > /dev/kmsg'"
      ```
    * invoke `sudo udevadm control --reload` afterwards to load the new rule
    * connect your gamepad and upload the `dmesg` output

josuetleoro, 2018-06-20:
1. Result of udevadm monitor:
[udevadm_monitor_output.txt](https://github.com/atar-axis/xpadneo/files/2118919/udevadm_monitor_output.txt)
2. dmesg output after changing the udev rule:
[dmesg_output3.txt](https://github.com/atar-axis/xpadneo/files/2118920/dmesg_output3.txt)

atar-axis, 2018-06-20:
After changing the udev-rule (and reloading it), you should see something like "UDEV RULE START" in your `dmesg` when the controller is connected - but there is no such message - which in return means that the udev rule is _not_ executed!

That's a huge step forward, thank you!

Can you now start `udevadm monitor --environment` and connect the gamepad again?

josuetleoro, 2018-06-20:
Here is the output of udevadm monitor:
[udevadm_monitor_output2.txt](https://github.com/atar-axis/xpadneo/files/2119092/udevadm_monitor_output2.txt)
I hope this can help anyone else that might have this issue :)

atar-axis, 2018-06-20:
Perfect, thank you again!
It looks like the udev rule, which is defined in `99-xpadneo.rules`, does in fact not fire on your system.
I have to come up with a new rule now - one which does work  on your system too.

In the meanwhile it would be very helpful if you **don not** update your kernel.
Thank you very much

atar-axis, 2018-06-20:
Okay, in order to find a fitting rule for you:
Can you please

0. shutdown the gamepad
1. `./uninstall.sh` xpadneo
2. echo "y" | sudo tee /sys/module/bluetooth/parameters/disable_ertm
3. connect the gamepad (without xpadneo)
4. * udevadm info --attribute-walk /dev/input/js0
    * upload the output

josuetleoro, 2018-06-21:
It is not a problem. I have to thank you for your time. You can find the output you required below
[udevadm_info_output.txt](https://github.com/atar-axis/xpadneo/files/2122212/udevadm_info_output.txt)
If you need anything else just let me know. And do not worry about the kernel driver, I am not planning to update any time soon :) . If in the future you would like to test in any other kernel version, please let me know. I can use the exact same system with other kernel versions. I have multiple machines to test. 

By the way, I saw that you created a project for WiFi support. I have an Elite version of the controller that I would love to be able to use. I read that some people have tried to implement the driver for the XBox One Wireless Adapter but gave up because there is not a good driver for the Mediatek mt7612u which is the one used in the adapter. This repo had some progress on it:

https://github.com/bytewarrior/xbox-one-wireless-adapter

But the project was abandoned because the Bluetooth version of the controller came out. If I can be of any help there also let me know.

atar-axis, 2018-06-21:
Thank you for your patience!

The output is somehow strange, because it looks exactly like it should...
In theory, the installed udev.rule should definitely fire on your system... Is it possible that the output is not the same every time? Especially the first section:

```
  looking at device '/devices/pci0000:00/0000:00:14.0/usb1/1-6/1-6:1.0/bluetooth/hci0/hci0:1/0005:045E:02FD.0001':
    KERNEL=="0005:045E:02FD.0001"
    SUBSYSTEM=="hid"
    DRIVER=="hid-generic"
    ATTR{country}=="21"
```

Could you maybe run the following commands and tell me if the first section alternates?

* `echo "y" | sudo tee /sys/module/bluetooth/parameters/disable_ertm`
* repeat for a couple of times:
  * connect the gamepad (without xpadneo)
  * `udevadm info --attribute-walk /sys/bus/hid/devices/0005:045E:02FD.00<HIT_TAB>/`

If it remains unchanged, could you then...

* replace the content of `/etc/udev/rules.d/99-xpadneo.rules` with
  ```
  ACTION=="add", KERNEL=="0005:045E:02FD.*|0005:045E:02E0.*", SUBSYSTEM=="hid", 
  DRIVER=="hid-generic", \
  RUN+="/bin/bash -c 'echo xpadneo: add w driver start > /dev/kmsg'", \
  RUN+="/bin/bash -c 'echo $kernel > /sys/bus/hid/drivers/hid-generic/unbind'", \
  RUN+="/bin/bash -c 'echo $kernel > /sys/bus/hid/drivers/xpadneo/bind'", \
  RUN+="/bin/bash -c 'echo xpadneo: add end > /dev/kmsg'", \
  GOTO="xpadneo_end"
  
  ACTION=="add", KERNEL=="0005:045E:02FD.*|0005:045E:02E0.*", SUBSYSTEM=="hid", \
  RUN+="/bin/bash -c 'echo xpadneo: add wo driver start > /dev/kmsg'", \
  RUN+="/bin/bash -c 'echo $kernel > /sys/bus/hid/drivers/hid-generic/unbind'", \
  RUN+="/bin/bash -c 'echo $kernel > /sys/bus/hid/drivers/xpadneo/bind'", \
  RUN+="/bin/bash -c 'echo xpadneo: add end > /dev/kmsg'", \
  GOTO="xpadneo_end"
  
  ACTION=="bind", KERNEL=="0005:045E:02FD.*|0005:045E:02E0.*", SUBSYSTEM=="hid", 
  DRIVER=="hid-generic", \
  RUN+="/bin/bash -c 'echo xpadneo: bind w driver start > /dev/kmsg'", \
  RUN+="/bin/bash -c 'echo $kernel > /sys/bus/hid/drivers/hid-generic/unbind'", \
  RUN+="/bin/bash -c 'echo $kernel > /sys/bus/hid/drivers/xpadneo/bind'", \
  RUN+="/bin/bash -c 'echo xpadneo: bind end > /dev/kmsg'", \
  GOTO="xpadneo_end"
  
  ACTION=="bind", KERNEL=="0005:045E:02FD.*|0005:045E:02E0.*", SUBSYSTEM=="hid", 
  DRIVER=="hid-generic", \
  RUN+="/bin/bash -c 'echo xpadneo: bind start > /dev/kmsg'", \
  RUN+="/bin/bash -c 'echo $kernel > /sys/bus/hid/drivers/hid-generic/unbind'", \
  RUN+="/bin/bash -c 'echo $kernel > /sys/bus/hid/drivers/xpadneo/bind'", \
  RUN+="/bin/bash -c 'echo xpadneo: bind event end > /dev/kmsg'", \
  GOTO="xpadneo_end"
  
  ACTION=="bind", KERNEL=="0005:045E:02FD.*|0005:045E:02E0.*", SUBSYSTEM=="hid", 
  DRIVER=="hid-generic", \
  RUN+="/bin/bash -c 'echo xpadneo: bind wo driver start > /dev/kmsg'", \
  RUN+="/bin/bash -c 'echo $kernel > /sys/bus/hid/drivers/hid-generic/unbind'", \
  RUN+="/bin/bash -c 'echo $kernel > /sys/bus/hid/drivers/xpadneo/bind'", \
  RUN+="/bin/bash -c 'echo xpadneo: bind end > /dev/kmsg'", \
  GOTO="xpadneo_end"
  
  LABEL="xpadneo_end"
  ```
* run `sudo udevadm control --reload`
* reconnect the gamepad and observe dmesg, upload the output

We can talk about the wifi dongle once the issue is solved :)

Thank you in advance!

atar-axis, 2018-06-21:
You can skip the first part, just do the latter and change the content of `99-xpadneo.rules`, ...
I think I know where the problem is:

atar-axis, 2018-06-22:
If I am right, then the following rule should work I think

```
ACTION=="bind", KERNEL=="0005:045E:02FD.*|0005:045E:02E0.*", SUBSYSTEM=="hid", 
DRIVER=="hid-generic", \
RUN+="/bin/bash -c 'echo $kernel > /sys/bus/hid/drivers/hid-generic/unbind'", \
RUN+="/bin/bash -c 'echo $kernel > /sys/bus/hid/drivers/xpadneo/bind'"
```

josuetleoro, 2018-06-22:
Sorry for the long wait. I could not test the udev rule sooner.

About hidp and hid_generic not being loaded yet, it actually makes sense on my system. I forgot to explain a little bit more, sorry about that. The system I am working on is a mobile robot. Therefore, there is no keyboard or mouse connected. I just tested to first connect a keyboard, to make sure hidp and hid_generic are loaded, and indeed this time the xpadneo driver is loaded correctly. This solves the mystery on my system :laughing:.

I really do not understand how to write Linux drivers or udev rules. But I am eager to learn if you have any useful source for me to read. Actually, I would like to learn more about writing Linux drivers and help in any way I can. Please let me know if there is any solution for my setup. Is forcing hidp and hid_generic to load at boot time a solution?

Thanks for all your help,

Josue

atar-axis, 2018-06-22:
Hey Josue, the solution is:

1. change the content of `/etc/udev/rules.d/99-xpadneo.rules` to:
```
ACTION=="add", KERNEL=="0005:045E:02FD.*|0005:045E:02E0.*", SUBSYSTEM=="hid",\
RUN+="/bin/bash -c 'echo $kernel > /sys/bus/hid/drivers/hid-generic/unbind'", \
RUN+="/bin/bash -c 'echo $kernel > /sys/bus/hid/drivers/xpadneo/bind'"
```
2. run `udevadm control --reload`

should work then, at least it does here on my test system - aaaaand it makes sense xD

josuetleoro, 2018-06-22:
It works! Thanks a lot. 

Sorry to bother you, just one more question. If I want to do the same in a different system (Most probably another mobile robot) I just have to add the rule above? Do I need to change anything in the rule?.

atar-axis, 2018-06-22:
Perfect! I will update the sourcecoe, so all you have to do is downloading the new driver ;)
But not before tomorrow - so long: yes, just do the same on the other system.

Btw.: Updating your system to at least kernel 4.16 would have solved the problem too, because they changes something in HID.ko (hid-core) - we do not need the udev rule anymore on more recent systems, but I am really happy that we fixed that bug anyhow.

josuetleoro, 2018-06-22:
I will wait for the new version then. Your driver works really well  by the way. Great job!

Hedronmx, 2018-10-04:
I am having problems with 4.18, lights com on but no vibration.
What do you need to troubleshoot this problem?

I had not issue running on 4.17

atar-axis, 2018-10-04:
please open a new issue and fill the template