atar-axis/xpadneo #40: Force Feedback not working

x0rzist, 2018-08-30:
I hope the problem is not on my side, but I don't get Force Feedback to work over Bluetooth on Arch Linux. The controller rumbles as expected when connectedbut it doesn't work in any game. When I tried to test it with fftest I found out that ther isn't any entry for the Controller in `/dev/input/by-id`.

When connected over USB Force Feedback is working as expected and an entry under `/dev/input/by-id` is present. 

I've attached the debug log file.
[xpadneo_log.txt](https://github.com/atar-axis/xpadneo/files/2335964/xpadneo_log.txt)

atar-axis, 2018-09-01:
Could you please upload a full dmesg log, there is nothing in yours which would explain that behaviour. Thanks

x0rzist, 2018-09-14:
Sorry, I didn't get back to it until now. And what can I say? It suddenly works now and I don't know what led to that. 
And thank you for wour great work.

atar-axis, 2018-09-14:
Perfect, you are welcome 😄