atar-axis/xpadneo #42: Problems with vibration

blacksparrow123, 2018-09-22:
**Describe the bug**
Sometimes controller freezes during vibration sequences.
And it also keeps sending last input signals.
(Character keep moving, constantly attacking, etc)

Sometimes controller just keeps vibrating, while working properly.

**To Reproduce**
No idea.
Happens to me during  

**Expected behavior**
Not freezing.

**Screenshots**
Not possible.

**System information:**
Linux anton-pc 4.14.70-1-MANJARO #1 SMP PREEMPT Sat Sep 15 20:38:15 UTC 2018 x86_64 GNU/Linux

- `dmesg` output
I could't file module required module in /sys/ folder.

**Additional context**
Installed via AUR.
If connected via usb, controller work fine.

atar-axis, 2018-09-23:
what does
modinfo hid-xpadneo
say?

atar-axis, 2018-09-23:
btw, there is another way to enable debugging output ;)

echo "options hid_xpadneo debug_level=3" | sudo tee /etc/modprobe.d/xpadneo.conf

blacksparrow123, 2018-09-25:
modinfo hid-xpadneo
filename:       /lib/modules/4.14.71-1-MANJARO/kernel/drivers/hid/hid-xpadneo.ko
version:        0.3.1
description:    Linux kernel driver for Xbox ONE S+ gamepads (BT), incl. FF
author:         Florian Dollinger <dollinger.florian@gmx.de>
license:        GPL
srcversion:     4F42AA1953F33EC02168084
alias:          hid:b0005g*v0000045Ep000002E0
alias:          hid:b0005g*v0000045Ep000002FD
depends:        hid,ff-memless
retpoline:      Y
name:           hid_xpadneo
vermagic:       4.14.71-1-MANJARO SMP preempt mod_unload modversions 
parm:           debug_level:(u8) Debug information level: 0 (none) to 3+ (most verbose). (byte)
parm:           dpad_to_buttons:(bool) Map the DPAD-buttons as BTN_DPAD_UP/RIGHT/DOWN/LEFT instead of as a hat-switch. Restart device to take effect. (bool)
parm:           trigger_rumble_damping:(u8) Damp the trigger: 1 (none) to 2^8+ (max) (byte)
parm:           fake_dev_version:(u16) Fake device version # to hide from SDL's mappings. 0x0001-0xFFFF: fake version, others: keep original (ushort)

[xpadneo_log.txt](https://github.com/atar-axis/xpadneo/files/2415971/xpadneo_log.txt)

atar-axis, 2018-09-25:
Under which circumstances have you logged the dmesg output?
Because there is not a single ``playing effect: ...`` message in it ;)

blacksparrow123, 2018-09-26:
Ow, sorry.
I send a proper dmesg later.

I played a few games via pcsx2.

atar-axis, 2018-09-26:
Have you ever tried any other gamepad - which has rumble features - in PCSX2?

blacksparrow123, 2018-09-26:
I tried dualshock 3, but it didn't work.
Honestly, ds3 interaction with were very buggy.
I couldn't even assign buttons properly. This is the driver issue, so far as i know.

atar-axis, 2018-09-26:
Sounds more like a PCSX2 to me, but I will give it a try anyway. Waiting for the dmesg now ;)

blacksparrow123, 2018-09-28:
[xpadneo_log.txt](https://github.com/atar-axis/xpadneo/files/2429344/xpadneo_log.txt)
Done this dmesg while pad froze and started vibrating.

blacksparrow123, 2018-09-29:
Whoops.
Wrong file. This one is correct.
[xpadneo_log.txt](https://github.com/atar-axis/xpadneo/files/2430940/xpadneo_log.txt)
Sorry. :(

atar-axis, 2018-10-02:
Hum, looks normal to me. Have you ever tried any other game outside PCSX2? (Like Steam)

blacksparrow123, 2018-10-06:
I did.
There are some sync issue or something.
Input stops working properly and pad keeps vibrating for too long.

But only if i go about 1m away from dongle.
When siting next to it, everything works fine.
The only issue is that vibration is too strong.

DS3 for example works fine even 2m away.

blacksparrow123, 2018-10-07:
:(
Well, i guess i'll be playing with a usb for now.
Thanks your help anyway.

May switching dongle to better one help?

atar-axis, 2018-10-07:
unfortunately you cannot use the dongle in linux since it is not yet supported by any driver.
have you ever tried to play in any other room or computer?

blacksparrow123, 2018-10-07:
I meant bluetooth adapter.

No, i didn't.
But i have notebook with build-in bluetooth.
So i might test it later.

closed 2018-10-18 by atar-axis