atar-axis/xpadneo #45: Controller recognized but doesn't work

johanpostema, 2018-09-26:
My controller seems to be recognized, but doesn't work or respond at all.
The strange thing is that the white X led keeps blinking instead of turning solid after connecting.
Basically the led keeps blinking and the controller automatically turns off after a while.

The controller does work properly when connecting it with my macbook.

Below is the dmesg with debug level set to 3.

Any ideas?

[   44.834234] xpadneo: hello there!
[  143.600005] Bluetooth: HIDP (Human Interface Emulation) ver 1.2
[  143.600101] Bluetooth: HIDP socket layer initialized
[  226.623696] hid-generic 0005:045E:02E0.0005: unknown main item tag 0x0
[  226.633900] input: Xbox Wireless Controller 7EED80192272 as /devices/platform/soc/3f201000.serial/tty/ttyAMA0/hci0/hci0:12/0005:045E:02E0.0005/input/input1
[  226.634837] hid-generic 0005:045E:02E0.0005: input,hidraw2: BLUETOOTH HID v9.03 Gamepad [Xbox Wireless Controller 7EED80192272] on b8:27:eb:87:e1:45
[  226.837865] xpadneo 0005:045E:02E0.0005: probing device: Xbox Wireless Controller 7EED80192272
[  226.837893] xpadneo 0005:045E:02E0.0005: REPORT (DESCRIPTOR) FIXUP HOOK
[  226.837905] xpadneo: report-descr: 05 01 09 05 a1 01 85 01 09 01 a1 00 09 30 09 31 15 00 27 ff ff 00 00 95 02 75 10 81 02 c0 09 01
[  226.837912] xpadneo: report-descr: a1 00 09 33 09 34 15 00 27 ff ff 00 00 95 02 75 10 81 02 c0 05 01 09 32 15 00 26 ff 03 95 01 75
[  226.837920] xpadneo: report-descr: 0a 81 02 15 00 25 00 75 06 95 01 81 03 05 01 09 35 15 00 26 ff 03 95 01 75 0a 81 02 15 00 25 00
[  226.837927] xpadneo: report-descr: 75 06 95 01 81 03 05 01 09 39 15 01 25 08 35 00 46 3b 01 66 14 00 75 04 95 01 81 42 75 04 95 01
[  226.837934] xpadneo: report-descr: 15 00 25 00 35 00 45 00 65 00 81 03 05 09 19 01 29 0a 15 00 25 01 75 01 95 0a 81 02 15 00 25 00
[  226.837940] xpadneo: report-descr: 75 06 95 01 81 03 05 01 09 80 85 02 a1 00 09 85 15 00 25 01 95 01 75 01 81 02 15 00 25 00 75 07
[  226.837947] xpadneo: report-descr: 95 01 81 03 c0 05 0f 09 21 85 03 a1 02 09 97 15 00 25 01 75 04 95 01 91 02 15 00 25 00 75 04 95
[  226.837954] xpadneo: report-descr: 01 91 03 09 70 15 00 25 64 75 08 95 04 91 02 09 50 66 01 10 55 0e 15 00 26 ff 00 75 08 95 01 91
[  226.837960] xpadneo: report-descr: 02 09 a7 15 00 26 ff 00 75 08 95 01 91 02 65 00 55 00 09 7c 15 00 26 ff 00 75 08 95 01 91 02 c0
[  226.837966] xpadneo: report-descr: 85 04 05 06 09 20 15 00 26 ff 00 75 08 95 01 81 02 c0 00
[  226.839725] xpadneo 0005:045E:02E0.0005: unknown main item tag 0x0
[  226.839811] xpadneo 0005:045E:02E0.0005: driver:
[  226.839821] xpadneo 0005:045E:02E0.0005: * version: 0.3.1
[  226.839829] xpadneo 0005:045E:02E0.0005: hdev:
[  226.839838] xpadneo 0005:045E:02E0.0005: * raw rdesc: (unfixed, see above)
[  226.839850] xpadneo 0005:045E:02E0.0005: * raw rsize: 307
[  226.839859] xpadneo 0005:045E:02E0.0005: * bus: 0x0005
[  226.839870] xpadneo 0005:045E:02E0.0005: * report group: 1
[  226.839881] xpadneo 0005:045E:02E0.0005: * vendor: 0x0000045E
[  226.839889] xpadneo 0005:045E:02E0.0005: * version: 0x00000903
[  226.839897] xpadneo 0005:045E:02E0.0005: * product: 0x000002E0
[  226.839908] xpadneo 0005:045E:02E0.0005: * country: 33
[  226.839916] xpadneo 0005:045E:02E0.0005: * driverdata: 0
[  226.839927] xpadneo 0005:045E:02E0.0005: * serial: 3a:66:33:3a:38:63
[  226.839938] xpadneo 0005:045E:02E0.0005: * physical location: 3a:37:32:3a:38:62
[  226.839969] xpadneo 0005:045E:02E0.0005: UP: 0x10000, USG: 0x0030 -> EV: 0x003, INP: 0x000
[  226.839985] xpadneo 0005:045E:02E0.0005: UP: 0x10000, USG: 0x0031 -> EV: 0x003, INP: 0x001
[  226.840015] xpadneo 0005:045E:02E0.0005: UP: 0x10000, USG: 0x0033 -> EV: 0x003, INP: 0x003
[  226.840027] xpadneo 0005:045E:02E0.0005: UP: 0x10000, USG: 0x0034 -> EV: 0x003, INP: 0x004
[  226.840042] xpadneo 0005:045E:02E0.0005: UP: 0x10000, USG: 0x0032 -> EV: 0x003, INP: 0x002
[  226.840054] xpadneo 0005:045E:02E0.0005: UP: 0x10000, USG: 0x0035 -> EV: 0x003, INP: 0x005
[  226.840064] xpadneo 0005:045E:02E0.0005: UP: 0x10000, USG: 0x0039 -> automatically
[  226.840081] xpadneo 0005:045E:02E0.0005: UP: 0x90000, USG: 0x0001 -> EV: 0x001, INP: 0x130
[  226.840092] xpadneo 0005:045E:02E0.0005: UP: 0x90000, USG: 0x0002 -> EV: 0x001, INP: 0x131
[  226.840103] xpadneo 0005:045E:02E0.0005: UP: 0x90000, USG: 0x0003 -> EV: 0x001, INP: 0x133
[  226.840114] xpadneo 0005:045E:02E0.0005: UP: 0x90000, USG: 0x0004 -> EV: 0x001, INP: 0x134
[  226.840153] xpadneo 0005:045E:02E0.0005: UP: 0x90000, USG: 0x0005 -> EV: 0x001, INP: 0x136
[  226.840164] xpadneo 0005:045E:02E0.0005: UP: 0x90000, USG: 0x0006 -> EV: 0x001, INP: 0x137
[  226.840175] xpadneo 0005:045E:02E0.0005: UP: 0x90000, USG: 0x0007 -> EV: 0x001, INP: 0x13A
[  226.840188] xpadneo 0005:045E:02E0.0005: UP: 0x90000, USG: 0x0008 -> EV: 0x001, INP: 0x13B
[  226.840199] xpadneo 0005:045E:02E0.0005: UP: 0x90000, USG: 0x0009 -> EV: 0x001, INP: 0x13D
[  226.840210] xpadneo 0005:045E:02E0.0005: UP: 0x90000, USG: 0x000A -> EV: 0x001, INP: 0x13E
[  226.840221] xpadneo 0005:045E:02E0.0005: UP: 0x10000, USG: 0x0085 -> EV: 0x001, INP: 0x13C
[  226.840232] xpadneo 0005:045E:02E0.0005: UP: 0x60000, USG: 0x0020 -> ignored
[  226.840246] xpadneo 0005:045E:02E0.0005: INPUT CONFIGURED HOOK
[  226.840256] xpadneo 0005:045E:02E0.0005: Fake device version: 0x1130
[  226.840573] input: Xbox Wireless Controller 7EED80192272 as /devices/platform/soc/3f201000.serial/tty/ttyAMA0/hci0/hci0:12/0005:045E:02E0.0005/input/input2
[  226.841205] xpadneo 0005:045E:02E0.0005: input,hidraw2: BLUETOOTH HID v9.03 Gamepad [Xbox Wireless Controller 7EED80192272] on b8:27:eb:87:e1:45
[  227.831631] xpadneo 0005:045E:02E0.0005: battery registered

atar-axis, 2018-09-26:
Try the following:
`sudo bluetoothctl`
`[bluetooth]# devices`

Is there any Xbox Wireless Controller liste? (C8:3F:26:XX:XX:XX)

If yes, remove it:
`[bluetooth]# remove C8:3F:26:<...>`

johanpostema, 2018-09-26:
Yes it's listed when I run devices. 

 But when I remove it I assume I have push the little connect button and run pair <mac>, trust <mac> and connect <mac> again while the driver is loaded? 
When I do that I get again the exact same result.

atar-axis, 2018-09-26:
Yes, please remove it and turn off the gamepad for a while.

Afterwards, do what you always do - after running 'trust ...' and 'connect ...', what does the console say?

atar-axis, 2018-09-26:
After removing it, switch off the gamepad for a little while and reboot, just to be sure.
After that, do what you always do. What does bluetoothctl say after 'connect ...'?

atar-axis, 2018-09-26:
When it is connected, but is still blinking, try to push the pair button just for a second or so. What hapens?

johanpostema, 2018-09-26:
So here's what I did, I removed the device and rebooted. Loaded the driver and set debugging to 3. My controller is turned of and the bateteries are removed for +/- 5 minutes
```
[bluetooth]# devices
[bluetooth]#
```
Now I first put the batteries back in the controller, push the little connect button and the X starts to blink fast. Inside bluetoothctl I do:
```
[bluetooth]# scan on
Discovery started
<... some other devices...>
[NEW] Device C8:3F:26:17:EB:71 Xbox Wireless Controller 7EED80192272
[CHG] Device 46:56:1F:80:F8:E8 ManufacturerData Key: 0x004c
[CHG] Device 46:56:1F:80:F8:E8 ManufacturerData Value: 0x0c
[CHG] Device 46:56:1F:80:F8:E8 ManufacturerData Value: 0x0e
<some more>
[bluetooth]# pair C8:3F:26:17:EB:71
Attempting to pair with C8:3F:26:17:EB:71
[CHG] Device C8:3F:26:17:EB:71 Connected: yes
[CHG] Device C8:3F:26:17:EB:71 Modalias: usb:v045Ep02E0d0903
[CHG] Device C8:3F:26:17:EB:71 UUIDs: 0000111e-0000-1000-8000-00805f9b34fb
[CHG] Device C8:3F:26:17:EB:71 UUIDs: 00001124-0000-1000-8000-00805f9b34fb
[CHG] Device C8:3F:26:17:EB:71 UUIDs: 00001200-0000-1000-8000-00805f9b34fb
[CHG] Device C8:3F:26:17:EB:71 ServicesResolved: yes
[CHG] Device C8:3F:26:17:EB:71 Paired: yes
Pairing successful
[CHG] Device C8:3F:26:17:EB:71 ServicesResolved: no
[CHG] Device C8:3F:26:17:EB:71 Connected: no
[CHG] Device 18:EE:69:0D:23:31 RSSI: -93
[CHG] Device 18:EE:69:0D:23:31 RSSI: -83
[CHG] Device 46:56:1F:80:F8:E8 ManufacturerData Key: 0x004c
[CHG] Device 46:56:1F:80:F8:E8 ManufacturerData Value: 0x0c
<...>
[bluetooth]# trust C8:3F:26:17:EB:71
[CHG] Device C8:3F:26:17:EB:71 Trusted: yes
Changing C8:3F:26:17:EB:71 trust succeeded
[CHG] Device 6C:32:36:1A:C3:4D RSSI: -59
[CHG] Device E5:0C:3A:1A:41:5B RSSI: -86
<...>
[Xbox Wireless Controller 7EED80192272]# connect C8:3F:26:17:EB:71
Attempting to connect to C8:3F:26:17:EB:71
[CHG] Device C8:3F:26:17:EB:71 ServicesResolved: yes
Connection successful
```

Now in dmesg I can see similar output as in my first post, but the controller doesn't respond or work. The X starts to blink slowly like it's not paired or something and turns of automatically after a while.

Now my controller is off for a few minutes and the batteries are removed, just in case.

johanpostema, 2018-09-26:
when I press it while it's blinking, nothing happens; it keeps blinking with the same rate.
Also nothing is added in dmesg, no additional lines.
When I hold the button long enough it shuts down.

First I thought it was perhaps the firmware in the controller, but when I connect it with my macbook the blinking light turns solid immediately after I connect.

johanpostema, 2018-09-26:
Unortunately I don't have a windows PC or an xbox, so I can't flash the firmware right now. 
But I'll take the controller somewhere in the next days to someone with windows in order to update the controller. I have to say that I have this controller for quite a while, so it might have an old firmware.

atar-axis, 2018-09-26:
> The X starts to blink slowly like it's not paired or something and turns of automatically after a while.

Does it start to blink slowly immediately after the 'connect' command?

johanpostema, 2018-09-26:
yes

johanpostema, 2018-09-26:
sorry, that's a no. It keeps flashing quickly after it's connected

atar-axis, 2018-09-26:
That's super strange.
Could you please try to hit the pair  button for a second INSTEAD of the 'connect' command?

johanpostema, 2018-09-26:
So I removed it first, then did pair + trust.
Turned of the controller as it didn't connect, also it didn't connect when pressing the pair button.
But after I turned off/on the controller, it automatically connects; but keeps blinking slowly

johanpostema, 2018-09-26:
I need to catch some sleep now. Ill try to update the firmware in the coming days and will see if the behaviour changes. I will update this thread to let you know. Tnx!

atar-axis, 2018-09-26:
If you can wait, I would prefer to find out why it does not work on your device instead of updating to the other fw. I am looking for a gamepad which still has the old fw for quite a long time now.

What I was asking for is the following (sorry I was on my phone before, what's why the answer was a bit short):
* `remove` the device as before
* `pair` and `trust` it
* now, while it is still online but not yet connected, press the little black button on the upper left side just for a second or so.

That's what you did, right?

atar-axis, 2018-10-02:
No problem, I was busy too ;)
Yes for sure, you have to set it into pairing mode and scan for it before - I forgot to mention that.

Before debugging, let us make sure that you have really done the following:
* `remove` the device
* set it into pairing mode by pressing and holding the upper little black button until the gamepad starts flashing fast
* `scan on`
* `pair ...` and `trust ...`
* do *not* `connect`
* press the **little black button on the upper left side** (the pairing button) just for a second or so. you wrote before that "When I hold the button long enough it shuts down." - was it maybe the (x) button you pressed?

If this does not work too, the next step would be to try another computer which has linux installed - just to make sure that it is not you system which is faulty. :smile:

johanpostema, 2018-10-18:
I tried the above as mentioned before, but doesn't work. Other linux computer same effect.
I will now update the firmware in the coming days and see if it makes a difference since it's an old firmware. Will post the results here.

atar-axis, 2018-10-19:
Thank you for your answer!
Your gamepad does behave really strange, even your HID report descriptor is totally different from what I ahave seen so far, it really looks like you have the old FW.

```
05 01 09 05 a1 01 85 01 09 01 a1 00 09 30 09 31 15 00 27 ff
ff 00 00 95 02 75 10 81 02 c0 09 01 a1 00 09 33 09 34 15 00
27 ff ff 00 00 95 02 75 10 81 02 c0 05 01 09 32 15 00 26 ff
03 95 01 75 0a 81 02 15 00 25 00 75 06 95 01 81 03 05 01 09
35 15 00 26 ff 03 95 01 75 0a 81 02 15 00 25 00 75 06 95 01
81 03 05 01 09 39 15 01 25 08 35 00 46 3b 01 66 14 00 75 04
95 01 81 42 75 04 95 01 15 00 25 00 35 00 45 00 65 00 81 03
05 09 19 01 29 0a 15 00 25 01 75 01 95 0a 81 02 15 00 25 00
75 06 95 01 81 03 05 01 09 80 85 02 a1 00 09 85 15 00 25 01
95 01 75 01 81 02 15 00 25 00 75 07 95 01 81 03 c0 05 0f 09
21 85 03 a1 02 09 97 15 00 25 01 75 04 95 01 91 02 15 00 25
00 75 04 95 01 91 03 09 70 15 00 25 64 75 08 95 04 91 02 09
50 66 01 10 55 0e 15 00 26 ff 00 75 08 95 01 91 02 09 a7 15
00 26 ff 00 75 08 95 01 91 02 65 00 55 00 09 7c 15 00 26 ff
00 75 08 95 01 91 02 c0
```

I will decrypt it later and take a look at it, looks similar to the USB descriptor at a first glance.

But the really weird thing is that your gamepad's LED is not switching to the continuos light. This is not really something which this driver is responsible for since the connection itself (via BT) is established by the kernel, not the (relatively abstract and high-level HID drivers).

Anyway, it is a bug nonetheless - one which we should fix if possible.

Can you tell me something about the other linux OS which you tried?

johanpostema, 2018-10-19:
I seem to be on firmware version "3.1.1209.0". Now Im setting up a windows 10 vm somewhere to be able to upgrade the firmware.
I tried different versions of raspbian on a few different raspberry pi's (pi 3, pi zero).

atar-axis, 2018-10-19:
Okay, would be helpful if you could capture the "connection procedure" using Wireshark before.

atar-axis, 2018-10-19:
Your decrypted report descriptor:

```
0x05, 0x01,        // Usage Page (Generic Desktop Ctrls)
0x09, 0x05,        // Usage (Game Pad)
0xA1, 0x01,        // Collection (Application)
0x85, 0x01,        //   Report ID (1)
0x09, 0x01,        //   Usage (Pointer)
0xA1, 0x00,        //   Collection (Physical)
0x09, 0x30,        //     Usage (X)
0x09, 0x31,        //     Usage (Y)
0x15, 0x00,        //     Logical Minimum (0)
0x27, 0xFF, 0xFF, 0x00, 0x00,  //     Logical Maximum (65534)
0x95, 0x02,        //     Report Count (2)
0x75, 0x10,        //     Report Size (16)
0x81, 0x02,        //     Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
0xC0,              //   End Collection
0x09, 0x01,        //   Usage (Pointer)
0xA1, 0x00,        //   Collection (Physical)
0x09, 0x33,        //     Usage (Rx)
0x09, 0x34,        //     Usage (Ry)
0x15, 0x00,        //     Logical Minimum (0)
0x27, 0xFF, 0xFF, 0x00, 0x00,  //     Logical Maximum (65534)
0x95, 0x02,        //     Report Count (2)
0x75, 0x10,        //     Report Size (16)
0x81, 0x02,        //     Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
0xC0,              //   End Collection
0x05, 0x01,        //   Usage Page (Generic Desktop Ctrls)
0x09, 0x32,        //   Usage (Z)
0x15, 0x00,        //   Logical Minimum (0)
0x26, 0xFF, 0x03,  //   Logical Maximum (1023)
0x95, 0x01,        //   Report Count (1)
0x75, 0x0A,        //   Report Size (10)
0x81, 0x02,        //   Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
0x15, 0x00,        //   Logical Minimum (0)
0x25, 0x00,        //   Logical Maximum (0)
0x75, 0x06,        //   Report Size (6)
0x95, 0x01,        //   Report Count (1)
0x81, 0x03,        //   Input (Const,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
0x05, 0x01,        //   Usage Page (Generic Desktop Ctrls)
0x09, 0x35,        //   Usage (Rz)
0x15, 0x00,        //   Logical Minimum (0)
0x26, 0xFF, 0x03,  //   Logical Maximum (1023)
0x95, 0x01,        //   Report Count (1)
0x75, 0x0A,        //   Report Size (10)
0x81, 0x02,        //   Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
0x15, 0x00,        //   Logical Minimum (0)
0x25, 0x00,        //   Logical Maximum (0)
0x75, 0x06,        //   Report Size (6)
0x95, 0x01,        //   Report Count (1)
0x81, 0x03,        //   Input (Const,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
0x05, 0x01,        //   Usage Page (Generic Desktop Ctrls)
0x09, 0x39,        //   Usage (Hat switch)
0x15, 0x01,        //   Logical Minimum (1)
0x25, 0x08,        //   Logical Maximum (8)
0x35, 0x00,        //   Physical Minimum (0)
0x46, 0x3B, 0x01,  //   Physical Maximum (315)
0x66, 0x14, 0x00,  //   Unit (System: English Rotation, Length: Centimeter)
0x75, 0x04,        //   Report Size (4)
0x95, 0x01,        //   Report Count (1)
0x81, 0x42,        //   Input (Data,Var,Abs,No Wrap,Linear,Preferred State,Null State)
0x75, 0x04,        //   Report Size (4)
0x95, 0x01,        //   Report Count (1)
0x15, 0x00,        //   Logical Minimum (0)
0x25, 0x00,        //   Logical Maximum (0)
0x35, 0x00,        //   Physical Minimum (0)
0x45, 0x00,        //   Physical Maximum (0)
0x65, 0x00,        //   Unit (None)
0x81, 0x03,        //   Input (Const,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
0x05, 0x09,        //   Usage Page (Button)
[... 48 lines cut ...]
0x91, 0x02,        //     Output (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position,Non-volatile)
0x09, 0x50,        //     Usage (0x50)
0x66, 0x01, 0x10,  //     Unit (System: SI Linear, Time: Seconds)
0x55, 0x0E,        //     Unit Exponent (-2)
0x15, 0x00,        //     Logical Minimum (0)
0x26, 0xFF, 0x00,  //     Logical Maximum (255)
0x75, 0x08,        //     Report Size (8)
0x95, 0x01,        //     Report Count (1)
0x91, 0x02,        //     Output (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position,Non-volatile)
0x09, 0xA7,        //     Usage (0xA7)
0x15, 0x00,        //     Logical Minimum (0)
0x26, 0xFF, 0x00,  //     Logical Maximum (255)
0x75, 0x08,        //     Report Size (8)
0x95, 0x01,        //     Report Count (1)
0x91, 0x02,        //     Output (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position,Non-volatile)
0x65, 0x00,        //     Unit (None)
0x55, 0x00,        //     Unit Exponent (0)
0x09, 0x7C,        //     Usage (0x7C)
0x15, 0x00,        //     Logical Minimum (0)
0x26, 0xFF, 0x00,  //     Logical Maximum (255)
0x75, 0x08,        //     Report Size (8)
0x95, 0x01,        //     Report Count (1)
0x91, 0x02,        //     Output (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position,Non-volatile)
0xC0,              //   End Collection
```

johanpostema, 2018-10-19:
I have 23k pcapng file which started with a clean setup (gamepad remove).
I started wireshark, started bluetoothctl, scan on, pair/trust/connect waited a while (pressed some buttons) untill the gamepad turned off again. According to bluetoothctl it was connected successfull,
Can I upload this pcapng somewhere?

johanpostema, 2018-10-19:
[scan_pair_trust_connect_xbox_one_s_controller.pcapng.gz](https://github.com/atar-axis/xpadneo/files/2495881/scan_pair_trust_connect_xbox_one_s_controller.pcapng.gz)

johanpostema, 2018-10-21:
I just updated the firmware and suddenly it started to work with Linux and the driver.

atar-axis, 2018-10-22:
I'm hald it works now, I will keep it in mind and update the README accordingly.