atar-axis/xpadneo #46: Your kernel headers for kernel 4.14.50-v7+ cannot be found

patrickemuller, 2018-10-03:
**Describe the bug**
Your kernel headers for kernel 4.14.50-v7+ cannot be found at
/lib/modules/4.14.50-v7+/build or /lib/modules/4.14.50-v7+/source.

**To Reproduce**
Just try to install using README

**Expected behavior**
When running `./install.sh` should install correctly

**Screenshots**
[image]

**System information:**
- `uname -a` Linux raspberrypi 4.14.50-v7+ #1122 SMP Tue Jun 19 12:26:26 BST 2018 armv7l GNU/Linux
- `dmesg` ... a very long log ...
  Please enable debugging output before: https://atar-axis.github.io/xpadneo/index.html#debugging

patrickemuller, 2018-10-03:
Closing this issue due to miss leading information that I provided.
I did update the raspberry packages, but forget to UPGRADE and reboot the hardware.

closed 2018-10-03 by patrickemuller

atar-axis, 2018-10-03:
perfect :)