atar-axis/xpadneo #47: Not detecting device at all

rektboi, 2018-10-04:
Hi,
So I followed the instructions, I did the ./install.sh and rebooted.
I have attached the install log for the record.
[install.log](https://github.com/atar-axis/xpadneo/files/2446104/install.log)

$ sudo bluetoothctl
[NEW] Controller 40:E2:30:3C:52:7E 5960x [default]
Agent registered
[bluetooth]# scan on
Discovery started
[CHG] Controller 40:E2:30:3C:52:7E Discovering: yes

(now I press the connect button, light starts flashing fast but no device appears.)
Running Linux Mint 19 fwiw.

What am I missing? Thanks in advance.

atar-axis, 2018-10-04:
does any other device appear?

rektboi, 2018-10-05:
I should have checked that before posting, my bad. Now I have been able to connect a phone and a mouse, but still the controller doesn't show up.

atar-axis, 2018-10-06:
what is the result if you type `devices`?

closed 2018-10-18 by atar-axis