atar-axis/xpadneo #48: Vibration not working

rcp87, 2018-10-26:
Hi, I follow the steps and install correctly the drivers, then I connect my xbox one controller by bluetooth following all steps, all OK, the gamepad vibrates on connect (and the triggers vibration too) without problem.

The problem is that vibration is not working in any game from Steam. I tested several games: Bastion, FEZ, Oddworld new and tasty, Outlast, Metro 2033 Redux...

I dont see anything wrong, all mappings are correct in all games, all axes, the triggers... all ok, except there is not vibration. I have been digging into the issues section but with no success...

I am running Steam on Manjaro Linux, with kernel 4.14.78-1. 
Any help would be appreciated, thanks for your time and congratulations for this great project for linux gamers :)

`uname -a` output:
Linux Cimmeria 4.14.78-1-MANJARO #1 SMP PREEMPT Sun Oct 21 07:57:51 UTC 2018 x86_64 GNU/Linux

- `dmesg` output:
https://pastebin.com/5hVDXzFN

rcp87, 2018-10-26:
I dont know if this could help but i forgot to comment that in Steam settings, if I click in Detect controller, the gamepad vibrates correctly, so this is very strange because is not working in all games.

rcp87, 2018-10-26:
Sorry i closed the issue accidentally :)

atar-axis, 2018-10-26:
hi there bernard ;)
could you please upload a dmesg log which is recorded while playing a game (one which has FF effects).

atar-axis, 2018-10-27:
Hey, unforunately there is not a single FF event in it, that means that Steam simply does not request one. Are you würd that those games have support for FF (I'm not a gamer so I don't know :D)

rcp87, 2018-10-27:
Yeah all those games have FF support. Mmm so Im sure that the problem is with Steam games, because the FF works when i connect the gamepad... It is strange because it vibrates too when i click on 'Identify' in steam controller settings.
I will try another game for more time to see if we are lucky and an event raise... By the way, you know if i can try the FF in other way?

rcp87, 2018-10-27:
I've doing more tests and i have a conclusion more or less. I realize that in game FEZ the vibration is working! But i test all the others games more time and in anyone can get it working.

One of these games is Oddworld: New and tasty. So i boot up windows and install this game to see a FF event, and when Abe chants the controller is vibrating continuously. So then i came back to linux and see that vibration not working in any time.

So i don't think is a problem with the drivers or with Steam. The problem as I understand is that the support of FF by bluetooth is very poor by the game developers in linux...

Its a pitty, because i love linux and i love playing games with this controller. 
I put the output of the dmesg when i was playing FEZ where i think you have to see the FF event:
https://pastebin.com/z6h1DD9J
Just out of curiosity, what is the event of the FF in the log??

atar-axis, 2018-10-27:
Hey there,
yeah that's what I was meant :)

For sude, one of those events is:
```[  431.291929] xpadneo 0005:045E:02FD.0005: playing effect: strong: 0x00, weak: 0x6666, direction: 0x00
[  431.291931] xpadneo 0005:045E:02FD.0005: max: 0x66, prop_left: 0x00, prop_right: 0x00, left trigger: 0x00, right: 0x00```

:)

closed 2018-10-27 by atar-axis

atar-axis, 2018-10-29:
you are welcome!