atar-axis/xpadneo #49: Disable rumble/vibration globally?

ipkpjersi, 2018-11-02:
**Is your feature request related to a problem? Please describe.**
I like rumble/vibration sometimes but other times I just want to turn it off. Turning it off on a per-game basis can be a bit tedious.

**Describe the solution you'd like**
I think it would be better if there was an option to toggle rumble at a global level and just disable it for all games instead of like say having to disable it on a per-game basis. Is this something that would be possible to implement in this project? This would literally make this project perfect for me.

atar-axis, 2018-11-02:
Hi there!
Shouldn't be a problem to add an option like that, controlling it would be the same then as it is for the other options (like trigger_dampening).

I will give it a try and get back to you within one or two days I think.

Cheers

atar-axis, 2018-11-02:
Hey there, would you please give it a try? https://github.com/atar-axis/xpadneo/tree/disable_ff

atar-axis, 2018-11-02:
ups, sorry the previous post was faulty - I updated it.

ipkpjersi, 2018-11-02:
Hi,

I just got home and tried it, and it says this: 

`` echo 1 | sudo tee /sys/module/hid_xpadneo/parameters/disable_ff
tee: /sys/module/hid_xpadneo/parameters/disable_ff: No such file or directory
1``

I noticed that in `/sys/module/` I only have `hid/` and `hid_generic/`, do you have any ideas why this might be?

atar-axis, 2018-11-02:
hey, could you please give me the output of `modinfo hid_xpadneo` and `lsmod | grep xpadneo`? looks like the driver is not loaded, in that case simply run `modprobe hid_xpadneo` (or connect the gamepad in advance).

ipkpjersi, 2018-11-02:
Hi,

``
filename:       /lib/modules/4.15.0-38-generic/updates/dkms/hid-xpadneo.ko
version:        0.3.2
description:    Linux kernel driver for Xbox ONE S+ gamepads (BT), incl. FF
author:         Florian Dollinger <dollinger.florian@gmx.de>
license:        GPL
srcversion:     4BF83FF3399184AEBF22919
alias:          hid:b0005g*v0000045Ep000002E0
alias:          hid:b0005g*v0000045Ep000002FD
depends:        hid,ff-memless
retpoline:      Y
name:           hid_xpadneo
vermagic:       4.15.0-38-generic SMP mod_unload 
parm:           debug_level:(u8) Debug information level: 0 (none) to 3+ (most verbose). (byte)
parm:           dpad_to_buttons:(bool) Map the DPAD-buttons as BTN_DPAD_UP/RIGHT/DOWN/LEFT instead of as a hat-switch. Restart device to take effect. (bool)
parm:           disable_ff:(bool) Disable all force-feedback effects (rumble). 1: disable ff, 0: enable ff. (bool)
parm:           trigger_rumble_damping:(u8) Damp the trigger: 1 (none) to 2^8+ (max) (byte)
parm:           fake_dev_version:(u16) Fake device version # to hide from SDL's mappings. 0x0001-0xFFFF: fake version, others: keep original (ushort)
``

``
hid_xpadneo            20480  0
ff_memless             16384  1 hid_xpadneo
hid                   118784  5 hidp,usbhid,hid_generic,hid_xpadneo,razermouse
``

I ran `modprobe hid_xpadneo` then ran the echo command, and it seems to have worked now. do you have any idea why it might not have loaded manually - and if there is anything I can do to load it automatically in the future?

I have a suggestion. I think you should include an `enable_vibration.sh` and `disable_vibration.sh` script, or possibly even just a `toggle_vibration.sh` script in the repo that way people can easily toggle vibration.

Also, when I turned on the controller - it still vibrated when I turned it on. Does that mean it is not working? Also, do I need to re-disable vibration upon reboot? Do you think there might be a way to automate that?

Thanks.

atar-axis, 2018-11-03:
a lot of questions^^

> I ran modprobe hid_xpadneo then ran the echo command, and it seems to have worked now. do you have any idea why it might not have loaded manually - and if there is anything I can do to load it automatically in the future?

xpadneo should always load automatically as soon as the gamepad is connected. please check again if it doesn't, if - then open a new issue for that problem :sunglasses: 

> I have a suggestion. I think you should include an enable_vibration.sh and disable_vibration.sh script, or possibly even just a toggle_vibration.sh script in the repo that way people can easily toggle vibration.

Feel free to commit such a script into the misc folder, should be easy - I will surely merge it.

> Also, when I turned on the controller - it still vibrated when I turned it on. Does that mean it is not working?

The gamepad is always rumbling on a successful connection to xpadneo, even if the `disable_ff` param is set to `1`. That's how the user easily gets informed that everything is fine. I am not sure if I am going to change that... let me think about it.

> Also, do I need to re-disable vibration upon reboot? Do you think there might be a way to automate that?

Yes, you need to do that everytime again. Alternatively you can set it permanent by doing something similar as described here: https://atar-axis.github.io/xpadneo/#debugging.

ipkpjersi, 2018-11-03:
Hi,

I will start working on the scripts now, but before I do - would this be the correct syntax: `echo "options hid_xpadneo disable_ff=1" | sudo tee /etc/modprobe.d/xpadneo.conf` for permanently disabling vibration?

I think the gamepad should not rumble if disable_ff is 1 because I think it would be better to not have it rumble at all. If you could implement this, that would be great. I can tell that this driver is working and loaded simply by my controller working properly, and not moving the mouse with the left stick.

I think I might be able to add a crontab: `@reboot modprobe hid_xpadneo` to ensure that this module is loaded on reboot.

Thanks again.

atar-axis, 2018-11-03:
> I will start working on the scripts now, but before I do - would this be the correct syntax: echo "options hid_xpadneo disable_ff=1" | sudo tee /etc/modprobe.d/xpadneo.conf for permanently disabling vibration?

Yes, it is.

> I think the gamepad should not rumble if disable_ff is 1 because I think it would be better to not have it rumble at all. If you could implement this, that would be great.

I absolutely see your point but - well, as said - I am not sure if this is how I want it to behave :grin: I will sleep a night over it before I decide. Thank you for you suggestion anyway.

> I think I might be able to add a crontab: @reboot modprobe hid_xpadneo to ensure that this module is loaded on reboot.

Possibly, but I would prefer to find the root cause if the driver is not loaded automatically. Please open a new issue and let us find the cause together. It is most probably because of a failing UDEV rule since you are on a kernel version before 4.16.

ipkpjersi, 2018-11-03:
Hi,

I have an idea - what about another option for disabling vibration on connect? I could also add that to the script.

I just realized there's no issue - it actually is loading the module when I turn the controller on.

Thanks.

atar-axis, 2018-11-03:
That would in the end lead to too much parameters. I disabled the rumble at connecting as you suggested, please give it a try and tell me if it works as expected.

ipkpjersi, 2018-11-03:
Hi,

It worked perfectly, thank you so much!! Everything is working perfectly just the way I like it, and even the config file to disable vibration works. I'll work on the script right now and then open a PR.

Just out of curiosity, do you happen to know if this driver works on Ubuntu 18.04? I'm hoping to upgrade to Ubuntu 18.04 next year when it is more stable, and I'd love for this to continue working on Ubuntu 18.04.

Also, an even more random question - do you happen to know if the Xbox One controller vibrates when connected in Windows (10)?

Thanks once again for all your help and hard work.

atar-axis, 2018-11-03:
Perfect :sunglasses:

> Just out of curiosity, do you happen to know if this driver works on Ubuntu 18.04? I'm hoping to upgrade to Ubuntu 18.04 next year when it is more stable, and I'd love for this to continue working on Ubuntu 18.04.

Yes it does!

> Also, an even more random question - do you happen to know if the Xbox One controller vibrates when connected in Windows (10)?

Hum, I believe it vibrates yes, but it's relatively short as far as I remember. But I am not sure.

> Thanks once again for all your help and hard work.

You are welcome!

ipkpjersi, 2018-11-03:
Hi,

I have created the script and it works perfectly for me. Feel free to test it out. Will all these changes including my script be making it into the master branch?

Thanks again!

atar-axis, 2018-11-03:
Yes, I will merge the branches after some testing :)
Regarding your PR: Thank you for the script! But i have a favor to ask you: could you please use the `disable_ff` branch instead of `master` for it - on both, you fork and the pull request. Thanks!

atar-axis, 2018-11-03:
Perfect, I just moved it to misc/tools ;) Thanks

atar-axis, 2018-11-03:
Just another question:
https://github.com/atar-axis/xpadneo/commit/b8ed79b260edf7fa20546ffb0320cc247e5000a9#r31157137

atar-axis, 2018-11-03:
give it a try: f12fb69c4d05773f016f55bcbc84fb479427e2c6

ipkpjersi, 2018-11-03:
Hi,

I'm giving it a quick look first, in enable_ff () I think you need to `rm 
-f /sys/module/hid_xpadneo/parameters/disable_ff` for it to work instantly, don't you?

Thanks.

atar-axis, 2018-11-03:
> I'm giving it a quick look first, in enable_ff () I think you need to rm -f /sys/module/hid_xpadneo/parameters/disable_ff for it to work instantly, don't you?

No, `disable_ff` is not a real file and moreover it always exists as long as the driver is loaded. What matters is only the content.

Hum, it script is yet untested, let's see...

atar-axis, 2018-11-03:
Give it a try ;)

atar-axis, 2018-11-03:
Thank you for testing!

ipkpjersi, 2018-11-03:
No problem, I am always very happy to help test such amazing software like this project. It works perfectly now, thank you!

atar-axis, 2018-11-03:
Perfect. Will let it open for some time - going to merge it after some testing is done.
Thank you again for a) your enhancement suggestions and b) testing.