atar-axis/xpadneo #55: Controller frequently disconnects whilst using

yerbestpal, 2018-11-19:
**Describe the bug**
I have this same issue on Ubuntu, Pop_OS, and now Fedora too. The install and first time connection of the controller goes smoothly. However, after using it for a while (usually while playing American Truck Simulator) the controller will disconnect at random. 

**To Reproduce**
To reproduce, it is simply a case of connecting the controller and using it.

**Expected behavior**
I expected the controller to remain connected.

**System information:**
- `uname -a` output
`Linux localhost.localdomain 4.19.2-300.fc29.x86_64 #1 SMP Wed Nov 14 19:05:24 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux
`
- `dmesg` output
https://hastebin.com/evujixajoz.coffeescript

**Additional context**
I have no idea if it could affect anything, but I also frequently connect bluetooth headphones while also using the controller, though I also experienced the problem without them.

ugly95, 2018-11-20:
I'm not sure it's the same issue, but I am having connection problems.

When I installed xpadneo, the first connection went fine.  But most of the time when I try to reconnect the controller, it takes a long time before it connects.  Sometimes it doesn't connect.  Sometimes it connects instantly.  So it's fairly random.

While it's trying to connect, if I type in `sudo bluetoothctl`, I see:
```
[CHG] Device 5C:BA:37:D4:46:6A Connected: no
[CHG] Device 5C:BA:37:D4:46:6A Connected: yes
[CHG] Device 5C:BA:37:D4:46:6A Connected: no
[CHG] Device 5C:BA:37:D4:46:6A Connected: yes
[CHG] Device 5C:BA:37:D4:46:6A Connected: no
```
etc.

Usually, if I type `connect 5C:BA:37:D4:46:6A` the controller will connect.  But every once in a while, I get `Failed to connect: org.bluez.Error.Failed`.

After it connects, it usually works fine.  Every once in a while the controller will randomly disconnect, similar to the description above.

`dmesg` output
[https://hastebin.com/ukafohoyut.coffeescript](https://hastebin.com/ukafohoyut.coffeescript)

Sorry, I don't know if this should be a separate issue.  I posted here because there is a similarity of random disconnects while using the controller.

atar-axis, 2018-11-22:
Sorry guys, I have bad news for you...
Connection problems are normally not related to this driver since it hooks at the HID layer, which is beyond the l2cap bluetooth thingy. I will give the dmesg a deeper look but I think this is the wrong place to complain about it :grin:

ipkpjersi, 2018-11-23:
I've had issues with Bluetooth disconnecting - not usually as much for my controller (I don't use my controller a ton so I haven't tested a ton though), but definitely with my Bluetooth headphones Edifier w800bt. I assumed that the cheap Bluetooth adapter I use (Orico BTA-408) or just Bluetooth in general on Linux is responsible for disconnecting - either because my distro is Ubuntu 16.04 and my kernel is 4.15 and main Linux is on like 4.20 I think, or something else. I don't think it's an issue specific to this driver. The Bluetooth random disconnects are scaring me away from buying a Bluetooth mouse even though I'd like to not have to use wired mice anymore.

I'd love any advice or suggestions anyone has, even though it's definitely out-of-scope of this project.

yerbestpal, 2018-11-23:
Most Logitech mice come with a tiny adapter specifically for them and work perfectly.

closed 2018-12-02 by atar-axis