atar-axis/xpadneo #56: Turn off Xbox One Controller (script?)

ipkpjersi, 2018-11-23:
**Is your feature request related to a problem? Please describe.**
I have to unplug the batteries to turn off my Xbox One controller when I am done playing, I believe it does not automatically disconnect. 

**Describe the solution you'd like**
I would like a script (if possible) to turn off my Xbox One controller. When I used Windows and I had an Xbox 360 controller, I had a script for turning off my Xbox 360 controller, I bound the script to a hotkey and it worked really well.

**Describe alternatives you've considered**
I have not thought of any alternatives.

**Additional context**
I am not sure if this is is even possible for this project to accomplish this, if this even operates at the level required to do this. If this is possible I'd definitely love to see this just so I don't have to unplug my batteries every time I want to turn my Xbox One controller off.

atar-axis, 2018-11-23:
Hey there,

have you tried to push the (X) button (the shiny one) for around 10 seconds?

Alternatively, the gamepad does turn off after some time when you leave it alone.

Cheers!

ipkpjersi, 2018-11-23:
Hi,

Do you mean the middle Xbox button? Does it have to be exactly 10 seconds - no more no less? I don't have an Xbox One controller with me right now (I will in 2 hours) but I don't recall that I was ever able to turn it off this way, although I don't recall counting exactly 10 seconds if that is required.

How long does it take for it to turn off? Maybe I'm just impatient but I don't remember it turning off automatically. That's definitely something I can try tonight, though.

If it isn't a lot of work, though, and if it's technically possible then I think adding in a quick script to turn it off would be pretty cool - then I could add a hotkey for turning it off on my keyboard, I'd definitely like that.

Thanks.

atar-axis, 2018-11-23:
Yes, right - the one in the middle.

I think it is around 10 seconds but it does not really matter, just push it and hold it down until the gamepad turns off.

The gamepad does definitely turn off itself after idling for a while, I think it is around 10 minutes.

Unfortunately, there is no way to turn the gp off by any command via HID over bluetooth. Thats a missing firmware feature.

Cheers

ipkpjersi, 2018-11-24:
Hi,

You were right! It does turn off after about 10 seconds, and also leaving it alone has it turn off automatically. 

This is not an issue, so I will now be closing it.

Thanks.

closed 2018-11-24 by ipkpjersi