atar-axis/xpadneo #57: Control Sticks not working

ipkpjersi, 2018-12-02:
**Describe the bug**
When I use my Xbox One controller with RPCS3, it does not work properly with the control sticks. For example, when I press left on the left control stick it looks like this: https://i.imgur.com/EEXovM7.png and when I press up on the left control stick it looks like this: https://i.imgur.com/LhhWiWt.png

Notice that it is always in the bottom right center of the stick. The other buttons work fine both in gamepad settings and in games. My Wireless 360 controller with the 360 PC adapter works perfectly 100% as expected in RPCS3.

**To Reproduce**
Steps to reproduce the behavior:
1. Download RPCS3 then run it and go to Gamepads Settings (Pads)
2. Connect your XB1 controller and change handler to Evdev
3. Move either of the control sticks in either direction.
4. See error

**Expected behavior**
I expect the control sticks to work properly.

**System information:**
Linux fast-desktop 4.15.0-39-generic #42~16.04.1-Ubuntu SMP Wed Oct 24 17:09:54 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux

**Additional Information**
Is this an RPCS3 issue or xpadneo issue?

atar-axis, 2018-12-02:
hey there!

please run `evtest /dev/input/eventX` (please replace X with the number your gamepad has - mostly the biggest one in this folder).

how does the output look like of you move the stick as you described?

ipkpjersi, 2018-12-02:
Here is what it looks like when I move the left stick to the left: 

https://hastebin.com/zovoveyawe.css

Here is what it looks like when I move the left stick up:
https://hastebin.com/vugomikegi.sql

I'm not an expert but it looks okay to me. The controller works great for games like Geometry Wars 3, Halo 1, Gears of War 1, etc.

atar-axis, 2018-12-02:
@ipkpjersi: could you please give me a log for the `evtest` output for the Wireless 360 gamepad?

atar-axis, 2018-12-02:
Hey again! I had contact to GalCiv in the support forum of RPCS3, he pointed me to the following issue: https://github.com/RPCS3/rpcs3/pull/3974

That means that the problem is on their side but there is a workaround described at the issue.

Maybe xpadneo should switch to a range from –32.768 to 32.767 in future version... I am not yet sure.
Anyway, I would be still interested in the `evtest` log for the other gamepad.

atar-axis, 2018-12-02:
Okay, I have to backpedal... looks like the problem is mainly on my side. I will definitely switch to a signed range in the next version!

I decided so because:
1) The linux gamepad spec does advice so: https://www.kernel.org/doc/html/v4.18/input/gamepad.html
2) Other drivers do that as well

ipkpjersi, 2018-12-02:
Sorry, I was sleeping while you made all this excellent progress! I'm glad you were able to narrow down the issue and will be fixing it. If you need me to test a beta build feel free to give me a holler. Thanks again for all your hard work.

atar-axis, 2018-12-03:
hey there, could you please give it a try?

`git clone https://github.com/atar-axis/xpadneo -b axis_offset`
I think you already know the other stuff like uninstalling the old version and so on...

I would really appreciate if you could test it at Steam, RPCS3 and other systems! Thanks!

atar-axis, 2018-12-04:
hey there! thanks for testing - on my machine it was working as expected. strange. I will take a closer look - brb

atar-axis, 2018-12-04:
hum, I have no idea at the moment where the problem is - it is perfectly running on my machine. the visualization is a bit slow but that's it - maybe the problem is just the visualization on your machine too? have you tried it already ingame?

ipkpjersi, 2018-12-04:
Hi,

In-game acts very strangely, it goes all the way to the left in RPCS3. I will delete all RPCS3 config files and try the latest version, I will also try other games and let you know how it goes.

Thanks.

atar-axis, 2018-12-04:
thanks! could you maybe also take a look at `evtest /dev/input/event<X> | grep ABS_RX`?

ipkpjersi, 2018-12-04:
Hi,

I tested that and both `ABS_RX` and `ABS_RY` were completely empty when I moved only the left stick.

I then tested it in Geometry Wars 3 and the button mappings were wrong - for example, pressing the start button made the game go left. I will uninstall, reboot, reinstall, reboot and then report back.

Thanks.

atar-axis, 2018-12-05:
> I tested that and both ABS_RX and ABS_RY were completely empty when I moved only the left stick, there is no output when I run that command.

It is the right stick which you need to move :grin: please compare the output (which should range from -32xxx to +32xxx) to the output of `evtest /dev/input/event<X> | grep ABS_X` while moving the *left* stick

ipkpjersi, 2018-12-05:
Here is the left stick:

https://hastebin.com/socebebili.bash

Here is the right stick:

https://hastebin.com/sabonifiku.bash

ipkpjersi, 2018-12-05:
Hi,

I'm actually not sure how to do debug_level 3

Thanks.

atar-axis, 2018-12-05:
Simply run `echo "options hid_xpadneo debug_level=3" | sudo tee /etc/modprobe.d/xpadneo.conf` and reboot :grin:

ipkpjersi, 2018-12-05:
Here is `evtest` without pressing any buttons before `debug_level 3`: https://hastebin.com/cadepezipi.bash

Here is `dmesg -w | grep shifted` first going up and down then going left and right: 
[image]

Here is `evtest` without pressing any buttons after `debug_level 3`:

atar-axis, 2018-12-05:
thank you very much! could you please give me a copy of the output of `dmesg -w | grep shifted` on pastebin? I have to analyze that

ipkpjersi, 2018-12-05:
https://hastebin.com/idijuzunan.cs

atar-axis, 2018-12-05:
anyway, looks like the problem is somewhere deeper - could you please share the whole *dmesg log* with me (please move the stick again as you described)?

ipkpjersi, 2018-12-05:
https://hastebin.com/gamusepiza.css

atar-axis, 2018-12-05:
Thanks, it will take some time to analyze that - stay tuned.

atar-axis, 2018-12-05:
I uploaded another version, but it is a shot in the dark - I don't expect it to work to be honest :grin: (not before I analyzed the dmesg)

atar-axis, 2018-12-05:
okay, I think I am on the scent.
would you please

1) pull the latest version
2) test it if it does already work
3) if not, run `dmesg -w | grep xpadneo > ~/xpadneo.log`
4) move the stick in question to the left and then to the right
5) upload ~/xpadneo.log

thanks for you patience!

ipkpjersi, 2018-12-07:
Hi,

My apologies for the delay, I have been busy.

It appears to be correct now - which definitely surprised me. 

I have attached a log just in case you need to see: 
[xpadneo.log](https://github.com/atar-axis/xpadneo/files/2655462/xpadneo.log)

Thanks.

atar-axis, 2018-12-07:
perfect, I just forgot to sync the input device - which seems to the rootcause of that strange behaviour.

thanks for testing

atar-axis, 2018-12-07:
hey there @kakra, would you mind giving it a try too? I would then merge it back if it works for you too :)

kakra, 2018-12-07:
@atar-axis Sure, wanted to reboot anyways. :-)

kakra, 2018-12-07:
Works fine for me, everything as expected. Here's the updated kernel patch I'm using:
https://github.com/kakra/linux/tree/rebase-4.19/xpadneo

atar-axis, 2018-12-07:
Perfect, thanks!

kakra, 2020-05-08:
If the issue still persists, we should bring the option back only as a module parameter as it conflicts with Wine / Proton / Steam expectations. Maybe a hwdb rule could fix it, too, but that needs a systemd installation. I'm open to any solution except hard-wiring it into the code.

ipkpjersi, 2020-05-08:
I'm still not super experienced with Linux even though I haven't used Windows in like 3 years, if this issue does persist, what would a module parameter be like?

kakra, 2020-05-08:
Something like `option hid-xpadneo shift_axis=Y` in `/etc/modprobe.conf`. We already have one for `combine_z_axis` which combines both triggers into a single axis.

ipkpjersi, 2020-05-09:
Just tested RPCS3, this issue does happen again, and yeah @kakra I think the module parameter is gonna be the best way to go.

If I understand modules correctly, I should be able to change whether this feature is enabled or disabled without having to reboot my entire computer, I can just dismod and then modprobe and that should work?

kakra, 2020-05-09:
Okay, let's reopen that. I've researched the issue a little, looked at the issue tracker of RPCS3 and at the kernel input layer specs. We should probably bring back the axis shifting and make it default. But I'd like to look at the Wine sources a little more first, they seem to craft their own HID description layers around what SDL tells about the gamepads. Maybe we can find a way to make all parties happy without manual configuration.

> If I understand modules correctly, ...

From your point of view, modules are just drivers similar to what you know from Windows. But it's called a module because the kernel is monolithic: Everything you load into the kernel will become part of the kernel and runs in kernel context / kernel space - thus it's a module (to put it simple, and xpadneo is no exception although it's not delivered as "part of the kernel"). The Windows kernel is different: It's a microkernel, it provides only a minimal set of very low-level functions, i.e. for drivers to talk to each other. Except for a few kernel level drivers, most drivers you load in Windows are services running in user space (similar to a daemon you are running in Linux). It's a completely different architectures. Windows drivers are not part of the kernel.

And actually your understanding is correct: You can just unload modules (at least most distro kernels allow that). So rebuilding xpadneo is a matter of `rmmod` followed by `modprobe` after reinstalling it. I've included a command in v0.7 that supports just that. Try running `make -C hid-xpadneo` from the source code project root (the directory you checked out with git): It should rebuild the driver (as long as `/usr/src/linux` exists), asks for the root password and reinstalls and reloads the driver.

You can even change parameters of the module while it is loaded, without unloading and reloading: Go to `/sys/modules/hid_xpadneo/parameters`: There a bunch of files and you can write parameters to those files: `echo 1 | sudo tee /sys/modules/hid_xpadneo/parameters/combine_z_axis` will turn the combine z axis feature on. The change isn't permanent across reboots/module reloads unless you put it into `/etc/modprobe.conf`.

ipkpjersi, 2020-05-09:
If this funuctionality become default (this is how I always used it for all games with Proton/Wine and of course RPCS3), I definitely think we should have the opposite functionality available as a module parameter - it's a great idea.

kakra, 2020-05-09:
Yes, but my credo is "convention over configuration". That means everything should work by default while still leaving everything open for configuration. So sure you'll get your parameters but I want to have it in a way that requires no manual configuration for the usual user.

kakra, 2020-05-21:
Note to myself:

- [ ] Investigate how this interacts with the Wine HID driver for gamepads

closed 2020-07-11 by kakra