atar-axis/xpadneo #61: Xpadneo on retropie "parse failed"

CommunicationAnimale, 2018-12-28:
**Describe the bug**
I'm trying to use my Xbox One controller on a Retropie. When I use my controller with a wire (and xpad drivers), I experience mapping issues, but I can navigate menus, play games, etc...
I therefore decided to install xpadneo :+1: 
<details>
  <summary> Click arrow to reveal</summary>

```
pi@retropie:~/src/xpadneo $ sudo ./install.sh
* copying module into /usr/src
* adding module to DKMS

Creating symlink /var/lib/dkms/hid-xpadneo/0.5.0/source ->
                 /usr/src/hid-xpadneo-0.5.0

DKMS: add completed.
* installing module (using DKMS)

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area....
make -j4 KERNELRELEASE=4.14.79-v7+ -C /lib/modules/4.14.79-v7+/build M=/var/lib/dkms/hid-xpadneo/0.5.0/build/src modules.....
cleaning build area...

DKMS: build completed.

hid-xpadneo.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/4.14.79-v7+/kernel/drivers/hid/

Running the post_install script:
Disabling ERTM permanently (requires reboot)...
Diabling ERTM instantly...
Installing UDEV rule to automatically bind xpadneo (necessary on kernel < 4.16)...

depmod...

Warning: Unable to find an initial ram disk that I know how to handle.
Will not try to make an initrd.

DKMS: install completed.
```
</details>

I then successfully connected the controller to the Retropie in the "Bluetooth" menu (the Xbox logo lit up, the controller did not rumble however). I am unable to interact with the Retropie using my Xbox controller (and Retropie diplays "No gamepads connected" in the main menu).

There are several errors in the dmesg log (full log with debugging below):
```dmesg
[  113.575872] hid-generic 0005:045E:02FD.0003: unknown main item tag 0x0
[  113.575950] hid-generic 0005:045E:02FD.0003: unbalanced collection at end of report description
[  113.576049] hid-generic: probe of 0005:045E:02FD.0003 failed with error -22
[  113.596876] hid_xpadneo: loading out-of-tree module taints kernel.
[  113.597871] xpadneo: hello there!
[  113.600563] xpadneo 0005:045E:02FD.0003: unknown main item tag 0x0
[  113.600757] xpadneo 0005:045E:02FD.0003: unbalanced collection at end of report description
[  113.600809] xpadneo 0005:045E:02FD.0003: parse failed
[  113.600856] xpadneo: probe of 0005:045E:02FD.0003 failed with error -22
```
 

**To Reproduce**
Steps to reproduce the behavior:
1. Install Xpadneo on Retropie
2. Connect the controller using the Bluetooth menu (or bluetoothctl)
4. Try to setup the controller using the "Configure input" option from the start menu

**Expected behavior**
[... 292 lines cut ...]
[   49.156448] xpadneo: report-descr: c0 05 0f 09 21 85 03 a1 02 09 97 15 00 25 01 75 04 95 01 91 02 15 00 25 00 75 04 95 01 91 03 09
[   49.156456] xpadneo: report-descr: 70 15 00 25 64 75 08 95 04 91 02 09 50 66 01 10 55 0e 15 00 26 ff 00 75 08 95 01 91 02 09 a7 15
[   49.156462] xpadneo: report-descr: 00 26 ff 00 75 08 95 01 91 02 65 00 55 00 09 7c 15 00 00
[   49.158257] xpadneo 0005:045E:02FD.0003: unknown main item tag 0x0
[   49.158330] xpadneo 0005:045E:02FD.0003: unbalanced collection at end of report description
[   49.158423] xpadneo 0005:045E:02FD.0003: parse failed
[   49.158506] xpadneo: probe of 0005:045E:02FD.0003 failed with error -22
[   49.191394] xpadneo 0005:045E:02FD.0003: probing device: Xbox Wireless Controller
[   49.191430] xpadneo 0005:045E:02FD.0003: REPORT (DESCRIPTOR) FIXUP HOOK
[   49.191446] xpadneo: report-descr: 05 01 09 05 a1 01 85 01 09 01 a1 00 09 30 09 31 15 00 27 ff ff 00 00 95 02 75 10 81 02 c0 09 01
[   49.191453] xpadneo: report-descr: a1 00 09 32 09 35 15 00 27 ff ff 00 00 95 02 75 10 81 02 c0 05 02 09 c5 15 00 26 ff 03 95 01 75
[   49.191619] xpadneo: report-descr: 0a 81 02 15 00 25 00 75 06 95 01 81 03 05 02 09 c4 15 00 26 ff 03 95 01 75 0a 81 02 15 00 25 00
[   49.191636] xpadneo: report-descr: 75 06 95 01 81 03 05 01 09 39 15 01 25 08 35 00 46 3b 01 66 14 00 75 04 95 01 81 42 75 04 95 01
[   49.191648] xpadneo: report-descr: 15 00 25 00 35 00 45 00 65 00 81 03 05 09 19 01 29 0f 15 00 25 01 75 01 95 0f 81 02 15 00 25 00
[   49.191660] xpadneo: report-descr: 75 01 95 01 81 03 05 0c 0a 24 02 15 00 25 01 95 01 75 01 81 02 15 00 25 00 75 07 95 01 81 03 05
[   49.191675] xpadneo: report-descr: 0c 09 01 85 02 a1 01 05 0c 0a 23 02 15 00 25 01 95 01 75 01 81 02 15 00 25 00 75 07 95 01 81 03
[   49.191685] xpadneo: report-descr: c0 05 0f 09 21 85 03 a1 02 09 97 15 00 25 01 75 04 95 01 91 02 15 00 25 00 75 04 95 01 91 03 09
[   49.191694] xpadneo: report-descr: 70 15 00 25 64 75 08 95 04 91 02 09 50 66 01 10 55 0e 15 00 26 ff 00 75 08 95 01 91 02 09 a7 15
[   49.191703] xpadneo: report-descr: 00 26 ff 00 75 08 95 01 91 02 65 00 55 00 09 7c 15 00 00
[   49.193844] xpadneo 0005:045E:02FD.0003: unknown main item tag 0x0
[   49.193923] xpadneo 0005:045E:02FD.0003: unbalanced collection at end of report description
[   49.193983] xpadneo 0005:045E:02FD.0003: parse failed
[   49.194045] xpadneo: probe of 0005:045E:02FD.0003 failed with error -22
```
</details>

CommunicationAnimale, 2018-12-28:
Hmm... I'm trying to debug the issue on another computer (my laptop running Fedora) and I can't get my controller to use Xpadneo: it is being recongnised as a generic device. What's super strange is that it is broadcasting a different device ID! (`0005:045E:02E0` vs `0005:045E:02FD.0003` in the log above)
I did not try to update the firmware (my controller was already on 3.1.1221.0).

What's more it seems that [this guy](https://www.youtube.com/watch?v=lF7ZFnw_PR0) was able to use his controller on a Retropie simply by disabling ERTM (i.e. only with xpad). I tried to follow the same procedure on a clean install, without success: I'm stuck in the exact same menu ("No gamepads detected"). Therefore I think that the error I am seeing might be a regression introduced in Retropie.

atar-axis, 2018-12-29:
Hi there!

The problem seems to be that your gamepad is somehow sending a HID device descriptor which hid-xpadneo is unable to parse (even hid-generic is unable to parse it).

Could you do me a favor and give me the output of
`/sys/kernel/debug/hid/0005:045E:<TAB>/rdesc`

CommunicationAnimale, 2018-12-29:
Thank you for the fast reply !

Here is the content of file:
```console
root@retropie:~# cat "/sys/kernel/debug/hid/0005:045E:02FD.0001/rdesc"
05 01 09 05 a1 01 85 01 09 01 a1 00 09 30 09 31 15 00 27 ff ff 00 00 95 02 75 10 81 02 c0 09 01 a1 00 09 32 09 35 15 00 27 ff ff 00 00 95 02 75 10 81 02 c0 05 02 09 c5 15 00 26 ff 03 95 01 75 0a 81 02 15 00 25 00 75 06 95 01 81 03 05 02 09 c4 15 00 26 ff 03 95 01 75 0a 81 02 15 00 25 00 75 06 95 01 81 03 05 01 09 39 15 01 25 08 35 00 46 3b 01 66 14 00 75 04 95 01 81 42 75 04 95 01 15 00 25 00 35 00 45 00 65 00 81 03 05 09 19 01 29 0f 15 00 25 01 75 01 95 0f 81 02 15 00 25 00 75 01 95 01 81 03 05 0c 0a 24 02 15 00 25 01 95 01 75 01 81 02 15 00 25 00 75 07 95 01 81 03 05 0c 09 01 85 02 a1 01 05 0c 0a 23 02 15 00 25 01 95 01 75 01 81 02 15 00 25 00 75 07 95 01 81 03 c0 05 0f 09 21 85 03 a1 02 09 97 15 00 25 01 75 04 95 01 91 02 15 00 25 00 75 04 95 01 91 03 09 70 15 00 25 64 75 08 95 04 91 02 09 50 66 01 10 55 0e 15 00 26 ff 00 75 08 95 01 91 02 09 a7 15 00 26 ff 00 75 08 95 01 91 02 65 00 55 00 09 7c 15 00 00

root@retropie:~#
```

atar-axis, 2018-12-29:
hey there, no problemo...

I am a bit busy right now but I found the problem, your descriptor is definitely corrupted:

unknown (yours):
```
05 01 09 05 a1 01 85 01 09 01 a1 00 09 30 09 31 15 00 27 ff ff 00 00 95 02 75 10 81 02 c0 09 01 a1 00 09 32 09 35 15 00 27 ff ff 00 00 95 02 75 10 81 02 c0 05 02 09 c5 15 00 26 ff 03 95 01 75 0a 81 02 15 00 25 00 75 06 95 01 81 03 05 02 09 c4 15 00 26 ff 03 95 01 75 0a 81 02 15 00 25 00 75 06 95 01 81 03 05 01 09 39 15 01 25 08 35 00 46 3b 01 66 14 00 75 04 95 01 81 42 75 04 95 01 15 00 25 00 35 00 45 00 65 00 81 03 05 09 19 01 29 0f 15 00 25 01 75 01 95 0f 81 02 15 00 25 00 75 01 95 01 81 03 05 0c 0a 24 02 15 00 25 01 95 01 75 01 81 02 15 00 25 00 75 07 95 01 81 03 05 0c 09 01 85 02 a1 01 05 0c 0a 23 02 15 00 25 01 95 01 75 01 81 02 15 00 25 00 75 07 95 01 81 03 c0 05 0f 09 21 85 03 a1 02 09 97 15 00 25 01 75 04 95 01 91 02 15 00 25 00 75 04 95 01 91 03 09 70 15 00 25 64 75 08 95 04 91 02 09 50 66 01 10 55 0e 15 00 26 ff 00 75 08 95 01 91 02 09 a7 15 00 26 ff 00 75 08 95 01 91 02 65 00 55 00 09 7c 15 00 00
```
windows:
```
05 01 09 05 a1 01 85 01 09 01 a1 00 09 30 09 31 15 00 27 ff ff 00 00 95 02 75 10 81 02 c0 09 01 a1 00 09 33 09 34 15 00 27 ff ff 00 00 95 02 75 10 81 02 c0 05 01 09 32 15 00 26 ff 03 95 01 75 0a 81 02 15 00 25 00 75 06 95 01 81 03 05 01 09 35 15 00 26 ff 03 95 01 75 0a 81 02 15 00 25 00 75 06 95 01 81 03 05 01 09 39 15 01 25 08 35 00 46 3b 01 66 14 00 75 04 95 01 81 42 75 04 95 01 15 00 25 00 35 00 45 00 65 00 81 03 05 09 19 01 29 0a 15 00 25 01 75 01 95 0a 81 02 15 00 25 00 75 06 95 01 81 03 05 01 09 80 85 02 a1 00 09 85 15 00 25 01 95 01 75 01 81 02 15 00 25 00 75 07 95 01 81 03 c0 05 0f 09 21 85 03 a1 02 09 97 15 00 25 01 75 04 95 01 91 02 15 00 25 00 75 04 95 01 91 03 09 70 15 00 25 64 75 08 95 04 91 02 09 50 66 01 10 55 0e 15 00 26 ff 00 75 08 95 01 91 02 09 a7 15 00 26 ff 00 75 08 95 01 91 02 65 00 55 00 09 7c 15 00 26 ff 00 75 08 95 01 91 02 c0 85 04 05 06 09 20 15 00 26 ff 00 75 08 95 01 81 02 c0 00
```
android:
```
05 01 09 05 a1 01 85 01 09 01 a1 00 09 30 09 31 15 00 27 ff ff 00 00 95 02 75 10 81 02 c0 09 01 a1 00 09 32 09 35 15 00 27 ff ff 00 00 95 02 75 10 81 02 c0 05 02 09 c5 15 00 26 ff 03 95 01 75 0a 81 02 15 00 25 00 75 06 95 01 81 03 05 02 09 c4 15 00 26 ff 03 95 01 75 0a 81 02 15 00 25 00 75 06 95 01 81 03 05 01 09 39 15 01 25 08 35 00 46 3b 01 66 14 00 75 04 95 01 81 42 75 04 95 01 15 00 25 00 35 00 45 00 65 00 81 03 05 09 19 01 29 0f 15 00 25 01 75 01 95 0f 81 02 15 00 25 00 75 01 95 01 81 03 05 0c 0a 24 02 15 00 25 01 95 01 75 01 81 02 15 00 25 00 75 07 95 01 81 03 05 0c 09 01 85 02 a1 01 05 0c 0a 23 02 15 00 25 01 95 01 75 01 81 02 15 00 25 00 75 07 95 01 81 03 c0 05 0f 09 21 85 03 a1 02 09 97 15 00 25 01 75 04 95 01 91 02 15 00 25 00 75 04 95 01 91 03 09 70 15 00 25 64 75 08 95 04 91 02 09 50 66 01 10 55 0e 15 00 26 ff 00 75 08 95 01 91 02 09 a7 15 00 26 ff 00 75 08 95 01 91 02 65 00 55 00 09 7c 15 00 26 ff 00 75 08 95 01 91 02 c0 85 04 05 06 09 20 15 00 26 ff 00 75 08 95 01 81 02 c0 00
```

atar-axis, 2018-12-29:
may I close the issue?

closed 2018-12-29 by atar-axis

CommunicationAnimale, 2018-12-29:
Yes, thank you. I updated the firmware and everything seems to work normally.
Thank you again, I learned many things today :+1:

atar-axis, 2018-12-29:
perfect - may I ask you if it is another version now?

you are welcome!

CommunicationAnimale, 2018-12-30:
No, I'm still using 3.1.1221. I used "Xbox accessories" on Windows 10; and selected "Update firmware". Apparently it did something (the logo on my controller blinked), but it told me I was already using the latest version.