atar-axis/xpadneo #62: Kernel lockdown prevents Xpadneo from starting

CommunicationAnimale, 2018-12-30:
**Describe the bug**
When kernel_lockdown is enabled Xpadneo fails to start with the error `Lockdown: systemd-udevd: Loading of unsigned module is restricted; see man kernel_lockdown.7` in `dmesg`.
Kernel_lockdown is enabled by default when secure boot is enabled.

**To Reproduce**
Steps to reproduce the behavior:
1.a Enable secure boot
1.b Connect gamepad with bluetoothctl, it is recognised as a generic device

2.a Disable kernel_lockdown.
  * Using Sysrq: [on Fedora: `echo "1" > /proc/sys/kernel/sysrq`](https://fedoraproject.org/wiki/QA/Sysrq#How_do_I_enable_the_magic_SysRq_key.3F) to enable temporarily [Sysrq key combinations](https://en.wikipedia.org/wiki/Magic_SysRq_key), then press `sysrq + alt + x`. This will print `Lifting lockdown` in `dmesg`
  * Simply disable secure boot, it will consequently disable kernel lockdown
2.b Connect gamepad with bluetoothctl, it rumbles and xpadneo shows up in dmesg

**Expected behavior**
Xpadneo should be used whether lockdown is enabled or not.

**System information**
*Note*: when kernel_lockdown is enables, debugfs is unavailable. For instance, it is not possible to `cat` in /sys/kernel/debug: `Lockdown: cat: debugfs is restricted; see man kernel_lockdown.7` shows up in `dmesg`.

- `uname -a` output
```
Linux 7480-fedora 4.19.10-300.fc29.x86_64 #1 SMP Mon Dec 17 15:34:44 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux
```
- `dmesg` output
<details>

```
[Dec30 13:48] Linux version 4.19.10-300.fc29.x86_64 (mockbuild@bkernel03.phx2.fedoraproject.org) (gcc version 8.2.1 20181105 (Red Hat 8.2.1-5) (GCC)) #1 SMP Mon Dec 17 15:34:44 UTC 2018
[  +0.000000] Command line: BOOT_IMAGE=/vmlinuz-4.19.10-300.fc29.x86_64 root=/dev/mapper/fedora-root ro resume=/dev/mapper/fedora-swap rd.lvm.lv=fedora/root rd.lvm.lv=fedora/swap rhgb quiet LANG=en_US.UTF-8
[  +0.000000] x86/fpu: Supporting XSAVE feature 0x001: 'x87 floating point registers'
[  +0.000000] x86/fpu: Supporting XSAVE feature 0x002: 'SSE registers'
[  +0.000000] x86/fpu: Supporting XSAVE feature 0x004: 'AVX registers'
[  +0.000000] x86/fpu: Supporting XSAVE feature 0x008: 'MPX bounds registers'
[  +0.000000] x86/fpu: Supporting XSAVE feature 0x010: 'MPX CSR'
[  +0.000000] x86/fpu: xstate_offset[2]:  576, xstate_sizes[2]:  256
[  +0.000000] x86/fpu: xstate_offset[3]:  832, xstate_sizes[3]:   64
[  +0.000000] x86/fpu: xstate_offset[4]:  896, xstate_sizes[4]:   64
[  +0.000000] x86/fpu: Enabled xstate features 0x1f, context size is 960 bytes, using 'compacted' format.
[  +0.000000] BIOS-provided physical RAM map:
[  +0.000000] BIOS-e820: [mem 0x0000000000000000-0x0000000000057fff] usable
[  +0.000000] BIOS-e820: [mem 0x0000000000058000-0x0000000000058fff] reserved
[  +0.000000] BIOS-e820: [mem 0x0000000000059000-0x000000000009dfff] usable
[  +0.000000] BIOS-e820: [mem 0x000000000009e000-0x00000000000fffff] reserved
[  +0.000000] BIOS-e820: [mem 0x0000000000100000-0x00000000bb6b3fff] usable
[  +0.000000] BIOS-e820: [mem 0x00000000bb6b4000-0x00000000bb6b4fff] ACPI NVS
[  +0.000000] BIOS-e820: [mem 0x00000000bb6b5000-0x00000000bb6b5fff] reserved
[  +0.000000] BIOS-e820: [mem 0x00000000bb6b6000-0x00000000ca503fff] usable
[  +0.000000] BIOS-e820: [mem 0x00000000ca504000-0x00000000ca879fff] reserved
[  +0.000000] BIOS-e820: [mem 0x00000000ca87a000-0x00000000ca8bbfff] ACPI data
[  +0.000000] BIOS-e820: [mem 0x00000000ca8bc000-0x00000000caf5ffff] ACPI NVS
[  +0.000000] BIOS-e820: [mem 0x00000000caf60000-0x00000000cb5fefff] reserved
[  +0.000000] BIOS-e820: [mem 0x00000000cb5ff000-0x00000000cb5fffff] usable
[  +0.000000] BIOS-e820: [mem 0x00000000cb600000-0x00000000cf7fffff] reserved
[  +0.000000] BIOS-e820: [mem 0x00000000f0000000-0x00000000f7ffffff] reserved
[  +0.000000] BIOS-e820: [mem 0x00000000fe000000-0x00000000fe010fff] reserved
[  +0.000000] BIOS-e820: [mem 0x00000000fec00000-0x00000000fec00fff] reserved
[  +0.000000] BIOS-e820: [mem 0x00000000fee00000-0x00000000fee00fff] reserved
[  +0.000000] BIOS-e820: [mem 0x00000000ff000000-0x00000000ffffffff] reserved
[  +0.000000] BIOS-e820: [mem 0x0000000100000000-0x000000022e7fffff] usable
[  +0.000000] NX (Execute Disable) protection: active
[  +0.000000] e820: update [mem 0xb9522018-0xb953fa57] usable ==> usable
[  +0.000000] e820: update [mem 0xb9522018-0xb953fa57] usable ==> usable
[  +0.000000] extended physical RAM map:
[  +0.000000] reserve setup_data: [mem 0x0000000000000000-0x0000000000057fff] usable
[  +0.000000] reserve setup_data: [mem 0x0000000000058000-0x0000000000058fff] reserved
[  +0.000000] reserve setup_data: [mem 0x0000000000059000-0x000000000009dfff] usable
[  +0.000000] reserve setup_data: [mem 0x000000000009e000-0x00000000000fffff] reserved
[  +0.000000] reserve setup_data: [mem 0x0000000000100000-0x00000000b9522017] usable
[  +0.000000] reserve setup_data: [mem 0x00000000b9522018-0x00000000b953fa57] usable
[... 975 lines cut ...]
[  +0.000061] input: Xbox Wireless Controller as /devices/pci0000:00/0000:00:14.0/usb1/1-7/1-7:1.0/bluetooth/hci0/hci0:256/0005:045E:02FD.0003/input/input22
[  +0.004563] xpadneo 0005:045E:02FD.0003: input,hidraw1: BLUETOOTH HID v9.03 Gamepad [Xbox Wireless Controller] on 1c:4d:70:4c:54:45
[  +0.990402] xpadneo 0005:045E:02FD.0003: battery registered
[  +1.317779] xpadneo: raw_event: 04 86
[  +0.000013] xpadneo 0005:045E:02FD.0003: hid-up: 60000, hid-usg: 20, input-code: 00, value: 86
[  +0.000003] xpadneo 0005:045E:02FD.0003: REPORT HOOK
[  +6.109833] Bluetooth: hci0: last event is not cmd complete (0x0f)
[  +5.488112] xpadneo 0005:045E:02FD.0003: Goodbye Xbox Wireless Controller!
[ +12.433888] Bluetooth: hci0: last event is not cmd complete (0x0f)
```
</details>
  
The following command need superuser privileges.
I recommend to run it as root directly (not using `sudo`), since this way the <TAB>-key will work as expected.
- `head -1 "/sys/kernel/debug/hid/0005:045E:<TAB>/rdesc" | tee /dev/tty | cksum`

```console
[root@7480-fedora ~]# head -1 "/sys/kernel/debug/hid/0005:045E:02FD.0004/rdesc" | tee /dev/tty | cksum
05 01 09 05 a1 01 85 01 09 01 a1 00 09 30 09 31 15 00 27 ff ff 00 00 95 02 75 10 81 02 c0 09 01 a1 00 09 32 09 35 15 00 27 ff ff 00 00 95 02 75 10 81 02 c0 05 02 09 c5 15 00 26 ff 03 95 01 75 0a 81 02 15 00 25 00 75 06 95 01 81 03 05 02 09 c4 15 00 26 ff 03 95 01 75 0a 81 02 15 00 25 00 75 06 95 01 81 03 05 01 09 39 15 01 25 08 35 00 46 3b 01 66 14 00 75 04 95 01 81 42 75 04 95 01 15 00 25 00 35 00 45 00 65 00 81 03 05 09 19 01 29 0f 15 00 25 01 75 01 95 0f 81 02 15 00 25 00 75 01 95 01 81 03 05 0c 0a 24 02 15 00 25 01 95 01 75 01 81 02 15 00 25 00 75 07 95 01 81 03 05 0c 09 01 85 02 a1 01 05 0c 0a 23 02 15 00 25 01 95 01 75 01 81 02 15 00 25 00 75 07 95 01 81 03 c0 05 0f 09 21 85 03 a1 02 09 97 15 00 25 01 75 04 95 01 91 02 15 00 25 00 75 04 95 01 91 03 09 70 15 00 25 64 75 08 95 04 91 02 09 50 66 01 10 55 0e 15 00 26 ff 00 75 08 95 01 91 02 09 a7 15 00 26 ff 00 75 08 95 01 91 02 65 00 55 00 09 7c 15 00 26 ff 00 75 08 95 01 91 02 c0 85 04 05 06 09 20 15 00 26 ff 00 75 08 95 01 81 02 c0 00
2579218694 1006
[root@7480-fedora ~]#
```

**Additional context**  
There are other reports of dkms modules failing to load when lockdown is enabled: e.g. https://bugzilla.redhat.com/show_bug.cgi?id=1599197. I don't understand everything in this bug report, but it seems that the maintainer says that this is not a bug in the kernel.

atar-axis, 2019-01-01:
First of all: Your bug reports are just awesome, precise and rich of useful information - thanks!
I will dig a bit deeper and let you know when I found the problem.

atar-axis, 2019-01-01:
Could you please give me the output of `modinfo hid_xpadneo` on your system?

atar-axis, 2019-01-01:
Btw, have you tried to self-sign it (as a temporary workaround)?
Like described here in these articles:

1. https://access.redhat.com/documentation/en-US/Red_Hat_Enterprise_Linux/7/html/Kernel_Administration_Guide/sect-signing-kernel-modules-for-secure-boot.html
2. https://github.com/Canonical-kernel/Ubuntu-kernel/blob/master/Documentation/module-signing.txt
3. https://askubuntu.com/questions/760671/could-not-load-vboxdrv-after-upgrade-to-ubuntu-16-04-and-i-want-to-keep-secur

atar-axis, 2019-01-01:
Here is a more automatic solution: https://computerlinguist.org/make-dkms-sign-kernel-modules-for-secure-boot-on-ubuntu-1604.html

I am not yet sure if this is something I can fix on the source-side.

atar-axis, 2019-01-03:
Unfortunately the script is not compatible with **ARCH** (since there is no `mokutil`), which is the linux distribution I am using. That's at least a hint that the script is not as portable as the maintainer hoped.

I cannot test the script therefore and cannot (and don't want to) add it to the driver, not as long as we haven't found a solution which works on (at least nearly) every distribution.

Furthermore, I never used `secure boot` or `kernel_lockdown`, I am therefore not experienced in this field - but what I found is that it somehow looks like there is not even a consistent opinion on how good or evil those techniques really are.

I think that the best way at the moment is to add some instructions to the README and hope for the future that Dell, who maintains DKMS, is going to add some signing functionality in the future. What do you think?

CommunicationAnimale, 2019-01-03:
Personally, I'm not even using Secure Boot anymore as I discovered yesterday that disabling it fixes a longstanding issue I had with hibernation... And you're completely right, feelings are mixed regarding Secure Boot.

I'm not sure that enabling auto-signing by default is a good idea, since it will be poorly tested. Once someone overcomes the hurdle of generating and enrolling the keys, enabling auto-signing is as simple as creating "/etc/dkms/hid-xpadneo.conf".

I think the least we can do is point out that Xpadneo will not work with Secure Boot OOTB in the README.md (especially considering that Secure Boot cannot be disabled on some computers, e.g. on "Windows 8 Ready" ARM computers), and add a link to [dop3j0e's gist](https://gist.github.com/dop3j0e/2a9e2dddca982c4f679552fc1ebb18df). My only gripes are:
* It does not support signing without password. On my computer, I removed the lines related to '$PROMPT', the function 'read_passphrase', the line where this function is called and added `-nodes` to the openssl incantation, and auto-signing is now passwordless.
* It was not immediately clear to me that `<module-name>` should be replaced with `hid-xpadneo` in the case of Xpadneo.
* It does not point out that the keyboard layout used when enrolling MOKs is QWERTY.

Final thoughts: maybe we should let ditributions handle this. I stated earlier that DKMS modules installed with apt on Ubuntu would be automatically signed, but it seems that *any* DKMS module will be automatically signed. From [Ubuntu wiki](https://wiki.ubuntu.com/UEFI/SecureBoot/DKMS):
> Simply install the package you need. Packages that make use of DKMS should prompt you to create a new Machine-Owner key (it will be done for you), and will guide you through the steps to enroll that key in your system's firmware.

atar-axis, 2019-01-07:
I am still confused by this...

1. SecureBoot and Kernel_Lockdown is not the same as we all know now (at least after the heated public dicussion).

2. Linus Torvalds said that he won't accept a patch that "automatically locks down the kernel" when SecureBoot is on

3. As far as I know Kernel_Lockdown is the mechanism that prevents loading unsigned modules, access to files like /dev/kmem and so on.

So why the heck is your kernel locked down when you enable SecureBoot?
And is the lockdown even yet merged to the official kernel? Couldn't find anything in elixir.bootlin.com