atar-axis/xpadneo #63: Combine LT and RT to form a single axis

piet206, 2019-01-01:
I installed xpadneo last night and all the controller inputs are working. However for my flight simulator, I need the LT and RT to be on the same axis for yaw control. 

For example in xboxdrv, there is an option in the config:
       **--trigger-as-zaxis**
              Combine LT and RT to form a zaxis instead

Any help would be much appreciated. Thanks.

atar-axis, 2019-01-01:
Hi there, I am quite busy at the moment but it shouldn't be a big deal to realize that - I hope I will find some time within next weeks.

piet206, 2019-01-01:
Thanks for the quick reply. Let's see if I can clarify what needs to be done.

Both LT and RT should be on axis ABS_Z
LT should be 0 to -1. RT should be 0 to 1. When both triggers are at rest, the value would be 0.

atar-axis, 2019-01-02:
Ah, ok I see - so they work against each other in a sense. Are you sure about the Values (0,-1,0,1)? Normally we are talking about ranges from -32.768 to 32.767 or 0 to 65.535 for axis.

piet206, 2019-01-02:
Exactly, they work against each other. No, I'm not sure about those values (0,-1,0,1). Your typical ranges that are used for the triggers sound good to me.

atar-axis, 2019-01-03:
Hey there, I uploaded a first draft, you can test it by using
`git clone https://github.com/atar-axis/xpadneo.git -b combine_triggers"
It is not yet working as expected - I am unsure about the range and the triggers do not go back to 0 as one would hope

piet206, 2019-01-03:
Awesome. I'll give it a try when I get home in a few hours and get back to you.

piet206, 2019-01-04:
I plugged in the controller via usb and ran the evtest. First pressing and releasing LT and then doing the same with RT:

```
p@p-desktop:~$ evtest /dev/input/event15 js0
Input driver version is 1.0.1
Input device ID: bus 0x3 vendor 0x45e product 0x2ea version 0x301
Input device name: "Microsoft X-Box One S pad"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 304 (BTN_SOUTH)
    Event code 305 (BTN_EAST)
    Event code 307 (BTN_NORTH)
    Event code 308 (BTN_WEST)
    Event code 310 (BTN_TL)
    Event code 311 (BTN_TR)
    Event code 314 (BTN_SELECT)
    Event code 315 (BTN_START)
    Event code 316 (BTN_MODE)
    Event code 317 (BTN_THUMBL)
    Event code 318 (BTN_THUMBR)
  Event type 3 (EV_ABS)
    Event code 0 (ABS_X)
      Value   2360
      Min   -32768
      Max    32767
      Fuzz      16
      Flat     128
    Event code 1 (ABS_Y)
      Value   1833
      Min   -32768
      Max    32767
      Fuzz      16
      Flat     128
    Event code 2 (ABS_Z)
      Value      0
      Min        0
      Max     1023
    Event code 3 (ABS_RX)
      Value  -1513
      Min   -32768
      Max    32767
      Fuzz      16
      Flat     128
    Event code 4 (ABS_RY)
      Value    919
      Min   -32768
      Max    32767
      Fuzz      16
      Flat     128
    Event code 5 (ABS_RZ)
      Value      0
      Min        0
      Max     1023
    Event code 16 (ABS_HAT0X)
      Value      0
      Min       -1
      Max        1
    Event code 17 (ABS_HAT0Y)
      Value      0
      Min       -1
      Max        1
  Event type 21 (EV_FF)
    Event code 80 (FF_RUMBLE)
    Event code 81 (FF_PERIODIC)
    Event code 88 (FF_SQUARE)
    Event code 89 (FF_TRIANGLE)
    Event code 90 (FF_SINE)
    Event code 96 (FF_GAIN)
Properties:
Testing ... (interrupt to exit)
Event: time 1546568904.403582, type 3 (EV_ABS), code 2 (ABS_Z), value 237
Event: time 1546568904.403582, -------------- SYN_REPORT ------------
Event: time 1546568904.411575, type 3 (EV_ABS), code 2 (ABS_Z), value 707
Event: time 1546568904.411575, -------------- SYN_REPORT ------------
Event: time 1546568904.419576, type 3 (EV_ABS), code 2 (ABS_Z), value 1023
Event: time 1546568904.419576, -------------- SYN_REPORT ------------
Event: time 1546568905.299580, type 3 (EV_ABS), code 2 (ABS_Z), value 650
Event: time 1546568905.299580, -------------- SYN_REPORT ------------
Event: time 1546568905.307550, type 3 (EV_ABS), code 2 (ABS_Z), value 135
Event: time 1546568905.307550, -------------- SYN_REPORT ------------
Event: time 1546568905.315576, type 3 (EV_ABS), code 2 (ABS_Z), value 0
Event: time 1546568905.315576, -------------- SYN_REPORT ------------
Event: time 1546568906.499586, type 3 (EV_ABS), code 5 (ABS_RZ), value 181
Event: time 1546568906.499586, -------------- SYN_REPORT ------------
Event: time 1546568906.507571, type 3 (EV_ABS), code 5 (ABS_RZ), value 804
Event: time 1546568906.507571, -------------- SYN_REPORT ------------
Event: time 1546568906.515550, type 3 (EV_ABS), code 5 (ABS_RZ), value 1023
Event: time 1546568906.515550, -------------- SYN_REPORT ------------
Event: time 1546568907.799575, type 3 (EV_ABS), code 5 (ABS_RZ), value 799
Event: time 1546568907.799575, -------------- SYN_REPORT ------------
Event: time 1546568907.807569, type 3 (EV_ABS), code 5 (ABS_RZ), value 254
Event: time 1546568907.807569, -------------- SYN_REPORT ------------
Event: time 1546568907.815547, type 3 (EV_ABS), code 5 (ABS_RZ), value 0
Event: time 1546568907.815547, -------------- SYN_REPORT ------------
```
Looks like you were spot on with the 1023 value. Thanks again.

atar-axis, 2019-01-04:
oh damn, i forgot to mention that... could you please run xboxdrv with --trigger-as-zaxis?!

piet206, 2019-01-04:
I've never been able to get xboxdrv to recognize my device which is what led me to xpadneo.

```
p@p-desktop:~$ xboxdrv --trigger-as-zaxis
xboxdrv 0.8.8 - http://pingus.seul.org/~grumbel/xboxdrv/ 
Copyright © 2008-2011 Ingo Ruhnke <grumbel@gmail.com> 
Licensed under GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html> 
This program comes with ABSOLUTELY NO WARRANTY. 
This is free software, and you are welcome to redistribute it under certain 
conditions; see the file COPYING for details. 

-- [ ERROR ] ------------------------------------------------------
No Xbox or Xbox360 controller found
```

atar-axis, 2019-01-04:
Ah I see. I will dig in the sourcecode of xboxdrv then. anyway, you said it works ingame and that's what counts :)

Btw, xpadneo is a wireless only driver until now - but I may add USB support someday.

atar-axis, 2019-01-04:
I will keep it open until I merged it ;)

piet206, 2019-01-04:
Ahh I see, very good.

atar-axis, 2019-01-04:
Okay I found the line I was searching for:
```
void
UInputOptions::trigger_as_zaxis()
{
  m_axis_map.push_back(AxisMapOption("gamepad.trigger", "ABS_Z:-255:255:0:0"));

  m_axis_map.push_back(AxisMapOption("gamepad.lt", "void"));
  m_axis_map.push_back(AxisMapOption("gamepad.rt", "void"));
}
```

So the range is [-255;255] in xboxdrv - it would make much sense to use this range since we would lose a lot of accuracy, therefore I will stay with  [-1024; 1023] for now.

atar-axis, 2019-01-04:
Merged to master, please give it a try!
https://github.com/atar-axis/xpadneo#installation
https://github.com/atar-axis/xpadneo#configuration

atar-axis, 2019-01-04:
I'll close this issue for now, if you are facing any problem - feel free to reopen it.

closed 2019-01-04 by atar-axis

atar-axis, 2019-02-20:
hey @piet206 - could you please update to the latest version and check if the gamepad still works as expected? I changed something which is related to this issue: https://github.com/atar-axis/xpadneo/issues/79

piet206, 2019-03-01:
@atar-axis yep still works

atar-axis, 2019-03-01:
Perfect, thanks!

kakra, 2020-07-12:
Would you mind testing if #226 works for your use-case?