atar-axis/xpadneo #66: Microsoft Xbox Elite Wireless Controller support

atar-axis, 2019-01-03:
Self explaining.

UriHerrera, 2019-01-22:
Hey!, I just stumbled upon your driver for the Xbox One controllers, thank you for your work. By the way, I would like to support your effort for the Elite controller compatibility, do you have other means to donate to you?.

atar-axis, 2019-01-22:
> Hey!, I just stumbled upon your driver for the Xbox One controllers, thank you for your work. 
You are welcome :grin: 

> By the way, I would like to support your effort for the Elite controller compatibility, do you have other means to donate to you?.

Oh thanks!!! Any suggestions? I would prefer some service that makes it possible to show the public who donated something over somehow anonymous donations. To each saint his candle ;)

UriHerrera, 2019-01-22:
> Oh thanks! Any suggestions? I would prefer some service that makes it possible to show the public who donated something over somehow anonymous donations. To each saint his candle :wink:

There's Patreon, Indiegogo, [Eliopay](https://eliopay.com/?ref=producthunt), [Ko-fi](https://ko-fi.com/).

atar-axis, 2019-01-23:
Here we go: https://github.com/atar-axis/xpadneo/blob/master/docs/README.md
:grin:

atar-axis, 2019-01-23:
@UriHerrera Thank you!

atar-axis, 2019-02-17:
Another one (Michael N.) donated 12€! Thank you very much!

atar-axis, 2019-04-25:
Received a 12€ donation by Yu-Jen S. (username: freehaha)! Thank you very much!

atar-axis, 2019-04-28:
I am currently looking for a used one - I already saw one for 50€. I will let you know as soon as there is any progress.

atar-axis, 2019-04-30:
Someone who wants to stay anonymous donated for this project. I will therefore buy the controller and start  development withing the next weeks. Thank you for everyone who donated!

atar-axis, 2019-05-03:
[image]

Stay tuned

atar-axis, 2019-05-03:
Bad news. **The XBOX One ELITE Wireless controller** arrived and I immediately tried to connect it via bluetooth. And guess what? It does not have a bluetooth interface...

The standard **XBOX One Controller** was the one *without* bluetooth, the **XBOX One Wireless Controller** was the one *with* - so I thought, and an E3 interview I watched strengthened that assumption, that the Elite controller has Bluetooth too. But I was wrong.

So the only way to connect it to the PC is via the Wireless Dongle.

I would like to keep it and continue working on the Wireless Dongle support, once finished - which will definitely take a while or two (if even possibile) - I will add the Elite Controller too. If any of you, who donated for the Elite controller, wants your money back: please tell me.

perillamint, 2019-05-11:
It uses raw 802.11 frames to communicate with PC and Xbox and its dongle is just an MT7612U Wi-Fi dongle which only has USB 2.0 interface. (actually, mt76 driver can use it as Wi-Fi dongle with appropriate firmware...)

atar-axis, 2019-05-11:
https://github.com/atar-axis/xpadneo/wiki