atar-axis/xpadneo #69: Add a configure.sh sript

atar-axis, 2019-01-04:
It would be easier for most users if we offer a configuration script which does the necessary sysfs settings 

`sudo ./configure.sh --combined_z_axis=true`

CodeCanna, 2019-03-12:
I wouldn't mind working on this, I'm a little new to contributing to open source, and contributing in general but I would love to help.

atar-axis, 2019-03-12:
Hey there! Thats nice, just open a pull request and I will merge it If it works 😊 If you have any kind of questions: feel free to ask :)

CodeCanna, 2019-03-12:
Cool I'm excited!  Hey so just to make sure, the desired behavior is for the user is:
```
./configure
make
make install
```

CodeCanna, 2019-03-12:
And I'm not clear on what you mean by:

1. shows the possible ones, and
2. does the necessary sysfs settings

Shows the possible ones what?
And does the necessary sysfs settings, does that mean set up the proper file system for install?

Thanks ;)

atar-axis, 2019-03-12:
Ah,... that's a misunderstanding :D It is not this Kind of configure script - What the config script should do is helping the users to configure the driver by changing the sysfs files - see the Configure section in the README :)

CodeCanna, 2019-03-12:
Ok I'm gonna check it out now.  Thanks for the pointer :)

CodeCanna, 2019-03-12:
Hey, would you rather have a cli menu, or cli args?

atar-axis, 2019-03-12:
Hey there, i would definitely prefer a shell script :) Just as the Install/Update/Uninstall scripts. :)

CodeCanna, 2019-03-12:
I know you want a shell script, I'm wondering if you want args to be passed to it, or for the use to choose options as the script runs.

atar-axis, 2019-03-12:
Ah sorry! I would then prefer Arguments to be passed, in case someone wants to automate it - but it's up to you! :)

CodeCanna, 2019-03-13:
Hi!  I think I almost have a working script, I just don't know how to parse the arguments from Fake Dev Version parameter.  It says the value ranges from 0 to 0xFFFF.  I don't know how to handle that value.

CodeCanna, 2019-03-13:
It's ready for review!! :smile:

atar-axis, 2019-03-13:
Hey :) Just Open a pull request, then we can discuss it or merge it directly

atar-axis, 2019-04-08:
see #82 and 769429bc91ca9883474008cc330721f2ef8ea638

closed 2019-04-08 by atar-axis