atar-axis/xpadneo #70: xpadneo on kernel 4.20

atar-axis, 2019-01-05:
Starting with 4.20 the linux kernel is going to support basic rumble for the Xbox One Wireless gamepad out of the box. How will this interact with xpadneo? Can we set a priority somewhere or do we need to write a new udev rule?

Is somebody already on 4.20RC and can try how it does behave? Maybe you @kakra? :grin: 

Cheers, Flo

kakra, 2019-01-05:
Apparently, kernel 4.20 has some problems with my system currently. I need to wait a little more. But it should not conflict with each other unless you do the driver binding wrong. I don't use the udev rules, instead I've removed the VID/PID from the other driver via kernel patch and compiled xpadneo statically into the kernel:
https://github.com/kakra/linux/commits/rebase-4.20/xpadneo

Porting to 4.20 went without conflicts - as did previous kernel versions. So I'm confident it should just work. I cannot speak for the udev rules, tho.

BTW: Why 4.20 RC? 4.20 is already final...

atar-axis, 2019-01-05:
> BTW: Why 4.20 RC? 4.20 is already final...

Yeah, missed the release somehow... I tested it and it still works on my system.

The problem is that, using DKMS, we now have two specialized drivers which support the Xbox One Wireless Gamepad. Somehow hid-microsoft does only yet support the 0x02fd version, https://elixir.bootlin.com/linux/latest/source/drivers/hid/hid-microsoft.c#L452 that is maybe one of reasons why it still works here.

Btw, I forgot that you don't  use DKMS :grin:

atar-axis, 2019-01-05:
We may need to do the same as the hid_generic does:

Whenever a new driver is attached, hid_generic checks if it fits any device which currectly hid_generic handles

```
static int __unmap_hid_generic(struct device *dev, void *data)
{
	struct hid_driver *hdrv = data;
	struct hid_device *hdev = to_hid_device(dev);

	/* only unbind matching devices already bound to hid-generic */
	if (hdev->driver != &hid_generic ||
	    hid_match_device(hdev, hdrv) == NULL)
		return 0;

	if (dev->parent)	/* Needed for USB */
		device_lock(dev->parent);
	device_release_driver(dev);
	if (dev->parent)
		device_unlock(dev->parent);

	return 0;
}

static void hid_generic_add_driver(struct hid_driver *hdrv)
{
	bus_for_each_dev(&hid_bus_type, NULL, hdrv, __unmap_hid_generic);
}
```

atar-axis, 2019-01-05:
Hum, I did some tests and it doesn't look good for the DKMS approach without an udev rule:
I installed two instances of xpadneo: hid-xpadneo**1** and hid-xpadneo**2**

Obviously both drivers do match the gamepad, hence both drivers get loaded by udev when the gamepad appears at the hid bus. The problem now is that only one the functions of one of both drivers are really called - which means nothing but: if hid-microsoft and hid-xpadneo are both claiming the same device-ids, only the functions of one (the one which is registered first) are executed.

kakra, 2019-01-05:
Maybe the udev rules for `xpadneo` needs to be ordered before any `hid-microsoft` rule. Rules in udev should be processed in order of their lexical naming, that is they are prefixed with double-digit values. It may need a stop marker to instruct udev to process other rules. OTOH, udev rules may override their values in order of processing. So it could also be ordered after the other rule to override driver binding. Maybe it needs to be split into two rules: One defining the driver, the other doing the binding.

Since I couldn't use 4.20 yet, I still need to look into how this works out with the new driver bindings in the kernel. I may need to amend the patch to also remove claiming the IDs in the other driver now.

Why couldn't Valve just use `xpadneo` instead of putting more stuff into the drivers which already mess up button order a lot (which is then corrected outside of the kernel by patches to SDL). That will never work out well because some driver paths (the old joystick API) would never see correct button/axis order then. Well, the problem in the first place is probably that SDL folks corrected it in the user-land instead of putting a quirks-fix into the kernel correcting it there. Live would be easier then now.

atar-axis, 2019-01-05:
the only solution I could find so far is another udev rule, I added it accordingly

atar-axis, 2019-01-05:
Maybe we should enhance the kernel itself with some kind of priority flag... Would be the cleanest solution

atar-axis, 2019-01-05:
related: #72

atar-axis, 2019-01-05:
I don't know yet if this works but wouldn't be possible to use the following in modprobes xpadneo.conf

```
remove hid_microsoft
remove hid_generic
install hid_microsoft
install hid_generic
```

This way xpadneo would be the first one in the list of registered drivers and therefore loaded whenever a fitting device appears. But I am unsure if `remove` works if there are active devices for hid_generic for example.

atar-axis, 2019-01-07:
sadly the approach above doesn't work because hid_generic is quite often builtin and therefore you cannot unload it.

kakra, 2019-01-07:
Yes, the most promising way would be to adjust the driver before udev does the binding. Removing hid drivers would also disconnect input devices during the process and may lead to lost input events during connection of the gamepad - this would be surprising at best but most probably very annoying. I'd recommend against that. Especially after a kernel rebuild you may end up with an unloaded module that cannot be inserted again afterwards until you rebooted.

Also, module reordering may not work because modules also have dependencies which may force specific load orders.

atar-axis, 2019-01-07:
What do you mean by 

> Yes, the most promising way would be to adjust the driver before udev does the binding.

kakra, 2019-01-07:
Udev rules allow to set `DRIVER=` early in the rule chain which finally defines the binding - according to the man-page. But you need to catch this early before udev walks up the parent device chain, because the device may have been bound by then. Using `udevadm monitor` and matching the first device node listed with a udev rules to set the driver should do the trick. I cannot properly test this here because my kernel is patched to exclude the conflicting device ids from the other "matching" drivers.

atar-axis, 2019-01-07:
Well this is nearly what we are doing at the moment, right?

```
ACTION=="add", KERNEL=="0005:045E:02FD.*|0005:045E:02E0.*", SUBSYSTEM=="hid",\
RUN+="/bin/bash -c 'echo $kernel > /sys/bus/hid/drivers/hid-generic/unbind'", \
RUN+="/bin/bash -c 'echo $kernel > /sys/bus/hid/drivers/xpadneo/bind'"

ACTION=="add", KERNEL=="0005:045E:02FD.*|0005:045E:02E0.*", SUBSYSTEM=="hid",\
RUN+="/bin/bash -c 'echo $kernel > /sys/bus/hid/drivers/hid-microsoft/unbind'", \
RUN+="/bin/bash -c 'echo $kernel > /sys/bus/hid/drivers/xpadneo/bind'"
```

Previously we had a rule that contained the `DRIVER=` thingy but I changed it to the curent one because it didn't worked on Raspbian for whatever reason.

atar-axis, 2019-01-07:
> PS: Using a final assignment DRIVER:= may prevent later rules from resetting the driver to something else.

Ah, that's interesting - I will give it a try

atar-axis, 2019-01-07:
I don't know exactly but at least we know that we cannot use it in a reliable way

atar-axis, 2019-01-07:
```
KERNEL[3502.705886] bind     /devices/pci0000:00/0000:00:14.0/usb1/1-5/1-5:1.0/bluetooth/hci0/hci0:256/0005:045E:02E0.0004 (hid)
UDEV  [3502.713605] bind     /devices/pci0000:00/0000:00:14.0/usb1/1-5/1-5:1.0/bluetooth/hci0/hci0:256/0005:045E:02E0.0004 (hid)
```

kakra, 2019-01-07:
Maybe bind is a thing only with dynamically loaded modules...

atar-axis, 2019-01-07:
Hum, that's a damn good idea and I think that's the solution since hid-generic was builtin on the Raspbian where I tried it.

atar-axis, 2019-01-07:
dkms does rebuild the initramfs too ;)

kakra, 2019-01-07:
> dkms does rebuild the initramfs too ;)

Until it doesn't... Murphy's law... you know... ;-)

atar-axis, 2019-01-08:
> PS: Using a final assignment DRIVER:= may prevent later rules from resetting the driver to something else.

Doesn't look like `DRIVER` is a key that accepts an assignment, unfortunately

Furthermore, I don't think that we can **prevent** binding to a driver within a udev-rule. As far as I know 
 udev does nothing more but creating symlinks in `/dev` and loading necessary modules via the following rule in `/usr/lib/udev/rules.d/80-drivers.rules`:
`DRIVER!="?*", ENV{MODALIAS}=="?*", RUN{builtin}="kmod load $env{MODALIAS}"`

Moreover, I think that the device is not bound to e.g. 'hid-generic' in the first instance but to the bus master 'hid' which then searches for a fitting driver in its list of registered drivers. The driver which was registered as first is served first.

atar-axis, 2019-01-08:
This is where `bind` and `unbind` uevents where introduced: https://github.com/torvalds/linux/commit/1455cf8dbfd06aa7651dcfccbadb7a093944ca65

atar-axis, 2019-01-08:
Just for documentation, this is how currently binding is done:

 * device_attach - try to attach device to a driver.
 * @dev: device.
 *
 * Walk the list of drivers that the bus has and call
 * driver_probe_device() for each pair. If a compatible
 * pair is found, break out and return.
 *
 * Returns 1 if the device was bound to a driver;
 * 0 if no matching driver was found;
 * -ENODEV if the device is not registered.

`device_attach()` ([here](https://elixir.bootlin.com/linux/latest/source/drivers/base/dd.c#L871) calls `__device_attach()`
 ([here](https://elixir.bootlin.com/linux/latest/source/drivers/base/dd.c#L804)) which does check if the device has a direct driver information (dev->driver),
if so it calls `device_bind_driver()` (see 1.), if not it calls `bus_for_each_drv(dev->bus, NULL, &data, __device_attach_driver)` (see 2.)

1. device_bind_driver() ([here](https://elixir.bootlin.com/linux/latest/source/drivers/base/dd.c#L429))
   actually binds the device to the driver by calling driver_sysfs_add() [here](https://elixir.bootlin.com/linux/latest/source/drivers/base/dd.c#L433),
   then it calls `driver_bound()` ([here](https://elixir.bootlin.com/linux/latest/source/drivers/base/dd.c#L329))
   where finally the 'bind' uevent is generated ([here](https://elixir.bootlin.com/linux/latest/source/drivers/base/dd.c#L356))
   which you can see via udevadm monitor.
  
2. `bus_for_each_drv(dev->bus, NULL, &data, __device_attach_driver)` (defined [here](https://elixir.bootlin.com/linux/latest/source/drivers/base/bus.c#L454)) does nothing
   else but calling `__device_attach_driver()` ([here](https://elixir.bootlin.com/linux/latest/source/drivers/base/dd.c#L737)) for each driver until the first one is found that returns '1'.
   
   In `__device_attach_driver()` then `driver_match_device(drv, dev)` and, if they match, `driver_probe_device(drv, dev)` is called.
   `driver_probe_device()` attempts to bind the device & driver together - to do that it calls `really_probe()` (defined [here](https://elixir.bootlin.com/linux/latest/source/drivers/base/dd.c#L455). `really_probe()` does a lot of things to bind device & driver together, it does e.g. call the drivers and busses probe() functions.
   When everything went fine and both do fit eachother, then `dev->driver` is set to the fitting driver which was registered at the bus, then the device is bound to the driver by `calling driver_sysfs_add()` and `driver_bound()` is called as above.

atar-axis, 2019-01-08:
the drivers are registered at the bus using [this function](https://github.com/torvalds/linux/blob/903b77c631673eeec9e9114e9524171cdf9a2646/drivers/base/bus.c#L639) which is called by [driver_register()](https://github.com/torvalds/linux/blob/6f0d349d922ba44e4348a17a78ea51b7135965b1/drivers/base/driver.c#L146) which is called by [__hid_register_driver()](https://github.com/torvalds/linux/blob/b5aef86e089a2d85a6d627372287785d08938cbe/drivers/hid/hid-core.c#L2455) which is another name for [hid_register_driver()](https://github.com/torvalds/linux/blob/b5aef86e089a2d85a6d627372287785d08938cbe/include/linux/hid.h#L868) which is called in xpadneo [here](https://github.com/atar-axis/xpadneo/blob/e1de350b679e07d987060a9a048d8c9b7dcd980c/hid-xpadneo/src/hid-xpadneo.c#L1362).

atar-axis, 2019-01-08:
Some information is also available here: https://github.com/torvalds/linux/blob/master/Documentation/driver-model/binding.txt

kakra, 2019-01-19:
I've updated my kernel branch: https://github.com/kakra/linux/commits/rebase-4.20/xpadneo

It includes a commit "xpadneo: Exclude xpadneo devices from hid-microsoft" to remove the interference with the new `hid-microsoft` driver, otherwise you'll see unsupported buttons (the battery indicator still works, tho).

atar-axis, 2019-01-19:
> otherwise you'll see unsupported buttons (the battery indicator still works, tho).

Sounds strange to me, really. How does the battery indicator work in that case?
Have they fixed it?

kakra, 2019-01-19:
It looks like the driver only implements a "rumble quirk", leaving the rest to the standard hid driver. `dmesg` states that `xpadneo` still gets activated when turning the gamepad on. But rumbling is already stolen away from xpadneo. I can only guess that hid input handling is also "stolen" by passing it to `hid-generic`. But `xpadneo` may still be used to read the battery. Does this make sense?

But if this is the case, `xpadneo` could try to hook the default drivers in a similar way, implementing just battery reading and button mapping. Tho, I'd like to see that we could make anything useful out of the trigger rumble. Currently, no game I tried can use it.

atar-axis, 2019-01-20:
Hum,... according to my [research](http://0x0001.de/linux-driver-loading-registration-and-binding/) I would say that only one driver is getting bounded to a device, hence xpadneo **or** hid-microsoft, but never both.
They are for sure both loaded, since they both fit.

But your observation is clearly contrary, so I think I may have to rethink that part...

---

Regarding the Trigger Feedback: Absolutely! But I think no game will ever use it as long as there is no native support in the kernel for that kind of rumble/ff. But I am always open to any kind of suggestions.

kakra, 2019-01-20:
Maybe ask the kernel input experts for any trigger rumble proposals?

kakra, 2019-01-20:
> Hum,... according to my [research](http://0x0001.de/linux-driver-loading-registration-and-binding/) I would say that only one driver is getting bounded to a device, hence xpadneo **or** hid-microsoft, but never both.
> They are for sure both loaded, since they both have the same modalias.
> 
> But your observation is clearly contrary to that, so I think I may have to rethink that part...

Maybe it's possible to pass events to an "upstream" driver so you could implement the special functions in one driver and register there but pass any non-handled events to another driver... I wonder how the ff-hook of 4.20 does it for the Xbox One S controller... The comment in the commit says "this only implements the ff-quirk, the rest is handled by the generic hid driver" (or something along that line).

atar-axis, 2019-01-20:
Ah, I remember now - it's like that:
Everything Input event that is not handled directly is handled by `hid-core`, hence it is still a single driver which is responsible for the device, but those drivers do only intervene if necessary. That's why hid-generic is so small, it does nearly nothing - leaving all the work to hid-core.

Hedronmx, 2019-01-20:
So should I downgrade to 4.19 to use my xbox one controllers atm?

atar-axis, 2019-01-20:
@hedronmx no, everything ist good! Don't worry. The Driver does Install a Set of udev rules which do the job

kakra, 2019-01-20:
> That's why `hid-generic` is so small, it does nearly nothing - leaving all the work to hid-core. I could in fact reduce the size of xpadneo too, there is a lot of code which isn't exactly necessary ;)

@atar-axis Cutting down the driver code to re-use existing kernel infrastructure may benefit the process of getting this driver into the kernel...

closed 2019-02-10 by atar-axis