atar-axis/xpadneo #73: Axis/Button order expected by HTML5 Gamepad API Game

potatoattack, 2019-01-06:
The order of axis/buttons expected by CrossCode cause problems making the game unplayable.
This game uses the HTML5 Gamepad API.
For example, the left stick is axis 0 and 1 (correct) but, the left trigger is axis 2 then the right stick is axis 3 and 4.
The way the controls are broken in the game leads me to believe it expects the right stick to be axis 2 and 3 and the left trigger to be 4.
You can view the button mapping on a controller with: https://gamepadviewer.com/
The way the controller responds in gamepadviewer is exactly the same as how the game plays.
There are also other button mapping issues as can be seen with gamepadviewer which are present in the game.

```
uname -a 
Linux awc-desktop 4.20.0-arch1-1-ARCH #1 SMP PREEMPT Mon Dec 24 03:00:40 UTC 2018 x86_64 GNU/Linux
```
[dmesg](https://gist.github.com/potatoattack/82c966751a094ef8250100434048a6af)

Another html5 gamepad tester for reference: http://html5gamepad.com/

atar-axis, 2019-01-06:
hey! could you please use one of the issue templates? I am not sure if this is a feature request to remap the buttons or a bug report, but I need more information in either case

closed 2019-01-06 by atar-axis

atar-axis, 2019-01-06:
by `issue template` I mean: please open a bug report (if you think this is a bug) and fill in all the things you are asked for :grin: