atar-axis/xpadneo #75: Xbox One Controller Crashes USB Bluetooth adapter when the adapter is plugged into USB 3.0

Hedronmx, 2019-01-28:
**Describe the bug**
When I turn on and connect my xbox one controller, my bluetooth device dies. But only if my usb adapter is connected to a usb 3.0 

**To Reproduce**
Connect bluetooth adapter to a usb 3.0 port
Turn on and connect xbox one controller
Watch it die

**Expected behavior**
Work

**Screenshots/Gifs**

**System information**
Please enable debugging output <https://atar-axis.github.io/xpadneo/index.html#debugging>
and add at least the following outputs
-   `uname -a`
`Linux carlos-pc 4.20.3-042003-generic #201901171122 SMP Thu Jan 17 16:25:26 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux`

-   `dmesg`
```
usb usb5: root hub lost power or was reset
[12602.648938] usb usb6: root hub lost power or was reset
[12603.121052] usb 5-2: reset full-speed USB device number 4 using xhci_hcd
[12615.960865] usb usb5: root hub lost power or was reset
[12615.960867] usb usb6: root hub lost power or was reset
[12616.432825] usb 5-2: reset full-speed USB device number 4 using xhci_hcd
[12635.416897] usb usb5: root hub lost power or was reset
[12635.416899] usb usb6: root hub lost power or was reset
[12635.888823] usb 5-2: reset full-speed USB device number 4 using xhci_hcd
[12639.512869] usb usb5: root hub lost power or was reset
[12639.512871] usb usb6: root hub lost power or was reset
[12639.984827] usb 5-2: reset full-speed USB device number 4 using xhci_hcd
[12677.269515] usb 3-2: USB disconnect, device number 2
[12685.368120] sky2 0000:07:00.0 enp7s0: Link is down
[12687.640851] usb usb3: root hub lost power or was reset
[12687.640854] usb usb4: root hub lost power or was reset

```
-   `sudo find "/sys/kernel/debug/hid/" -name "0005:045E:*" -exec sh -c 'echo "{}" && head -1 "{}/rdesc" | tee /dev/tty | cksum && echo' \;`
```
/sys/kernel/debug/hid/0005:045E:0917.0018
05 01 09 06 a1 01 85 01 15 00 25 01 05 07 1a e0 00 2a e7 00 75 01 95 08 81 02 05 07 19 00 2a 91 00 16 00 00 26 ff 00 75 08 95 0a 81 00 05 0c 0a c0 02 a1 02 1a c1 02 2a c6 02 95 06 b1 03 c0 05 08 75 01 95 03 19 01 29 03 25 01 91 02 95 05 91 01 c0 05 0c 09 01 a1 01 85 10 19 00 2a ff 03 75 0c 95 01 15 00 26 ff 03 81 00 75 04 95 01 81 01 c0 06 07 ff 09 82 a1 01 15 00 26 ff 00 75 08 95 13 85 27 09 c6 82 02 01 85 22 09 c8 b2 02 01 85 24 09 cb b2 02 01 95 10 85 28 09 cc 82 02 01 09 d0 92 02 01 09 d4 b2 02 01 c0 06 07 ff 09 80 a1 01 85 30 15 00 25 ff 95 01 75 08 09 00 91 02 c0 09 81 a1 01 85 2a 15 00 25 ff 75 08 95 48 09 c6 82 02 01 09 c7 92 02 01 95 3c 09 c8 b2 02 01 85 2b 09 c9 82 02 01 09 ca 92 02 01 09 cb b2 02 01 85 2c 75 20 95 04 19 cc 29 cf 81 02 19 d0 29 d3 91 02 19 d4 29 d7 b1 02 85 2d 19 d8 29 db 81 02 19 dc 29 df 91 02 19 e0 29 e3 b1 02 85 2e 19 e4 29 e7 81 02 19 e8 29 eb 91 02 19 ec 29 ef b1 02 85 2f 19 f0 29 f3 81 02 19 f4 29 f7 91 02 19 f8 29 fb b1 02 c0 
2313906022 976

/sys/kernel/debug/hid/0005:045E:0805.0017
05 01 09 02 a1 01 05 01 09 02 a1 02 85 01 09 01 a1 00 05 09 19 01 29 05 15 00 25 01 95 05 75 01 81 02 95 01 75 03 81 01 05 01 09 30 09 31 95 02 75 10 16 01 80 26 ff 7f 81 06 a1 02 85 12 09 48 95 01 75 02 15 00 25 01 35 01 45 78 b1 02 85 01 09 38 35 00 45 00 95 01 75 10 16 01 80 26 ff 7f 81 06 c0 a1 02 85 12 09 48 95 01 75 02 15 00 25 01 35 01 45 78 b1 02 35 00 45 00 75 04 b1 01 85 01 05 0c 0a 38 02 35 00 45 00 95 01 75 10 16 01 80 26 ff 7f 81 06 c0 c0 c0 c0 05 01 09 06 a1 01 85 02 15 00 25 01 05 07 1a e0 00 2a e7 00 75 01 95 08 81 02 05 07 19 00 2a 91 00 16 00 00 26 ff 00 75 08 95 0a 81 00 c0 05 0c 09 01 a1 01 85 03 95 01 75 10 15 00 26 ff 03 19 00 2a ff 03 81 00 c0 06 07 ff 0a 10 02 a1 01 85 05 15 00 26 ff 00 75 08 95 01 09 00 91 02 c0 0a 11 02 a1 01 85 2a 15 00 25 ff 75 08 95 48 09 c6 82 02 01 09 c7 92 02 01 09 c8 b2 02 01 85 2b 09 c9 82 02 01 09 ca 92 02 01 09 cb b2 02 01 85 2c 75 20 95 04 19 cc 29 cf 81 02 19 d0 29 d3 91 02 19 d4 29 d7 b1 02 85 2d 19 d8 29 db 81 02 19 dc 29 df 91 02 19 e0 29 e3 b1 02 85 2e 19 e4 29 e7 81 02 19 e8 29 eb 91 02 19 ec 29 ef b1 02 85 2f 19 f0 29 f3 81 02 19 f4 29 f7 91 02 19 f8 29 fb b1 02 c0 0a 12 02 a1 01 85 04 06 00 ff 0a 0a fa 15 00 25 ff 95 08 75 08 81 02 85 24 06 02 ff 15 00 26 ff 00 75 08 95 48 0a 0a fa b1 02 85 27 75 08 95 13 0a 0a fa 81 02 c0 
1891733236 1306
```

**Additional context**  
Running on Elementary Juno. I am using bluetooth keyboard, mouse and headset. They all work fine

atar-axis, 2019-01-28:
Hey there! Bad news: Bluetooth issues are not caused by xpadneo because this driver is only the topmost layer, all the bluetooth thing happen in the kernel itself.

Anyhow, have you ever tried another bluetooth dongle?

closed 2019-01-28 by atar-axis

Hedronmx, 2019-01-28:
That's weird that it only dies when the xbox one controller is connected. I can use xpad just fine on usb 2.0, but I get better connection off of 3.0

atar-axis, 2019-01-28:
> I can use xpad just fine on usb 2.0

What do you mean by that? As far as I know **xpad** has no Wireless support, right? 
Do you mean **xpadneo**?

> it also works correctly on windows, even when connected to 3.0 port

 Yeah, you are right - that's weird, then the problem might be somewhere at the usb bt dongle driver. As said, it cannot be caused by xpadneo since the communication is done by the kernel. I am sorry :/

ugly95, 2019-01-28:
It seems that some BT dongles don't play well with the Xbox One S controller.  I mentioned in another issue here that my BT dongle doesn't easily connect to the controller (it does eventually, but takes a long time).  I have another PC with an internal BT controller, and it connects instantly with the xpadneo driver.  So it's not the controller, and it's not the driver.  I've also tried connecting on Windows with the dongle, and it seems to have the same issue.  Other BT devices connect without issue.

I know this is outside the scope of this project, but I'd be curious to know what BT dongles people are using that connect successfully.  Or, maybe it is a USB 2.0/3.0 issue, or the dongle not being able to get enough power from particular ports.

And it might be worth noting in the readme that connection issues are outside the scope of the driver, and maybe even list some known problematic dongles.

Hedronmx, 2019-01-28:
@atar-axis Yes i meant xpadneo.

@ugly95 I am using asus bt 400 usb adapter as well as internal lenovo flex 5 bluetooth. on my laptop since im not using any other bluetooth device, the controller works great but on my pc everything is really bad, If I use usb 3.0 port then the adapter dies, and if I use usb 2.0 then my xbox controller lags a lot, especially when using the dpad.