atar-axis/xpadneo #76: Installation Success, Connecting Fails, no /sys/module/hid_xpadneo

jslay88, 2019-02-09:
**Describe the bug**
Ubuntu 18.10 Desktop
Disabled Secure Boot
I have installed prereqs, can scan, pair, and trust the controller. Pairing Fails.
Cannot enable debug as `/sys/module/hid_xpadneo` does not exist.
```
[bluetooth]# connect EC:83:50:C3:D6:BC
Attempting to connect to EC:83:50:C3:D6:BC
[CHG] Device EC:83:50:C3:D6:BC Connected: yes
Failed to connect: org.bluez.Error.Failed
[CHG] Device EC:83:50:C3:D6:BC Connected: no
```

**To Reproduce**
Steps to reproduce the behavior:
Fresh install of Ubuntu 18.10, EFI, Disabled Secure Boot
* `git clone` repo
* `apt install` prereqs
* `cd` into `xpadneo`
* `sudo ./install.sh` -- reports success
* `sudo bluetoothctl`
* `scan on`
* `pair EC:83:50:C3:D6:BC`
* `trust EC:83:50:C3:D6:BC`
* `connect EC:83:50:C3:D6:BC`

**Expected behavior**
A connected controller

**System information**
Please enable debugging output <https://atar-axis.github.io/xpadneo/index.html#debugging>
and add at least the following outputs
-   `uname -a`
-   `dmesg`  -- 
```
[    6.898091] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[    6.898093] Bluetooth: BNEP filters: protocol multicast
[    6.898096] Bluetooth: BNEP socket layer initialized
[    7.202324] IPv6: ADDRCONF(NETDEV_UP): ens33: link is not ready
[    7.208148] IPv6: ADDRCONF(NETDEV_UP): ens33: link is not ready
[    7.212372] e1000: ens33 NIC Link is Up 1000 Mbps Full Duplex, Flow Control: None
[    7.213568] IPv6: ADDRCONF(NETDEV_CHANGE): ens33: link becomes ready
[    9.963240] [drm:vmw_stdu_crtc_page_flip [vmwgfx]] *ERROR* Page flip error -16.
[   17.363676] Bluetooth: RFCOMM TTY layer initialized
[   17.363681] Bluetooth: RFCOMM socket layer initialized
[   17.363686] Bluetooth: RFCOMM ver 1.11
[   18.401305] rfkill: input handler disabled
[   43.608428] Bluetooth: hci0: last event is not cmd complete (0x0f)
[   50.702037] perf: interrupt took too long (2518 > 2500), lowering kernel.perf_event_max_sample_rate to 79250
[   59.364316] Bluetooth: hci0: last event is not cmd complete (0x0f)
[   70.744831] perf: interrupt took too long (3182 > 3147), lowering kernel.perf_event_max_sample_rate to 62750
[   75.610273] Bluetooth: HIDP (Human Interface Emulation) ver 1.2
[   75.610276] Bluetooth: HIDP socket layer initialized
[   81.121089] Bluetooth: hci0: ACL packet for unknown connection handle 12
[   86.617443] Bluetooth: hci0: last event is not cmd complete (0x0f)
```

-   `sudo find "/sys/kernel/debug/hid/" -name "0005:045E:*" -exec sh -c 'echo "{}" && head -1 "{}/rdesc" | tee /dev/tty | cksum && echo' \;` -- Returns nothing

atar-axis, 2019-02-10:
Hi There, have you read the section about secure boot? And is It maybe active? Sounds like!

jslay88, 2019-02-10:
As you can see from my output, secure boot is disabled.

atar-axis, 2019-02-10:
Sorry, I must have overread that somehow...

It's clear that `/sys/module/hid_xpadneo` does not exist when your controller is not even connected - but since the connection is not done by xpadneo, the problem lies not in the driver.

Let me try to help you anyway, please give me the output of:
* `dkms status`
* `/etc/modprobe.d/99-xpadneo-bluetooth.conf`
* `/sys/module/bluetooth/parameters/disable_ertm`

jslay88, 2019-02-10:
```
dkms status
hid-xpadneo, 0.5.4, 4.18.0-15-generic, x86_64: installed
```

atar-axis, 2019-02-10:
* Is the firmware of your gamepad the latest? If not, please update on Windows or Xbox
* Was your gamepad connected to an windows system or xbox before?

atar-axis, 2019-02-10:
Furthermore, can you please do what is described here? https://stackoverflow.com/questions/37003147/i-want-to-enable-debug-messages-on-bluez

jslay88, 2019-02-10:
I just received this controller earlier this week. I synced it up with my Windows 10 PC and played around with it for a bit (both with a USB cable, and the Windows Wireless Adapter, not Bluetooth). I am trying to build a project using xinput in python for something else down the road.

As far as I know, its as up-to-date as I can make it by doing the basics so far. Haven't seen any lit on how to update the controller itself (if you have a link, wouldn't mind trying to update it).

I am the first to admit, I am no guru in Linux, especially when it comes to the hardware side of it.

```
Feb 10 10:14:52 ubuntu bluetoothd[32357]: src/adapter.c:trigger_start_discovery()
Feb 10 10:14:52 ubuntu bluetoothd[32357]: src/adapter.c:cancel_passive_scanning()
Feb 10 10:14:54 ubuntu bluetoothd[32357]: src/adapter.c:connected_callback() hci0 device EC:83:50:C3:D6:BC connected eir_len 5
Feb 10 10:14:54 ubuntu bluetoothd[32357]: extract_int: Unexpected end of packet
Feb 10 10:14:54 ubuntu bluetoothd[32357]: src/device.c:device_probe_profiles() Probing profiles for device EC:83:50:C3:D6:BC
Feb 10 10:14:54 ubuntu bluetoothd[32357]: profiles/input/device.c:input_device_register() /org/bluez/hci0/dev_EC_83_50_C3_D6_BC
Feb 10 10:14:54 ubuntu bluetoothd[32357]: src/service.c:btd_service_ref(): ref=2
Feb 10 10:14:54 ubuntu bluetoothd[32357]: src/service.c:change_state(): device EC:83:50:C3:D6:BC profile input-hid state changed: unavailable -> disconnected (0)
Feb 10 10:14:54 ubuntu bluetoothd[32357]: src/device.c:device_svc_resolved() /org/bluez/hci0/dev_EC_83_50_C3_D6_BC err 0
Feb 10 10:14:54 ubuntu bluetoothd[32357]: src/device.c:connect_profiles() /org/bluez/hci0/dev_EC_83_50_C3_D6_BC (all), client :1.356
Feb 10 10:14:54 ubuntu bluetoothd[32357]: profiles/input/device.c:input_device_connect()
Feb 10 10:14:54 ubuntu bluetoothd[32357]: Can't get HIDP connection info
Feb 10 10:14:54 ubuntu bluetoothd[32357]: src/service.c:change_state(): device EC:83:50:C3:D6:BC profile input-hid state changed: disconnected -> connecting (0)
Feb 10 10:14:54 ubuntu bluetoothd[32357]: src/adapter.c:bonding_attempt_complete() hci0 bdaddr EC:83:50:C3:D6:BC type 0 status 0x5
Feb 10 10:14:54 ubuntu bluetoothd[32357]: src/device.c:device_bonding_complete() bonding (nil) status 0x05
Feb 10 10:14:54 ubuntu bluetoothd[32357]: src/device.c:device_bonding_failed() status 5
Feb 10 10:14:54 ubuntu bluetoothd[32357]: src/adapter.c:resume_discovery()
Feb 10 10:14:54 ubuntu bluetoothd[32357]: src/adapter.c:trigger_start_discovery()
Feb 10 10:14:54 ubuntu bluetoothd[32357]: src/adapter.c:cancel_passive_scanning()
Feb 10 10:14:54 ubuntu bluetoothd[32357]: connect error: Connection refused (111)
Feb 10 10:14:54 ubuntu bluetoothd[32357]: src/service.c:change_state(): device EC:83:50:C3:D6:BC profile input-hid state changed: connecting -> disconnected (-5)
Feb 10 10:14:54 ubuntu bluetoothd[32357]: src/device.c:device_profile_connected() input-hid Input/output error (5)
Feb 10 10:14:54 ubuntu bluetoothd[32357]: src/device.c:device_profile_connected() returning response to :1.356
Feb 10 10:14:58 ubuntu bluetoothd[32357]: src/adapter.c:dev_disconnected() Device EC:83:50:C3:D6:BC disconnected, reason 4
Feb 10 10:14:58 ubuntu bluetoothd[32357]: src/adapter.c:adapter_remove_connection()
Feb 10 10:14:58 ubuntu bluetoothd[32357]: plugins/policy.c:disconnect_cb() reason 4
Feb 10 10:14:58 ubuntu bluetoothd[32357]: src/adapter.c:bonding_attempt_complete() hci0 bdaddr EC:83:50:C3:D6:BC type 0 status 0xe
Feb 10 10:14:58 ubuntu bluetoothd[32357]: src/device.c:device_bonding_complete() bonding (nil) status 0x0e
Feb 10 10:14:58 ubuntu bluetoothd[32357]: src/device.c:device_bonding_failed() status 14
Feb 10 10:14:58 ubuntu bluetoothd[32357]: src/adapter.c:resume_discovery()
Feb 10 10:14:58 ubuntu bluetoothd[32357]: src/adapter.c:trigger_start_discovery()
Feb 10 10:14:58 ubuntu bluetoothd[32357]: src/adapter.c:cancel_passive_scanning()
Feb 10 10:15:03 ubuntu bluetoothd[32357]: src/adapter.c:start_discovery_timeout()
Feb 10 10:15:03 ubuntu bluetoothd[32357]: src/adapter.c:start_discovery_timeout() adapter->current_discovery_filter == 0
Feb 10 10:15:03 ubuntu bluetoothd[32357]: src/adapter.c:start_discovery_complete() status 0x00
Feb 10 10:15:03 ubuntu bluetoothd[32357]: src/adapter.c:discovering_callback() hci0 type 7 discovering 1 method 0
Feb 10 10:15:16 ubuntu bluetoothd[32357]: src/device.c:connect_profiles() /org/bluez/hci0/dev_EC_83_50_C3_D6_BC (all), client :1.356
Feb 10 10:15:16 ubuntu bluetoothd[32357]: profiles/input/device.c:input_device_connect()
Feb 10 10:15:16 ubuntu bluetoothd[32357]: Can't get HIDP connection info
Feb 10 10:15:16 ubuntu bluetoothd[32357]: src/service.c:change_state(): device EC:83:50:C3:D6:BC profile input-hid state changed: disconnected -> connecting (0)
Feb 10 10:15:17 ubuntu bluetoothd[32357]: src/adapter.c:device_found_callback() hci0 addr EC:83:50:C3:D6:BC, rssi -69 flags 0x0001 eir_len 5
Feb 10 10:15:17 ubuntu bluetoothd[32357]: src/device.c:device_set_legacy() legacy 0
Feb 10 10:15:17 ubuntu bluetoothd[32357]: src/adapter.c:device_found_callback() hci0 addr EC:83:50:C3:D6:BC, rssi -70 flags 0x0000 eir_len 35
Feb 10 10:15:17 ubuntu bluetoothd[32357]: src/device.c:device_set_legacy() legacy 0
Feb 10 10:15:17 ubuntu bluetoothd[32357]: src/adapter.c:discovering_callback() hci0 type 7 discovering 0 method 0
Feb 10 10:15:17 ubuntu bluetoothd[32357]: src/adapter.c:trigger_start_discovery()
[... cut ...]
Feb 10 10:15:25 ubuntu bluetoothd[32357]: src/device.c:device_set_flags() flags 0
Feb 10 10:15:30 ubuntu bluetoothd[32357]: src/adapter.c:device_found_callback() hci0 addr EC:83:50:C3:D6:BC, rssi -63 flags 0x0001 eir_len 5
Feb 10 10:15:30 ubuntu bluetoothd[32357]: src/device.c:device_set_legacy() legacy 0
Feb 10 10:15:31 ubuntu bluetoothd[32357]: src/adapter.c:device_found_callback() hci0 addr EC:83:50:C3:D6:BC, rssi -70 flags 0x0000 eir_len 35
Feb 10 10:15:31 ubuntu bluetoothd[32357]: src/device.c:device_set_legacy() legacy 0
```

atar-axis, 2019-02-10:
So is it by any chance the same PC (native Windows and Linux in VM?).

> Xbox Accessories in Windows 10 reports that it is up-to-date, and no updates are available.
v3.1.1221.0

Please hit the upgrade/update button anyways if possible (the gamepad should then "shutdown" for a moment - sometimes the gamepad has a little problem to connect to another OS after connecting it to another, especially if it is the same PC.  By "updating" it - even if it was already the latest fw - you reset the internals.

atar-axis, 2019-02-10:
Moreover, please type `remove <MAC>` before you you try to connect part/trust/connect it again

jslay88, 2019-02-10:
I have hit the upgrade button, the controller reboots. Still unable to connect with Bluetooth in Linux.

Yes, Native Windows, Workstation Pro 15 Trial, using hardware level pass-through. 
I am able to connect both of my esp32's in Windows and in Linux for a Bluetooth Serial Profile. 

They had to be added independently between OS's, as one would expect. 

I did a remove, I see a DEL message, did a pair, says it still exists, did a trust, says succeeded, did a connect, now its looping.

```
[CHG] Device EC:83:50:C3:D6:BC Connected: yes
[CHG] Device EC:83:50:C3:D6:BC Connected: no
[CHG] Device EC:83:50:C3:D6:BC Connected: yes
[CHG] Device EC:83:50:C3:D6:BC Connected: no
[CHG] Device EC:83:50:C3:D6:BC Connected: yes
[CHG] Device EC:83:50:C3:D6:BC Connected: no
[CHG] Device EC:83:50:C3:D6:BC Connected: yes
[CHG] Device EC:83:50:C3:D6:BC Connected: no
[CHG] Device EC:83:50:C3:D6:BC Connected: yes
[CHG] Device EC:83:50:C3:D6:BC Connected: no
[CHG] Device EC:83:50:C3:D6:BC Connected: yes
[CHG] Device EC:83:50:C3:D6:BC Connected: no
[CHG] Device EC:83:50:C3:D6:BC Connected: yes
[CHG] Device EC:83:50:C3:D6:BC Connected: no
[CHG] Device EC:83:50:C3:D6:BC Connected: yes
```

atar-axis, 2019-02-10:
Ah! looping is at least a bit more common problem - that's mostly because of the **ERTM** thingy which should gets disabled while installing xpadneo. Can you give me again the output of:

* `/etc/modprobe.d/99-xpadneo-bluetooth.conf`
* `/sys/module/bluetooth/parameters/disable_ertm`

jslay88, 2019-02-10:
Sorry I don't see a `DEL` message, I see `Device has been removed`

```
[Xbox Wireless Controller]# remove EC:83:50:C3:D6:BC
Failed to pair: org.bluez.Error.AuthenticationCanceled
[CHG] Device EC:83:50:C3:D6:BC Paired: yes
[CHG] Device EC:83:50:C3:D6:BC ServicesResolved: no
Device has been removed
[CHG] Device EC:83:50:C3:D6:BC Connected: no
[bluetooth]# pair EC:83:50:C3:D6:BC
Attempting to pair with EC:83:50:C3:D6:BC
Failed to pair: org.bluez.Error.AlreadyExists
[bluetooth]# trust EC:83:50:C3:D6:BC
Changing EC:83:50:C3:D6:BC trust succeeded
[bluetooth]# connect EC:83:50:C3:D6:BC
Attempting to connect to EC:83:50:C3:D6:BC
Failed to connect: org.bluez.Error.Failed
[CHG] Device EC:83:50:C3:D6:BC Connected: yes
[CHG] Device EC:83:50:C3:D6:BC Connected: no
[CHG] Device EC:83:50:C3:D6:BC Connected: yes
[CHG] Device EC:83:50:C3:D6:BC Connected: no
[CHG] Device EC:83:50:C3:D6:BC Connected: yes
[CHG] Device EC:83:50:C3:D6:BC Connected: no
[CHG] Device EC:83:50:C3:D6:BC Connected: yes
[CHG] Device EC:83:50:C3:D6:BC Connected: no
[CHG] Device EC:83:50:C3:D6:BC Connected: yes
[CHG] Device EC:83:50:C3:D6:BC Connected: no
```

```
sudo cat /etc/modprobe.d/99-xpadneo-bluetooth.conf 
options bluetooth disable_ertm=y
```

```
sudo cat /sys/module/bluetooth/parameters/disable_ertm 
Y
```

atar-axis, 2019-02-10:
hum, that's a hard one,... have you ensured that the gamepad is disconnected from the native OS? And have you rebooted the system after the `remove` thingy?

jslay88, 2019-02-10:
Yes, there is no way for it to be talking to the native OS right now, as the Bluetooth adapter is passed through to the VM, I don't have it connected with USB, and I have the Windows Wireless Adapter pulled out sitting on my desk :).

Having said that, I detatched the Bluetooth adapter again (probably the 100th time), did a couple reboots, ran the delete command again, this time getting the `DEL` message, and was finally able to connect.

I have to assume it "auto-pairs" because the controller is actively looking for that adapters MAC on power on, presenting the `AlreadyExists` issue after removal, when it detects it on `scan on`.

Ultimately, who knows what was going on, I'll report back if there are any further issues.

```
sudo bluetoothctl
Agent registered
[bluetooth]# remove EC:83:50:C3:D6:BC
[DEL] Device EC:83:50:C3:D6:BC Xbox Wireless Controller
Device has been removed
[bluetooth]# scan on
Discovery started
[NEW] Device EC:83:50:C3:D6:BC Xbox Wireless Controller
[CHG] Device EC:83:50:C3:D6:BC Modalias: usb:v045Ep02E0d0903
[CHG] Device EC:83:50:C3:D6:BC UUIDs: 00001124-0000-1000-8000-00805f9b34fb
[CHG] Device EC:83:50:C3:D6:BC UUIDs: 00001200-0000-1000-8000-00805f9b34fb
[CHG] Device EC:83:50:C3:D6:BC ServicesResolved: yes
[CHG] Device EC:83:50:C3:D6:BC Paired: yes
[CHG] Device EC:83:50:C3:D6:BC ServicesResolved: no
[CHG] Device EC:83:50:C3:D6:BC Connected: no
[CHG] Device EC:83:50:C3:D6:BC RSSI: -63
[bluetooth]# pair EC:83:50:C3:D6:BC
Attempting to pair with EC:83:50:C3:D6:BC
Failed to pair: org.bluez.Error.AlreadyExists
[bluetooth]# trust EC:83:50:C3:D6:BC
[CHG] Device EC:83:50:C3:D6:BC Trusted: yes
Changing EC:83:50:C3:D6:BC trust succeeded
[bluetooth]# connect EC:83:50:C3:D6:BC
Attempting to connect to EC:83:50:C3:D6:BC
[CHG] Device EC:83:50:C3:D6:BC Connected: yes
Connection successful
[CHG] Device EC:83:50:C3:D6:BC Modalias: usb:v045Ep02FDd0903
[CHG] Device EC:83:50:C3:D6:BC ServicesResolved: yes
[Xbox Wireless Controller]# 
```

atar-axis, 2019-02-10:
perfect, this device is definitely naughty sometimes... anyhow I am happy it works now - I will add the given advices to the README later on.

closed 2019-02-10 by atar-axis

jslay88, 2019-02-10:
I think a lot of it also has to do with the fact this is the first time on discovery that it picked up the UUIDs. I think its pretty hard for it to determine what profile/module to select/load when it doesn't have those.

ugly95, 2019-02-10:
This seems a lot like the problem I've been having with my controller.  I have not been able to get it to reliably connect.

Usually, I have to go to the terminal and manually connect it in bluetoothctl.

As a workaround, I've made an easy script (replace the MAC address with your own):
```
#!/bin/bash

echo -e 'connect 5C:BA:37:D4:46:6A' | bluetoothctl
```
And running that a couple times usually gets it to connect.

The only time I can connect 100% of the time, is at boot.  If I power on the XBox controller before I start my PC.  When the desktop loads up, it will connect every time.  But if I wait until after the desktop loads up, I have to use the above script, once or sometimes a couple of times to connect.

I'm thinking it has something to do with some USB Bluetooth dongles don't play as well with Linux and the Xbox controller.  If I use the dongle on Windows, it seems to work fine, on the same PC and on other PCs I've tested.  If I use the same Xbox controller using a Linux PC with an internal bluetooth adapter, it connects perfectly.  I  have a generic bluetooth gamepad that works perfectly with this dongle on Mint and in Windows.  But using the USB dongle with the Xbox controller with Linux Mint on two different PCs have the same issue.    It seems to work fine once it's connected, but it's a real chore to actually connect.

I've tried removing the controller, and pairing again.  I've tried unplugging and replugging in the dongle.  I've tried this tip: [https://unix.stackexchange.com/questions/255509/bluetooth-pairing-on-dual-boot-of-windows-linux-mint-ubuntu-stop-having-to-p](https://unix.stackexchange.com/questions/255509/bluetooth-pairing-on-dual-boot-of-windows-linux-mint-ubuntu-stop-having-to-p).  Nothing works reliably, other than immediately connecting at boot.

jslay88, 2019-02-10:
Yeah, I am having issues getting it to connect without running the same. 

The other main issue I am experiencing right now, is js0 just seems to stop streaming data, and I am forced to go through reconnecting again. 

When plugging in with USB, js0 streams like a charm.

atar-axis, 2019-02-11:
I am really sorry guys to hear that, but it is definitely not a problem with xpadneo itself, believe me. Xpadneo does **nothing** as long as the gamepad is not connected, it is not even loaded.

I cannot help you therefore by fixing a bug or sth like that, I fear. But feel free to discuss here and exchange experiences :)

jslay88, 2019-02-11:
Heh, I’m giving up I think. Something about this setup is entirely unreliable, and I just don’t know enough to know what piece is the issue. 

Still having issues with input from the controller just stopping in the middle after a successful connection, randomly, while on Bluetooth. As well as auto connecting issues. 

My PS3 controller seems to work just fine over Bluetooth. Had an input streaming test running for about 30m with no hiccups. Auto connecting seems to work with it as well, no issues. Even after reboot. 

And in Windows, both work perfectly on Bluetooth.

atar-axis, 2019-02-11:
I know that it can be a pain in the ..., but believe me - the gamepad is worth it - once it runs.
I would start by borrowing me another BT dongle from a friend, a unreliable connection is - as already stated by @ugly95 - mostly a problem caused by the BT-stick which is somehow incompatible with the rest.

jslay88, 2019-02-11:
It is a Broadcom BCM20702A0. 

I just find it strange, that if it was the adapter, then why does the PS3 controller not have its input hang after a successful connection as the Xbox one does?

atar-axis, 2019-02-11:
We still don't know what exactly is causing the connection problems, and I don't say that it is the adapter, it is most probably the adapter in *combination* with the GP under linux.

Do you still have enabled the debug output? And if, what is the output right before the connection hangs - if there is any.

jslay88, 2019-02-11:
Yeah, js0 is just easy to read right off the bat for basic this button, this value @ 8 byte values.

I do have everything still setup, I am just out of time for now. I will likely have to revisit this with you later in the week. I am not even against setting up an SSH account for you and working with you directly over slack or similar. I'd love to see/learn this side of linux.

atar-axis, 2019-02-11:
I would be happy to help you, but before we can examine the problem any further, we should localize the problem a bit more by trying another bt-dongle first ;)

ugly95, 2019-02-11:
If it helps anybody, here are the Bluetooth USB dongles I have.

[Panda Bluetooth 4.0 USB Nano Adapter (Amazon link)](https://www.amazon.com/gp/product/B00BCU4TZE/)

Using: `lsusb | grep -i 'blue'`, I get:

```
Bus 001 Device 006: ID 0a12:0001 Cambridge Silicon Radio, Ltd Bluetooth Dongle (HCI mode)
```
I have a second one:
[MIATONE Bluetooth Adapter Bluetooth CSR 4.0 (Amazon link)](https://www.amazon.com/gp/product/B00M1ATR4C/)

Using: `lsusb | grep -i 'blue'`, I get:

```
Bus 001 Device 007: ID 0a12:0001 Cambridge Silicon Radio, Ltd Bluetooth Dongle (HCI mode)
```

So, both dongles I have result in the same unreliable behaviour, and it looks like they use the same Cambridge Silicon Radio chipset (probably should have done some research before buying two dongles with the same chipset).

On my miniPC that has an internal Bluetooth chip, and the controller connects immediately, I don't know how to find the exact model, but `dmesg` gives me some indication of an Intel bluetooth adapter.

atar-axis, 2019-02-12:
Unfortunately both are not available in Germany, I will see If I can get one somewhere else cheaply

atar-axis, 2019-02-12:
Btw, great article to localize the Problem: https://wiki.ubuntu.com/DebuggingBluetooth

ugly95, 2019-02-16:
I decided to buy a different USB adapter.  I found one that advertises a different chipset.

[Plugable USB Bluetooth 4.0 Low Energy Micro Adapter](https://www.amazon.com/Plugable-Bluetooth-Adapter-Raspberry-Compatible/dp/B009ZIILLI/)

From the link provided by @atar-axis it seems one command is `hciconfig -a`

```
	BD Address: 5C:F3:70:8C:99:37  ACL MTU: 1021:8  SCO MTU: 64:1
	UP RUNNING 
	RX bytes:5454 acl:60 sco:0 events:161 errors:0
	TX bytes:4595 acl:41 sco:0 commands:97 errors:0
	Features: 0xbf 0xfe 0xcf 0xfe 0xdb 0xff 0x7b 0x87
	Packet type: DM1 DM3 DM5 DH1 DH3 DH5 HV1 HV2 HV3 
	Link policy: RSWITCH SNIFF 
	Link mode: SLAVE ACCEPT 
	Name: 'rickpc'
	Class: 0x1c0104
	Service Classes: Rendering, Capturing, Object Transfer
	Device Class: Computer, Desktop workstation
	HCI Version: 4.0 (0x6)  Revision: 0x1000
	LMP Version: 4.0 (0x6)  Subversion: 0x220e
	Manufacturer: Broadcom Corporation (15)
```
And ``lsusb | grep -i 'blue'`` gives me:
```
Bus 001 Device 005: ID 0a5c:21e8 Broadcom Corp. BCM20702A0 Bluetooth 4.0
```
This appears to be the same chipset mentioned by @jslay88.

With this new Broadcom chipset adapter, the Xbox One S controller connects perfectly.  Usually by the 2nd or third time the connection light on the controller blinks it connects.  I have no issues reconnecting.  It works exactly the way I would expect it to.

For completeness, here is what I get with `hciconfig -a` for my Panda Bluetooth dongle:
```
        Type: Primary  Bus: USB
	BD Address: 00:1A:7D:DA:71:11  ACL MTU: 310:10  SCO MTU: 64:8
	UP RUNNING PSCAN ISCAN 
	RX bytes:1810249 acl:90 sco:0 events:258312 errors:0
	TX bytes:115115063 acl:394622 sco:0 commands:64 errors:0
	Features: 0xff 0xff 0x8f 0xfe 0xdb 0xff 0x5b 0x87
	Packet type: DM1 DM3 DM5 DH1 DH3 DH5 HV1 HV2 HV3 
	Link policy: RSWITCH HOLD SNIFF PARK 
	Link mode: SLAVE ACCEPT 
	Name: 'rickpc'
	Class: 0x1c0104
	Service Classes: Rendering, Capturing, Object Transfer
	Device Class: Computer, Desktop workstation
	HCI Version: 4.0 (0x6)  Revision: 0x22bb
	LMP Version: 4.0 (0x6)  Subversion: 0x22bb
	Manufacturer: Cambridge Silicon Radio (10)
```
The Cambridge Silicon chipset provides a link policty of 'HOLD' and 'PARK', but I have no idea if that has anything to do with anything.

So, for me at least, the issue seems to be the dongle.  I have a feeling there might be something more to the issue.  On a quick look at Amazon, it seems that most of the cheap dongles list CSR (for Cambridge Silicon Radio) as the chipset.  So if the chipset was the only issue, I'd expect more people to be running into this issue.

As an unrelated side note, I find the Pluggable USB dongle is shitty for audio (tends to be laggy and doesn't connect as well to my headphones).  I really like the Panda dongle since it has a discreet green light instead of the bright blue LED that most dongles have.  For now I'm actually using both.  The Pluggable dongle only for the controller, and the Panda dongle for everything else.

@atar-axis if you want to test out one of my Cambridge Silicon Radio chipsets, I'll send you one.  I have a spare one.

atar-axis, 2019-02-17:
I am super happy that it works now, at least for you @ugly95 .
`HOLD` and `PARK` are just power-saving modes, initiated by the master - I don't know if this has something to do with the connection problems, but I doubt it to be honest.

I would appreciate it if you would send me one of your spare ones, where are you from?
Maybe shipping is a bit high to Germany? :grin: 
Btw, I am not an bluetooth-expert, I would love to try the adapter on my machine and dig a bit deeper if it does not work, but I cannot promise anything :)

ugly95, 2019-03-02:
Just a follow-up on this after testing for a week or two.  With the Broadcom-based adapter, the controller still works fine.  But it is quite sensitive.

For example, if I plug the adapter into a cheap 4-to-1 USB hub, the adapter does not work at all.  If I plug the adapter in one of my monitor USB ports (which essentially is a USB hub that connects to one of the rear panel USB ports on my PC), the controller connects fine and generally works, but has issues.  For example, it's a bit laggy at times, especially if I don't press a button for a minute or two.  And sometimes the rumble will get stuck on until I power cycle the gamepad.

For best results I have a USB extension cable running from a USB port on the rear panel (I think it's a USB 2.0 port), and up to my desk.  Even though this is physically farther away from me than the monitor USB port, this works best.  No lag, no rumble issues and connects well.

I did try testing the adapters in a different distro (Manjaro).  I didn't see any different results, and when I went back to my normal distro (Mint), none of the adapters would connect.

To fix I purged bluez and reinstalled:
`sudo apt purge bluez`
then I deleted everything in /var/lib/bluetooth then ran:
`sudo apt install blueberry* bluetooth* bluez* gnome-bluetooth* pulseaudio-module-bluetooth*`
After re-pairing my devices, all is well again.

@atar-axis 
I am in Canada (I could also use the US postal service since I'm near the border).  If I can ship to Germany relatively inexpensively, I'll send it.  It's tiny, so hopefully it would only be a $10 shipping cost.

msdos, 2020-10-17:
I just followed https://github.com/atar-axis/xpadneo/issues/76#issuecomment-462176703:

> Having said that, I detatched the Bluetooth adapter again (probably the 100th time), did a couple reboots, ran the delete command again, this time getting the DEL message (think its best to not be scanning when you remove a device), and was finally able to connect.

I removed the controllers in bluetoothctl, rebooted, than run `scan on` to be able to connect the controllers (just running pair directly even having the mac address wasn't working).

kakra, 2020-10-17:
Yeah, maybe we should add that to the docs: Currently, the controller can only be added with scan mode on I believe. I'm not sure if this is due to updated firmware in the controller, or due to changes in the Linux Bluetooth stack, tho.