atar-axis/xpadneo #77: Center of joysticks off

superluig164, 2019-02-18:
The centers of the joysticks seem to be off.
I am attempting to use my Xbox One controller with Dolphin.  In the setup menu, it shows the sticks being off center when neutral, but they still seem to travel the whole way they're being moved (i.e. when I move the stick physically from center to outside, the dot moves from the "false" center to the outside, regardless of the distance)

GIF explaining this a bit better (please read the description on Imgur as well)
https://imgur.com/a/YzWf4lL

uname -a:
Linux TommyROG-UB 4.18.0-14-generic #15-Ubuntu SMP Mon Jan 14 09:01:02 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux

sudo find "/sys/kernel/debug/hid/" -name "0005:045E:*" -exec sh -c 'echo "{}" && head -1 "{}/rdesc" | tee /dev/tty | cksum && echo' \:
/sys/kernel/debug/hid/0005:045E:02FD.0007
05 01 09 05 a1 01 85 01 09 01 a1 00 09 30 09 31 15 00 27 ff ff 00 00 95 02 75 10 81 02 c0 09 01 a1 00 09 32 09 35 15 00 27 ff ff 00 00 95 02 75 10 81 02 c0 05 02 09 c5 15 00 26 ff 03 95 01 75 0a 81 02 15 00 25 00 75 06 95 01 81 03 05 02 09 c4 15 00 26 ff 03 95 01 75 0a 81 02 15 00 25 00 75 06 95 01 81 03 05 01 09 39 15 01 25 08 35 00 46 3b 01 66 14 00 75 04 95 01 81 42 75 04 95 01 15 00 25 00 35 00 45 00 65 00 81 03 05 09 19 01 29 0f 15 00 25 01 75 01 95 0f 81 02 15 00 25 00 75 01 95 01 81 03 05 0c 0a 24 02 15 00 25 01 95 01 75 01 81 02 15 00 25 00 75 07 95 01 81 03 05 0c 09 01 85 02 a1 01 05 0c 0a 23 02 15 00 25 01 95 01 75 01 81 02 15 00 25 00 75 07 95 01 81 03 c0 05 0f 09 21 85 03 a1 02 09 97 15 00 25 01 75 04 95 01 91 02 15 00 25 00 75 04 95 01 91 03 09 70 15 00 25 64 75 08 95 04 91 02 09 50 66 01 10 55 0e 15 00 26 ff 00 75 08 95 01 91 02 09 a7 15 00 26 ff 00 75 08 95 01 91 02 65 00 55 00 09 7c 15 00 26 ff 00 75 08 95 01 91 02 c0 85 04 05 06 09 20 15 00 26 ff 00 75 08 95 01 81 02 c0 00 
2579218694 1006

atar-axis, 2019-02-18:
It would be so much easier to help, if ppl would fill out the whole bug report template - I cannot even tell if xpadneo is used in your case :)

superluig164, 2019-02-18:
I must have missed something, I'm sorry - could you tell me what other information you need?  I will provide it ASAP

atar-axis, 2019-02-18:
`dmesg` is missing, at least ;) no problem.

atar-axis, 2019-02-19:
Btw, the output of "evtest /dev/input/event<highest#>" would be useful too, please quickly move the sticks (left one first) like that:
middle (idle) > left > right > middle (idle)

And give me the whole Output Afterwards.

superluig164, 2019-02-19:
This is the most my terminal allowed for dmesg to output... 

`[    0.000000] BRK [0x427568000, 0x427568fff] PGTABLE
[    0.000000] BRK [0x427569000, 0x427569fff] PGTABLE
[    0.000000] BRK [0x42756a000, 0x42756afff] PGTABLE
[    0.000000] BRK [0x42756b000, 0x42756bfff] PGTABLE
[    0.000000] RAMDISK: [mem 0x3344d000-0x35a1dfff]
[    0.000000] ACPI: Early table checksum verification disabled
[    0.000000] ACPI: RSDP 0x00000000000F05B0 000024 (v02 _ASUS_)
[    0.000000] ACPI: XSDT 0x000000005678B0A8 0000CC (v01 _ASUS_ Notebook 01072009 AMI  00010013)
[    0.000000] ACPI: FACP 0x00000000567B0C88 00010C (v05 _ASUS_ Notebook 01072009 AMI  00010013)
[    0.000000] ACPI: DSDT 0x000000005678B200 025A85 (v02 _ASUS_ Notebook 01072009 INTL 20120913)
[    0.000000] ACPI: FACS 0x0000000057325F80 000040
[    0.000000] ACPI: APIC 0x00000000567B0D98 0000BC (v03 _ASUS_ Notebook 01072009 AMI  00010013)
[    0.000000] ACPI: FPDT 0x00000000567B0E58 000044 (v01 _ASUS_ Notebook 01072009 AMI  00010013)
[    0.000000] ACPI: FIDT 0x00000000567B0EA0 00009C (v01 _ASUS_ Notebook 01072009 AMI  00010013)
[    0.000000] ACPI: MCFG 0x00000000567B0F40 00003C (v01 _ASUS_ Notebook 01072009 MSFT 00000097)
[    0.000000] ACPI: HPET 0x00000000567B0F80 000038 (v01 _ASUS_ Notebook 01072009 AMI. 0005000B)
[    0.000000] ACPI: SSDT 0x00000000567B0FB8 000315 (v01 SataRe SataTabl 00001000 INTL 20120913)
[    0.000000] ACPI: ECDT 0x00000000567B12D0 0000C1 (v01 _ASUS_ Notebook 01072009 AMI. 00000005)
[    0.000000] ACPI: LPIT 0x00000000567B1398 000094 (v01 INTEL  SKL      00000000 MSFT 0000005F)
[    0.000000] ACPI: SSDT 0x00000000567B1430 000248 (v02 INTEL  sensrhub 00000000 INTL 20120913)
[    0.000000] ACPI: DBGP 0x00000000567B1678 000034 (v01 INTEL           00000000 MSFT 0000005F)
[    0.000000] ACPI: DBG2 0x00000000567B16B0 000054 (v00 INTEL           00000000 MSFT 0000005F)
[    0.000000] ACPI: SSDT 0x00000000567B1708 003D2D (v02 DptfTa DptfTabl 00001000 INTL 20120913)
[    0.000000] ACPI: SSDT 0x00000000567B5438 006685 (v02 SaSsdt SaSsdt   00003000 INTL 20120913)
[    0.000000] ACPI: UEFI 0x00000000567BBAC0 000042 (v01                 00000000      00000000)
[    0.000000] ACPI: SSDT 0x00000000567BBB08 000E58 (v02 CpuRef CpuSsdt  00003000 INTL 20120913)
[    0.000000] ACPI: SSDT 0x00000000567BC960 0000A5 (v02 SgRef  SgPeg    00001000 INTL 20120913)
[    0.000000] ACPI: DMAR 0x00000000567BCA08 0000F0 (v01 INTEL  SKL      00000001 INTL 00000001)
[    0.000000] ACPI: TPM2 0x00000000567BCAF8 000034 (v03        Tpm2Tabl 00000001 AMI  00000000)
[    0.000000] ACPI: SSDT 0x00000000567BCB30 001A3F (v01 OptRef OptTabl  00001000 INTL 20120913)
[    0.000000] ACPI: MSDM 0x0000000056102F18 000055 (v03 _ASUS_ Notebook 00000000 ASUS 00000001)
[    0.000000] ACPI: Local APIC address 0xfee00000
[    0.000000] No NUMA configuration found
[    0.000000] Faking a node at [mem 0x0000000000000000-0x0000000485ffffff]
[    0.000000] NODE_DATA(0) allocated [mem 0x485fd5000-0x485ffffff]
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000001000-0x0000000000ffffff]
[    0.000000]   DMA32    [mem 0x0000000001000000-0x00000000ffffffff]
[    0.000000]   Normal   [mem 0x0000000100000000-0x0000000485ffffff]
[    0.000000]   Device   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000001000-0x000000000009bfff]
[    0.000000]   node   0: [mem 0x0000000000100000-0x0000000051ce6fff]
[    0.000000]   node   0: [mem 0x0000000051d32000-0x0000000051d83fff]
[    0.000000]   node   0: [mem 0x00000000521c5000-0x00000000558d7fff]
[    0.000000]   node   0: [mem 0x0000000056369000-0x0000000056682fff]
[    0.000000]   node   0: [mem 0x0000000057f7f000-0x0000000057ffefff]
[    0.000000]   node   0: [mem 0x0000000100000000-0x0000000485ffffff]
[    0.000000] Reserved but unavailable: 10367 pages
[    0.000000] Initmem setup node 0 [mem 0x0000000000001000-0x0000000485ffffff]
[    0.000000] On node 0 totalpages: 4044673
[    0.000000]   DMA zone: 64 pages used for memmap
[    0.000000]   DMA zone: 21 pages reserved
[    0.000000]   DMA zone: 3995 pages, LIFO batch:0
[    0.000000]   DMA32 zone: 5408 pages used for memmap
[    0.000000]   DMA32 zone: 346086 pages, LIFO batch:31
[    0.000000]   Normal zone: 57728 pages used for memmap
[    0.000000]   Normal zone: 3694592 pages, LIFO batch:31
[    0.000000] Reserving Intel graphics memory at [mem 0x59000000-0x78ffffff]
[    0.000000] ACPI: PM-Timer IO Port: 0x1808
[    0.000000] ACPI: Local APIC address 0xfee00000
[    0.000000] ACPI: LAPIC_NMI (acpi_id[0x01] high edge lint[0x1])
[    0.000000] ACPI: LAPIC_NMI (acpi_id[0x02] high edge lint[0x1])
[    0.000000] ACPI: LAPIC_NMI (acpi_id[0x03] high edge lint[0x1])
[    0.000000] ACPI: LAPIC_NMI (acpi_id[0x04] high edge lint[0x1])
[    0.000000] ACPI: LAPIC_NMI (acpi_id[0x05] high edge lint[0x1])
[    0.000000] ACPI: LAPIC_NMI (acpi_id[0x06] high edge lint[0x1])
[... 1654 lines cut ...]
Event: time 1550548793.007753, -------------- SYN_REPORT ------------
Event: time 1550548793.014676, type 3 (EV_ABS), code 4 (ABS_RY), value 3903
Event: time 1550548793.014676, -------------- SYN_REPORT ------------
Event: time 1550548793.022647, type 3 (EV_ABS), code 4 (ABS_RY), value 3363
Event: time 1550548793.022647, -------------- SYN_REPORT ------------
Event: time 1550548793.029551, type 3 (EV_ABS), code 4 (ABS_RY), value 3147
[... cut ...]
Event: time 1550548793.097571, -------------- SYN_REPORT ------------
Event: time 1550548793.119567, type 3 (EV_ABS), code 4 (ABS_RY), value 1443
Event: time 1550548793.119567, -------------- SYN_REPORT ------------
Event: time 1550548793.134766, type 3 (EV_ABS), code 4 (ABS_RY), value 1407
Event: time 1550548793.134766, -------------- SYN_REPORT ------------
Event: time 1550548793.164656, type 3 (EV_ABS), code 4 (ABS_RY), value 1355
Event: time 1550548793.164656, -------------- SYN_REPORT ------------`

atar-axis, 2019-02-19:
According to the evtest output, the gamepad works just as expected, the range is set up correctly from -32768 to 32767 and the values are in fact within this range. The problem therefore is not xpadneo, but Dolphin which most probably expects another range without reading the evdev information about the range.

closed 2019-02-19 by atar-axis

superluig164, 2019-02-19:
It doesn't have the ability to chance it AFAIK, but I'll submit a bug report on their end - thank you.

atar-axis, 2019-02-19:
See NOTE in the third section: https://wiki.dolphin-emu.org/index.php?title=Configuring_Controllers

> NOTE: If the controller is significantly off center in Dolphin but nowhere else, check all controller axis to see if one has a "+-" on it. This is not supposed to happen and is the source of your bug.

superluig164, 2019-02-19:
Welp, I didn't see that 😅 Thank you

atar-axis, 2019-02-19:
Please tell me if/once it works ;)

superluig164, 2019-02-20:
Yep, it works great now :D  I haven't tried it with any Steam games but I'm sure it works great.  Thank you!

atar-axis, 2019-02-20:
Perfect ;)