atar-axis/xpadneo #80: Controller connects then immediatly

cbelcher, 2019-03-02:


cbelcher, 2019-03-02:
Going through the debug procedures and I don't see the hid_xpadneo directory?  Should it be there?  Sent you 3 cups of Joe!

atar-axis, 2019-03-02:
Hi There!
Thank you very much for the donation :)

No, this directory is not created as long as the devices ist not successfully connected.
The connection loop you describe is usually caused by a ERTM bug in Linux. Therefore the driver installation tool does disable ERTM.

Please upload a log of the Installation process and reboot your Pi after installation.

atar-axis, 2019-03-02:
Btw, same Problem with solution: https://github.com/atar-axis/xpadneo/issues/76#issuecomment-462175632

cbelcher, 2019-03-02:
I bet you’re right, I had to disable ERTM with the other XBOX driver, let me try that!  What part of Germany?  The wife’s family is originally from Hesse and now reside in Bad Amstal (that’s where the wife was raised), her brother is in Düsseldorf and works in Cologne and her sister lives in Marsburg..   Such a beautiful country!

cbelcher, 2019-03-02:
Thanks guys!

Btw, same Problem with solution: #76 (comment)<https://github.com/atar-axis/xpadneo/issues/76#issuecomment-462175632>

cbelcher, 2019-03-02:
Worked like a champ, thanks guys!

cbelcher, 2019-03-02:
Alter-Axis came to the rescue and got me working.  Steps to resolve issue:
Edit the autostart.sh script to add one line
sudo nano /opt/retropie/configs/all/autostart.sh
On the first line enter add:
sudo bash -c 'echo 1 > /sys/module/bluetooth/parameters/disable_ertm'
Save the file Cntl X then Y then Enter 
sudo reboot
sudo bluetoothctl
# remove the existing controller, FYI your MAC will be unique!
[bluetooth]# remove EC:83:50:12:35:2F
Then start the pairing process.
[bluetooth]# scan on
Push the connect button on upper side of the gamepad, and hold it down, until the light starts flashing fast.
Wait for the gamepad to show up in bluetoothctl, remember the MAC address (EC:83:50:12:35:2F)
[bluetooth]# pair EC:83:50:12:35:2F
[bluetooth]# trust EC:83:50:12:35:2F
[bluetooth]# connect EC:83:50:12:35:2F
Done.

cbelcher, 2019-03-02:
What would cause /dev/input/js0 not to show up?  The controller shows connected.

atar-axis, 2019-03-02:
Hey Craig!

> What part of Germany?

The most southern :) (Upper Bavaria)

> Edit the autostart.sh script to add one line (...)

This is not necessary since the _post_install script_ should already configure modprobe to do something silimar, see here: https://raw.githubusercontent.com/atar-axis/xpadneo/master/hid-xpadneo/dkms.post_install
But I am happy if it works for you anyway 😄 

> What would cause /dev/input/js0 not to show up?

Hum, it depends. Can you please upload again the output of...
* `dmesg` and
* `lsmod | grep xpadneo`
after the connection is established. Thanks!

cbelcher, 2019-03-02:
pi@retropie:~$ lsmod
Module                  Size  Used by
hidp                   24576  1
cmac                   16384  1
bnep                   20480  2
hci_uart               24576  1
bluetooth             368640  28 hidp,hci_uart,bnep
ecdh_generic           28672  1 bluetooth
brcmfmac              307200  0
brcmutil               16384  1 brcmfmac
evdev                  24576  2
joydev                 20480  0
cfg80211              573440  1 brcmfmac
rfkill                 28672  6 bluetooth,cfg80211
snd_bcm2835            32768  1
snd_pcm                98304  2 snd_bcm2835
snd_timer              32768  1 snd_pcm
snd                    69632  4 snd_timer,snd_bcm2835,snd_pcm
fixed                  16384  0
uio_pdrv_genirq        16384  0
uio                    20480  1 uio_pdrv_genirq
uinput                 20480  0
ip_tables              24576  0
x_tables               28672  1 ip_tables
ipv6                  434176  44

pi@retropie:~$ lsmod | grep xpadneo
pi@retropie:~$

cbelcher, 2019-03-02:
The wife works for the German National Tourist Board aka GNT, she gets Americans to come visit Germany and she loves it.

atar-axis, 2019-03-02:
So it looks like xpadneo is not bounded to the gamepad, could you please upload the `dmesg` too? ;)

cbelcher, 2019-03-02:
> If you want me to analyze why it didn't worked out of the box, then please upload a log of the installation 😃 and the ouput of `cat /etc/modprobe.d/99-xpadneo-bluetooth.conf`

What directory is the log or it's name?  And there is no file on my PI with the name 99-xpadneo-bluetooth.conf.  I'm guessing we should see some output in the lsmod output?

cbelcher, 2019-03-02:
Schlaf gut!

atar-axis, 2019-03-02:
Aha! It becomes clearer and clearer :) Looks like sth. went wrong while installing the driver!

`dmesg` is just a command you type into the shell ;)

Anyway, please Reinstall the driver!

1. Switch into the directory where you cloned the Repo to
2. Type `sudo ./uninstall.sh`
3. `sudo ./install.sh`
4. Upload the output of both commands ;)

Danke ;) Ihr auch!

cbelcher, 2019-03-02:
pi@retropie:/etc/modprobe.d$ dmesg
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.14.30-v7+ (dc4@dc4-XPS13-9333) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611)) #1102 SMP Mon Mar 26 16:45:49 BST 2018
[    0.000000] CPU: ARMv7 Processor [410fd034] revision 4 (ARMv7), cr=10c5383d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi 3 Model B Plus Rev 1.3
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] cma: Reserved 8 MiB at 0x1f400000
[    0.000000] On node 0 totalpages: 131072
[    0.000000] free_area_init_node: node 0, pgdat 80c7edc0, node_mem_map 9ef80000
[    0.000000]   Normal zone: 1152 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 131072 pages, LIFO batch:31
[    0.000000] random: fast init done
[    0.000000] percpu: Embedded 17 pages/cpu @9ff95000 s38720 r8192 d22720 u69632
[    0.000000] pcpu-alloc: s38720 r8192 d22720 u69632 alloc=17*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 129920
[    0.000000] Kernel command line: 8250.nr_uarts=0 bcm2708_fb.fbwidth=1440 bcm2708_fb.fbheight=900 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  dwc_otg.lpm_enable=0 console=ttyS0,115200 console=tty3 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait quiet loglevel=3 logo.nologo consoleblank=0 plymouth.enable=0
[    0.000000] PID hash table entries: 2048 (order: 1, 8192 bytes)
[    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes)
[    0.000000] Memory: 498116K/524288K available (7168K kernel code, 550K rwdata, 2048K rodata, 1024K init, 706K bss, 17980K reserved, 8192K cma-reserved)
[    0.000000] Virtual kernel memory layout:
                   vector  : 0xffff0000 - 0xffff1000   (   4 kB)
                   fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
                   vmalloc : 0xa0800000 - 0xff800000   (1520 MB)
                   lowmem  : 0x80000000 - 0xa0000000   ( 512 MB)
                   modules : 0x7f000000 - 0x80000000   (  16 MB)
                     .text : 0x80008000 - 0x80800000   (8160 kB)
                     .init : 0x80b00000 - 0x80c00000   (1024 kB)
                     .data : 0x80c00000 - 0x80c8986c   ( 551 kB)
                      .bss : 0x80c90f40 - 0x80d417f4   ( 707 kB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 24918 entries in 74 pages
[    0.000000] Hierarchical RCU implementation.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] arch_timer: cp15 timer(s) running at 19.20MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
[    0.000007] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
[    0.000018] Switching to timer-based delay loop, resolution 52ns
[    0.000268] Console: colour dummy device 80x30
[    0.000285] console [tty3] enabled
[    0.000310] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=192000)
[    0.000325] pid_max: default: 32768 minimum: 301
[    0.000636] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.000647] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.001581] Disabling memory control group subsystem
[    0.001661] CPU: Testing write buffer coherency: ok
[    0.002056] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
[    0.002434] Setting up static identity map for 0x100000 - 0x10003c
[    0.002550] Hierarchical SRCU implementation.
[    0.003181] smp: Bringing up secondary CPUs ...
[    0.003913] CPU1: thread -1, cpu 1, socket 0, mpidr 80000001
[    0.004709] CPU2: thread -1, cpu 2, socket 0, mpidr 80000002
[    0.005493] CPU3: thread -1, cpu 3, socket 0, mpidr 80000003
[    0.005596] smp: Brought up 1 node, 4 CPUs
[    0.005608] SMP: Total of 4 processors activated (153.60 BogoMIPS).
[    0.005613] CPU: All CPU(s) started in HYP mode.
[    0.005617] CPU: Virtualization extensions available.
[    0.006390] devtmpfs: initialized
[    0.017047] VFP support v0.3: implementor 41 architecture 3 part 40 variant 3 rev 4
[    0.017269] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.017286] futex hash table entries: 1024 (order: 4, 65536 bytes)
[    0.017841] pinctrl core: initialized pinctrl subsystem
[    0.018587] NET: Registered protocol family 16
[    0.021310] DMA: preallocated 1024 KiB pool for atomic coherent allocations
[    0.026195] hw-breakpoint: found 5 (+1 reserved) breakpoint and 4 watchpoint registers.
[    0.026202] hw-breakpoint: maximum watchpoint size is 8 bytes.
[... 190 lines cut ...]
[    5.869959] Adding 102396k swap on /var/swap.  Priority:-2 extents:1 across:102396k SSFS
[    5.941711] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[    5.941766] brcmfmac: power management disabled
[    7.038673] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
[    7.040939] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[    7.041141] brcmfmac: power management disabled
[... cut ...]
[   11.741580] Bluetooth: BNEP filters: protocol multicast
[   11.741594] Bluetooth: BNEP socket layer initialized
[14133.117932] Bluetooth: HIDP (Human Interface Emulation) ver 1.2
[14133.117972] Bluetooth: HIDP socket layer initialized
[14133.122885] hid-generic 0005:045E:02FD.0003: unknown main item tag 0x0
[14133.122955] hid-generic 0005:045E:02FD.0003: unbalanced collection at end of report description
[14133.123048] hid-generic: probe of 0005:045E:02FD.0003 failed with error -22
pi@retropie:/etc/modprobe.d$

cbelcher, 2019-03-02:
uninstalling and reinstalling.

cbelcher, 2019-03-02:
output from the uninstall and reinstall.

pi@retropie:~$ cd /home/pi/xpadneo/
pi@retropie:~/xpadneo$ sudo ./uninstall.sh 
* unloading current driver module
modprobe: FATAL: Module hid_xpadneo not found.
* looking for registered instances
found 1 registered instance(s) on your system
* 0.5.4
  * uninstalling and removing 0.5.4 from DKMS

------------------------------
Deleting module version: 0.5.4
completely from the DKMS tree.
------------------------------
Done.
  * removing 0.5.4 folder from /usr/src
pi@retropie:~/xpadneo$ sudo ./install
sudo: ./install: command not found
pi@retropie:~/xpadneo$ sudo ./install.sh
* replacing version string if necessary
* copying module into /usr/src
* adding module to DKMS

Creating symlink /var/lib/dkms/hid-xpadneo/0.5.4/source ->
                 /usr/src/hid-xpadneo-0.5.4

DKMS: add completed.
* installing module (using DKMS)

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area....
make -j4 KERNELRELEASE=4.14.30-v7+ -C /lib/modules/4.14.30-v7+/build M=/var/lib/dkms/hid-xpadneo/0.5.4/build/src modules....
cleaning build area...

DKMS: build completed.

hid-xpadneo.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/4.14.30-v7+/kernel/drivers/hid/

Running the post_install script:
Disabling ERTM permanently (requires reboot)...
Diabling ERTM instantly...
Installing UDEV rule to automatically bind xpadneo (necessary on kernel < 4.16 and >= 4.20)...

depmod....

Warning: Unable to find an initial ram disk that I know how to handle.
Will not try to make an initrd.

DKMS: install completed.
pi@retropie:~/xpadneo$

cbelcher, 2019-03-02:
Still no /dev/input/js0, lsmod | grep xpadneo has a
 null output.  The controller is paired but nothing see's it.  Bottom line no change.

atar-axis, 2019-03-02:
OK perfect! Now please turn off the gamepad, reboot the Pi, type `sudo dmesg -C`.

Now connect your gamepad and share the `dmesg` output with me again.

Thank you for your patience!

cbelcher, 2019-03-02:
sudo dmesg -C output is null.

cbelcher, 2019-03-02:
pi@retropie:~$ dmesg
[  131.755570] Bluetooth: HIDP (Human Interface Emulation) ver 1.2
[  131.755609] Bluetooth: HIDP socket layer initialized
[  131.761101] hid-generic 0005:045E:02FD.0003: unknown main item tag 0x0
[  131.761190] hid-generic 0005:045E:02FD.0003: unbalanced collection at end of report description
[  131.761344] hid-generic: probe of 0005:045E:02FD.0003 failed with error -22
[  131.788067] hid_xpadneo: loading out-of-tree module taints kernel.
[  131.788528] hid_xpadneo: disagrees about version of symbol dev_err
[  131.788539] hid_xpadneo: Unknown symbol dev_err (err -22)
[  131.788592] hid_xpadneo: disagrees about version of symbol dev_printk
[  131.788599] hid_xpadneo: Unknown symbol dev_printk (err -22)

atar-axis, 2019-03-02:
Thanks! I will have to so some research since this is a totally new error for me too, I report back as soon as I understand it a bit more ;)

cbelcher, 2019-03-02:
No worries mate, gute nacht :)

atar-axis, 2019-03-02:
Looks like the new issue (not the connection problem but the 'unbalanced' error in dmesg) is the same as reported in #61.

Could you therefore please Update your gamepad before we continue? Either via an Xbox or Windows 10 OS (Xbox Accessoires App). It doesnt matter If your fw is already the latest btw. ;)

cbelcher, 2019-03-03:
I did before I started with this driver install, needed them with another driver that I got working.  It’s a Model 1708 with the latest firmware.

cbelcher, 2019-03-03:
Isn’t it time for bed in Germany, like midnight?

cbelcher, 2019-03-03:
Go get some sleep this there is no rush at all!

cbelcher, 2019-03-03:
So you’re telling me to reinstall the latest firmware, don’t think I read the last part 😊  Trying it now.

cbelcher, 2019-03-03:
Ya just comes back when I try to update it and says No updates needed.  It’s running 3.1.1221.0

atar-axis, 2019-03-03:
Yeah right, this resets the gamepad :)
After updating, please reboot the Pi again, connect the gamepad and give me the output of `sudo find "/sys/kernel/debug/hid/" -name "0005:045E:*" -exec sh -c 'echo "{}" && head -1 "{}/rdesc" | tee /dev/tty | cksum && echo' \;`

cbelcher, 2019-03-03:
pi@retropie:~$ sudo find "/sys/kernel/debug/hid/" -name "0005:045E:*" -exec sh -c 'echo "{}" && head -1 "{}/rdesc" | tee /dev/tty | cksum && echo' \;
/sys/kernel/debug/hid/0005:045E:02FD.0003
05 01 09 05 a1 01 85 01 09 01 a1 00 09 30 09 31 15 00 27 ff ff 00 00 95 02 75 10 81 02 c0 09 01 a1 00 09 32 09 35 15 00 27 ff ff 00 00 95 02 75 10 81 02 c0 05 02 09 c5 15 00 26 ff 03 95 01 75 0a 81 02 15 00 25 00 75 06 95 01 81 03 05 02 09 c4 15 00 26 ff 03 95 01 75 0a 81 02 15 00 25 00 75 06 95 01 81 03 05 01 09 39 15 01 25 08 35 00 46 3b 01 66 14 00 75 04 95 01 81 42 75 04 95 01 15 00 25 00 35 00 45 00 65 00 81 03 05 09 19 01 29 0f 15 00 25 01 75 01 95 0f 81 02 15 00 25 00 75 01 95 01 81 03 05 0c 0a 24 02 15 00 25 01 95 01 75 01 81 02 15 00 25 00 75 07 95 01 81 03 05 0c 09 01 85 02 a1 01 05 0c 0a 23 02 15 00 25 01 95 01 75 01 81 02 15 00 25 00 75 07 95 01 81 03 c0 05 0f 09 21 85 03 a1 02 09 97 15 00 25 01 75 04 95 01 91 02 15 00 25 00 75 04 95 01 91 03 09 70 15 00 25 64 75 08 95 04 91 02 09 50 66 01 10 55 0e 15 00 26 ff 00 75 08 95 01 91 02 09 a7 15 00 26 ff 00 75 08 95 01 91 02 65 00 55 00 09 7c 15 00 26 ff 00 75 08 95 01 91 02 c0 85 04 05 06 09 20 15 00 26 ff 00 75 08 95 01 81 02 c0 00
2579218694 1006

pi@retropie:~$

atar-axis, 2019-03-03:
Perfect, the hid report descriptor is correct, can you please add the dmesg output too - I would like to check if it still reports the same problem ;)

cbelcher, 2019-03-03:
pi@retropie:~$ dmesg
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.14.30-v7+ (dc4@dc4-XPS13-9333) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611)) #1102 SMP Mon Mar 26 16:45:49 BST 2018
[    0.000000] CPU: ARMv7 Processor [410fd034] revision 4 (ARMv7), cr=10c5383d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi 3 Model B Plus Rev 1.3
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] cma: Reserved 8 MiB at 0x1f400000
[    0.000000] On node 0 totalpages: 131072
[    0.000000] free_area_init_node: node 0, pgdat 80c7edc0, node_mem_map 9ef80000
[    0.000000]   Normal zone: 1152 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 131072 pages, LIFO batch:31
[    0.000000] random: fast init done
[    0.000000] percpu: Embedded 17 pages/cpu @9ff95000 s38720 r8192 d22720 u69632
[    0.000000] pcpu-alloc: s38720 r8192 d22720 u69632 alloc=17*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 129920
[    0.000000] Kernel command line: 8250.nr_uarts=0 bcm2708_fb.fbwidth=1440 bcm2708_fb.fbheight=900 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  dwc_otg.lpm_enable=0 console=ttyS0,115200 console=tty3 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait quiet loglevel=3 logo.nologo consoleblank=0 plymouth.enable=0
[    0.000000] PID hash table entries: 2048 (order: 1, 8192 bytes)
[    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes)
[    0.000000] Memory: 498116K/524288K available (7168K kernel code, 550K rwdata, 2048K rodata, 1024K init, 706K bss, 17980K reserved, 8192K cma-reserved)
[    0.000000] Virtual kernel memory layout:
                   vector  : 0xffff0000 - 0xffff1000   (   4 kB)
                   fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
                   vmalloc : 0xa0800000 - 0xff800000   (1520 MB)
                   lowmem  : 0x80000000 - 0xa0000000   ( 512 MB)
                   modules : 0x7f000000 - 0x80000000   (  16 MB)
                     .text : 0x80008000 - 0x80800000   (8160 kB)
                     .init : 0x80b00000 - 0x80c00000   (1024 kB)
                     .data : 0x80c00000 - 0x80c8986c   ( 551 kB)
                      .bss : 0x80c90f40 - 0x80d417f4   ( 707 kB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 24918 entries in 74 pages
[    0.000000] Hierarchical RCU implementation.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] arch_timer: cp15 timer(s) running at 19.20MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
[    0.000006] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
[    0.000018] Switching to timer-based delay loop, resolution 52ns
[    0.000268] Console: colour dummy device 80x30
[    0.000286] console [tty3] enabled
[    0.000310] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=192000)
[    0.000325] pid_max: default: 32768 minimum: 301
[    0.000638] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.000648] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.001582] Disabling memory control group subsystem
[    0.001663] CPU: Testing write buffer coherency: ok
[    0.002056] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
[    0.002435] Setting up static identity map for 0x100000 - 0x10003c
[    0.002552] Hierarchical SRCU implementation.
[    0.003183] smp: Bringing up secondary CPUs ...
[    0.003912] CPU1: thread -1, cpu 1, socket 0, mpidr 80000001
[    0.004712] CPU2: thread -1, cpu 2, socket 0, mpidr 80000002
[    0.005495] CPU3: thread -1, cpu 3, socket 0, mpidr 80000003
[    0.005599] smp: Brought up 1 node, 4 CPUs
[    0.005609] SMP: Total of 4 processors activated (153.60 BogoMIPS).
[    0.005614] CPU: All CPU(s) started in HYP mode.
[    0.005619] CPU: Virtualization extensions available.
[    0.006396] devtmpfs: initialized
[    0.017061] VFP support v0.3: implementor 41 architecture 3 part 40 variant 3 rev 4
[    0.017282] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.017300] futex hash table entries: 1024 (order: 4, 65536 bytes)
[    0.017858] pinctrl core: initialized pinctrl subsystem
[    0.018605] NET: Registered protocol family 16
[    0.021331] DMA: preallocated 1024 KiB pool for atomic coherent allocations
[    0.026216] hw-breakpoint: found 5 (+1 reserved) breakpoint and 4 watchpoint registers.
[    0.026223] hw-breakpoint: maximum watchpoint size is 8 bytes.
[... 201 lines cut ...]
[   12.024272] Bluetooth: L2CAP socket layer initialized
[   12.024314] Bluetooth: SCO socket layer initialized
[   12.038223] Bluetooth: HCI UART driver ver 2.3
[   12.038239] Bluetooth: HCI UART protocol H4 registered
[   12.038245] Bluetooth: HCI UART protocol Three-wire (H5) registered
[   12.209408] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   12.209417] Bluetooth: BNEP filters: protocol multicast
[... cut ...]
[  500.121436] systemd[1]: apt-daily.timer: Adding 8h 16min 27.686505s random time.
[  500.122058] systemd[1]: apt-daily-upgrade.timer: Adding 31min 25.798210s random time.
[ 1024.044187] systemd[1]: apt-daily.timer: Adding 4h 44min 21.944464s random time.
[ 1024.044961] systemd[1]: apt-daily-upgrade.timer: Adding 32min 14.750916s random time.
[ 1026.746838] systemd[1]: apt-daily.timer: Adding 3h 45min 19.941871s random time.
[ 1026.747655] systemd[1]: apt-daily-upgrade.timer: Adding 14min 29.155516s random time.
pi@retropie:~$

atar-axis, 2019-03-03:
Looking good! Second issue solved - one left ;)

cbelcher, 2019-03-03:
Lol 😊

atar-axis, 2019-03-03:
I think the (hopefully) last problem is most probably related to the kernel headers used by dkms, I will report back tomorrow 😴

I am sorry that you are having such a hard time using this driver, it usually works "out of the box"

cbelcher, 2019-03-03:
No worries mate!  If there’s an issue I’m the one to find it!  I’ve got the best luck!

atar-axis, 2019-03-03:
* `uname -r` 
* `apt list --installed | grep -i header`

atar-axis, 2019-03-03:
And `dpkg -s raspberrypi-kernel-headers`

cbelcher, 2019-03-03:
pi@retropie:~$ uname -r
4.14.30-v7+
pi@retropie:~$ apt list --installed | grep -i header

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

libxmu-headers/stable,now 2:1.1.2-2 all [installed,automatic]
raspberrypi-kernel-headers/stable,now 1.20190215-1 armhf [installed]
pi@retropie:~$

cbelcher, 2019-03-03:
It’s working from the last time we touched base I upgraded from:

Linux 4.14.30-v7+ armv7l GNU/Linux
To
Linux 4.14.98-v7+ armv7l GNU/Linux

atar-axis, 2019-03-03:
Working? Like "Problems solved"? :) Did it rumble?

cbelcher, 2019-03-03:
Can’t say I remember it rumble, but js0 is now there and Retroarch see’s it.  I call it done!

Put that in your database of fixes, looks like some issue with that version of RetroPie?

For the record these are the steps I used update RetroPie.

Launch RetroPie setup.
pi@retropie:~$ sudo /home/pi/RetroPie-Setup/retropie_setup.sh

Select Update
Took at least 30 minutes to update.
Reboot, starting working!

atar-axis, 2019-03-03:
It should rumble every time you connect it - does it? :)
If not, I fear we aren't done because then xpadneo isn't used and the gamepad will not work reliable ;)

cbelcher, 2019-03-03:
Let me test.

cbelcher, 2019-03-03:
Confirmed, never rumbles!  So it’s using something other than xpadneo?

atar-axis, 2019-03-03:
Yes, the question is: is it hid-generic or hid-microsoft?
I guess it is the former, we can simply check that by:

* `sudo dmesg -C` (clears the dmesg output)
* connect gamepad
* `dmesg` (please upload the output)

cbelcher, 2019-03-03:
pi@retropie:~$ sudo dmesg -C
pi@retropie:~$ dmesg
[  531.042681] hid-generic 0005:045E:02FD.0004: unknown main item tag 0x0
[  531.043634] input: Xbox Wireless Controller as /devices/platform/soc/3f201000.serial/tty/ttyAMA0/hci0/hci0:11/0005:045E:02FD.0004/input/input3
[  531.048099] hid-generic 0005:045E:02FD.0004: input,hidraw2: BLUETOOTH HID v9.03 Gamepad [Xbox Wireless Controller] on b8:27:eb:35:fc:c7
pi@retropie:~$

cbelcher, 2019-03-03:
Looks like you guessed correctly!

cbelcher, 2019-03-03:
Got it working.  I feel the rumble.  Let me check the dsmeg.

pi@retropie:~$ sudo dmesg
[  295.997631] hid-generic 0005:045E:02FD.0004: unknown main item tag 0x0
[  295.998644] input: Xbox Wireless Controller as /devices/platform/soc/3f201000.serial/tty/ttyAMA0/hci0/hci0:11/0005:045E:02FD.0004/input/input4
[  296.000132] hid-generic 0005:045E:02FD.0004: input,hidraw2: BLUETOOTH HID v9.03 Gamepad [Xbox Wireless Controller] on b8:27:eb:35:fc:c7
[  296.199877] xpadneo 0005:045E:02FD.0004: unknown main item tag 0x0
[  296.201850] input: Xbox Wireless Controller as /devices/platform/soc/3f201000.serial/tty/ttyAMA0/hci0/hci0:11/0005:045E:02FD.0004/input/input5
[  296.205541] xpadneo 0005:045E:02FD.0004: input,hidraw2: BLUETOOTH HID v9.03 Gamepad [Xbox Wireless Controller] on b8:27:eb:35:fc:c7
pi@retropie:~$

That looks good!

So How I got this fixed.

First thing I noticed Raspian was asking for login on reboot, this was caused by the RetroPie Upgrade.  The forums say it’s systemd?

Fixed that with

at command prompt, type sudo raspi-config
select option 3 in menu (Boot Options)
select option B1 (Desktopp/CLI)
select option B2 (Console Autologin)

reboot

Then I tried to connect controller and nothing, launched bluetoothctl and saw the controller connect and disconnect like I did when we first started this.

Uninstalled xpadneo and reinstalled
Output of that:
pi@retropie:~/xpadneo$ sudo ./uninstall.sh
* unloading current driver module
modprobe: FATAL: Module hid_xpadneo not found.
* looking for registered instances
found 0 registered instance(s) on your system
pi@retropie:~/xpadneo$ sudo ./install.sh
* replacing version string if necessary
* copying module into /usr/src
* adding module to DKMS

Creating symlink /var/lib/dkms/hid-xpadneo/0.5.4/source ->
                 /usr/src/hid-xpadneo-0.5.4

DKMS: add completed.
* installing module (using DKMS)

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area....
make -j4 KERNELRELEASE=4.14.98-v7+ -C /lib/modules/4.14.98-v7+/build M=/var/lib/dkms/hid-xpadneo/0.5.4/build/src modules.....
cleaning build area...

DKMS: build completed.

hid-xpadneo.ko:
Running module version sanity check.
- Original module
   - No original module exists within this kernel
- Installation
   - Installing to /lib/modules/4.14.98-v7+/kernel/drivers/hid/

Running the post_install script:
Disabling ERTM permanently (requires reboot)...
Diabling ERTM instantly...
Installing UDEV rule to automatically bind xpadneo (necessary on kernel < 4.16 and >= 4.20)...

depmod....

Warning: Unable to find an initial ram disk that I know how to handle.
Will not try to make an initrd.

DKMS: install completed.

Tried to connect the controller and it rummbed and the Xbox controller light went solid.  js0 showed up int /dev/input and RetroPie saw the controller.

Done now I think?

atar-axis, 2019-03-03:
Perfect, that's what I would have suggested next ;)
Yeah, now it's done 😁 

You had three problems I think
1. The installation went wrong somehow, that's why ERTM was still enabled
2. Your gamepads firmware was in a faulty state where the HID report descriptor is cropped
3. Your headers and kernel did not fit somehow, that was fixed after the update. Moreover DKMS did not automatically upgrade xpadneo - no idea why.

That took some doing xD

closed 2019-03-03 by atar-axis

cbelcher, 2019-03-03:
But we did it.  Thanks guys

cbelcher, 2019-03-04:
Just a heads up I reformatted my SD card on the Pi and took it back to the place you and I started.  This time I updated RetroPie first and then followed your procedure.  No issues at all, just flawless.

Thanks Florian!

Closed #80<https://github.com/atar-axis/xpadneo/issues/80>.

atar-axis, 2019-03-04:
You are Welcome Craig!

mlyszczek, 2019-12-12:
I got same problem on Gentoo, I could pair with pad but right after I connected I got disconnect. In my case I was missing 'hidp' module (BT_HIDP). I leave it here as this is easily found by search engines with 'xpadneo disconnect' keywrods.