atar-axis/xpadneo #81: Messed keymaps on Xbox One Wireless Controller White

Teashrock, 2019-03-02:
**Describe the bug**
All keymaps on gamepad are mixed up with each other. Menu Button acts like RB, View is like LB, RT like LT, et cetera.

**To Reproduce**
Steps to reproduce the behavior:
1. Install driver.
2. Connect via Bluetooth.
3. Try to play (unsuccesfully).

**Expected behavior**
A normal keymapping where each button/stick fits the place.

**Screenshots/Gifs**
https://youtu.be/mxuF3whtBF8 (PLEASE, TURN OFF SOUND)

**System information**
Please enable debugging output <https://atar-axis.github.io/xpadneo/index.html#debugging>
and add at least the following outputs
-   `uname -a`: Linux G31M-ES2L 4.15.0-45-generic #48-Ubuntu SMP Tue Jan 29 16:28:13 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux
-   `dmesg`: https://pastebin.com/Bai5EG2u
-   `sudo find "/sys/kernel/debug/hid/" -name "0005:045E:*" -exec sh -c 'echo "{}" && head -1 "{}/rdesc" | tee /dev/tty | cksum && echo' \;`: 
/sys/kernel/debug/hid/0005:045E:02FD.0002
05 01 09 05 a1 01 85 01 09 01 a1 00 09 30 09 31 15 00 27 ff ff 00 00 95 02 75 10 81 02 c0 09 01 a1 00 09 32 09 35 15 00 27 ff ff 00 00 95 02 75 10 81 02 c0 05 02 09 c5 15 00 26 ff 03 95 01 75 0a 81 02 15 00 25 00 75 06 95 01 81 03 05 02 09 c4 15 00 26 ff 03 95 01 75 0a 81 02 15 00 25 00 75 06 95 01 81 03 05 01 09 39 15 01 25 08 35 00 46 3b 01 66 14 00 75 04 95 01 81 42 75 04 95 01 15 00 25 00 35 00 45 00 65 00 81 03 05 09 19 01 29 0f 15 00 25 01 75 01 95 0f 81 02 15 00 25 00 75 01 95 01 81 03 05 0c 0a 24 02 15 00 25 01 95 01 75 01 81 02 15 00 25 00 75 07 95 01 81 03 05 0c 09 01 85 02 a1 01 05 0c 0a 23 02 15 00 25 01 95 01 75 01 81 02 15 00 25 00 75 07 95 01 81 03 c0 05 0f 09 21 85 03 a1 02 09 97 15 00 25 01 75 04 95 01 91 02 15 00 25 00 75 04 95 01 91 03 09 70 15 00 25 64 75 08 95 04 91 02 09 50 66 01 10 55 0e 15 00 26 ff 00 75 08 95 01 91 02 09 a7 15 00 26 ff 00 75 08 95 01 91 02 65 00 55 00 09 7c 15 00 26 ff 00 75 08 95 01 91 02 c0 85 04 05 06 09 20 15 00 26 ff 00 75 08 95 01 81 02 c0 00 
2579218694 1006

atar-axis, 2019-03-02:
Hi there!
Could you please give me the **generated events**, like described here:
https://atar-axis.github.io/xpadneo/#debugging

Teashrock, 2019-03-02:
Here, please! https://pastebin.com/gqakTz75

atar-axis, 2019-03-02:
> Please press the buttons in question, tell me the order you pressed them and upload the ouput 😁

atar-axis, 2019-03-02:
Look, the output is useless as long as I don't know the order in which you pressed the buttons ;D
Furthermore, your output is a bit long, please do not press/move any other button or axis!

Teashrock, 2019-03-03:
RAxis: https://pastebin.com/rBk8j3Lk
LAxis: https://pastebin.com/Ag8YERiz
LT: https://pastebin.com/Ha6WRxxz
RT: https://pastebin.com/ikb0X89R
LB: https://pastebin.com/jfQ1jiEG
RB: https://pastebin.com/8npCPM5Y
View: https://pastebin.com/6mxChFSX
Menu: https://pastebin.com/p6bRdhWB
X: https://pastebin.com/NsU52Ttv
Y: https://pastebin.com/Mrqa2uca
B: https://pastebin.com/6KGE2K49
A: https://pastebin.com/ZDmhKtx3
Left-D-Pad: https://pastebin.com/ep3B6pfN
Up-D-Pad: https://pastebin.com/ddgfd6FY
Right-D-Pad: https://pastebin.com/PZuN8J7P
Down-D-Pad: https://pastebin.com/9UT6WK5h

atar-axis, 2019-03-03:
Looks like everything is alright, so the oroblem is not the driver but the Game you are playing - which Emulator/Game is it?

atar-axis, 2019-03-03:
I will close this issue since the problem is not the driver but the emulators/games mapping. I will try to help you anyway!

Teashrock, 2019-03-04:
Sorry, did not respond earlier. The game I playingis not from any emulator, it's called "CrossCode" and available via Steam. I'll give record of graphical JStest of my gamepad, so you'll see that issue exists. If I wasn't sure, won't submit a bug report at all.

Teashrock, 2019-03-04:
P. S. I tried also another driver here: https://github.com/paroj/xpad but it's incomplete and sort of abandoned. Most of keys with it works correctly in the same game, but the driver itself glithces often, and View button isn't working at all. Even turn it on is problem.

Then, I saw this driver and that it's actively developed, so wrote here.

atar-axis, 2019-03-04:
Well we an reooen it if the promised output is convicing ;)

Teashrock, 2019-03-10:
At least, here: https://youtu.be/4ClP9vuZjYs

Teashrock, 2019-03-12:
Are you still there?

atar-axis, 2019-03-12:
Yes, but I am quite busy atm, I will try to help you as soon as possible

atar-axis, 2019-03-13:
It's hard to tell - from your video - which buttons you are pressing. I cannot reproduce the problem here and your evtest output is right, which means that the first layer above the driver is working correctly. The problem has to be somewhere else, from your video I cannot say if the output is correct.

Teashrock, 2019-03-13:
Okay, I'll provide you a description later.

Teashrock, 2019-03-13:
0:40 Left Axis. Behaves totally normal, as i can tell?
0:43 Right axis. Yes, this is one right axis, not the button with it! As you can see, it jammed to the left by default (LT-button is responsible); onscreen up/down moves in reality are left/right respectively; onscreen LT pressing up/down is actually down/up (!!!) of Right Axis!
0:54 Left Trigger (LT) and Right Trigger (RT). Onscreen moving of Right Axis to right side depends on pressing the LT and I can't say that much about RT, 'cus I don't know which value should be normal.
1:04 So, there is buttons.

0 is A
1 is B
2 is X
3 is Y
4 is LB (Left Button)
5 is RB (Right Button)
6 is View Button
7 is Menu Button
8 button was not found! What's this?
9 is LAxis pressing
10 is RAxis pressing

Teashrock, 2019-03-20:
So, what's up?

Teashrock, 2019-03-20:
Did you found the bug already (sorry if annoying)?

atar-axis, 2019-03-20:
does `jstest` (non graphical) behave the same?

atar-axis, 2019-03-20:
Okay I was able to reproduce it, the problem is a wrong mapping!
This is **not** caused by xpadneo but the joystick interface...
Just use the "Mapping" button in jstest-gtk to fix it on your system ;)

closed 2019-03-20 by atar-axis

Teashrock, 2019-03-20:
How can I remap it permanently? Also, cannot drag and drop between two lists, but this is required.

atar-axis, 2019-03-20:
> How can I remap it permanently?

http://lmgtfy.com/?q=jstest+permanent+mapping
First Hit...

atar-axis, 2019-03-20:
Btw, if you do not answer my questions, how can I help you then?

> does jstest (non graphical) behave the same?

Teashrock, 2019-03-20:
Sorry, I just didn't saw the message. Will try to answer soon.

atar-axis, 2019-03-20:
Have you tried the above mapping?