atar-axis/xpadneo #83: Kernel stable:	5.x.x

TeknoMobil, 2019-03-13:
Hello there
Will there be Kernel 5.x.x support

atar-axis, 2019-03-13:
Yes, for sure. Should, in theory, work out of the box - but I haven't yet tested it. Do you have any issues?

TeknoMobil, 2019-03-13:
The problem originated from me, thanks
kernel headers for kernel 5.0.0-300.fc30.x86_64 -- Did not come with this update

[serdar@fedora xpadneo-master]$ sudo ./install.sh
[sudo] password for serdar: 
* replacing version string if necessary
* copying module into /usr/src
* adding module to DKMS

```
Creating symlink /var/lib/dkms/hid-xpadneo/0.5.4/source ->
                 /usr/src/hid-xpadneo-0.5.4

DKMS: add completed.
* installing module (using DKMS)
Error! echo
Your kernel headers for kernel 5.0.0-300.fc30.x86_64 cannot be found at
/lib/modules/5.0.0-300.fc30.x86_64/build or /lib/modules/5.0.0-300.fc30.x86_64/source.
```

atar-axis, 2019-03-13:
Well, you need fitting headers in order to use DKMS for installation - if those headers aren't installed by default, then you need to install them by hand, have you tried ``sudo dnf install dkms bluez bluez-tools kernel-devel-`uname -r` kernel-headers-`uname -r``` ?
And, have you rebooted after the update to 5.0?

TeknoMobil, 2019-03-13:
Yes, the problem is Fedora.
kernel-headers-5.0.0-300.fc30.x86_64 not found

[serdar@fedora xpadneo-master]$ uname -a
Linux fedora 5.0.0-300.fc30.x86_64.
-----------------------------------------------------------------------------
Thank you
regards

atar-axis, 2019-03-13:
Hum, I am not used to Fedora - which Release do you use? F29?

closed 2019-03-13 by atar-axis

TeknoMobil, 2019-03-13:
I am using Fedora 30 Test version :)
Thanks

atar-axis, 2019-03-13:
> I am using Fedora 30 Test version

That's the problem :D

TeknoMobil, 2019-03-23:
Header updated with new kernel, and works perfectly :)

atar-axis, 2019-03-23:
Thank you for getting Back ;)

ElazarHatesScreenNames, 2019-04-30:
@TeknoMobil Hi! I am having the same issue, where did you get the headers? I get 
No match for argument: kernel-devel-5.0.0-300.fc30.x86_64
No match for argument: kernel-headers-5.0.0-300.fc30.x86_64

and cannot find the rpm files or any other files. 

help would be much appreciated!

TeknoMobil, 2019-04-30:
Hi
Open terminal: 
sudo dnf update --refresh

install updates, kernel will be updated to the latest version