atar-axis/xpadneo #84: Installing xpadneo doesn't take effect on retropie after rpi-update

aannenko, 2019-03-17:
**Describe the bug**
As of March 17 2019, on Retropie after rpi-update `sudo ./install.sh` shows the following error:

```bash
pi@retropie:~/downloads/xpadneo $ sudo ./install.sh
* replacing version string if necessary
* copying module into /usr/src
* adding module to DKMS
Creating symlink /var/lib/dkms/hid-xpadneo/0.5.4/source -> 
/usr/src/hid-xpadneo-0.5.4
DKMS: add completed.
* installing module (using DKMS)
Error! echo
Your kernel headers for kernel 4.19.27-v7+ cannot be found at
/lib/modules/4.19.27-v7+/build or /lib/modules/4.19.27-v7+/source.
```

After installation Xbox Wireless Controller can be connected and will use the usual xpad driver (the controller doesn't rumble when connected).

**To Reproduce**
Steps to reproduce the behavior:
1. [Install RetroPie](https://retropie.org.uk/docs/First-Installation/#install)
2. `sudo rpi-update`
3. `sudo apt update`
4. `sudo apt dist-upgrade --auto-remove --purge`
5. `sudo apt autoclean`
6. Install xpad-neo as described in [README.md](https://github.com/atar-axis/xpadneo)
7. Connect Xbox Wireless Controller from emulationstation -> retropie -> bluetooth -> Register and Connect to Bluetooth Device.

**Expected behavior**
Controller rubles when connected, xpadneo driver is used instead of the usual xpad driver.

**System information**

```bash
pi@retropie:/ $ uname -a
Linux retropie 4.19.27-v7+ #1206 SMP Wed Mar 6 14:40:18 GMT 2019 armv7l GNU/Linux

pi@retropie:~ $ sudo find "/sys/kernel/debug/hid/" -name "0005:045E:*" -exec sh -c 'echo "{}" && head -1 "{}/rdesc" | tee /dev/tty | cksum && echo' \;
/sys/kernel/debug/hid/0005:045E:02FD.0005
05 01 09 05 a1 01 85 01 09 01 a1 00 09 30 09 31 15 00 27 ff ff 00 00 95 02 75 10 81 02 c0 09 01 a1 00 09 32 09 35 15 00 27 ff ff 00 00 95 02 75 10 81 02 c0 05 02 09 c5 15 00 26 ff 03 95 01 75 0a 81 02 15 00 25 00 75 06 95 01 81 03 05 02 09 c4 15 00 26 ff 03 95 01 75 0a 81 02 15 00 25 00 75 06 95 01 81 03 05 01 09 39 15 01 25 08 35 00 46 3b 01 66 14 00 75 04 95 01 81 42 75 04 95 01 15 00 25 00 35 00 45 00 65 00 81 03 05 09 19 01 29 0f 15 00 25 01 75 01 95 0f 81 02 15 00 25 00 75 01 95 01 81 03 05 0c 0a 24 02 15 00 25 01 95 01 75 01 81 02 15 00 25 00 75 07 95 01 81 03 05 0c 09 01 85 02 a1 01 05 0c 0a 23 02 15 00 25 01 95 01 75 01 81 02 15 00 25 00 75 07 95 01 81 03 c0 05 0f 09 21 85 03 a1 02 09 97 15 00 25 01 75 04 95 01 91 02 15 00 25 00 75 04 95 01 91 03 09 70 15 00 25 64 75 08 95 04 91 02 09 50 66 01 10 55 0e 15 00 26 ff 00 75 08 95 01 91 02 09 a7 15 00 26 ff 00 75 08 95 01 91 02 65 00 55 00 09 7c 15 00 26 ff 00 75 08 95 01 91 02 c0 85 04 05 06 09 20 15 00 26 ff 00 75 08 95 01 81 02 c0 00
2579218694 1006
```

[dmesg.txt](https://github.com/atar-axis/xpadneo/files/2975636/dmesg.txt).

**Additional context**  
Unlike the usual xpad driver, xpadneo allows Kodi 18.1 Leia detect and work with my Xbox Wireless Controller. Xpadneo also enables rumble in Nintendo64 games on the mupen64plus-based emulation cores (which is awesome; a huge THANK YOU to the driver's author and maintainers!).

atar-axis, 2019-03-17:
Please read the Note about an Update in the Prerequisites section under Raspbian ;)

Install RetroPie
sudo rpi-update
sudo apt update
sudo apt autoclean

atar-axis, 2019-03-18:
Please read the **Raspbian** section here: https://github.com/atar-axis/xpadneo#prerequisites
Especially the note about Updates.

closed 2019-03-18 by atar-axis