atar-axis/xpadneo #85: Wrong mapping(fresh install)

Alfik0, 2019-03-24:
**Describe the bug**
The controller has wrong mapping after installation of the driver according to the instruction. Tested on Bastion.(For example X button was working as Y button)

**System information**
Fresh installation of lubuntu 18.10

**Additional context**  
The issue was resolved by following guide on archwiki(https://wiki.archlinux.org/index.php/Gamepad#Connect_Xbox_Wireless_Controller_with_Bluetooth), following instructions on askubuntu(https://askubuntu.com/questions/998144/how-can-i-use-my-xbox-one-s-controller-via-bluetooth) and by disconnecting/connecting the controller manually using bluetoothctl in unknown order. (maybe it would also help with #81 ?)

I see from #80 that installation script should do it itself, but i don't know why i had to do it also manually. I don't really expect fix as I think that problem is with bluetooth connection, but mention in instruction(or adding troubleshooting page on wiki) would be nice.

atar-axis, 2019-03-25:
hum,... so after the installation (and without trying to fixing it) the `/sys/module/bluetooth/parameters/disable_ertm` is empty?

Alfik0, 2019-03-25:
Sorry, that I won't answer your questions as I don't have time to do the installation of entire system again and don't remember everything what I have done to fix it, but there was more adventures after what i have described above. I will write everything just in case someone will have similar problem.

1) Next morning, after reboot, the controller has stopped working again. I have tried many things but none of them worked.
2) In the end Bluetooth got stuck "connected" with xbox controller, even when the controller was completely disabled and it was persistent though reboots.
3) I have updated the kernel which "unstuck" the bluetooth.
4) I have reinstalled the xpadneo and rebooted system. The installation was without any errors. I have saved the output and can attach it if needed.
5) I have very carefully followed connection steps and the xbox pad started working correctly. 

By the way I have noticed that it is possible to get the controller working without using bluetoothctl trust. It will connect the controller once, it will work correctly, but after restart of either pc or controller it will be in loop of connecting/disconnecting. 

Looking at everything what I have written, it is possible that the first time I did not restart the system after driver installation and the wrong mapping was due to either default driver or/and the ertm setting which were not applied. Then I have done something which has worked only until reboot but at the same has fucked up bluetooth for the next connections.

Maybe in like 2 weeks I will have time to do some testing and will try to reproduce it in exactly the same way.

atar-axis, 2019-03-25:
I have myself installed xpadneo on 20+ systems and never had such issues, moreover: normally it isn't necessary to reboot the system - since the driver also changes the current ertm settings.

It's hard to reproduce that.

I thank you for reporting this issue. I think I will close it for now since you solved it in the end - until you, me or sb else can reproduce it. Is that OK for you?

Alfik0, 2019-03-25:
Of course, sorry that I was not really helpful.

closed 2019-03-25 by Alfik0

atar-axis, 2019-03-25:
No Problem :) Thank you for reporting anyway!