atar-axis/xpadneo #87: Uninstall issue

julien-f, 2019-03-30:
```
# ./uninstall.sh 
* unloading current driver module
* looking for registered instances
found 3 registered instance(s) on your system
* 0.5.4
  * uninstalling and removing 0.5.4 from DKMS

-------- Uninstall Beginning --------
Module:  hid-xpadneo
Version: 0.5.4
Kernel:  4.20.16-200.fc29.x86_64 (x86_64)
-------------------------------------

Status: Before uninstall, this module version was ACTIVE on this kernel.
Removing any linked weak-modules

hid-xpadneo.ko.xz:
 - Uninstallation
   - Deleting from: /lib/modules/4.20.16-200.fc29.x86_64/extra/
rmdir: failed to remove 'extra': Directory not empty
 - Original module
   - No original module was found for this module on this kernel.
   - Use the dkms install command to reinstall any previous module version.

Running the post_remove script:
Removing Modules Hooks and Configurations
depmod....

dracut........

DKMS: uninstall completed.

-------- Uninstall Beginning --------
Module:  hid-xpadneo
Version: 0.5.4
Kernel:  5.0.3-200.fc29.x86_64 (x86_64)
-------------------------------------

Status: Before uninstall, this module version was ACTIVE on this kernel.
Removing any linked weak-modules

hid-xpadneo.ko.xz:
 - Uninstallation
   - Deleting from: /lib/modules/5.0.3-200.fc29.x86_64/extra/
rmdir: failed to remove 'extra': Directory not empty
 - Original module
   - No original module was found for this module on this kernel.
   - Use the dkms install command to reinstall any previous module version.

Running the post_remove script:
Removing Modules Hooks and Configurations
depmod....

dracut........

DKMS: uninstall completed.

-------- Uninstall Beginning --------
Module:  hid-xpadneo
Version: 0.5.4
Kernel:  5.0.4-200.fc29.x86_64 (x86_64)
-------------------------------------

Status: Before uninstall, this module version was ACTIVE on this kernel.
Removing any linked weak-modules

hid-xpadneo.ko.xz:
 - Uninstallation
   - Deleting from: /lib/modules/5.0.4-200.fc29.x86_64/extra/
rmdir: failed to remove 'extra': Directory not empty
 - Original module
   - No original module was found for this module on this kernel.
   - Use the dkms install command to reinstall any previous module version.

Running the post_remove script:
Removing Modules Hooks and Configurations
depmod....

dracut.......

DKMS: uninstall completed.
  * removing 0.5.4 folder from /usr/src
* 0.5.4
  * uninstalling and removing 0.5.4 from DKMS
Error! Could not locate dkms.conf file.
File: /var/lib/dkms/hid-xpadneo/0.5.4/source/dkms.conf does not exist.
  * removing 0.5.4 folder from /usr/src
rm: cannot remove '/usr/src/hid-xpadneo-0.5.4/': No such file or directory
* 0.5.4
  * uninstalling and removing 0.5.4 from DKMS
Error! Could not locate dkms.conf file.
File: /var/lib/dkms/hid-xpadneo/0.5.4/source/dkms.conf does not exist.
  * removing 0.5.4 folder from /usr/src
rm: cannot remove '/usr/src/hid-xpadneo-0.5.4/': No such file or directory
```

**System information**
Please enable debugging output <https://atar-axis.github.io/xpadneo/index.html#debugging>
and add at least the following outputs
-   `uname -a`: `Linux prometheus 5.0.4-200.fc29.x86_64 #1 SMP Mon Mar 25 02:27:33 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux` on Fedora 29
-   `dmesg`
-   `sudo find "/sys/kernel/debug/hid/" -name "0005:045E:*" -exec sh -c 'echo "{}" && head -1 "{}/rdesc" | tee /dev/tty | cksum && echo' \;`

**Additional context**  
Add any other context about the problem here.

atar-axis, 2019-03-30:
Thank you for reporting! Looks like there is a Parameter Problem - i will have a look

atar-axis, 2019-03-30:
should be - hopefully - fixed now.

closed 2019-03-30 by atar-axis

julien-f, 2019-03-31:
It appears to be working now, thank you :slightly_smiling_face:

atar-axis, 2019-04-01:
Nice to hear that, looks like the problem was the following:

You had installed version `0.5.4` for more than one kernel (which happens if you upgrade your kernel).
While running `uninstall.sh` the script is looking for all installed versions and removes them one by one via `dkms uninstall ... --all` . The `--all` means **for all kernels**, therefore it isn't necessary to run the uninstall command more than once for the same version.

Fixed that by piping the array of installed versions to `sort -u`

commit 306e733f (2019-03-30): removed duplicate entries when the same version is/was installed on different kernels (uninstall.sh), fixes #87