atar-axis/xpadneo #88: Request: Option to disable trigger rumble

Zowayix, 2019-03-30:
**Is your feature request related to a problem? Please describe.**
Some applications (read: Dolphin) have issues with the triggers having force feedback as well as the normal controller force feedback. The regular rumble (what's the proper word for that? Whole-controller-rumble?) works just fine, but Dolphin gets confused and rumbles the left trigger only when I would expect it to rumble the whole thing. That's just a thing that only happens with xpadneo, and it happens no matter what I set trigger_rumble_damping to, and didn't happen with other rumbly games (e.g. Tomb Raider was tested and was fine).

Since I don't know what I'm doing, I don't know about that issue there and whether to report it to them and which is the cause of the incompatibility, it's just an example as to why you might want to disable trigger force feedback. Let's say "for troubleshooting purposes". It's possible there are other applications which interact with gamepads in that way where they allow you to set the specific force feedback effect used, and have the same issue.

Anyway, I went into the hid-xpadneo.c file myself and started commenting out things and setting variables like `fraction_TL` to 0 to see what happens. This did end up fixing the issue, and I figure nothing I would play would use the trigger rumble anyway, and that works in everything else that rumbles too. Yay!

"Fiddle with random bits of code in a kernel driver" is probably not a great solution though, and probably not something I think normal people would do. I'm not even a C programmer, so I probably wouldn't be able to make my ad-hoc modifications into an option myself.

**Describe the solution you'd like**
A configuration option like `disable_ff` except it leaves normal rumble enabled, but disables the trigger rumble entirely (so all rumble is sent to the main body of the controller, not either of the triggers), and maybe it can be called `disable_trigger_ff`, or some nice name like that.

**Describe alternatives you've considered**
Attempting to get used to only 1 trigger rumbling (it feels wrong and lopsided), or disabling rumble altogether with disable_ff (which does stop the triggers rumbling, but it's not as enjoyable; I like feeling the action)

**Additional context**
Can't think of any additional context right now, although since it's 2:18am, I inevitably am not making sense right now and missed something.

atar-axis, 2019-03-30:
I will add it in te next Version, thank you for your Suggestion! I don't Know why dolphin gets confused but I will Take a look into it too

atar-axis, 2019-03-30:
Can you tell me, step by step, how to replicate this in Dolphin? I would like to investigate that.

atar-axis, 2019-03-31:
We have quite a few options to offer a setting to disable FF.

Since this driver is stil in `0.x.x` state I don't care too much about backwards compatibility. Therefore I would suggest the following way to let the user disable FF:

```
option disable_ff

0 = do not disable
1 = disable only triggers ff
2 = disable only main ff
4 = disable all ff
```

Zowayix, 2019-03-31:
That suggestion does seem fine to me. I had a bit of a shower thought, would it also make sense to have a "redirect trigger rumble to main" or "redirect main rumble to triggers' feature, or would that be too much to implement?

As for replicating the Dolphin thing, there's not really any specific steps, you just go to Controllers to configure a standard GameCube controller and then click "Motor" to change that. The dialog looks like this:

[image]

This happens no matter which "effect" is chosen or combination of effects, and doesn't happen when you press "Test" in that dialog (it also doesn't happen with fftest). But if you play a game with that uses a GameCube controller and rumbles it, that rumble will happen on the triggers (I tested this with Super Smash Bros. Brawl since simply walking around or doing anything generates rumble).

atar-axis, 2019-03-31:
there is already some kind of redirection from main rumble to triggers, thing is that linux does not support trigger rumble natively. we (@kakra and me) decided therefore to use the `direction` to decided if the triggers should rumble or not, so if the right trigger rumbles, then Dolphin sent an rumble message with and direction value which is indicating that the rumble direction should be "on the right".

therefore redirecting does not make too much sense in my eyes, because you will never have any rumbling at the triggers without the main rotors rumbling.

I will implement the above mentioned parameters and tell if if it is ready for testing ;)

atar-axis, 2019-03-31:
give it a try: https://github.com/atar-axis/xpadneo/tree/enable_ff

i decided to remove the old option `disable_ff` and added `enable_ff`,
using this option, you can easily turn on/off every of the four rumble motors.

```
enum {
	FF_ENABLE_NONE          = 0x00,
	FF_ENABLE_RIGHT         = 0x01,
	FF_ENABLE_LEFT          = 0x02,
	FF_ENABLE_RIGHT_TRIGGER = 0x04,
	FF_ENABLE_LEFT_TRIGGER  = 0x08,
	FF_ENABLE_ALL           = 0x0F
};
```

To disable the FF on the triggers, simply put `0x03` (0x01+0x02) in the parameter :)

kakra, 2019-03-31:
> @Zowayix: If we would introduce `disable_trigger_ff`, we would have to introduce `disable_main_ff` too. I don't like the idea to have two (disable_trigger_ff, disable_main_ff)) or even three (if we keep disable_ff for backwards compatibility), to do a single job.
> 
> I would suggest the following way to let the user disable FF:
> 
> ```
> option disable_ff
> 
> 0 = do not disable
> 1 = disable only trigger ff
> 2 = disable only main ff
> 4 = disable all ff
> ```
> Thing is, that this is **not** backwards-compatible. But I don't care too much about backwards compatibility since this driver is stil in `0.x.x` state (see [here](https://semver.org/#spec-item-4)).
> 
> Any thoughts or suggestions @kakra @Zowayix?

There's no game currently (at least none I know of) which could use the trigger rumble. So we actually have a backwards-compatible change here:

```
0: do not disable
1: disable main ff
2: disable trigger ff
3: disable main + trigger ff
```

The value 1 remains to have the same observed effect. And it should be parsed as a bitfield. I'm not sure why @atar-axis suggested 4 (thus bit 2^2) as a separate option to disable all ff (which is also mutually exclusive to bit 2^0 and 2^1). If anything, the docs could suggest to write 255 to disable all ff and it would be future proof for not-yet-released controllers adding more rumble motors.

atar-axis, 2019-03-31:
As always: good points!

atar-axis, 2019-03-31:
done as suggested

> 0: do not disable
1: disable main ff
2: disable trigger ff
3: disable main + trigger ff

kakra, 2019-03-31:
BTW: Do we have a game/app here which actually could support directional rumble? (Dolphin)

Could you provide a link?

atar-axis, 2019-03-31:
I havent tested it yet, but yes maybe. Looks like it does at least set a direction

Zowayix, 2019-04-06:
Apologies for late reply. Had a test of that enable_ff branch just now.

I did an update.sh from the master branch to make sure the issue still occurs, which it does.

Switched to the enable_ff branch, it didn't like me doing update.sh (some error about wanting me to do a git pull or git hard reset), so I did an uninstall.sh and then install.sh.

Leaving disable_ff at 0, everything is the same: Dolphin rumbles the left trigger, everything else rumbles the main body as normal.

I then set the disable_ff parameter to 2, with `sudo sh -c 'echo 2 > /sys/module/hid_xpadneo/parameters/param_disable_ff'`

Tested again. This time I get no rumble from anything, Dolphin or fftest or otherwise.

atar-axis, 2019-04-07:
Hey there! I am sorry about that, I will try to fix it now - brb

atar-axis, 2019-04-07:
Okay, should be ready now... I made a really unmindful failure.
Anyway, it does at least work here on my system using the directional rumble test tool in `misc/`
can you give it a try too? Btw, I renamed the paramters back to what you are used to (without the leading `param_` thing).

atar-axis, 2019-04-11:
looking for tester #94

atar-axis, 2019-04-19:
Have you found some time to test the new version @Zowayix ? :)

Zowayix, 2019-04-19:
Oh! I have not, actually. Long story short my computer's having issues lately so I've been restoring everything from a backup for the last few days and it's not quite done yet. I'll see how it goes once that's done in hopefully the next few days though.

atar-axis, 2019-04-19:
Oh! Good luck then :)

Zowayix, 2019-04-27:
Finally was able to get around to testing this again. Repeated the same test steps as before, it rumbles the main body of the controller now. Yaaaay!

atar-axis, 2019-04-27:
Thanks for confirming, I will then merge it ;)

atar-axis, 2019-04-28:
I will close this issue now since the PR is pending ;)
Cheers!

closed 2019-04-28 by atar-axis

awsms, 2024-07-30:
Was the option to disable Force Feedback removed?

kakra, 2024-07-30:
Yep, it's called "rumble_attenuation" now... Setting it to 100 will essentially disable force feedback. There was no point in keeping both.