atar-axis/xpadneo #89: Emulation station crashes when controller turned off

ausmurp, 2019-04-01:
**Describe the bug**
A clear and concise description of what the bug is.

ES crashes when connecting a second controller and then turning controller off.

**To Reproduce**
Steps to reproduce the behavior:

Connect 2 controllers. Then turn one off by holding Xbox home button.

**Expected behavior**
A clear and concise description of what you expected to happen.

ES should not crash. 

**Screenshots/Gifs**
If applicable, add screenshots or animated gifs to help explain your problem.

**System information**
Please enable debugging output <https://atar-axis.github.io/xpadneo/index.html#debugging>
and add at least the following outputs
-   `uname -a`
-   `dmesg`
-   `sudo find "/sys/kernel/debug/hid/" -name "0005:045E:*" -exec sh -c 'echo "{}" && head -1 "{}/rdesc" | tee /dev/tty | cksum && echo' \;`

**Additional context**  
Add any other context about the problem here.

Love this driver thanks! Ran into this today.

atar-axis, 2019-04-01:
Hey there, I cannot reproduce that because I so not own a second gamepad. Would you please add the *System information*?

ausmurp, 2019-04-01:
Yep I'll add that later tomorrow, didn't realize I missed it. I know I'm on Stretch version with latest retropie version, latest kernel doe 3b. I just upped everything before I got the driver you created, which is awesome btw.

atar-axis, 2019-04-01:
Fine :grin: Please let the station crash before you gather the information (if possible). Thanks!

cmitu, 2019-04-03:
I don't think this is driver related, but SDL issue - see https://github.com/RetroPie/EmulationStation/issues/552.
@ausmurp - are you using RetroPie ?

atar-axis, 2019-04-03:
Thanks @cmitu

closed 2019-04-03 by atar-axis