atar-axis/xpadneo #90: Force Feedback just working on one of two Xbox controllers

germangoergs, 2019-04-01:
**Describe the bug**
A clear and concise description of what the bug is.

I do own two Xbox One Controllers, one black and one red (the red can be found here: https://www.amazon.de/dp/B01MY13UKE/ref=twister_B00I03MYC6?_encoding=UTF8&psc=1), the other one was given to me to Christmas.
The strange behavior I could notice is that when I connect the black one, it rumbles as it should, but the red one doesn't when connected, even though I checked everything and also re-connected it.
But there's no ff.

**To Reproduce**
Steps to reproduce the behavior:
1. Connect red controller to Linux
2. No rumble :(
**Expected behavior**
A clear and concise description of what you expected to happen.
Rumbles when connected
**Screenshots/Gifs**
If applicable, add screenshots or animated gifs to help explain your problem.

**System information**
Please enable debugging output <https://atar-axis.github.io/xpadneo/index.html#debugging>
and add at least the following outputs
-   `uname -a`
`Linux Tobias-Linux 5.0.3-1-default #1 SMP Fri Mar 22 17:30:35 UTC 2019 (2a31831) x86_64 x86_64 x86_64 GNU/Linux`
-   `dmesg`
```
[    0.000000] microcode: microcode updated early to revision 0xc6, date = 2018-04-17
[    0.000000] Linux version 5.0.3-1-default (geeko@buildhost) (gcc version 8.3.1 20190226 [gcc-8-branch revision 269204] (SUSE Linux)) #1 SMP Fri Mar 22 17:30:35 UTC 2019 (2a31831)
[    0.000000] Command line: BOOT_IMAGE=/vmlinuz-5.0.3-1-default root=UUID=d8a7a9bd-d6bb-4a01-a551-e8f65721140f splash=silent quiet bluetooth.disable_ertm=1
[    0.000000] x86/fpu: Supporting XSAVE feature 0x001: 'x87 floating point registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x002: 'SSE registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x004: 'AVX registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x008: 'MPX bounds registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x010: 'MPX CSR'
[    0.000000] x86/fpu: xstate_offset[2]:  576, xstate_sizes[2]:  256
[    0.000000] x86/fpu: xstate_offset[3]:  832, xstate_sizes[3]:   64
[    0.000000] x86/fpu: xstate_offset[4]:  896, xstate_sizes[4]:   64
[    0.000000] x86/fpu: Enabled xstate features 0x1f, context size is 960 bytes, using 'compacted' format.
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000009c7ff] usable
[    0.000000] BIOS-e820: [mem 0x000000000009c800-0x000000000009ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000000e0000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x000000007da2afff] usable
[    0.000000] BIOS-e820: [mem 0x000000007da2b000-0x000000007da2bfff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x000000007da2c000-0x000000007da75fff] reserved
[    0.000000] BIOS-e820: [mem 0x000000007da76000-0x000000007dac8fff] usable
[    0.000000] BIOS-e820: [mem 0x000000007dac9000-0x000000007e7f9fff] reserved
[    0.000000] BIOS-e820: [mem 0x000000007e7fa000-0x0000000086e54fff] usable
[    0.000000] BIOS-e820: [mem 0x0000000086e55000-0x00000000877adfff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000877ae000-0x0000000087f99fff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x0000000087f9a000-0x0000000087ffefff] ACPI data
[    0.000000] BIOS-e820: [mem 0x0000000087fff000-0x0000000087ffffff] usable
[    0.000000] BIOS-e820: [mem 0x0000000088000000-0x00000000880fffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000e0000000-0x00000000efffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fe000000-0x00000000fe010fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fec00000-0x00000000fec00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fee00000-0x00000000fee00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ff000000-0x00000000ffffffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x00000006767fffff] usable
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] SMBIOS 2.8 present.
[    0.000000] DMI: To Be Filled By O.E.M. To Be Filled By O.E.M./Z170 Pro4S, BIOS P7.50 01/23/2018
[    0.000000] tsc: Detected 4000.000 MHz processor
[    0.000828] tsc: Detected 4008.000 MHz TSC
[    0.000828] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.000829] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.000833] last_pfn = 0x676800 max_arch_pfn = 0x400000000
[    0.000836] MTRR default type: write-back
[    0.000836] MTRR fixed ranges enabled:
[    0.000837]   00000-9FFFF write-back
[    0.000837]   A0000-BFFFF uncachable
[    0.000838]   C0000-FFFFF write-protect
[... 831 lines cut ...]
[   18.439662] IPv6: ADDRCONF(NETDEV_CHANGE): enp0s31f6: link becomes ready
[   18.488432] NET: Registered protocol family 17
[   22.725224] snd_hda_codec_hdmi hdaudioC1D0: HDMI: invalid ELD data byte 39
[   29.736214] snd_hda_codec_hdmi hdaudioC1D0: HDMI: invalid ELD data byte 2
[   34.038783] Bluetooth: RFCOMM TTY layer initialized
[   34.038788] Bluetooth: RFCOMM socket layer initialized
[   34.038793] Bluetooth: RFCOMM ver 1.11
[   85.084568] Bluetooth: HIDP (Human Interface Emulation) ver 1.2
[   85.084578] Bluetooth: HIDP socket layer initialized
[   85.086492] hid-generic 0005:045E:02FD.0005: unknown main item tag 0x0
[   85.086742] input: Xbox Wireless Controller as /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3.2/1-3.2:1.0/bluetooth/hci0/hci0:70/0005:045E:02FD.0005/input/input25
[   85.087026] input: Xbox Wireless Controller Consumer Control as /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3.2/1-3.2:1.0/bluetooth/hci0/hci0:70/0005:045E:02FD.0005/input/input26
[   85.087207] hid-generic 0005:045E:02FD.0005: input,hidraw3: BLUETOOTH HID v9.03 Gamepad [Xbox Wireless Controller] on 00:1a:7d:da:71:11
[   85.216273] microsoft 0005:045E:02FD.0005: unknown main item tag 0x0
[   85.216520] input: Xbox Wireless Controller as /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3.2/1-3.2:1.0/bluetooth/hci0/hci0:70/0005:045E:02FD.0005/input/input27
[   85.217118] microsoft 0005:045E:02FD.0005: input,hidraw3: BLUETOOTH HID v9.03 Gamepad [Xbox Wireless Controller] on 00:1a:7d:da:71:11
[   85.221164] xpadneo: hello there!
[   85.328939] xpadneo 0005:045E:02FD.0005: unknown main item tag 0x0
[   85.329061] input: Xbox Wireless Controller as /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3.2/1-3.2:1.0/bluetooth/hci0/hci0:70/0005:045E:02FD.0005/input/input28
[   85.329296] xpadneo 0005:045E:02FD.0005: input,hidraw3: BLUETOOTH HID v9.03 Gamepad [Xbox Wireless Controller] on 00:1a:7d:da:71:11
```
-   `sudo find "/sys/kernel/debug/hid/" -name "0005:045E:*" -exec sh -c 'echo "{}" && head -1 "{}/rdesc" | tee /dev/tty | cksum && echo' \;`
[no output]
**Additional context**  
Add any other context about the problem here.

atar-axis, 2019-04-01:
Hey there!
1) When did you made the dmesg output? Have you had already connected both gamepads?
2) What happens if you change the order in which you connect the gamepads? Is it always the same (physical) gamepad that doesn't rumble or is it always the second one connected?

germangoergs, 2019-04-01:
> Hey there!
> 
>     1. When did you made the dmesg output? Have you had already connected both gamepads?
> 
>     2. What happens if you change the order in which you connect the gamepads? Is it always the same (physical) gamepad that doesn't rumble or is it always the second one connected?

Yep, I think that both should have been connected, but I can gather new dmesg if you want.
I took the dmesg right after I had exactly this problem :D 

It's always the black one, no matter which I connect first. ;)

atar-axis, 2019-04-01:
> It's always the black one, no matter which I connect first. ;)

Ah, that's interesting!
Could you please do the following for me:

1) Connect both gamepads as always
2) Go to `/sys/bus/hid/devices/`
3) Show me the output of `ls -la` in there
4) Open  each folder that starts with `0005:045E` and show me again the output of `ls -la` there

atar-axis, 2019-04-01:
btw, can you maybe confirm this issue: #89 ?

germangoergs, 2019-04-02:
> Ah, that's interesting!
> Could you please do the following for me:
>
>     1. Connect both gamepads as always
>
>     2. Go to `/sys/bus/hid/devices/`
>
>     3. Show me the output of `ls -la` in there [...]

Oh wait, I was wrong, sorry. 
This time the red one vibrated, sorry.

So it seems like the second connected vibrates.

```
/sys/bus/hid/devices> ls -la
insgesamt 0
drwxr-xr-x 2 root root 0  2. Apr 18:27 .
drwxr-xr-x 4 root root 0  2. Apr 18:27 ..
lrwxrwxrwx 1 root root 0  2. Apr 18:27 0003:046D:C335.0003 -> ../../../devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3.3/1-3.3:1.0/0003:046D:C335.0003
lrwxrwxrwx 1 root root 0  2. Apr 18:27 0003:046D:C531.0001 -> ../../../devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3.1/1-3.1:1.0/0003:046D:C531.0001
lrwxrwxrwx 1 root root 0  2. Apr 18:27 0003:046D:C531.0002 -> ../../../devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3.1/1-3.1:1.1/0003:046D:C531.0002
lrwxrwxrwx 1 root root 0  2. Apr 18:31 0005:045E:02FD.0005 -> ../../../devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3.2/1-3.2:1.0/bluetooth/hci0/hci0:62/0005:045E:02FD.0005
lrwxrwxrwx 1 root root 0  2. Apr 18:31 0005:045E:02FD.0006 -> ../../../devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3.2/1-3.2:1.0/bluetooth/hci0/hci0:69/0005:045E:02FD.0006
```

```
/sys/bus/hid/devices/0005:045E:02FD.0005> ls -la
insgesamt 0
drwxr-xr-x 6 root root    0  2. Apr 18:31 .
drwxr-xr-x 4 root root    0  2. Apr 18:31 ..
-r--r--r-- 1 root root 4096  2. Apr 18:34 country
lrwxrwxrwx 1 root root    0  2. Apr 18:31 driver -> ../../../../../../../../../../../bus/hid/drivers/microsoft
drwxr-xr-x 3 root root    0  2. Apr 18:31 hidraw
drwxr-xr-x 3 root root    0  2. Apr 18:31 input
-r--r--r-- 1 root root 4096  2. Apr 18:34 modalias
drwxr-xr-x 2 root root    0  2. Apr 18:31 power
drwxr-xr-x 3 root root    0  2. Apr 18:31 power_supply
-r--r--r-- 1 root root 4096  2. Apr 18:34 report_descriptor
lrwxrwxrwx 1 root root    0  2. Apr 18:31 subsystem -> ../../../../../../../../../../../bus/hid
-rw-r--r-- 1 root root 4096  2. Apr 18:31 uevent
```

```
/sys/bus/hid/devices/0005:045E:02FD.0006> ls -la
insgesamt 0
drwxr-xr-x 6 root root    0  2. Apr 18:31 .
drwxr-xr-x 4 root root    0  2. Apr 18:31 ..
-r--r--r-- 1 root root 4096  2. Apr 18:35 country
lrwxrwxrwx 1 root root    0  2. Apr 18:31 driver -> ../../../../../../../../../../../bus/hid/drivers/xpadneo
drwxr-xr-x 3 root root    0  2. Apr 18:31 hidraw
drwxr-xr-x 3 root root    0  2. Apr 18:31 input
-r--r--r-- 1 root root 4096  2. Apr 18:35 modalias
drwxr-xr-x 2 root root    0  2. Apr 18:31 power
drwxr-xr-x 3 root root    0  2. Apr 18:31 power_supply
-r--r--r-- 1 root root 4096  2. Apr 18:35 report_descriptor
lrwxrwxrwx 1 root root    0  2. Apr 18:31 subsystem -> ../../../../../../../../../../../bus/hid
-rw-r--r-- 1 root root 4096  2. Apr 18:31 uevent
```

These are the outputs. :)

And thanks for providing help. :)

germangoergs, 2019-04-02:
> btw, can you maybe confirm this issue: #89 ?

Unfortunately, I don't know what this "Emulation Station" shall be, sorry. O.o

atar-axis, 2019-04-02:
okay perfect, thing is that only one of your gamepads is using the correct driver.
The question now is: why?

The one which vibrates is the one which is connected to xpadneo.
Can you please make sure that it is always the second one? I need something like a pattern

germangoergs, 2019-04-02:
> okay perfect, thing is that only one of your gamepads is using the correct driver.
> The question now is: why?
> 
> The one which vibrates is the one which is connected to xpadneo.
> Can you please make sure that it is always the second one? I need something like a pattern

Ok ok, I think I got how it works (mostly).

So it's just the black controller that's vibrating when connected, the red one didn't in like 10 tries, so I think there's just a little probability that the red one will vibrate again, maybe I mixed both up and the black vibrated back then in reality.
So it seems that it's the black one.

Interestingly though is that I also saw that my black one was connected, but there wasn't a driver "set" to it. It also didn't vibrate then.
A "simple" re-connection fixed that though.

atar-axis, 2019-04-02:
I think we can have a look at this issue in the other one, let's try to fix them together. I will close this for now in favor of #91

closed 2019-04-02 by atar-axis