atar-axis/xpadneo #91: Re-connection of gamepad sometimes/mostly not working

germangoergs, 2019-04-01:
**Describe the bug**
A clear and concise description of what the bug is.
When I disconnected my controller or shut my PC down, I'm having some troubles to re-connect it via bluetooth.
It shows that it's trying to connect, but goes back to un-connected state and so on.
**To Reproduce**
Steps to reproduce the behavior:
1. Connect Controller
2. Dis-connect it
3. try to re-connect it again
**Expected behavior**
A clear and concise description of what you expected to happen.
The controller should be connected immediately again
**Screenshots/Gifs**
If applicable, add screenshots or animated gifs to help explain your problem.

**System information**
Please enable debugging output <https://atar-axis.github.io/xpadneo/index.html#debugging>
and add at least the following outputs
-   `uname -a`
`Linux Tobias-Linux 5.0.3-1-default #1 SMP Fri Mar 22 17:30:35 UTC 2019 (2a31831) x86_64 x86_64 x86_64 GNU/Linux`
-   `dmesg`
```
[    0.000000] microcode: microcode updated early to revision 0xc6, date = 2018-04-17
[    0.000000] Linux version 5.0.3-1-default (geeko@buildhost) (gcc version 8.3.1 20190226 [gcc-8-branch revision 269204] (SUSE Linux)) #1 SMP Fri Mar 22 17:30:35 UTC 2019 (2a31831)
[    0.000000] Command line: BOOT_IMAGE=/vmlinuz-5.0.3-1-default root=UUID=d8a7a9bd-d6bb-4a01-a551-e8f65721140f splash=silent quiet bluetooth.disable_ertm=1
[    0.000000] x86/fpu: Supporting XSAVE feature 0x001: 'x87 floating point registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x002: 'SSE registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x004: 'AVX registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x008: 'MPX bounds registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x010: 'MPX CSR'
[    0.000000] x86/fpu: xstate_offset[2]:  576, xstate_sizes[2]:  256
[    0.000000] x86/fpu: xstate_offset[3]:  832, xstate_sizes[3]:   64
[    0.000000] x86/fpu: xstate_offset[4]:  896, xstate_sizes[4]:   64
[    0.000000] x86/fpu: Enabled xstate features 0x1f, context size is 960 bytes, using 'compacted' format.
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000009c7ff] usable
[    0.000000] BIOS-e820: [mem 0x000000000009c800-0x000000000009ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000000e0000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x000000007da2afff] usable
[    0.000000] BIOS-e820: [mem 0x000000007da2b000-0x000000007da2bfff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x000000007da2c000-0x000000007da75fff] reserved
[    0.000000] BIOS-e820: [mem 0x000000007da76000-0x000000007dac8fff] usable
[    0.000000] BIOS-e820: [mem 0x000000007dac9000-0x000000007e7f9fff] reserved
[    0.000000] BIOS-e820: [mem 0x000000007e7fa000-0x0000000086e54fff] usable
[    0.000000] BIOS-e820: [mem 0x0000000086e55000-0x00000000877adfff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000877ae000-0x0000000087f99fff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x0000000087f9a000-0x0000000087ffefff] ACPI data
[    0.000000] BIOS-e820: [mem 0x0000000087fff000-0x0000000087ffffff] usable
[    0.000000] BIOS-e820: [mem 0x0000000088000000-0x00000000880fffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000e0000000-0x00000000efffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fe000000-0x00000000fe010fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fec00000-0x00000000fec00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fee00000-0x00000000fee00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ff000000-0x00000000ffffffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x00000006767fffff] usable
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] SMBIOS 2.8 present.
[    0.000000] DMI: To Be Filled By O.E.M. To Be Filled By O.E.M./Z170 Pro4S, BIOS P7.50 01/23/2018
[    0.000000] tsc: Detected 4000.000 MHz processor
[    0.000828] tsc: Detected 4008.000 MHz TSC
[    0.000828] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.000829] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.000833] last_pfn = 0x676800 max_arch_pfn = 0x400000000
[    0.000836] MTRR default type: write-back
[    0.000836] MTRR fixed ranges enabled:
[    0.000837]   00000-9FFFF write-back
[    0.000837]   A0000-BFFFF uncachable
[    0.000838]   C0000-FFFFF write-protect
[    0.000838] MTRR variable ranges enabled:
[    0.000839]   0 base 00C0000000 mask 7FC0000000 uncachable
[... 830 lines cut ...]
[   18.488432] NET: Registered protocol family 17
[   22.725224] snd_hda_codec_hdmi hdaudioC1D0: HDMI: invalid ELD data byte 39
[   29.736214] snd_hda_codec_hdmi hdaudioC1D0: HDMI: invalid ELD data byte 2
[   34.038783] Bluetooth: RFCOMM TTY layer initialized
[   34.038788] Bluetooth: RFCOMM socket layer initialized
[   34.038793] Bluetooth: RFCOMM ver 1.11
[   85.084568] Bluetooth: HIDP (Human Interface Emulation) ver 1.2
[   85.084578] Bluetooth: HIDP socket layer initialized
[   85.086492] hid-generic 0005:045E:02FD.0005: unknown main item tag 0x0
[   85.086742] input: Xbox Wireless Controller as /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3.2/1-3.2:1.0/bluetooth/hci0/hci0:70/0005:045E:02FD.0005/input/input25
[   85.087026] input: Xbox Wireless Controller Consumer Control as /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3.2/1-3.2:1.0/bluetooth/hci0/hci0:70/0005:045E:02FD.0005/input/input26
[   85.087207] hid-generic 0005:045E:02FD.0005: input,hidraw3: BLUETOOTH HID v9.03 Gamepad [Xbox Wireless Controller] on 00:1a:7d:da:71:11
[   85.216273] microsoft 0005:045E:02FD.0005: unknown main item tag 0x0
[   85.216520] input: Xbox Wireless Controller as /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3.2/1-3.2:1.0/bluetooth/hci0/hci0:70/0005:045E:02FD.0005/input/input27
[   85.217118] microsoft 0005:045E:02FD.0005: input,hidraw3: BLUETOOTH HID v9.03 Gamepad [Xbox Wireless Controller] on 00:1a:7d:da:71:11
[   85.221164] xpadneo: hello there!
[   85.328939] xpadneo 0005:045E:02FD.0005: unknown main item tag 0x0
[   85.329061] input: Xbox Wireless Controller as /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3.2/1-3.2:1.0/bluetooth/hci0/hci0:70/0005:045E:02FD.0005/input/input28
[   85.329296] xpadneo 0005:045E:02FD.0005: input,hidraw3: BLUETOOTH HID v9.03 Gamepad [Xbox Wireless Controller] on 00:1a:7d:da:71:11
```
-   `sudo find "/sys/kernel/debug/hid/" -name "0005:045E:*" -exec sh -c 'echo "{}" && head -1 "{}/rdesc" | tee /dev/tty | cksum && echo' \;`
[no output]
**Additional context**  
Add any other context about the problem here.
You still did a great job, I hope you can make the "real" wireless adapter work. :)

atar-axis, 2019-04-01:
A connection loop is mostly caused by ERTM,
please provide me the follwoing outputs:

* `cat /sys/module/bluetooth/parameters/disable_ertm`
* `cat /etc/udev/rules.d/99-xpadneo.rules`

germangoergs, 2019-04-02:
> A connection loop is mostly caused by ERTM,
> please provide me the following outputs:
> 
>     * `cat /sys/module/bluetooth/parameters/disable_ertm`
> 
>     * `cat /etc/udev/rules.d/99-xpadneo.rules`
> 
> 
> Thanks!

```
cat /sys/module/bluetooth/parameters/disable_ertm
Y
```

```
cat /etc/udev/rules.d/99-xpadneo.rules
ACTION=="add", KERNEL=="0005:045E:02FD.*|0005:045E:02E0.*", SUBSYSTEM=="hid",\
RUN+="/bin/bash -c 'echo $kernel > /sys/bus/hid/drivers/hid-generic/unbind'", \
RUN+="/bin/bash -c 'echo $kernel > /sys/bus/hid/drivers/xpadneo/bind'"

ACTION=="add", KERNEL=="0005:045E:02FD.*|0005:045E:02E0.*", SUBSYSTEM=="hid",\
RUN+="/bin/bash -c 'echo $kernel > /sys/bus/hid/drivers/microsoft/unbind'", \
RUN+="/bin/bash -c 'echo $kernel > /sys/bus/hid/drivers/xpadneo/bind'"
```

I have to thank you. :)

atar-axis, 2019-04-02:
Okay so ERTM is disabled, that's right. What's in the folder /etc/modprobe.d/? If there is anything with xpadneo, please Show me it's content too :)

germangoergs, 2019-04-02:
> Okay so ERTM is disabled, that's right. What's in the folder /etc/modprobe.d/? If there is anything called xpadneo..., then please Show me it's content too :)

Yes, there's a file with that name.
Well, the content of the file is: `options bluetooth disable_ertm=y` :D

atar-axis, 2019-04-02:
okay, so it is not a problem while installing the driver, all files are where they should be.
Is this issue happening on both of your gamepads (black and red)?

If so, then please run
* ` sudo dmesg -C`
* `bluetoothctl`
* `[bluetooth]# devices` -> note the MAC of your gamepad(s)
* `[bluetooth]# remove <MAC>` 
* shut down your gamepad
* reboot your system
* do the connection procedure described in the README (bluetoothctl, scan, pair, trust....)
* show me output of bluetoothctl
* shut down your gamepad
* turn it on again
* show me the output of bluetoothctl once again

germangoergs, 2019-04-02:
```
bluetoothctl
Agent registered
[bluetooth]# scan on
Discovery started
[CHG] Controller 00:1A:7D:DA:71:11 Discovering: yes
[NEW] Device 28:24:FF:70:5A:DD Panasonic Viera
[NEW] Device 59:8D:87:09:64:35 Micar L8
[CHG] Device 59:8D:87:09:64:35 RSSI: -71
[CHG] Device 28:24:FF:70:5A:DD RSSI: -93
[CHG] Device 28:24:FF:70:5A:DD RSSI: -82
[CHG] Device 59:8D:87:09:64:35 RSSI: -80
[CHG] Device 59:8D:87:09:64:35 RSSI: -72
[NEW] Device EC:83:50:88:75:E7 Xbox Wireless Controller
[CHG] Device EC:83:50:88:75:E7 Modalias: usb:v045Ep02E0d0903
[CHG] Device EC:83:50:88:75:E7 UUIDs: 00001124-0000-1000-8000-00805f9b34fb
[CHG] Device EC:83:50:88:75:E7 UUIDs: 00001200-0000-1000-8000-00805f9b34fb
[CHG] Device EC:83:50:88:75:E7 ServicesResolved: yes
[CHG] Device EC:83:50:88:75:E7 Paired: yes
[CHG] Device EC:83:50:88:75:E7 ServicesResolved: no
[CHG] Device EC:83:50:88:75:E7 Connected: no
[CHG] Device 59:8D:87:09:64:35 RSSI: -80
[bluetooth]# pair EC:83:50:88:75:E7
Attempting to pair with EC:83:50:88:75:E7
Failed to pair: org.bluez.Error.AlreadyExists
[CHG] Device 59:8D:87:09:64:35 RSSI: -72
[bluetooth]# pair EC:83:50:88:75:E7
Attempting to pair with EC:83:50:88:75:E7
Failed to pair: org.bluez.Error.AlreadyExists
[CHG] Device 59:8D:87:09:64:35 RSSI: -81
[CHG] Device 59:8D:87:09:64:35 RSSI: -71
[CHG] Device 28:24:FF:70:5A:DD RSSI: -74
[CHG] Device EC:83:50:88:75:E7 RSSI: -46
[CHG] Device EC:83:50:88:75:E7 RSSI: -93
[CHG] Device 59:8D:87:09:64:35 RSSI: -80
[CHG] Device 59:8D:87:09:64:35 RSSI: -71
[CHG] Device EC:83:50:88:75:E7 RSSI: -45
[CHG] Device 28:24:FF:70:5A:DD RSSI: -93
[CHG] Device 28:24:FF:70:5A:DD RSSI: -78
[CHG] Device EC:83:50:88:75:E7 RSSI: -85
[CHG] Device EC:83:50:88:75:E7 RSSI: -93
[CHG] Device 28:24:FF:70:5A:DD RSSI: -59
[CHG] Device 28:24:FF:70:5A:DD RSSI: -80
[CHG] Device EC:83:50:88:75:E7 RSSI: -46
[CHG] Device EC:83:50:88:75:E7 RSSI: -93
[CHG] Device 59:8D:87:09:64:35 RSSI: -79
[bluetooth]# pair EC:83:50:88:75:E7
Attempting to pair with EC:83:50:88:75:E7
Failed to pair: org.bluez.Error.AlreadyExists
[CHG] Device 59:8D:87:09:64:35 RSSI: -53
[CHG] Device EC:83:50:88:75:E7 RSSI: -46
[CHG] Device 59:8D:87:09:64:35 RSSI: -74
[CHG] Device EC:83:50:88:75:E7 RSSI: -96
[CHG] Device EC:83:50:88:75:E7 RSSI: -44
[CHG] Device EC:83:50:88:75:E7 RSSI: -93
[CHG] Device 28:24:FF:70:5A:DD RSSI: -93
[CHG] Device 28:24:FF:70:5A:DD RSSI: -84
[CHG] Device EC:83:50:88:75:E7 RSSI: -43
[bluetooth]# trust EC:83:50:88:75:E7
[CHG] Device EC:83:50:88:75:E7 Trusted: yes
Changing EC:83:50:88:75:E7 trust succeeded
[CHG] Device 28:24:FF:70:5A:DD RSSI: -93
[CHG] Device EC:83:50:88:75:E7 RSSI: -93
[CHG] Device 28:24:FF:70:5A:DD RSSI: -83
[CHG] Device 28:24:FF:70:5A:DD RSSI: -54
[CHG] Device 28:24:FF:70:5A:DD RSSI: -80
[CHG] Device EC:83:50:88:75:E7 RSSI: -64
[CHG] Device EC:83:50:88:75:E7 Connected: yes
[CHG] Device EC:83:50:88:75:E7 Connected: no
[CHG] Device EC:83:50:88:75:E7 Connected: yes
[... 257 lines cut ...]
[CHG] Device 59:8D:87:09:64:35 RSSI: -82
[CHG] Device 28:24:FF:70:5A:DD RSSI: -93
[CHG] Device 28:24:FF:70:5A:DD RSSI: -83
[CHG] Device 59:8D:87:09:64:35 RSSI: -66
[CHG] Device 28:24:FF:70:5A:DD RSSI: -69
[CHG] Device 59:8D:87:09:64:35 RSSI: -50
[CHG] Device 59:8D:87:09:64:35 RSSI: -77
[CHG] Device 28:24:FF:70:5A:DD RSSI: -80
[CHG] Device 28:24:FF:70:5A:DD RSSI: -93
[CHG] Device 28:24:FF:70:5A:DD RSSI: -81
[CHG] Device EC:83:50:88:75:E7 ServicesResolved: no
[CHG] Device EC:83:50:88:75:E7 Connected: no
[CHG] Device EC:83:50:88:75:E7 Connected: yes
[CHG] Device EC:83:50:88:75:E7 Connected: no
[CHG] Device EC:83:50:88:75:E7 Connected: yes
[CHG] Device EC:83:50:EC:FB:5E ServicesResolved: no
[CHG] Device EC:83:50:EC:FB:5E Paired: no
[CHG] Device EC:83:50:EC:FB:5E Connected: no
[CHG] Device EC:83:50:EC:FB:5E Connected: yes
[CHG] Device EC:83:50:EC:FB:5E Paired: yes
[CHG] Device 59:8D:87:09:64:35 RSSI: -60
[CHG] Device 59:8D:87:09:64:35 RSSI: -72
```

Now the black controller vibrated, the red one didn't.

germangoergs, 2019-04-02:
Oh and btw., I thought for about 10 minutes that I destroyed it completely, cause I updated my kernel and that's why the driver wasn't loaded, even though it said that (program identified that it was installed on old kernel).

So it would be cool, if the driver was directly activated in a new kernel. ;)

atar-axis, 2019-04-02:
This driver is using DKMS which does, if you system is healthy, automatically rebuild the driver on every kernel update. If this doesn't work, then it is another issue - most probably not caused by a broken driver but a broken system.

Your output is showing that you do a lot of things which are not correct, therefore let's try it again, step by step:

Firstly, please start `bluetoothctl` and type in `devices`, show me the output then.

germangoergs, 2019-04-03:
> This driver is using DKMS which does, if you system is healthy, automatically rebuild the driver on every kernel update. If this doesn't work, then it is another issue - most probably not caused by [...]

Oh Ok, good to know that my five days old system is already broken, even though every other kernel module updated just fine, but OK, you're the dev. :D 

Here you go:

```
bluetoothctl
Agent registered
[bluetooth]# devices
Device EC:83:50:EC:FB:5E Xbox Wireless Controller
Device EC:83:50:88:75:E7 Xbox Wireless Controller
[bluetooth]# 
```

atar-axis, 2019-04-03:
I am not saying that the root cause **is** a broken system, but ist is not very unlikely. Btw, just because your system is "new" does not mean that there aren't bugs and compatibility problems in it...

Anyways, if the update of the module fails ever again, then please copy the output and open a new issue.

---

> Here you go

please run the following in `bluetoothctl` and give me the output
```
remove EC:83:50:EC:FB:5E
remove EC:83:50:88:75:E7
```
now please remove `xpadneo` by running the `./uninstall.sh` script.
Again, please provide me the output.

germangoergs, 2019-04-03:
> I am not saying that the root cause **is** a broken system, but ist is not very unlikely. Btw, just because your system is "new" it does not mean that there aren't bugs and compatibility problems in it...
> 
> Anyways, if the update of the module fails ever again, then please copy the output and open a new issue.
> 
> > Here you go
> 
> please run the following in `bluetoothctl` and give me the output
> 
> ```
> remove EC:83:50:EC:FB:5E
> remove EC:83:50:88:75:E7
> ```
> 
> now please remove `xpadneo` by running the `./uninstall.sh` script.
> Again, please provide me the output.

```
devices
Device EC:83:50:EC:FB:5E Xbox Wireless Controller
Device EC:83:50:88:75:E7 Xbox Wireless Controller
[bluetooth]# remove EC:83:50:EC:FB:5E
[DEL] Device EC:83:50:EC:FB:5E Xbox Wireless Controller
Device has been removed
[bluetooth]# remove EC:83:50:88:75:E7
[DEL] Device EC:83:50:88:75:E7 Xbox Wireless Controller
Device has been removed
```

```
~/git/xpadneo> sudo ./uninstall.sh 
* unloading current driver module
* looking for registered instances
found 1 registered instance(s) on your system
* 0.5.10
  * uninstalling and removing 0.5.10 from DKMS

-------- Uninstall Beginning --------
Module:  hid-xpadneo
Version: 0.5.10
Kernel:  5.0.5-1-default (x86_64)
-------------------------------------

Status: Before uninstall, this module version was ACTIVE on this kernel.

hid-xpadneo.ko:
 - Uninstallation
   - Deleting from: /lib/modules/5.0.5-1-default/updates/
rmdir: konnte 'updates' nicht entfernen: Das Verzeichnis ist nicht leer
 - Original module
   - No original module was found for this module on this kernel.
   - Use the dkms install command to reinstall any previous module version.

Running the post_remove script:
Removing Modules Hooks and Configurations
depmod...

dracut........

DKMS: uninstall completed.

------------------------------
Deleting module version: 0.5.10
completely from the DKMS tree.
------------------------------
Done.
  * removing 0.5.10 folder from /usr/src
```

I'm ready to go on now.

atar-axis, 2019-04-03:
If you haven't already, please reboot your system now.

For the next step:
Do you have an Xbox or Windows OS nearby?

germangoergs, 2019-04-03:
> If you haven't already, please reboot your system now.
> 
> For the next step:
> Do you have an Xbox or Windows OS nearby?

I've already rebooted and yes.

atar-axis, 2019-04-03:
Please update the firmware of both gamepads!
(Even if it is already the latest fw which is installed, this somehow resets the internals of your gamepads)

germangoergs, 2019-04-03:
> Please update the firmware of both gamepads!
> (Even if it is already the latest fw which is installed, this somehow resets the internals of your gamepads)
> 
> Do not(!) connect them via bluetooth to any other system, use a cable please (in Windows you need an USB connection anyway).

I "updated" it, it already was latest firmware.

atar-axis, 2019-04-03:
okay fine - as said this is not a problem - updating poved to be effective on setting the internals to the default state. We already fixed a bunch of problems this way.

Now please reinstall xpadneo and give me the output.

germangoergs, 2019-04-03:
> okay fine - as said, this is not a problem - updating poved to be effective on setting the internals back to the default state. We already fixed a bunch of problems this way.
> 
> Now please reinstall xpadneo and give me the output. Reboot afterwards.

Here you go:
```
:~/git/xpadneo> sudo ./install.sh   
* replacing version string if necessary
* copying module into /usr/src
* adding module to DKMS

Creating symlink /var/lib/dkms/hid-xpadneo/0.5.10/source ->
                 /usr/src/hid-xpadneo-0.5.10

DKMS: add completed.
* installing module (using DKMS)

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
make -j8 KERNELRELEASE=5.0.5-1-default -C /lib/modules/5.0.5-1-default/build M=/var/lib/dkms/hid-xpadneo/0.5.10/build/src modules...
cleaning build area...

DKMS: build completed.

hid-xpadneo.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/5.0.5-1-default/updates/

Running the post_install script:
Disabling ERTM permanently (requires reboot)...
Diabling ERTM instantly...
Installing UDEV rule to automatically bind xpadneo (necessary on kernel < 4.16 and >= 4.20)...

depmod....

Backing up initrd-5.0.5-1-default to /boot/initrd-5.0.5-1-default.old-dkms
Making new initrd-5.0.5-1-default
(If next boot fails, revert to initrd-5.0.5-1-default.old-dkms image)
dracut.........

DKMS: install completed.
```

atar-axis, 2019-04-03:
again: if you haven't rebooted yet, please reboot now.

After that, please run `bluetoothctl` and type in `scan on`. Wait for all active devices to appear, give them some minutes, after that - please turn on your first gamepad and press the pairing button until the light flashes fast. Wait for the gamepad to appear in `bluetoothctl`. Run `pair <MAC>`.

Give me the output ;)

germangoergs, 2019-04-03:
> again: if you haven't rebooted yet, please reboot now.
> 
> After that, please run `bluetoothctl` (`sudo bluetoothctl` on raspbian systems) and type in `scan on`. Wait for all active devices to appear, give them some minutes, after that - please turn on your first gamepad and press the pairing button until the light flashes fast. Wait for the gamepad to appear in `bluetoothctl`. Run `pair <MAC>`.
> 
> Give me the output ;)

```
bluetoothctl
Agent registered
[bluetooth]# scan on
Discovery started
[CHG] Controller 00:1A:7D:DA:71:11 Discovering: yes
[CHG] Device EC:83:50:EC:FB:5E RSSI: -93
[CHG] Device EC:83:50:EC:FB:5E RSSI: -38
[CHG] Device EC:83:50:EC:FB:5E RSSI: -93
[bluetooth]# pair EC:83:50:EC:FB:5E
Attempting to pair with EC:83:50:EC:FB:5E
Failed to pair: org.bluez.Error.AlreadyExists
```

atar-axis, 2019-04-03:
okay this is the first problem, the gamepad definitely not exists anymore since we removed it.
Please make sure that you removed the existing gamepads from your KDE/Gnome Bluetooth manager too. 
what does `devices` say?

germangoergs, 2019-04-03:
> okay this is the first problem, the gamepad definitely not exists anymore since we removed it.
> Please make sure that you removed the existing gamepads from your KDE/Gnome Bluetooth manager too.
> what does `devices` say?

As far as I could notice, a device that was found by the bluetooth manager is automatically added.
I can't do nothing against it.
For example the other controller isn't shown, but whenever I search for a device, these are also added automatically until I reboot my PC.

atar-axis, 2019-04-03:
what does `devices` say? And what `paired-devices` ? both in bluetoothctl

germangoergs, 2019-04-03:
> what does `devices` say? And what `paired-devices` ? both in bluetoothctl

I added devices at the end of my last message.

```
paired-devices
Device EC:83:50:EC:FB:5E Xbox Wireless Controller
```

atar-axis, 2019-04-03:
Okay, let's keep that in mind for now.

Please run `trust EC:83:50:EC:FB:5E` now, let's see what this results in

germangoergs, 2019-04-03:
> Okay, let's keep that in mind for now.
> 
> Please run `trust EC:83:50:EC:FB:5E` now, let's see what this results in

```
trust EC:83:50:EC:FB:5E
[CHG] Device EC:83:50:EC:FB:5E Trusted: yes
Changing EC:83:50:EC:FB:5E trust succeeded
```

atar-axis, 2019-04-03:
Now please run `sudo dmesg -C`, back to bluetoothctl, run `connect ...` (make sure the gamepad is still in pairing mode)

germangoergs, 2019-04-03:
I tried multiple times to connect it, but always this error:
```
connect EC:83:50:EC:FB:5E
Attempting to connect to EC:83:50:EC:FB:5E
[CHG] Device EC:83:50:EC:FB:5E Connected: yes
Failed to connect: org.bluez.Error.Failed
[CHG] Device EC:83:50:EC:FB:5E Connected: no
```
dmesg gives me:
```
dmesg
[ 1188.363189] Bluetooth: HIDP (Human Interface Emulation) ver 1.2
[ 1188.363192] Bluetooth: HIDP socket layer initialized
```

atar-axis, 2019-04-03:
Do you use an USB-Bluetooth stick oder builtin bluetooth?
What is the output of `sudo btmon` while trying to connect the gamepad?

germangoergs, 2019-04-03:
> Do you use an USB-Bluetooth stick oder builtin bluetooth?
> What is the output of `sudo btmon` while trying to connect the gamepad?

I use USB-Bluetooth. :D 

`sudo btmon` output:
https://pastebin.com/awXM0t0x

atar-axis, 2019-04-03:
> I use USB-Bluetooth
This seem to be a problem. Some dongles are not reliable in connecting with this gamepad. This behaviour was reported more than once, e.g. by @ugly95  in https://github.com/atar-axis/xpadneo/issues/76

Anyway, try to connect it a couple of times again, some users reported that this helps. For a long-term solution I would suggest to use another USB BT stick.

atar-axis, 2019-04-03:
Btw, the output of `cat /sys/module/bluetooth/parameters/disable_ertm` is still `Y`, right?

germangoergs, 2019-04-03:
Yes, it's still yes.

I did some testing with my controllers to maybe find some kind of pattern.
Actually, I found out that the red controller seems to have problems connecting to my PC more often than the red one, on which I changed battery, since they seemed to be empty (then it mostly also rumbled when connected, BUT still not always, as well as the black).

BUT I made notes regarding my observations, so have a look at your own.
I also began to catch btmon at some point and I added dmesg. For free! :D 
I hope it's some kind of helpful. :D 
I just hadn't thought about that the cli just saves a specific number of lines, so the btmon is just shows the last few tries, I hope they're helpful anyway.

So here are my "observations":
https://www.magentacloud.de/lnk/6YAKnMT9

And here is the btmon:
https://pastebin.com/RT6D9yR1

here dmesg:
https://pastebin.com/B3BP3ztv

Thanks again for your support. :)

germangoergs, 2019-04-03:
Oh um btw., when I use fftest to test Force Feedback it shows me the following:
```
sudo fftest
Force feedback test program.
HOLD FIRMLY YOUR WHEEL OR JOYSTICK TO PREVENT DAMAGES

Device /dev/input/event0 opened
Features:
  * Absolute axes: 
    [00 00 00 00 00 00 00 00 ]
  * Relative axes: 
    [00 00 ]
  * Force feedback effects types: 
    Force feedback periodic effects: 
    [00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ]
  * Number of simultaneous effects: 0

Uploading effect #0 (Periodic sinusoidal) ... Error:: Function not implemented
Uploading effect #1 (Constant) ... Error: Function not implemented
Uploading effect #2 (Spring) ... Error: Function not implemented
Uploading effect #3 (Damper) ... Error: Function not implemented
Uploading effect #4 (Strong rumble, with heavy motor) ... Error: Function not implemented
Uploading effect #5 (Weak rumble, with light motor) ... Error: Function not implemented
```

atar-axis, 2019-04-04:
Weak Battery ... :see_no_evil: :blush:
Are you able to borrow another (or two) USB BT Sticks from a friend? To confirm that the Connection issue is because of the compatibility? (Not the Rumble thing)

germangoergs, 2019-04-04:
> Weak Battery ... see_no_evil blush
> 
> Do you have the possibility to borrow another (or two) USB BT Sticks from a friend? To confirm that the Connection issue is because of the compatibility?
> 
> (we would have to solve only the rumble/binding thing then, which is maybe caused by udev)
> 
> Btw, fftest needs the correct event file, which is most probably not event0 but a higher number

So...
I tested with a notebook with integrated bluetooth and I could find similar behaviour as on my PC, except connection times, I think they were slightly faster on the notebook.
Anyway, both controllers vibrate now (most of the times) when connected, probably the empty battery wasn't really helpful (sorry for that again).
But still, the red controller also had on then notebook more connection issues than the black one and when the black one/red one is already connected, the other one is able to connect/connects much faster (dunno why).
So I'd say that long connection times are occurred by my bt stick.
Do you know a "good" one?
But as my testing also shows, the red controller sometimes has problems connecting via bluetooth, no matter to which device.

Oh Ok, sorry. O.o

germangoergs, 2019-04-04:
Oh and there's also no ff in games. :(

atar-axis, 2019-04-04:
Okay, since you are able to connect both now, at least more or less, I would suggest that we try to solve your rumble problems now - ok? 😊 We can come back to the connection Thing later, but I really think it is a Hardware Problem.

When both are connected: to which drivers are they bound?

germangoergs, 2019-04-04:
> Okay, since you are able to connect both now, at least more or less, I would suggest that we try to solve your rumble problems now - ok? blush We can come back to the connection Thing later, but I really think it is a Hardware Problem.
> 
> When both are connected: to which drivers are they bound?

Yep, that's fine.

Well, sometimes no driver is shown (then it doesn't rumble), sometimes Microsoft driver (doesnm't rumble as well), but mostly Xpadneo drivers.

atar-axis, 2019-04-04:
Okay, so whenever one of your gamepads is connected to microsoft next time,
try to bind it manually to xpadneo by running the following commands:

```
[KERNEL_STRING] > /sys/bus/hid/drivers/microsoft/unbind
[KERNEL_STRING] > /sys/bus/hid/drivers/xpadneo/bind
```

You find the `[KERNEL_STRING]` in `dmesg`, it looks something like this: `0005:045E:02FD.00XX`.
Take the one which is mentioned beneath `microsoft` in dmesg.

germangoergs, 2019-04-04:
> Okay, so whenever one of your gamepads is connected to microsoft next time,
> try to bind it manually to xpadneo by running the following commands:
>
> ```
> [KERNEL_STRING] > [...]

OK, cool, I'll do that. :)

atar-axis, 2019-04-04:
Tell me If it works manually, then  we can examine why it doesn't automatically

germangoergs, 2019-04-05:
Well, to be honest, the last times a controller wasn't bound to the xpadneo driver, it was bound to no driver, not even the Microsoft ones.
And then when I try to bind it to Xpad, I'm told that I do not have the required permissions.

And I've tested a game on which FF works (via Proton), so it seems that FF does work in games. :)

atar-axis, 2019-04-05:
You have to run it as root / sudo ;) Even If it works, I would be glad If you could solve the issue with me anyway :)

germangoergs, 2019-04-05:
I run it as root, I thought it was funny. :D 

OK. :)

atar-axis, 2019-04-05:
You can use the following command to run it using `sudo`:
```
echo [KERNEL_STRING] | sudo tee /sys/bus/hid/drivers/microsoft/unbind
echo [KERNEL_STRING] | sudo tee /sys/bus/hid/drivers/xpadneo/bind
```

germangoergs, 2019-04-05:
Ah OK, now it worked, even with unbound driver, sorry for my incompetence. O.o

atar-axis, 2019-04-05:
Would running `udevadm monitor` while connecting your gamepad? This way we can examine the binding problem a bit better.

I am especially interested in the result when a GP gets bound to Microsoft or nothing.

germangoergs, 2019-04-05:
Here you go:
```
KERNEL[13475.963779] add      /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3.2/1-3.2:1.0/bluetooth/hci0/hci0:70 (bluetooth)
UDEV  [13475.969080] add      /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3.2/1-3.2:1.0/bluetooth/hci0/hci0:70 (bluetooth)
KERNEL[13476.072721] add      /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3.2/1-3.2:1.0/bluetooth/hci0/hci0:70/0005:045E:02FD.000B (hid)
KERNEL[13476.072844] add      /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3.2/1-3.2:1.0/bluetooth/hci0/hci0:70/0005:045E:02FD.000B/input/input37 (input)
KERNEL[13476.072904] add      /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3.2/1-3.2:1.0/bluetooth/hci0/hci0:70/0005:045E:02FD.000B/input/input37/event19 (input)
KERNEL[13476.072998] add      /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3.2/1-3.2:1.0/bluetooth/hci0/hci0:70/0005:045E:02FD.000B/input/input37/js1 (input)
KERNEL[13476.073028] add      /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3.2/1-3.2:1.0/bluetooth/hci0/hci0:70/0005:045E:02FD.000B/hidraw/hidraw4 (hidraw)
KERNEL[13476.073051] bind     /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3.2/1-3.2:1.0/bluetooth/hci0/hci0:70/0005:045E:02FD.000B (hid)
KERNEL[13476.080644] remove   /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3.2/1-3.2:1.0/bluetooth/hci0/hci0:70/0005:045E:02FD.000B/power_supply/hid-ec:83:50:88:75:e7-battery (power_supply)
KERNEL[13476.102981] remove   /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3.2/1-3.2:1.0/bluetooth/hci0/hci0:70/0005:045E:02FD.000B/input/input37/event19 (input)
KERNEL[13476.134997] remove   /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3.2/1-3.2:1.0/bluetooth/hci0/hci0:70/0005:045E:02FD.000B/input/input37/js1 (input)
KERNEL[13476.159009] remove   /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3.2/1-3.2:1.0/bluetooth/hci0/hci0:70/0005:045E:02FD.000B/input/input37 (input)
KERNEL[13476.159123] remove   /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3.2/1-3.2:1.0/bluetooth/hci0/hci0:70/0005:045E:02FD.000B/hidraw/hidraw4 (hidraw)
KERNEL[13476.159214] unbind   /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3.2/1-3.2:1.0/bluetooth/hci0/hci0:70/0005:045E:02FD.000B (hid)
UDEV  [13476.160029] add      /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3.2/1-3.2:1.0/bluetooth/hci0/hci0:70/0005:045E:02FD.000B (hid)
UDEV  [13476.167659] add      /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3.2/1-3.2:1.0/bluetooth/hci0/hci0:70/0005:045E:02FD.000B/input/input37 (input)
UDEV  [13476.169269] add      /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3.2/1-3.2:1.0/bluetooth/hci0/hci0:70/0005:045E:02FD.000B/hidraw/hidraw4 (hidraw)
UDEV  [13476.176145] add      /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3.2/1-3.2:1.0/bluetooth/hci0/hci0:70/0005:045E:02FD.000B/input/input37/event19 (input)
UDEV  [13476.177399] add      /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3.2/1-3.2:1.0/bluetooth/hci0/hci0:70/0005:045E:02FD.000B/input/input37/js1 (input)
UDEV  [13476.182163] bind     /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3.2/1-3.2:1.0/bluetooth/hci0/hci0:70/0005:045E:02FD.000B (hid)
UDEV  [13476.186370] remove   /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3.2/1-3.2:1.0/bluetooth/hci0/hci0:70/0005:045E:02FD.000B/power_supply/hid-ec:83:50:88:75:e7-battery (power_supply)
UDEV  [13476.187030] remove   /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3.2/1-3.2:1.0/bluetooth/hci0/hci0:70/0005:045E:02FD.000B/input/input37/event19 (input)
UDEV  [13476.187919] remove   /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3.2/1-3.2:1.0/bluetooth/hci0/hci0:70/0005:045E:02FD.000B/input/input37/js1 (input)
UDEV  [13476.190023] remove   /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3.2/1-3.2:1.0/bluetooth/hci0/hci0:70/0005:045E:02FD.000B/hidraw/hidraw4 (hidraw)
UDEV  [13476.192759] remove   /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3.2/1-3.2:1.0/bluetooth/hci0/hci0:70/0005:045E:02FD.000B/input/input37 (input)
UDEV  [13476.197538] unbind   /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3.2/1-3.2:1.0/bluetooth/hci0/hci0:70/0005:045E:02FD.000B (hid)
KERNEL[13477.605957] change   /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3.2/1-3.2:1.0/bluetooth/hci0/hci0:69/0005:045E:02FD.000A/power_supply/xpadneo_batt_3a:33:38:3a:63:65_0 (power_supply)
UDEV  [13477.612091] change   /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3.2/1-3.2:1.0/bluetooth/hci0/hci0:69/0005:045E:02FD.000A/power_supply/xpadneo_batt_3a:33:38:3a:63:65_0 (power_supply)
KERNEL[13495.126486] remove   /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3.2/1-3.2:1.0/bluetooth/hci0/hci0:69/0005:045E:02FD.000A/input/input36/event18 (input)
UDEV  [13495.132809] remove   /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3.2/1-3.2:1.0/bluetooth/hci0/hci0:69/0005:045E:02FD.000A/input/input36/event18 (input)
KERNEL[13495.146954] remove   /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3.2/1-3.2:1.0/bluetooth/hci0/hci0:69/0005:045E:02FD.000A/input/input36/js0 (input)
UDEV  [13495.152058] remove   /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3.2/1-3.2:1.0/bluetooth/hci0/hci0:69/0005:045E:02FD.000A/input/input36/js0 (input)
KERNEL[13495.170954] remove   /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3.2/1-3.2:1.0/bluetooth/hci0/hci0:69/0005:045E:02FD.000A/input/input36 (input)
KERNEL[13495.171077] remove   /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3.2/1-3.2:1.0/bluetooth/hci0/hci0:69/0005:045E:02FD.000A/hidraw/hidraw3 (hidraw)
KERNEL[13495.171261] remove   /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3.2/1-3.2:1.0/bluetooth/hci0/hci0:69/0005:045E:02FD.000A/power_supply/xpadneo_batt_3a:33:38:3a:63:65_0 (power_supply)
[... cut ...]
UDEV  [13546.527326] add      /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3.2/1-3.2:1.0/bluetooth/hci0/hci0:70/0005:045E:02FD.000E/input/input41/event18 (input)
UDEV  [13546.528601] bind     /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3.2/1-3.2:1.0/bluetooth/hci0/hci0:70/0005:045E:02FD.000E (hid)
UDEV  [13546.529932] change   /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3.2/1-3.2:1.0/bluetooth/hci0/hci0:70/0005:045E:02FD.000E/power_supply/xpadneo_batt_3a:33:38:3a:63:65_0 (power_supply)
```

Last output was xpadneo driver, the second last without a driver bound.

atar-axis, 2019-04-05:
Thank you! Could you please do the same again - but this time using `udevadm monitor -p`?
And, could you note me somewhere what you did when? Udev isn't really to examine (at least for me), not knowing what happend when makes it even harder :blush:

atar-axis, 2019-04-05:
btw, what kind of system/distribution are you on?

germangoergs, 2019-04-05:
```
UDEV  [32262.906894] add      /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3.2/1-3.2:1.0/bluetooth/hci0/hci0:70/0005:045E:02FD.0012/input/input47 (input)
ACTION=add
DEVPATH=/devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3.2/1-3.2:1.0/bluetooth/hci0/hci0:70/0005:045E:02FD.0012/input/input47
SUBSYSTEM=input
PRODUCT=5/45e/2fd/903
NAME="Xbox Wireless Controller"
PHYS="00:1a:7d:da:71:11"
UNIQ="ec:83:50:88:75:e7"
PROP=0
EV=1b
KEY=7fff000000000000 0 100040000000 0 0
ABS=30627
MSC=10
MODALIAS=input:b0005v045Ep02FDe0903-e0,1,3,4,k9E,AC,130,131,132,133,134,135,136,137,138,139,13A,13B,13C,13D,13E,ra0,1,2,5,9,A,10,11,m4,lsfw
SEQNUM=5093
USEC_INITIALIZED=32262906707
ID_BUS=bluetooth
ID_PATH=pci-0000:00:14.0-usb-0:3.2:1.0
ID_PATH_TAG=pci-0000_00_14_0-usb-0_3_2_1_0
ID_FOR_SEAT=input-pci-0000_00_14_0-usb-0_3_2_1_0
TAGS=:seat:

UDEV  [32262.907353] add      /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3.2/1-3.2:1.0/bluetooth/hci0/hci0:70/0005:045E:02FD.0012/hidraw/hidraw3 (hidraw)
ACTION=add
DEVPATH=/devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3.2/1-3.2:1.0/bluetooth/hci0/hci0:70/0005:045E:02FD.0012/hidraw/hidraw3
SUBSYSTEM=hidraw
DEVNAME=/dev/hidraw3
SEQNUM=5096
USEC_INITIALIZED=32262907274
MAJOR=246
MINOR=3

UDEV  [32262.909209] add      /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3.2/1-3.2:1.0/bluetooth/hci0/hci0:70/0005:045E:02FD.0012/input/input47/js0 (input)
ACTION=add
DEVPATH=/devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3.2/1-3.2:1.0/bluetooth/hci0/hci0:70/0005:045E:02FD.0012/input/input47/js0
SUBSYSTEM=input
DEVNAME=/dev/input/js0
SEQNUM=5095
USEC_INITIALIZED=32262909116
ID_BUS=bluetooth
MAJOR=13
MINOR=0

KERNEL[32262.922729] change   /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3.2/1-3.2:1.0/bluetooth/hci0/hci0:70/0005:045E:02FD.0012/power_supply/xpadneo_batt_3a:33:38:3a:63:65_0 (power_supply)
ACTION=change
DEVPATH=/devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3.2/1-3.2:1.0/bluetooth/hci0/hci0:70/0005:045E:02FD.0012/power_supply/xpadneo_batt_3a:33:38:3a:63:65_0
SUBSYSTEM=power_supply
POWER_SUPPLY_NAME=(null)
POWER_SUPPLY_PRESENT=0
POWER_SUPPLY_CAPACITY_LEVEL=Full
POWER_SUPPLY_SCOPE=Device
POWER_SUPPLY_STATUS=Unknown
POWER_SUPPLY_MANUFACTURER=Microsoft
POWER_SUPPLY_MODEL_NAME=Xbox Wireless Controller
POWER_SUPPLY_ONLINE=0
SEQNUM=5110

UDEV  [32262.972207] add      /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3.2/1-3.2:1.0/bluetooth/hci0/hci0:70/0005:045E:02FD.0012/input/input47/event18 (input)
ACTION=add
DEVPATH=/devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3.2/1-3.2:1.0/bluetooth/hci0/hci0:70/0005:045E:02FD.0012/input/input47/event18
SUBSYSTEM=input
DEVNAME=/dev/input/event18
SEQNUM=5094
USEC_INITIALIZED=32262972008
ID_BUS=bluetooth
MAJOR=13
MINOR=82

UDEV  [32262.974326] bind     /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3.2/1-3.2:1.0/bluetooth/hci0/hci0:70/0005:045E:02FD.0012 (hid)
[... 1843 lines cut ...]
[32638.531305] usb 1-3.7-port1: unable to enumerate USB device
[32640.103118] usb 1-3.7.1: new full-speed USB device number 20 using xhci_hcd
[32640.303129] usb 1-3.7.1: device descriptor read/64, error -32
[32640.611101] usb 1-3.7.1: device descriptor read/64, error -32
[32640.919089] usb 1-3.7.1: new full-speed USB device number 21 using xhci_hcd
[32641.119077] usb 1-3.7.1: device descriptor read/64, error -32
[32641.427646] usb 1-3.7.1: device descriptor read/64, error -32
[32641.535089] usb 1-3.7-port1: attempt power cycle
[32642.239054] usb 1-3.7.1: new full-speed USB device number 22 using xhci_hcd
[32642.239153] usb 1-3.7.1: Device not responding to setup address.
[32642.447114] usb 1-3.7.1: Device not responding to setup address.
[32642.655050] usb 1-3.7.1: device not accepting address 22, error -71
[32642.855067] usb 1-3.7.1: new full-speed USB device number 23 using xhci_hcd
[32642.855168] usb 1-3.7.1: Device not responding to setup address.
[32643.063102] usb 1-3.7.1: Device not responding to setup address.
[32643.271038] usb 1-3.7.1: device not accepting address 23, error -71
[32643.271113] usb 1-3.7-port1: unable to enumerate USB device
[32644.851066] usb 1-3.7.1: new full-speed USB device number 24 using xhci_hcd
[32645.051010] usb 1-3.7.1: device descriptor read/64, error -32
[32645.359156] usb 1-3.7.1: device descriptor read/64, error -32
[32645.666994] usb 1-3.7.1: new full-speed USB device number 25 using xhci_hcd
[32645.866998] usb 1-3.7.1: device descriptor read/64, error -32
```

Same one here with dmesg, but for any reason I can't change maximum length. O.o

germangoergs, 2019-04-05:
> btw, what kind of system/distribution are you on? maybe hid-generic is builtin on your system and not shown - could be the reason why sometimes you do not see any driver.

And I'm on Opensuse Tumbleweed.

atar-axis, 2019-04-07:
Ah, yeah - try to use `udevadm monitor -p > /tmp/xpadneo_log1` then and please upload the log afterwards. But please, whenever you do something (like disconnecting or connecting a new gp), start a new log (abort the old log before with <kbd>STRG</kbd> + <kbd>c</kbd> and start a new one). Just like this:

* connected red gampad
* was bound to microsoft
* output was: [result of devadm monitor -p > /tmp/xpadneo_log**1**]
---
* connected black gamepad too
* was bound to xpadneo
* output was: [result of devadm monitor -p > /tmp/xpadneo_log**2**]
...

atar-axis, 2019-04-08:
https://github.com/atar-axis/xpadneo/issues/93

germangoergs, 2019-04-08:
> take a look here: #93

Yep, I have an unreliable device. :D 

Then I'll buy a new Bluetooth dongle, yey. O.o
Or I wait until you make the Wifi dongle work. :D 

And sorry that I didn't answer, haven't had time the last days, but you'll get some new logs tomorrow. ;)

germangoergs, 2019-04-10:
Hey, sorry that I'm just sending logs now, but here you go. I connected my controllers 3 times, I numerated them by order of connections.
After second log, I didn't turn the red one off, after the first I did.

I hope, they help you. :)

[1XpadConnectRed_XPadNeo.txt](https://github.com/atar-axis/xpadneo/files/3065007/1XpadConnectRed_XPadNeo.txt)
[2XpadConnectRed_nodriver.txt](https://github.com/atar-axis/xpadneo/files/3065008/2XpadConnectRed_nodriver.txt)
[3XpadConnectBlack_nodriver.txt](https://github.com/atar-axis/xpadneo/files/3065009/3XpadConnectBlack_nodriver.txt)

atar-axis, 2019-04-10:
Do you have any kind of problems using your Keyboard?
I am just asking because there are around 100 bind/unbind events for it in your log - looks really weird! :blush:

I will continue analyzing the log tomorow or so - stay tuned ;)

germangoergs, 2019-04-10:
Umm nope, I don't have problems with my Keyboard, I just have an additional driver to make the macro keys work.
But thanks for letting me know!

OK, thanks. :)

atar-axis, 2019-04-10:
Your logs are really useful now, thank you! Could you do me another favor?
Please replace the content of `/etc/udev/rules.d/99-xpadneo.rule`
with the following:

```
ACTION=="add", KERNEL=="0005:045E:02FD.*|0005:045E:02E0.*", SUBSYSTEM=="hid",\
RUN+="/bin/bash -c 'echo xpadneo 1 > /dev/kmsg && echo $kernel > /sys/bus/hid/drivers/hid-generic/unbind'", \
RUN+="/bin/bash -c 'echo xpadneo 2 > /dev/kmsg && echo $kernel > /sys/bus/hid/drivers/microsoft/unbind'", \
RUN+="/bin/bash -c 'echo xpadneo 3 > /dev/kmsg && echo $kernel > /sys/bus/hid/drivers/xpadneo/bind'"
```

Please reconnect the gamepad then until it gets again to "no driver" - show me `dmesg` then.
Thanks :)

atar-axis, 2019-04-10:
UPDATE:
Please use the following rule instead:

```
# if the device is an xbox one wireless gamepad, stop processing of the automatic
# driver loading in /usr/lib/udev/rules.d/80-drivers.rules but load xpadneo by using the "RUN:=" operator

ACTION=="add", ENV{MODALIAS}=="hid:b0005g0001v0000045Ep000002E0|hid:b0005g0001v0000045Ep000002FD", \
RUN{builtin}:="kmod load hid_xpadneo", \
RUN+="/bin/bash -c 'echo \"xpadneo udev: loading only xpadneo\" > /dev/kmsg'"

# no other (fitting) driver gets loaded and registers himself at the bus master. Therefore the device is always bound to hid_xpadneo
# the only problem is: this does not work if hid_microsoft was already loaded by another device before!

# on newer kernel version (>= 4.14, see https://github.com/atar-axis/xpadneo/issues/70#issuecomment-452139479)
# udev added an bind/unbind uevent, this is the one we should normally use to recognize binding to hid_microsoft or hid_generic.
# support for the (02fd) gamepad was added to hid_microsoft in kernel 4.20, therefore we should be able to detect binding to microsoft
# by using the bind event there

ACTION=="bind", KERNEL=="0005:045E:02FD.*|0005:045E:02E0.*", SUBSYSTEM=="hid", DRIVER=="microsoft", \
RUN+="/bin/bash -c 'echo \"xpadneo udev: binding detected\" > /dev/kmsg'", \
RUN+="/bin/bash -c 'echo \"xpadneo udev: unbinding microsoft\" > /dev/kmsg && echo $kernel > /sys/bus/hid/drivers/microsoft/unbind'", \
RUN+="/bin/bash -c 'echo \"xpadneo udev: binding xpadneo\" > /dev/kmsg && echo $kernel > /sys/bus/hid/drivers/xpadneo/bind'", \
GOTO="end"

ACTION=="bind", KERNEL=="0005:045E:02FD.*|0005:045E:02E0.*", SUBSYSTEM=="hid", DRIVER=="hid-generic", \
RUN+="/bin/bash -c 'echo \"xpadneo udev: binding detected\" > /dev/kmsg'", \
RUN+="/bin/bash -c 'echo \"xpadneo udev: unbinding hid-generic\" > /dev/kmsg && echo $kernel > /sys/bus/hid/drivers/hid_generic/unbind'", \
RUN+="/bin/bash -c 'echo \"xpadneo udev: binding xpadneo\" > /dev/kmsg && echo $kernel > /sys/bus/hid/drivers/xpadneo/bind'", \
GOTO="end"

# if the bind event did not fire, then we have most probably installed a kernel version before 4.14
# which does not have gamepad support in microsoft
# TODO: the problem is, that the add event happens before the bind event, and it happens always and on every kernel
#       the only solution I can think of is to utilize uname -a

#ACTION=="add", KERNEL=="0005:045E:02FD.*|0005:045E:02E0.*", SUBSYSTEM=="hid",\
#RUN+="/bin/bash -c 'echo \"xpadneo udev: add detected\" > /dev/kmsg'", \
#RUN+="/bin/bash -c 'echo \"xpadneo udev: unbinding hid-generic\" > /dev/kmsg && echo $kernel > /sys/bus/hid/drivers/hid-generic/unbind'", \
#RUN+="/bin/bash -c 'echo \"xpadneo udev: binding xpadneo\" > /dev/kmsg && echo $kernel > /sys/bus/hid/drivers/xpadneo/bind'"

LABEL="end"
```

[44 more replies]

closed 2019-04-19 by atar-axis