atar-axis/xpadneo #94: HELP! Looking for testers

atar-axis, 2019-04-11:
Hey there,
If anyone is interested in helping me testing new functions, please tell me.
Thanks!

Gigabyte29, 2019-04-13:
I can help if you want

atar-axis, 2019-04-13:
That would be great! What kernel are you on?

Gigabyte29, 2019-04-13:
I'm on Antergos with kernel 5.0.7.arch1

atar-axis, 2019-04-13:
Ah, as myself - but that's just fine :) First thing you could try - if you want - is the `enable_ff` branch, see the discussion [here](https://github.com/atar-axis/xpadneo/issues/88)

Lysdestic, 2019-04-16:
I've been following this project loosely as I've got a couple of original Xbox One controllers that function via the old wireless adapter (i.e., not Bluetooth). I understand that is of lesser priority, but if you ever make any headway in that area, I'd be glad to help test. :)

atar-axis, 2019-04-19:
We need testers for:

1) https://github.com/atar-axis/xpadneo/tree/new_udev_rule
2) https://github.com/atar-axis/xpadneo/tree/enable_ff

plafue, 2019-04-28:
I tested the new rule of the `new_udev_rule` branch and the **microsoft** module grabs control of the gamepad every time. 

With `master` at least every second time or so the udev rule works and xpadneo binds with the controller. So i'm sticking with the master version for now.

`uname -a` : `5.0.0-13-generic #14-Ubuntu SMP Mon Apr 15 14:59:14 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux`

this is what dmesg says about my usb dongle:
```
[    1.513611] usb 3-3: New USB device found, idVendor=0a12, idProduct=0001, bcdDevice=88.91
[    1.513612] usb 3-3: New USB device strings: Mfr=0, Product=2, SerialNumber=0
[    1.513613] usb 3-3: Product: CSR8510 A10
```

atar-axis, 2019-04-28:
Thank you for Testing!!! What does dmesg say when hid-microsoft grabs it?

plafue, 2019-04-28:
```[   64.322241] Bluetooth: HIDP (Human Interface Emulation) ver 1.2
[   64.322244] Bluetooth: HIDP socket layer initialized
[   64.322754] hid-generic 0005:045E:02FD.0003: unknown main item tag 0x0
[   64.322819] input: Xbox Wireless Controller as /devices/pci0000:00/0000:00
:14.0/usb3/3-3/3-3:1.0/bluetooth/hci0/hci0:71/0005:045E:02FD.0003/input/input
11
[   64.322886] input: Xbox Wireless Controller Consumer Control as /devices/p
ci0000:00/0000:00:14.0/usb3/3-3/3-3:1.0/bluetooth/hci0/hci0:71/0005:045E:02FD
.0003/input/input12
[   64.322933] hid-generic 0005:045E:02FD.0003: input,hidraw2: BLUETOOTH HID
v9.03 Gamepad [Xbox Wireless Controller] on 00:1a:7d:da:71:04
[   64.431667] microsoft 0005:045E:02FD.0003: unknown main item tag 0x0
[   64.431725] input: Xbox Wireless Controller as /devices/pci0000:00/0000:00
:14.0/usb3/3-3/3-3:1.0/bluetooth/hci0/hci0:71/0005:045E:02FD.0003/input/input
13
[   64.431806] microsoft 0005:045E:02FD.0003: input,hidraw2: BLUETOOTH HID v9
.03 Gamepad [Xbox Wireless Controller] on 00:1a:7d:da:71:04
[   64.432410] xpadneo: hello there!
```

plafue, 2019-04-28:
And this when xpadneo takes control:
```
[  251.270100] xpadneo 0005:045E:02FD.0005: unknown main item tag 0x0
[  251.270140] input: Xbox Wireless Controller as /devices/pci0000:00/0000:00
:14.0/usb3/3-3/3-3:1.0/bluetooth/hci0/hci0:71/0005:045E:02FD.0005/input/input
16
[  251.270311] xpadneo 0005:045E:02FD.0005: input,hidraw2: BLUETOOTH HID v9.0
3 Gamepad [Xbox Wireless Controller] on 00:1a:7d:da:71:04
```

atar-axis, 2019-04-28:
Can you give it another try? I updated the branch.

plafue, 2019-04-28:
One file left behind... Classic ;)

It seems to work reliably now :tada: yay!

Thanks a lot for all the effort you put in xpadneo @atar-axis! :heart:

atar-axis, 2019-04-28:
Hehe, happy to hear that - thank you for reporting the issue and testing!

plafue, 2019-04-28:
Mention me anytime if you need testers :+1:

Nachti!

atar-axis, 2019-04-28:
Perfect, thank you for that offer! Nacht ;)

atar-axis, 2019-04-29:
both PR's/branches are merged into master now, hopefully stable. Thanks for testing.

gordongnarf, 2019-05-05:
Hi, i installed xpadneo in order to be able to rumble my new xbox controller attached to my raspberry pi 3 (the controller is new and only called xbox controller, not "one" or "360" or anything).
Before installation i had the controller running via Bluetooth using python and evdev to read the buttons and analog sticks (worked fine). I followed the installation description in 
https://pimylifeup.com/xbox-controllers-raspberry-pi/ (installing xboxdrv).
When i connect with the controller, it seems, that only the hid-generic takes over control - see dmesg below.
Now it seems that this is a general problem - but i cannot see the solution in any of the issues ...
Help would be appreciated - i can also act as a tester for raspberry pi :-)

[Mai 5 20:42] Bluetooth: HIDP (Human Interface Emulation) ver 1.2
[  +0,000030] Bluetooth: HIDP socket layer initialized
[  +0,006041] hid-generic 0005:045E:02FD.0001: unknown main item tag 0x0
[  +0,001050] input: Xbox Wireless Controller as /devices/platform/soc/3f201000.serial/tty/ttyAMA0/hci0/hci0:11/0005:045E:02FD.0001/input/input0
[  +0,000694] hid-generic 0005:045E:02FD.0001: input,hidraw0: BLUETOOTH HID v9.03 Gamepad [Xbox Wireless Controller] on b8:27:eb:73:c7:4f
[  +0,046279] hid_xpadneo: loading out-of-tree module taints kernel.
[  +0,000994] xpadneo: hello there!

uname -a
Linux raspberrypi 4.14.98-v7+ #1200 SMP Tue Feb 12 20:27:48 GMT 2019 armv7l GNU/Linux

atar-axis, 2019-05-05:
Please open a new Issue @gordongnarf ;) Solving this issue would flood this one here which is dedicated to something else :)

atar-axis, 2019-05-06:
Even the new udev_rules seems to be unstable, I am therefore looking for testers for this one: https://github.com/atar-axis/xpadneo/issues/100#issuecomment-489569419

atar-axis, 2019-05-10:
I am looking for testers for the new `mouse simulation mode`, see #99 for instructions

atar-axis, 2019-05-15:
> Even the new udev-rule seems to be unstable, I am therefore looking for testers for a new one: #100 (comment)
> 
> So far tested on:
> 
> 5.0.10-arch1-1-ARCH (my local machine)
> 4.14.98-v7+ (@gordongnarf)

I switched to another udev rule since more an more users reported issues. I hope it works for all now

atar-axis, 2019-07-03:
I will close this since nobody is testing, looks like this is the wrong way to call for help :grin:

closed 2019-07-03 by atar-axis