atar-axis/xpadneo #95: connect script

atar-axis, 2019-04-19:
As CSR dongles do have problems to connect to the gamepad, we could offer a script which automates the connection. We could use `coproc` to automate the binding process and loop through a `connect <MAC>` loop until it is connected.

Coproc does work something like this:
```
#!/bin/bash

coproc bluetoothctl
echo "devices" >& "${COPROC[1]}"

DONE=false
until $DONE ;do
read var <& "${COPROC[0]}" || DONE=true
if [[ -z "$var"  ]]; then
  
echo $var
done
``

germangoergs, 2019-04-19:
Could it be possible to make bluetoothctl generally use this script to connect when the controller is found? 🤔
I don't know, sorry. 😅

atar-axis, 2019-04-19:
I don't think so - maybe we can somehow automate that, but it is more of an workaround really a solution. I will try to debug the connection problem anyway.

germangoergs, 2019-04-19:
Yep, I know, but still better than nothing/having to do it manually every time, when you know what I mean.

Ah OK, thanks. :)

atar-axis, 2019-04-19:
Doesn't yet work, we need to remove unprintable chars from the coproc output, but as a starting point:

```
#!/bin/bash

read -p "Turn off your gamepad and press ENTER to continue..."

coproc bluetoothctl

read -p "Turn on your gamepad and set it into pairing mode (fast flashing LED), press ENTER afterwards..."
echo "scan on" >& "${COPROC[1]}"
echo "Scanning..."

while read var; do # read var <& "${COPROC[0]}" || DONE=true

    xpad="$var" | grep "Xbox Wireless Controller" | sed -i 's/[\d128-\d255]//g'

    echo "xpad: $xpad, * $var"

    if [[ ! -z "$xpad" ]]; then
    mac=$(echo "$xpad" | sed -ne "s/.*\(\([A-F0-9]\{2\}\(:\|\)\)\{6\}\).*/\1/p" )
    echo "Xbox Wireless Controller found: $mac"
    DONE=true
    break;
    fi

    echo "next line"

done <& "${COPROC[0]}"
```

atar-axis, 2019-04-19:
```
#!/bin/bash

coproc bluetoothctl

read -p "remove your gamepad from bluetoothctl, press ENTER afterwards" # automate

echo "scan on" >& "${COPROC[1]}"
echo "Scanning... (turn your GP into pairing mode now)"

while read line; do # read var <& "${COPROC[0]}" || DONE=true

    xpad=$(echo "$line" | grep "Xbox Wireless Controller")

    echo "$line" >> log.txt

    if [[ ! -z "$xpad" ]]; then
        mac=$(echo "$xpad" | sed -ne "s/.*\(\([A-F0-9]\{2\}\(:\|\)\)\{6\}\).*/\1/p" )
        echo "Xbox Wireless Controller found: $mac"
        break;
    fi

done <& "${COPROC[0]}"

echo "pairing..."
echo "pair $mac" >& "${COPROC[1]}"

# TODO: wait until paired

echo "trusting..."
echo "trust $mac" >& "${COPROC[1]}"

# TODO: wait until trusted

echo "connecting..."
echo "connect $mac" >& "${COPROC[1]}"

# TODO: wait until connected, repeat if necessary

```

atar-axis, 2019-04-19:
okay, here we go, give it a try @germangoergs  @ugly95
It's just a first draft ;)

```
#!/usr/bin/expect

set timeout 120
spawn /usr/bin/bluetoothctl

send_user "please turn off your gamepad and remove it from bluetoothctl\n"
expect_user -re "(.*)\n"
send_user "press ENTER and then turn on your gamepad and set it into pairing mode immediately\n"
expect_user -re "(.*)\n"

expect {
    -re ".*\#" {
        send "scan on\r"
    }
}

expect {
    -re "\\\[.*NEW.*\\\].*Device (.*) Xbox Wireless Controller" {
        set mac $expect_out(1,string)
        send_user "found gamepad $mac\n"
        send "pair $mac\r"
    }
}

while true {    
    expect {
        -re "Paired: yes" {
            #send_user "sucessfully paired"
            send "trust $mac\r"
            break
        }
        
        -re "Failed to pair" {
            send "pair $mac\r"
        }
    }
}

expect {
    -re "trust succeeded" {
        #send_user "trust succeeded"
        send "connect $mac\r"
    }
}

while true {    
    expect {
        -re "Connection successful" {
            #send_user "trust succeeded"
            send "connect $mac\r"
            break
        }
        
        -re "AuthenticationFailed" {
            send "connect $mac\r"
        }
    }
}
```

germangoergs, 2019-04-19:
OK, cool, thanks. :)
Will test it later, if that's OK. :D

So we need to disconnect the controller from bluetoothctl every time we want to connect it, right?

atar-axis, 2019-04-19:
Yeah for Sure!

No, it's just a draft - If it works this way, then we can change it so you don't need to remove it ;)

ugly95, 2019-04-19:
It seems to work.  

It can take a while to connect, about 10-15 seconds.

I did need to install expect on Linux Mint 19.1

```
sudo apt install expect
```

Here is the output when I run the script:

```
spawn /usr/bin/bluetoothctl
please turn off your gamepad and remove it from bluetoothctl

press ENTER and then turn on your gamepad and set it into pairing mode immediately

[NEW] Controller 00:1A:7D:DA:71:11 rickpc [default]
[NEW] Device 5C:BA:37:D4:46:6A Xbox Wireless Controller
[NEW] Device 34:DE:1A:97:BA:B6 rick-ZBOX
[NEW] Device 00:1A:7D:21:8B:D4 Kinivo BTH240
Agent registered
[CHG] Device 5C:BA:37:D4:46:6A ServicesResolved: no
[CHG] Device 5C:BA:37:D4:46:6A Connected: no
[DEL] Device 5C:BA:37:D4:46:6A Xbox Wireless Controller
[bluetooth]# scan on
Discovery started
[CHG] Controller 00:1A:7D:DA:71:11 Discovering: yes
[NEW] Device 5C:BA:37:D4:46:6A Xbox Wireless Controller
[bluetooth]# found gamepad 5C:BA:37:D4:46:6A
pair 5C:BA:37:D4:46:6A
Attempting to pair with 5C:BA:37:D4:46:6A
[CHG] Device 5C:BA:37:D4:46:6A Connected: yes
[CHG] Device 5C:BA:37:D4:46:6A Modalias: usb:v045Ep02E0d0903
[CHG] Device 5C:BA:37:D4:46:6A Modalias: usb:v045Ep02FDd0903
[CHG] Device 5C:BA:37:D4:46:6A UUIDs: 00001124-0000-1000-8000-00805f9b34fb
[CHG] Device 5C:BA:37:D4:46:6A UUIDs: 00001200-0000-1000-8000-00805f9b34fb
[CHG] Device 5C:BA:37:D4:46:6A ServicesResolved: yes
[CHG] Device 5C:BA:37:D4:46:6A Paired: yes
Pairing successful
[Xbox Wireless Controller]# trust 5C:BA:37:D4:46:6A
[CHG] Device 5C:BA:37:D4:46:6A Trusted: yes
Changing 5C:BA:37:D4:46:6A trust succeeded
[Xbox Wireless Controller]# connect 5C:BA:37:D4:46:6A
Attempting to connect to 5C:BA:37:D4:46:6A
Connection successful
```

If I remember, correctly, with the CSR dongle, it usually does connect successfully the first time you pair, trust and connect.  The connection issue is more of a problem when I re-connect the controller after it has been powered off.

NoXPhasma, 2019-04-19:
> If I remember, correctly, with the CSR dongle, it usually does connect successfully the first time you pair, trust and connect. The connection issue is more of a problem when I re-connect the controller after it has been powered off.

This is my experience as well. However, just running this command one second after enabling the gamepad works reliable for me:
```
echo -e 'connect 5C:BA:37:F7:23:0E' | bluetoothctl
```

germangoergs, 2019-04-20:
> It seems to work.
>
> It can take a while to connect, about 10-15 seconds.
>
> I did need to install expect on Linux Mint 19.1
>
> ```
> sudo apt install expect
> ```
>
> Here is the output when I run the script: [...]

Yep, he's right.
I can actually connect the controller when set into pairing mode reliably after waiting about 2 seconds maximum and without removing it from bluetoothctl.

atar-axis, 2019-04-20:
Ha,... Okay, i see. Time wasted! 😁

closed 2019-04-20 by atar-axis

NoXPhasma, 2019-04-20:
Doesn't matter if I run the script command or execute `connect 5C:BA:37:F7:23:0E` inside of bluetoothctl. But running the script is obviously much more convenient :)

atar-axis, 2019-04-20:
Yeah! Thank you. The dongles which @ugly95 sent me arrived, so I could test it myself too :) I opened another issue where I will collect information. We can then create a patch for bluez if we found the problem