atar-axis/xpadneo #96: CSR reconnection problems

atar-axis, 2019-04-20:
# Problem

Dongles with Cambridge Silicon Radio 8510 Chipset do have problems to with to connect to the gamepad on linux. This seems to happen only when the dongle is in slave mode, therefore it is a **re**connection problem.

# `btmon` log

<details>
<p>

```
< ACL Data TX: Handle 70 flags 0x00 dlen 16                                                                                                                              #174 [hci0] 25.943375
      L2CAP: Connection Response (0x03) ident 4 len 8
        Destination CID: 0
        Source CID: 64
        Result: Connection refused - security block (0x0003)
        Status: No further information available (0x0000)
> HCI Event: Encryption Change (0x08) plen 4                                                                                                                             #175 [hci0] 25.944598
        Status: Success (0x00)
        Handle: 70
        Encryption: Enabled with E0 (0x01)
> HCI Event: Command Status (0x0f) plen 4                                                                                                                                #176 [hci0] 25.945596
      Remote Name Request (0x01|0x0019) ncmd 1
        Status: Success (0x00)
< HCI Command: Read Encryption Key Size (0x05|0x0008) plen 2                                                                                                             #177 [hci0] 25.945683
        Handle: 70
> HCI Event: Number of Completed Packets (0x13) plen 5                                                                                                                   #178 [hci0] 25.947607
        Num handles: 1
        Handle: 70
        Count: 1
> ACL Data RX: Handle 70 flags 0x02 dlen 16                                                                                                                              #179 [hci0] 25.948344
      L2CAP: Information Response (0x0b) ident 1 len 8
        Type: Extended features supported (0x0002)
        Result: Success (0x0000)
        Features: 0x00000010
          Streaming Mode
> HCI Event: Number of Completed Packets (0x13) plen 5                                                                                                                   #180 [hci0] 25.948590
        Num handles: 1
        Handle: 70
        Count: 1
> HCI Event: Command Complete (0x0e) plen 7                                                                                                                              #181 [hci0] 25.949606
      Read Encryption Key Size (0x05|0x0008) ncmd 1
        Status: Success (0x00)
        Handle: 70
        Key size: 16
> HCI Event: Remote Name Req Complete (0x07) plen 255                                                                                                                    #182 [hci0] 25.972694
        Status: Success (0x00)
        Address: C8:3F:26:AB:64:C0 (Microsoft Corporation)
        Name: Xbox Wireless Controller
< HCI Command: Disconnect (0x01|0x0006) plen 3                                                                                                                           #183 [hci0] 30.080154
        Handle: 70
        Reason: Authentication Failure (0x05)
> HCI Event: Command Status (0x0f) plen 4                                                                                                                                #184 [hci0] 30.192879
      Disconnect (0x01|0x0006) ncmd 1
        Status: Success (0x00)
> HCI Event: Disconnect Complete (0x05) plen 4                                                                                                                           #185 [hci0] 30.270815
        Status: Success (0x00)
        Handle: 70
        Reason: Connection Terminated By Local Host (0x16)
@ MGMT Event: Device Disconnected (0x000c) plen 8                                                                                                                    {0x0002} [hci0] 30.270895
        BR/EDR Address: C8:3F:26:AB:64:C0 (Microsoft Corporation)
        Reason: Connection terminated by local host (0x02)
@ MGMT Event: Device Disconnected (0x000c) plen 8                                                                                                                    {0x0001} [hci0] 30.270895
        BR/EDR Address: C8:3F:26:AB:64:C0 (Microsoft Corporation)
        Reason: Connection terminated by local host (0x02)
< HCI Command: Write Scan Enable (0x03|0x001a) plen 1                                                                                                                    #186 [hci0] 30.303513
        Scan enable: Page Scan (0x02)
> HCI Event: Command Complete (0x0e) plen 4                                                                                                                              #187 [hci0] 30.305600
      Write Scan Enable (0x03|0x001a) ncmd 1
        Status: Success (0x00)
> HCI Event: Connect Request (0x04) plen 10                                                                                                                              #188 [hci0] 30.336607
[... 946 lines cut ...]
        Type: Extended features supported (0x0002)
        Result: Success (0x0000)
        Features: 0x00000010
          Streaming Mode
> HCI Event: Number of Completed Packets (0x13) plen 5                                                                                                                   #360 [hci0] 49.429609
        Num handles: 1
        Handle: 70
        Count: 1
> HCI Event: Command Complete (0x0e) plen 7                                                                                                                              #361 [hci0] 49.430616
      Read Encryption Key Size (0x05|0x0008) ncmd 1
        Status: Success (0x00)
        Handle: 70
        Key size: 16
> HCI Event: Remote Name Req Complete (0x07) plen 255                                                                                                                    #362 [hci0] 49.452651
        Status: Success (0x00)
        Address: C8:3F:26:AB:64:C0 (Microsoft Corporation)
        Name: Xbox Wireless Controller
```

</p>
</details>

# Trace

The Problems is that BlueZ somehow thinks the Authentication was wrong, interestingly this does only happen on CSR chipsets. A very similar, and already fixed, authentication failure can be found [here](https://lkml.org/lkml/2014/4/24/730).

atar-axis, 2019-04-20:
https://github.com/torvalds/linux/blob/master/drivers/bluetooth/btusb.c

atar-axis, 2019-04-22:
added wireshark log for CSR, took some time to establish the connection.
[csr_long.pcapng.log](https://github.com/atar-axis/xpadneo/files/3103253/csr_long.pcapng.log)

Looks like the prolem is this one: https://linux-bluetooth.vger.kernel.narkive.com/U8xZU5ie/patch-bluetooth-fix-for-security-block-issue

The conenction fails as long as `Encryption Change` comes after the `Connect Request`.

atar-axis, 2019-04-22:
here is a usbmon log from my onboard BT controller:
<details>
<p>

```
> HCI Event: Connect Request (0x04) plen 10                                                                                                                               #18 [hci1] 16.402089
        Address: C8:3F:26:AB:64:C0 (Microsoft Corporation)
        Class: 0x000508
          Major class: Peripheral (mouse, joystick, keyboards)
          Minor class: 0x02
        Link type: ACL (0x01)
< HCI Command: Accept Connection Request (0x01|0x0009) plen 7                                                                                                             #19 [hci1] 16.468595
        Address: C8:3F:26:AB:64:C0 (Microsoft Corporation)
        Role: Master (0x00)
> HCI Event: Command Status (0x0f) plen 4                                                                                                                                 #20 [hci1] 16.470087
      Accept Connection Request (0x01|0x0009) ncmd 1
        Status: Success (0x00)
> HCI Event: Role Change (0x12) plen 8                                                                                                                                    #21 [hci1] 16.658087
        Status: Success (0x00)
        Address: C8:3F:26:AB:64:C0 (Microsoft Corporation)
        Role: Master (0x00)
> HCI Event: Vendor (0xff) plen 4                                                                                                                                         #22 [hci1] 16.866082
      Intel PTT Switch Notification (0x26)
        Handle: 512
        Packet type table: Enhanced data rate (0x01)
> HCI Event: Connect Complete (0x03) plen 11                                                                                                                              #23 [hci1] 16.870092
        Status: Success (0x00)
        Handle: 512
        Address: C8:3F:26:AB:64:C0 (Microsoft Corporation)
        Link type: ACL (0x01)
        Encryption: Disabled (0x00)
< HCI Command: Read Remote Supported Features (0x01|0x001b) plen 2                                                                                                        #24 [hci1] 16.870337
        Handle: 512
> HCI Event: Command Status (0x0f) plen 4                                                                                                                                 #25 [hci1] 16.871078
      Read Remote Supported Features (0x01|0x001b) ncmd 1
        Status: Success (0x00)
> HCI Event: Max Slots Change (0x1b) plen 3                                                                                                                               #26 [hci1] 16.878084
        Handle: 512
        Max slots: 5
> HCI Event: Read Remote Supported Features (0x0b) plen 11                                                                                                                #27 [hci1] 16.880075
        Status: Success (0x00)
        Handle: 512
        Features: 0xbf 0x3e 0x8d 0xfe 0xdb 0xfd 0x7b 0x87
          3 slot packets
          5 slot packets
          Encryption
          Slot offset
          Timing accuracy
          Role switch
          Sniff mode
          Power control requests
          Channel quality driven data rate (CQDDR)
          SCO link
          HV2 packets
          HV3 packets
          CVSD synchronous data
          Power control
          Transparent synchronous data
          Broadcast Encryption
          Enhanced Data Rate ACL 2 Mbps mode
          Enhanced Data Rate ACL 3 Mbps mode
          Enhanced inquiry scan
          Interlaced inquiry scan
          Interlaced page scan
          RSSI with inquiry results
          Extended SCO link (EV3 packets)
          EV4 packets
          EV5 packets
          AFH capable slave
          AFH classification slave
[... 937 lines cut ...]
        a1 04 89                                         ...             
> HCI Event: Inquiry Complete (0x01) plen 1                                                                                                                              #167 [hci1] 80.210189
        Status: Success (0x00)
< HCI Command: LE Set Scan Enable (0x08|0x000c) plen 2                                                                                                                   #168 [hci1] 80.345141
        Scanning: Disabled (0x00)
        Filter duplicates: Disabled (0x00)
> HCI Event: Command Complete (0x0e) plen 4                                                                                                                              #169 [hci1] 80.346188
      LE Set Scan Enable (0x08|0x000c) ncmd 2
        Status: Success (0x00)
@ MGMT Event: Discovering (0x0013) plen 2                                                                                                                            {0x0002} [hci1] 80.346257
        Address type: 0x07
          BR/EDR
          LE Public
          LE Random
        Discovery: Disabled (0x00)
@ MGMT Event: Discovering (0x0013) plen 2                                                                                                                            {0x0001} [hci1] 80.346257
        Address type: 0x07
          BR/EDR
          LE Public
          LE Random
        Discovery: Disabled (0x00)
```

</p>
</details>

germangoergs, 2019-04-22:
Great work, thank you! :)

There is a fix provided, is there a way to implement it? O.o (sorry noob here)

atar-axis, 2019-04-22:
I'll try to port this patch to the current sourcecode (the old patch will not works since it is so damn old), recompile the bluetooth module and see if it works then. But the underlying problem seems to be the CSR chipset, not linux.

germangoergs, 2019-04-22:
Ah OK, I see.

Good luck then! :)

atar-axis, 2019-04-23:
I had a closer look at the Patch and I really don't like it, what it does ist essentially delaying the Connection request Processing for 20ms in order to give the Encryption Change Message a chance to get processed before. That's very hackish, unfortunately.

atar-axis, 2019-04-24:
I now understand a little bit better what @holtmann meant when he said what he said.
First of all, the whole communication to and from bluetooth dongles over USB is specified in `PART B: USB TRANSPORT LAYER` of the Bluetooth Specification 5.1. That's really nice.

The Endpoints about which he talked are the following:
[image]
So it is the `HCI event` and the `ACL data` pipe.

atar-axis, 2019-04-26:
According to USBmon the packages do already arrive in the wrong order at the usb port:

[image]

I will have to confirm that using a real usb analzer (hardware, like BEAGLE)

atar-axis, 2019-05-15:
Is somebody still interested in fixing this? As said it is a problem which is caused by the Chipset itself, I would therefore suggest to try another one instead of working around by delaying the `Connection request` until the `Encryption Change` arrived.

I will close it for now - tell me if we should still try to fix it

closed 2019-05-15 by atar-axis

Zeioth, 2019-07-26:
Same issue here. The gemepad connects correctly and works great 100% of the time. But if I turn it off and then turn on again (or if the gamepad goes to sleep mode), then it won't be detected by the game until I re-open the game.

I've only tested Wine games thou. I'll try to test the issue with a native game in the next days.

Zeioth, 2019-07-26:
I can confirm is a driver/dongle issue. It happen with both native and wine games.