atar-axis/xpadneo #100: connection to driver xpadneo on raspberry fails with error -22 or only hid-generic is started

gordongnarf, 2019-05-05:
**Describe the bug**
I try to connect the xbox controller to my raspberry pi 3.
For me it does work with the hid-generic, since i only want to read out buttons and analog sticks with "python evdev"
In order to be able to "rumble" the controller i would like to use the xpadneo driver.
After installation of the xpadneo driver i either get:

[  +0,163060] hid-generic 0005:045E:02FD.0001: unknown main item tag 0x0
[  +0,000071] hid-generic 0005:045E:02FD.0001: unbalanced collection at end of report description
[  +0,000095] hid-generic: probe of 0005:045E:02FD.0001 failed with error -22
[  +0,052215] hid_xpadneo: loading out-of-tree module taints kernel.
[  +0,000973] xpadneo: hello there!
[  +0,002379] xpadneo 0005:045E:02FD.0001: unknown main item tag 0x0
[  +0,000078] xpadneo 0005:045E:02FD.0001: unbalanced collection at end of report description
[  +0,000054] xpadneo 0005:045E:02FD.0001: parse failed
[  +0,000051] xpadneo: probe of 0005:045E:02FD.0001 failed with error -22

with the above situation i do not get any files in /dev/input

or (after turning the controller off/on) the xpadneo driver does not start at all:

[Mai 5 23:33] hid-generic 0005:045E:02FD.0002: unknown main item tag 0x0
[  +0,000512] input: Xbox Wireless Controller as /devices/platform/soc/3f201000.serial/tty/ttyAMA0/hci0/hci0:12/0005:045E:02FD.0002/input/input0
[  +0,010544] hid-generic 0005:045E:02FD.0002: input,hidraw0: BLUETOOTH HID v9.03 Gamepad [Xbox Wireless Controller] on b8:27:eb:73:c7:4f

but i do have files in /dev/input/:
pi@raspberrypi:~ $ ls /dev/input/
event0  js0
With python i read out event0 with evdev and everything is fine - except the fact, that it does not rumble :-(

**To Reproduce**
I tried two time to: 
remove the device from bluetoothctl
uninstall xpadneo
reboot
install xpadneo
reboot
Install current FW with Win10 (was no update necessary, but something was done to the device)
bluetoothctl scan on, pair mac, trust mac, connect mac

after the FW-update with Windows i could not connect to the device anymore (connect and immediate unconnect). So i uninstalled the xpadneo and tried again, still no connect. Reboot, remove from bluetoothctl, etc. Finally it connected again, but i got the error -22 (see above).
The whole procedure seems to be critical, depending on not so clear steps.

**Expected behavior**
A rumble and a stable connect :-)

**Screenshots/Gifs**

**System information**
-   `uname -a` : 
Linux raspberrypi 4.14.98-v7+ #1200 SMP Tue Feb 12 20:27:48 GMT 2019 armv7l GNU/Linux

-   `dmesg`
see above

-   `sudo find "/sys/kernel/debug/hid/" -name "0005:045E:*" -exec sh -c 'echo "{}" && head -1 "{}/rdesc" | tee /dev/tty | cksum && echo' \;`
/sys/kernel/debug/hid/0005:045E:02FD.0002
05 01 09 05 a1 01 85 01 09 01 a1 00 09 30 09 31 15 00 27 ff ff 00 00 95 02 75 10 81 02 c0 09 01 a1 00 09 32 09 35 15 00 27 ff ff 00 00 95 02 75 10 81 02 c0 05 02 09 c5 15 00 26 ff 03 95 01 75 0a 81 02 15 00 25 00 75 06 95 01 81 03 05 02 09 c4 15 00 26 ff 03 95 01 75 0a 81 02 15 00 25 00 75 06 95 01 81 03 05 01 09 39 15 01 25 08 35 00 46 3b 01 66 14 00 75 04 95 01 81 42 75 04 95 01 15 00 25 00 35 00 45 00 65 00 81 03 05 09 19 01 29 0f 15 00 25 01 75 01 95 0f 81 02 15 00 25 00 75 01 95 01 81 03 05 0c 0a 24 02 15 00 25 01 95 01 75 01 81 02 15 00 25 00 75 07 95 01 81 03 05 0c 09 01 85 02 a1 01 05 0c 0a 23 02 15 00 25 01 95 01 75 01 81 02 15 00 25 00 75 07 95 01 81 03 c0 05 0f 09 21 85 03 a1 02 09 97 15 00 25 01 75 04 95 01 91 02 15 00 25 00 75 04 95 01 91 03 09 70 15 00 25 64 75 08 95 04 91 02 09 50 66 01 10 55 0e 15 00 26 ff 00 75 08 95 01 91 02 09 a7 15 00 26 ff 00 75 08 95 01 91 02 65 00 55 00 09 7c 15 00 26 ff 00 75 08 95 01 91 02 c0 85 04 05 06 09 20 15 00 26 ff 00 75 08 95 01 81 02 c0 00 
2579218694 1006

gordongnarf, 2019-05-05:
I tried to reproduce the error:
remove the device from bluetoothctl
uninstall xpadneo
reboot
install xpadneo
reboot
Install current FW with Win10 (was no update necessary, but something was done to the device)
bluetoothctl scan on, pair mac, trust mac, connect mac

Then i get the error:
[  +5,203454] hid-generic 0005:045E:02FD.0001: unknown main item tag 0x0
[  +0,000068] hid-generic 0005:045E:02FD.0001: unbalanced collection at end of report description
[  +0,000066] hid-generic: probe of 0005:045E:02FD.0001 failed with error -22
[  +0,052439] hid_xpadneo: loading out-of-tree module taints kernel.
[  +0,001159] xpadneo: hello there!
[  +0,001886] xpadneo 0005:045E:02FD.0001: unknown main item tag 0x0
[  +0,000074] xpadneo 0005:045E:02FD.0001: unbalanced collection at end of report description
[  +0,000067] xpadneo 0005:045E:02FD.0001: parse failed
[  +0,000060] xpadneo: probe of 0005:045E:02FD.0001 failed with error -22

After turning the controller off/on i get again:
[Mai 6 00:13] hid-generic 0005:045E:02FD.0002: unknown main item tag 0x0
[  +0,000750] input: Xbox Wireless Controller as /devices/platform/soc/3f201000.serial/tty/ttyAMA0/hci0/hci0:11/0005:045E:02FD.0002/input/input0
[  +0,000321] hid-generic 0005:045E:02FD.0002: input,hidraw0: BLUETOOTH HID v9.03 Gamepad [Xbox Wireless Controller] on b8:27:eb:73:c7:4f

so its reproducible ... time to go to bed anyway ...

atar-axis, 2019-05-05:
Hey! Do you get the `22` error only the first time after updating the fw?

atar-axis, 2019-05-05:
https://github.com/atar-axis/xpadneo/issues/91#issuecomment-480565026

Please upload this log :) thanks!

atar-axis, 2019-05-06:
It may be possible to simplify the whole udev-rule monster

* remove `/etc/udev/rules.d/xpadneo.sh`
* replace the content of `/etc/udev/rules.d/99-xpadneo.rules` with the following:
    ```
    ACTION=="add", \
    KERNEL=="0005:045E:02FD.*|0005:045E:02E0.*", \
    SUBSYSTEM=="hid", \
    RUN:="/bin/sh -c 'echo xpadneo udev: $kernel > /dev/kmsg; modprobe hid_xpadneo; echo $kernel > /sys/bus/hid/drivers/hid-generic/unbind; echo $kernel > /sys/bus/hid/drivers/microsoft/unbind; echo $kernel > /sys/bus/hid/drivers/xpadneo/bind; echo xpadneo udev: ok > /dev/kmsg'"
    ```

gordongnarf, 2019-05-06:
> Hey! Do you get the `22` error only the first time after updating the fw?
> 
> Looks like the udev rule does not work on your machine, once again... Give me a Minute to add instructions here

Hi, thanks for the quick reply!
It seems that i get the error even without updating the firmware. In order to produce the logs that you want i removed the device from bluetoothctl, unistalled your driver, reboot, then with using log1 i connected to the controller again - see **dmesg**:

[Mai 6 23:03] Bluetooth: HIDP (Human Interface Emulation) ver 1.2
[  +0,000036] Bluetooth: HIDP socket layer initialized
[  +0,005508] hid-generic 0005:045E:02FD.0001: unknown main item tag 0x0
[  +0,000069] hid-generic 0005:045E:02FD.0001: unbalanced collection at end of report description
[  +0,000080] hid-generic: probe of 0005:045E:02FD.0001 failed with error -22
[Mai 6 23:06] hid-generic 0005:045E:02FD.0002: unknown main item tag 0x0
[  +0,000114] hid-generic 0005:045E:02FD.0002: unbalanced collection at end of report description
[  +0,000088] hid-generic: probe of 0005:045E:02FD.0002 failed with error -22

**/tmp/xpadneolog1**

monitor will print the received events for:
UDEV - the event which udev sends out after rule processing
KERNEL - the kernel uevent

KERNEL[8894.393686] add      /devices/platform/soc/3f201000.serial/tty/ttyAMA0/hci0/hci0:11 (bluetooth)
ACTION=add
DEVPATH=/devices/platform/soc/3f201000.serial/tty/ttyAMA0/hci0/hci0:11
DEVTYPE=link
SEQNUM=1198
SUBSYSTEM=bluetooth

UDEV  [8894.401583] add      /devices/platform/soc/3f201000.serial/tty/ttyAMA0/hci0/hci0:11 (bluetooth)
ACTION=add
DEVPATH=/devices/platform/soc/3f201000.serial/tty/ttyAMA0/hci0/hci0:11
DEVTYPE=link
SEQNUM=1198
SUBSYSTEM=bluetooth
SYSTEMD_ALIAS=/sys/subsystem/bluetooth/devices/hci0:11
SYSTEMD_WANTS=bluetooth.target
TAGS=:systemd:
USEC_INITIALIZED=8894400825

KERNEL[8903.190409] remove   /devices/platform/soc/3f201000.serial/tty/ttyAMA0/hci0/hci0:11 (bluetooth)
ACTION=remove
DEVPATH=/devices/platform/soc/3f201000.serial/tty/ttyAMA0/hci0/hci0:11
DEVTYPE=link
SEQNUM=1199
SUBSYSTEM=bluetooth

UDEV  [8903.196005] remove   /devices/platform/soc/3f201000.serial/tty/ttyAMA0/hci0/hci0:11 (bluetooth)
ACTION=remove
DEVPATH=/devices/platform/soc/3f201000.serial/tty/ttyAMA0/hci0/hci0:11
DEVTYPE=link
SEQNUM=1199
SUBSYSTEM=bluetooth
SYSTEMD_ALIAS=/sys/subsystem/bluetooth/devices/hci0:11
SYSTEMD_WANTS=bluetooth.target
TAGS=:systemd:
USEC_INITIALIZED=8894400825

KERNEL[8939.224732] add      /devices/platform/soc/3f201000.serial/tty/ttyAMA0/hci0/hci0:12 (bluetooth)
ACTION=add
DEVPATH=/devices/platform/soc/3f201000.serial/tty/ttyAMA0/hci0/hci0:12
DEVTYPE=link
SEQNUM=1200
SUBSYSTEM=bluetooth

UDEV  [8939.232688] add      /devices/platform/soc/3f201000.serial/tty/ttyAMA0/hci0/hci0:12 (bluetooth)
ACTION=add
DEVPATH=/devices/platform/soc/3f201000.serial/tty/ttyAMA0/hci0/hci0:12
DEVTYPE=link
SEQNUM=1200
[... 357 lines cut ...]
POWER_SUPPLY_CAPACITY=52
POWER_SUPPLY_MODEL_NAME=Xbox Wireless Controller
POWER_SUPPLY_NAME=hid-9c:aa:1b:e7:fc:a2-battery
POWER_SUPPLY_ONLINE=1
POWER_SUPPLY_PRESENT=1
POWER_SUPPLY_SCOPE=Device
POWER_SUPPLY_STATUS=Discharging
SEQNUM=1214
SUBSYSTEM=power_supply

UDEV  [259.983524] change   /devices/platform/soc/3f201000.serial/tty/ttyAMA0/hci0/hci0:12/0005:045E:02FD.0001/power_supply/hid-9c:aa:1b:e7:fc:a2-battery (power_supply)
ACTION=change
DEVPATH=/devices/platform/soc/3f201000.serial/tty/ttyAMA0/hci0/hci0:12/0005:045E:02FD.0001/power_supply/hid-9c:aa:1b:e7:fc:a2-battery
POWER_SUPPLY_CAPACITY=52
POWER_SUPPLY_MODEL_NAME=Xbox Wireless Controller
POWER_SUPPLY_NAME=hid-9c:aa:1b:e7:fc:a2-battery
POWER_SUPPLY_ONLINE=1
POWER_SUPPLY_PRESENT=1
POWER_SUPPLY_SCOPE=Device
POWER_SUPPLY_STATUS=Discharging
SEQNUM=1214
SUBSYSTEM=power_supply
USEC_INITIALIZED=259983052

**hope you can read the matrix without decoder ;-)**

atar-axis, 2019-05-06:
Thanks! I will take a look later, have you tried the instructions above too? Would be interesting too :)

gordongnarf, 2019-05-06:
> **Note**: It may be possible to simplify the whole udev-rule monster
> 
>     * remove `/etc/udev/rules.d/xpadneo.sh`
> 
>     * replace the content of `/etc/udev/rules.d/99-xpadneo.rules` with the following:
>       ```
>       ACTION=="add", \
>       KERNEL=="0005:045E:02FD.*|0005:045E:02E0.*", \
>       SUBSYSTEM=="hid", \
>       RUN:="/bin/sh -c 'echo xpadneo udev: $kernel > /dev/kmsg; modprobe hid_xpadneo; echo $kernel > /sys/bus/hid/drivers/hid-generic/unbind; echo $kernel > /sys/bus/hid/drivers/microsoft/unbind; echo $kernel > /sys/bus/hid/drivers/xpadneo/bind; echo xpadneo udev: ok > /dev/kmsg'"
>       ```
> 
> 
> @gordongnarf: If you want, you can give this a try too, but it would be great if you could send me the log anyway.

**did what you wrote above and i got my rumble :-) see dmesg below**
Thanks a lot! hope with my testing i could help you somehow!
If you have further questions, need logs, ... just ask!

[Mai 6 23:47] hid-generic 0005:045E:02FD.0003: unknown main item tag 0x0
[  +0,000746] input: Xbox Wireless Controller as /devices/platform/soc/3f201000.serial/tty/ttyAMA0/hci0/hci0:13/0005:045E:02FD.0003/input/input2
[  +0,000711] hid-generic 0005:045E:02FD.0003: input,hidraw0: BLUETOOTH HID v9.03 Gamepad [Xbox Wireless Controller] on b8:27:eb:73:c7:4f
[  +0,011425] xpadneo udev: 0005:045E:02FD.0003
[  +0,142775] xpadneo 0005:045E:02FD.0003: unknown main item tag 0x0
[  +0,000374] input: Xbox Wireless Controller as /devices/platform/soc/3f201000.serial/tty/ttyAMA0/hci0/hci0:13/0005:045E:02FD.0003/input/input3
[  +0,000517] xpadneo 0005:045E:02FD.0003: input,hidraw0: BLUETOOTH HID v9.03 Gamepad [Xbox Wireless Controller] on b8:27:eb:73:c7:4f
[  +0,990708] xpadneo udev: ok

atar-axis, 2019-05-06:
> did what you wrote above and i got my rumble :-) see dmesg below

that's good to hear, it also works on my local machine. But I will wait for some others to report back first, if it works for them too then I will merge it into the master :grin: 
Thanks for testing and reporting the issue!

closed 2019-05-06 by atar-axis

gordongnarf, 2019-05-08:
Hi,
would it be asked too much if I ask you for help to bring to work a python
script, that makes the controller Rumble?
Fftest works fine - so it should be possible at least.
In the net I can find the evdev documentation example - no error but also
no Rumble .
Then there is an old example class that also does not work for me - error.
I tried to understand the c-code of Fftest and transfer it to python - no
success.

I can provide more information.
I guess a running python-Rumble example would be interesting for many
people...
The information : which data must be transferred, would also be big help.
Thomas

atar-axis, 2019-05-08:
Hey Thomas, I am not a well trained python programmer, but I am sure we will find a solution - just open a new issue please ;)

flagoss, 2019-05-15:
I have been trying for days to make my controllers use Xpadneo on connection ( Connecting with vibration ) and I have not been able. The only thing that worked for me is using that methode:
```
* remove `/etc/udev/rules.d/xpadneo.sh`
* replace the content of `/etc/udev/rules.d/99-xpadneo.rules` with the following wich as worked pretty well:
  ```
  ACTION=="add", \
  KERNEL=="0005:045E:02FD.*|0005:045E:02E0.*", \
  SUBSYSTEM=="hid", \
  RUN:="/bin/sh -c 'echo xpadneo udev: $kernel > /dev/kmsg; modprobe hid_xpadneo; echo $kernel > /sys/bus/hid/drivers/hid-generic/unbind; echo $kernel > /sys/bus/hid/drivers/microsoft/unbind; echo $kernel > /sys/bus/hid/drivers/xpadneo/bind; echo xpadneo udev: ok > /dev/kmsg'"
  ```
```
I am using a Raspberry Pi 3 B+ and did a fresh install of Retropie. Did a full system update. Uninstalled Xpad. Then I installed Xpadneo.

atar-axis, 2019-05-15:
@flagoss: Thank you for reporting that! I will Change that in master today

atar-axis, 2019-05-15:
@flagoss I change the udev rule to the one which where mentioned in the issue you quoted:
https://github.com/atar-axis/xpadneo/commit/f2729dc96c55439c6dad3a4672e08a4173e79a98

Could you please give it a try?
* `./uninstall.sh`
* clone the new version (master)
* `./install.sh`

flagoss, 2019-05-17:
Thank you... I will try that a try tonight